Louisvdw/dbus-serialbattery #652: Multiple Seplos via RS485

schindlerit1984, 2023-05-20:
### Describe the bug

Hi,

getting an error when i try to connect two Seplos Batteries individually via Rs485 - two USB Devices and no daisy chaining - individually the work fine - once both are connected only one of them is working.  

`ERROR:SerialBattery:Unexpected err=SerialException('read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'), type(err)=<class 'serial.serialutil.SerialException'>`

is what i am getting on the second usb then once first one is running. 

any idea how to fix it ? for me its mainly to pull the rs485 infos out of the pack like cell voltages etc.  

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

hope to get both bms visible - this was only working once a few versions ago ( i am not 100% sure) 

### Driver version

1.0.20230518dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.042

### BMS type

Seplos

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 194.0
MAX_BATTERY_DISCHARGE_CURRENT = 194.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.550
FLOAT_CELL_VOLTAGE = 3.45

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; Example with 1 BMS: Jkbms_Ble C8:47:8C:00:00:00
; Example with 3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =
[... 188 lines cut ...]
; NOTE: SOC_LOW_WARNING is also used to calculate the Time-To-Go even if you are not using a LltJbd BMS
SOC_LOW_WARNING = 20
SOC_LOW_ALARM   = 10

; -- Daly settings
; Battery capacity (amps) if the BMS does not support reading it
BATTERY_CAPACITY = 50
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = 1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT = 15
```

### Relevant log output

```shell
2023-05-20 09:35:52.607457500 ERROR:SerialBattery:Unexpected err=SerialException('read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'), type(err)=<class 'serial.serialutil.SerialException'>
2023-05-20 09:35:53.108557500 INFO:SerialBattery:Testing Seplos
2023-05-20 09:35:55.101010500 ERROR:SerialBattery:Unexpected err=SerialException('read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'), type(err)=<class 'serial.serialutil.SerialException'>
2023-05-20 09:35:55.602083500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
```

mr-manuel, 2023-05-20:
Multiple devices on one RS485 adapter are not supported yet. See https://github.com/Louisvdw/dbus-serialbattery/issues/8

schindlerit1984, 2023-05-20:
this is via two adapters not one

mr-manuel, 2023-05-20:
Oh ok. That you meaned with indivitually. Only connected one at a time over USB to serial adapter. If you connect both at the same time, with one adapter each it do not work. Is this correct?

@wollew maybe you can take a look?

schindlerit1984, 2023-05-20:
yeah so once the first one is detected the second one gets ignored / either on boot or boot once a second one is connected either as usb0 / 1 / 2 ... it does not matter.

the once i am using is shown as : 

lrwxrwxrwx    1 root     root            13 Jan  1  1970 usb-FTDI_FT232R_USB_UART_A9HJMHE8-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx    1 root     root            13 May 20 13:49 usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB7

in my case ttyUSB0 works just fine - both converters work fine individually  - also it does not matter if they come up as ttyUSB 1-7 . both packs have no dip switch enabled - switching between the packs and all various combinations bring the same result 

if i try to manually start them with ./start-serialbattery.sh ttyUSB7 i am getting ->

```
UTC-2023.05.20-13:53:57 Starting dbus-serialbattery.py on ttyUSB7
INFO:SerialBattery:
INFO:SerialBattery:Starting dbus-serialbattery
INFO:SerialBattery:dbus-serialbattery v1.0.20230518dev
INFO:SerialBattery:Testing Seplos
WARNING:SerialBattery:checksum error
INFO:SerialBattery:Testing Seplos
ERROR:SerialBattery:Unexpected err=SerialException('read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'), type(err)=<class 'serialserialutil.SerialException'>
INFO:SerialBattery:Testing Seplos
ERROR:SerialBattery:Unexpected err=SerialException('read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'), type(err)=<class 'serialserialutil.SerialException'>
ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB7
```

wollew, 2023-06-05:
> Oh ok. That you meaned with indivitually. Only connected one at a time over USB to serial adapter. If you connect both at the same time, with one adapter each it do not work. Is this correct?
> 
> @wollew maybe you can take a look?

I will ty to look into it, sorry for the delay, it seems I did not get a notification.

wollew, 2023-06-05:
Also, if you unplug usb-FTDI_FT232R_USB_UART_A9HJMHE8-if00-port0, does the other adapter still show up as something other than ttyUSB0 but then works?

schindlerit1984, 2023-06-05:
> @schindlerit1984 Are you saying you're getting the above log output when you already have a running (and working) dbus-serialbattery.py on /dev/ttyUSB0 ? And you then manually start python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB7 ?

Yes thats correct. 

Once the other is disconnected the other one shows up ( Takes Like 1-2 mins )

wollew, 2023-06-05:
I guess you already checked this always happens for the second adapter, not matter which adapter you use as the second one (looking at https://github.com/Louisvdw/dbus-serialbattery/issues/652#issuecomment-1555916894 you own two different ones and also no matter which physical BMS is connected to which adapter? If so, I have no idea what could be the issue here because the two python process should not interfere with each other. Would you be able to change the loglevel to DEBUG and then give us the output (check /opt/victronenergy/dbus-serialbattery/utils.py)? Not that I really know how that could help but maybe I'm missing something here.

schindlerit1984, 2023-06-05:
Yes tried all the various ways. Different Adapter / Same Type Adapter , different cables , swapped them and so on. 

Checking on the loglevel

schindlerit1984, 2023-06-05:
sometimes both are there an then 20 secs later one is gone and the other one is there

both then end up in checksum error - during loading this happens 

```
2023-06-05 13:46:41.500149500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-05 13:46:41.500960500 INFO:SerialBattery:dbus-serialbattery v1.0.20230526dev
2023-06-05 13:46:41.501372500 INFO:SerialBattery:Testing Seplos
2023-06-05 13:46:43.390286500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('ascii', b'\x80\x00\x06f', 0, 1, 'ordinal not in range(128)'), type(err)=<class 'UnicodeDecodeError'>
2023-06-05 13:46:43.390515500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-05 13:46:43.891299500 INFO:SerialBattery:Testing Seplos
2023-06-05 13:46:46.147632500 WARNING:SerialBattery:checksum error
2023-06-05 13:46:46.154293500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-05 13:46:46.655379500 INFO:SerialBattery:Testing Seplos
2023-06-05 13:46:47.788607500 INFO:SerialBattery:Connection established to Seplos
2023-06-05 13:46:47.789085500 INFO:SerialBattery:Battery Seplos connected to dbus from /dev/ttyUSB7
2023-06-05 13:46:47.789384500 INFO:SerialBattery:========== Settings ==========
2023-06-05 13:46:47.789842500 INFO:SerialBattery:> Connection voltage: 54.44V | Current: 25.59A | SoC: 98.1%
2023-06-05 13:46:47.790152500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 0
2023-06-05 13:46:47.790700500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-06-05 13:46:47.791094500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 194.0A | MAX BATTERY DISCHARGE CURRENT: 194.0A
2023-06-05 13:46:47.791392500 INFO:SerialBattery:> CVCM:     True
2023-06-05 13:46:47.791718500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
2023-06-05 13:46:47.792035500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-06-05 13:46:47.792343500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   True
2023-06-05 13:46:47.792632500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: True
2023-06-05 13:46:47.826477500 INFO:SerialBattery:DeviceInstance = 1
2023-06-05 13:46:47.826921500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB7
2023-06-05 13:46:47.845993500 INFO:SerialBattery:publish config values = 1
2023-06-05 13:46:56.690850500 Traceback (most recent call last):
2023-06-05 13:46:56.690854500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 361, in publish_battery
2023-06-05 13:46:56.690858500     success = self.battery.refresh_data()
2023-06-05 13:46:56.690861500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 117, in refresh_data
2023-06-05 13:46:56.690865500     result_status = self.read_status_data()
2023-06-05 13:46:56.690868500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 194, in read_status_data
2023-06-05 13:46:56.690961500     data = self.read_serial_data_seplos(
2023-06-05 13:46:56.690964500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 293, in read_serial_data_seplos
2023-06-05 13:46:56.690968500     if not Seplos.is_valid_frame(data):
2023-06-05 13:46:56.690971500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 267, in is_valid_frame
2023-06-05 13:46:56.690975500     if chksum != Seplos.int_from_2byte_hex_ascii(data, -5):
2023-06-05 13:46:56.691005500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 32, in int_from_2byte_hex_ascii
2023-06-05 13:46:56.691010500     bytes.fromhex(data[offset : offset + 4].decode("ascii")),
2023-06-05 13:46:56.691013500 UnicodeDecodeError: 'ascii' codec can't decode byte 0x94 in position 1: ordinal not in range(128)
2023-06-05 13:46:56.691017500

```

schindlerit1984, 2023-06-05:
reboot gave debug infos - let me check

schindlerit1984, 2023-06-05:
rebooted with usb0 + 4 now : 

the moment usb4 goes out it posts 

2023-06-05 13:52:15.555094500 DEBUG:SerialBattery:wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004644E00201FD34\r'

```

3.341V [16]3.342V
2023-06-05 13:51:44.395782500 DEBUG:SerialBattery:read status data
2023-06-05 13:51:44.396171500 DEBUG:SerialBattery:read serial data seplos
2023-06-05 13:51:44.406723500 DEBUG:SerialBattery:wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
2023-06-05 13:51:45.521412500 DEBUG:SerialBattery:return info data of length 150 : b'0000100D7A0D210D5C0D280D3E0D1C0D780D360D0E0D3A0D0D0D3B0D150D180D0D0D0E060C0D0BF80BFA0C0C0C340C19FFCC151A6B4F0A6D6003D56D60001603E8151CFFF800000BC80B5B'
2023-06-05 13:51:45.529937500 DEBUG:SerialBattery:Voltage cell[0]=3.45V
2023-06-05 13:51:45.529942500 DEBUG:SerialBattery:Voltage cell[1]=3.361V
2023-06-05 13:51:45.529945500 DEBUG:SerialBattery:Voltage cell[2]=3.42V
2023-06-05 13:51:45.531220500 DEBUG:SerialBattery:Voltage cell[3]=3.368V
2023-06-05 13:51:45.531225500 DEBUG:SerialBattery:Voltage cell[4]=3.39V
2023-06-05 13:51:45.531228500 DEBUG:SerialBattery:Voltage cell[5]=3.356V
2023-06-05 13:51:45.531485500 DEBUG:SerialBattery:Voltage cell[6]=3.448V
2023-06-05 13:51:45.531854500 DEBUG:SerialBattery:Voltage cell[7]=3.382V
2023-06-05 13:51:45.532211500 DEBUG:SerialBattery:Voltage cell[8]=3.342V
2023-06-05 13:51:45.532563500 DEBUG:SerialBattery:Voltage cell[9]=3.386V
2023-06-05 13:51:45.532915500 DEBUG:SerialBattery:Voltage cell[10]=3.341V
2023-06-05 13:51:45.533255500 DEBUG:SerialBattery:Voltage cell[11]=3.387V
2023-06-05 13:51:45.533611500 DEBUG:SerialBattery:Voltage cell[12]=3.349V
2023-06-05 13:51:45.533968500 DEBUG:SerialBattery:Voltage cell[13]=3.352V
2023-06-05 13:51:45.534363500 DEBUG:SerialBattery:Voltage cell[14]=3.341V
2023-06-05 13:51:45.534672500 DEBUG:SerialBattery:Voltage cell[15]=3.342V
2023-06-05 13:51:45.535012500 DEBUG:SerialBattery:Temp cell[0]=35.4°C
2023-06-05 13:51:45.535344500 DEBUG:SerialBattery:Temp cell[1]=33.3°C
2023-06-05 13:51:45.535654500 DEBUG:SerialBattery:Temp cell[2]=33.5°C
2023-06-05 13:51:45.535955500 DEBUG:SerialBattery:Temp cell[3]=35.3°C
2023-06-05 13:51:45.536360500 DEBUG:SerialBattery:Current = -0.52A , Voltage = 54.02V
2023-06-05 13:51:45.536635500 DEBUG:SerialBattery:Capacity = 274.71/280.0Ah , SOC = 98.1%
2023-06-05 13:51:45.536895500 DEBUG:SerialBattery:Cycles = 22
2023-06-05 13:51:45.537178500 DEBUG:SerialBattery:Environment temp = 39.3°C ,  Power temp = 36.6°C
2023-06-05 13:51:45.537423500 DEBUG:SerialBattery:HW:Seplos BMS 16 cells
2023-06-05 13:51:45.537768500 DEBUG:SerialBattery:read serial data seplos
2023-06-05 13:51:45.551755500 DEBUG:SerialBattery:wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004644E00201FD34\r'
2023-06-05 13:51:47.615803500 WARNING:SerialBattery:checksum error
2023-06-05 13:51:47.639070500 DEBUG:SerialBattery:logged to dbus [98.1]
2023-06-05 13:51:47.639636500 DEBUG:SerialBattery:Cells:[1]3.45V [2]3.361V [3]3.42V [4]3.368V [5]3.39V [6]3.356V [7]3.448V [8]3.382V [9]3.342V [10]3.386V [11]3.341V [12]3.387V [13]3.349V [14]3.352V [15]3.341V [16]3.342V
2023-06-05 13:51:49.397852500 DEBUG:SerialBattery:read status data
2023-06-05 13:51:49.398227500 DEBUG:SerialBattery:read serial data seplos
2023-06-05 13:51:49.411417500 DEBUG:SerialBattery:wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
2023-06-05 13:51:50.523913500 DEBUG:SerialBattery:return info data of length 150 : b'0000100D7A0D210D5C0D280D3E0D1D0D7A0D360D0F0D3A0D0E0D3B0D150D190D0D0D0E060C0D0BF80BFA0C0C0C340C190000151A6B4F0A6D6003D56D60001603E8151DFFFBFEEA0BC80B5B'
2023-06-05 13:51:50.529828500 DEBUG:SerialBattery:Voltage cell[0]=3.45V
2023-06-05 13:51:50.530183500 DEBUG:SerialBattery:Voltage cell[1]=3.361V
2023-06-05 13:51:50.530650500 DEBUG:SerialBattery:Voltage cell[2]=3.42V
2023-06-05 13:51:50.531096500 DEBUG:SerialBattery:Voltage cell[3]=3.368V
2023-06-05 13:51:50.531537500 DEBUG:SerialBattery:Voltage cell[4]=3.39V
2023-06-05 13:51:50.531988500 DEBUG:SerialBattery:Voltage cell[5]=3.357V
2023-06-05 13:51:50.532428500 DEBUG:SerialBattery:Voltage cell[6]=3.45V
2023-06-05 13:51:50.532896500 DEBUG:SerialBattery:Voltage cell[7]=3.382V
2023-06-05 13:51:50.533331500 DEBUG:SerialBattery:Voltage cell[8]=3.343V
2023-06-05 13:51:50.533757500 DEBUG:SerialBattery:Voltage cell[9]=3.386V
2023-06-05 13:51:50.534173500 DEBUG:SerialBattery:Voltage cell[10]=3.342V
2023-06-05 13:51:50.534674500 DEBUG:SerialBattery:Voltage cell[11]=3.387V
2023-06-05 13:51:50.535140500 DEBUG:SerialBattery:Voltage cell[12]=3.349V
2023-06-05 13:51:50.535520500 DEBUG:SerialBattery:Voltage cell[13]=3.353V
2023-06-05 13:51:50.535889500 DEBUG:SerialBattery:Voltage cell[14]=3.341V
2023-06-05 13:51:50.536235500 DEBUG:SerialBattery:Voltage cell[15]=3.342V
2023-06-05 13:51:50.536650500 DEBUG:SerialBattery:Temp cell[0]=35.4°C
2023-06-05 13:51:50.537013500 DEBUG:SerialBattery:Temp cell[1]=33.3°C
2023-06-05 13:51:50.537320500 DEBUG:SerialBattery:Temp cell[2]=33.5°C
2023-06-05 13:51:50.537619500 DEBUG:SerialBattery:Temp cell[3]=35.3°C
2023-06-05 13:51:50.538027500 DEBUG:SerialBattery:Current = 0.0A , Voltage = 54.02V
2023-06-05 13:51:50.538305500 DEBUG:SerialBattery:Capacity = 274.71/280.0Ah , SOC = 98.1%
2023-06-05 13:51:50.538563500 DEBUG:SerialBattery:Cycles = 22
[... 146 lines cut ...]
[... cut ...]
2023-06-05 13:52:20.542443500 DEBUG:SerialBattery:read serial data seplos
2023-06-05 13:52:20.554174500 DEBUG:SerialBattery:wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004644E00201FD34\r'
2023-06-05 13:52:22.633602500 WARNING:SerialBattery:checksum error
2023-06-05 13:52:24.409452500 DEBUG:SerialBattery:read status data
2023-06-05 13:52:24.409832500 DEBUG:SerialBattery:read serial data seplos
2023-06-05 13:52:24.419963500 DEBUG:SerialBattery:wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
```

wollew, 2023-06-05:
To me, this looks like more than one serial batterydriver is trying to read from the same port. 
As you've already edited utils.py, can you change 

```
logging.basicConfig()
```

into

```
FORMAT = '%(process)d %(message)s'
logging.basicConfig(format=FORMAT)
```
 and the restart the driver (you don't need to reboot, it is enough to kill all dbus-serialbattery.py processes.

schindlerit1984, 2023-06-05:
done it  

definitly runs longer in parallel but still second usb ( usb4 goes lost after ~1-2 mins) 

```

2023-06-05 14:55:08.385408500 3180 Voltage cell[13]=3.375V
2023-06-05 14:55:08.385413500 3180 Voltage cell[14]=3.346V
2023-06-05 14:55:08.385416500 3180 Voltage cell[15]=3.347V
2023-06-05 14:55:08.385419500 3180 Temp cell[0]=34.9°C
2023-06-05 14:55:08.385421500 3180 Temp cell[1]=32.9°C
2023-06-05 14:55:08.385424500 3180 Temp cell[2]=33.0°C
2023-06-05 14:55:08.385426500 3180 Temp cell[3]=34.8°C
2023-06-05 14:55:08.385428500 3180 Current = 4.07A , Voltage = 54.86V
2023-06-05 14:55:08.385431500 3180 Capacity = 274.92/280.0Ah , SOC = 98.1%
2023-06-05 14:55:08.385434500 3180 Cycles = 22
2023-06-05 14:55:08.385542500 3180 Environment temp = 38.5°C ,  Power temp = 35.6°C
2023-06-05 14:55:08.385546500 3180 HW:Seplos BMS 16 cells
2023-06-05 14:55:08.385548500 3180 read serial data seplos
2023-06-05 14:55:08.402166500 3180 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004644E00201FD34\r'
2023-06-05 14:55:10.502105500 3180 checksum error
2023-06-05 14:55:10.553582500 3180 logged to dbus [98.1]
2023-06-05 14:55:10.557532500 3180 Cells:[1]3.507V [2]3.424V [3]3.508V [4]3.429V [5]3.491V [6]3.391V [7]3.506V [8]3.508V [9]3.348V [10]3.479V [11]3.347V [12]3.484V [13]3.365V [14]3.375V [15]3.346V [16]3.347V
2023-06-05 14:55:12.241564500 3180 read status data
2023-06-05 14:55:12.241945500 3180 read serial data seplos
2023-06-05 14:55:12.256513500 3180 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
2023-06-05 14:55:13.368884500 3180 return info data of length 150 : b'0000100DB60D620DB60D670DA60D400DB60DB70D150D990D140D9F0D250D300D120D14060C080BF40BF50C070C2C0C0F018315706B640A6D6003D56D60001603E81575FFF700000BC90B5C'
2023-06-05 14:55:13.378267500 3180 Voltage cell[0]=3.51V
2023-06-05 14:55:13.378271500 3180 Voltage cell[1]=3.426V
2023-06-05 14:55:13.378274500 3180 Voltage cell[2]=3.51V
2023-06-05 14:55:13.378277500 3180 Voltage cell[3]=3.431V
2023-06-05 14:55:13.378279500 3180 Voltage cell[4]=3.494V
2023-06-05 14:55:13.378282500 3180 Voltage cell[5]=3.392V
2023-06-05 14:55:13.378385500 3180 Voltage cell[6]=3.51V
2023-06-05 14:55:13.379239500 3180 Voltage cell[7]=3.511V
2023-06-05 14:55:13.379243500 3180 Voltage cell[8]=3.349V
2023-06-05 14:55:13.379517500 3180 Voltage cell[9]=3.481V
2023-06-05 14:55:13.379919500 3180 Voltage cell[10]=3.348V
2023-06-05 14:55:13.380318500 3180 Voltage cell[11]=3.487V
2023-06-05 14:55:13.380720500 3180 Voltage cell[12]=3.365V
2023-06-05 14:55:13.381149500 3180 Voltage cell[13]=3.376V
2023-06-05 14:55:13.381522500 3180 Voltage cell[14]=3.346V
2023-06-05 14:55:13.389640500 3180 Voltage cell[15]=3.348V
2023-06-05 14:55:13.389645500 3180 Temp cell[0]=34.9°C
2023-06-05 14:55:13.389648500 3180 Temp cell[1]=32.9°C
2023-06-05 14:55:13.389650500 3180 Temp cell[2]=33.0°C
2023-06-05 14:55:13.389653500 3180 Temp cell[3]=34.8°C
2023-06-05 14:55:13.389655500 3180 Current = 3.87A , Voltage = 54.88V
2023-06-05 14:55:13.389658500 3180 Capacity = 274.92/280.0Ah , SOC = 98.1%
2023-06-05 14:55:13.389661500 3180 Cycles = 22
2023-06-05 14:55:13.389663500 3180 Environment temp = 38.5°C ,  Power temp = 35.6°C
2023-06-05 14:55:13.389710500 3180 HW:Seplos BMS 16 cells
2023-06-05 14:55:13.389714500 3180 read serial data seplos
2023-06-05 14:55:13.402686500 3180 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004644E00201FD34\r'
2023-06-05 14:55:15.510185500 3180 checksum error
2023-06-05 14:55:15.528398500 3180 logged to dbus [98.1]
2023-06-05 14:55:15.529041500 3180 Cells:[1]3.51V [2]3.426V [3]3.51V [4]3.431V [5]3.494V [6]3.392V [7]3.51V [8]3.511V [9]3.349V [10]3.481V [11]3.348V [12]3.487V [13]3.365V [14]3.376V [15]3.346V [16]3.348V
2023-06-05 14:55:17.244022500 3180 read status data
2023-06-05 14:55:17.244027500 3180 read serial data seplos
2023-06-05 14:55:17.258428500 3180 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
2023-06-05 14:55:18.370444500 3180 return info data of length 150 : b'0000100DB80D630DB80D680DA80D410DB80DB90D150D9A0D140DA00D260D300D130D14060C080BF40BF50C070C2C0C0F010115726B650A6D6003D56D60001603E81576FFF400000BC90B5C'
2023-06-05 14:55:18.380468500 3180 Voltage cell[0]=3.512V
2023-06-05 14:55:18.380473500 3180 Voltage cell[1]=3.427V
2023-06-05 14:55:18.380476500 3180 Voltage cell[2]=3.512V
2023-06-05 14:55:18.380478500 3180 Voltage cell[3]=3.432V
2023-06-05 14:55:18.380481500 3180 Voltage cell[4]=3.496V
2023-06-05 14:55:18.380483500 3180 Voltage cell[5]=3.393V
2023-06-05 14:55:18.380947500 3180 Voltage cell[6]=3.512V
2023-06-05 14:55:18.381767500 3180 Voltage cell[7]=3.513V
2023-06-05 14:55:18.382865500 3180 Voltage cell[8]=3.349V
[... 113 lines cut ...]
2023-06-05 14:56:00.524328500 3000 Voltage cell[13]=3.375V
2023-06-05 14:56:00.524634500 3000 Voltage cell[14]=3.346V
2023-06-05 14:56:00.524933500 3000 Voltage cell[15]=3.347V
2023-06-05 14:56:00.525264500 3000 Temp cell[0]=34.9°C
2023-06-05 14:56:00.525576500 3000 Temp cell[1]=32.9°C
2023-06-05 14:56:00.525888500 3000 Temp cell[2]=33.0°C
2023-06-05 14:56:00.526252500 3000 Temp cell[3]=34.8°C
2023-06-05 14:56:00.526699500 3000 Current = -0.82A , Voltage = 54.9V
2023-06-05 14:56:00.527040500 3000 Capacity = 274.94/280.0Ah , SOC = 98.1%
2023-06-05 14:56:00.527373500 3000 Cycles = 22
2023-06-05 14:56:00.527687500 3000 Environment temp = 38.4°C ,  Power temp = 35.5°C
2023-06-05 14:56:00.527952500 3000 HW:Seplos BMS 16 cells
[... cut ...] x01'
2023-06-05 14:56:01.645677500 3000 logged to dbus [98.1]
2023-06-05 14:56:01.646345500 3000 Cells:[1]3.513V [2]3.427V [3]3.514V [4]3.432V [5]3.496V [6]3.392V [7]3.512V [8]3.513V [9]3.348V [10]3.483V [11]3.345V [12]3.489V [13]3.364V [14]3.375V [15]3.346V [16]3.347V
2023-06-05 14:56:04.383969500 3000 read status data
2023-06-05 14:56:04.384657500 3000 read serial data seplos
2023-06-05 14:56:04.400117500 3000 wrote 20 bytes to serial port /dev/ttyUSB0, command=b'~20004642E00201FD36\r'
^C

```

schindlerit1984, 2023-06-05:
first one also does with the same error but not in parallel and randomly

wollew, 2023-06-05:
So, if the log output in https://github.com/Louisvdw/dbus-serialbattery/issues/652#issuecomment-1576959884 is complete (what happens between 14:55:25.494761500 and 14:55:25.494761500 ?) we have both PID 3000 and PID 3180 working on the same tty, this cannot work. Are you starting these processes manually or are you getting the messages from the log?

schindlerit1984, 2023-06-05:
nothing manual  - let me restart again and pull the log again

schindlerit1984, 2023-06-05:
what is @40000000647dfd4703548424 5036 wrote 20 bytes to serial port /dev/ttyUSB0, command=b'~20004644E00201FD34\r'
 doing ?

schindlerit1984, 2023-06-05:
maybe this helps 
[
@40000000647dfe8818c7ecec     if chksum != Seplos.int_from_2byte_hex_ascii(data, -5):
@40000000647dfe8833c1810c   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 32, in int_from_2byte_hex_ascii
@40000000647dfe8833c1987c     bytes.fromhex(data[offset : offset + 4].decode("ascii")),
@40000000647dfe8833c1a81c UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 0: ordinal not in range(128)](url)

schindlerit1984, 2023-06-05:
the 2899 was in log for usb0 and 3073 was for usb4 - 

currently running for a few minutes with 0 or 4 going out after each other with the same messages

seems like the driver restarts  - pid changes

schindlerit1984, 2023-06-05:
thats what i am getting on usb4 the moment it dies and comes back 

```
2023-06-05 15:32:55.363738500 8376 HW:Seplos BMS 16 cells
2023-06-05 15:32:55.364218500 8376 read serial data seplos
2023-06-05 15:32:55.382417500 8376 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004644E00201FD34\r'
2023-06-05 15:32:56.474438500 8376 return info data of length 98 : b'00001000000000000000000000000000000000060000000000000000140000000000000300000100000000000000000001'
2023-06-05 15:32:56.480428500 8376 alarm info raw b'00001000000000000000000000000000000000060000000000000000140000000000000300000100000000000000000001'
2023-06-05 15:32:56.481375500 8376 alarm info decoded b'\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x00\x00\x00\x00\x00\x14\x00\x00\x00\x00\x00\x00\x03\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01'
2023-06-05 15:32:56.497176500 8376 logged to dbus [98.0]
2023-06-05 15:32:56.498718500 8376 Cells:[1]3.422V [2]3.383V [3]3.413V [4]3.391V [5]3.423V [6]3.366V [7]3.424V [8]3.409V [9]3.334V [10]3.423V [11]3.332V [12]3.423V [13]3.346V [14]3.355V [15]3.333V [16]3.334V
2023-06-05 15:32:59.229948500 8376 read status data
2023-06-05 15:32:59.230320500 8376 read serial data seplos
2023-06-05 15:32:59.244784500 8376 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
2023-06-05 15:33:00.721588500 Traceback (most recent call last):
2023-06-05 15:33:00.721593500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 361, in publish_battery
2023-06-05 15:33:00.721597500     success = self.battery.refresh_data()
2023-06-05 15:33:00.721600500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 117, in refresh_data
2023-06-05 15:33:00.721604500     result_status = self.read_status_data()
2023-06-05 15:33:00.721607500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 194, in read_status_data
2023-06-05 15:33:00.721688500     data = self.read_serial_data_seplos(
2023-06-05 15:33:00.721692500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 293, in read_serial_data_seplos
2023-06-05 15:33:00.721696500     if not Seplos.is_valid_frame(data):
2023-06-05 15:33:00.721699500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 267, in is_valid_frame
2023-06-05 15:33:00.721702500     if chksum != Seplos.int_from_2byte_hex_ascii(data, -5):
2023-06-05 15:33:00.721730500   File "/opt/victronenergy/dbus-serialbattery/bms/seplos.py", line 32, in int_from_2byte_hex_ascii
2023-06-05 15:33:00.721734500     bytes.fromhex(data[offset : offset + 4].decode("ascii")),
2023-06-05 15:33:00.721738500 UnicodeDecodeError: 'ascii' codec can't decode byte 0x80 in position 0: ordinal not in range(128)
2023-06-05 15:33:00.721742500
2023-06-05 15:33:18.559215500 10080 Starting dbus-serialbattery
2023-06-05 15:33:18.559821500 10080 dbus-serialbattery v1.0.20230526dev
2023-06-05 15:33:18.560131500 10080 Testing Seplos
2023-06-05 15:33:18.560642500 10080 read status data
2023-06-05 15:33:18.561062500 10080 read serial data seplos
2023-06-05 15:33:18.574779500 10080 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
2023-06-05 15:33:19.691020500 10080 return info data of length 150 : b'0000100D6D0D460D660D4F0D700D350D6F0D630D0F0D6F0D150D6F0D1E0D250D0E0D0F060C040BF20BF40C030C230C050661153A6B480A6D6003D46D60001603E81542000800000BCA0B5C'
2023-06-05 15:33:19.697305500 10080 Current = 16.33A , Voltage = 54.34V
2023-06-05 15:33:19.697663500 10080 Capacity = 274.64/280.0Ah , SOC = 98.0%
2023-06-05 15:33:19.697984500 10080 Cycles = 22
2023-06-05 15:33:19.698324500 10080 Environment temp = 37.6°C ,  Power temp = 34.6°C
2023-06-05 15:33:19.698720500 10080 HW:Seplos BMS 16 cells
2023-06-05 15:33:19.699013500 10080 Connection established to Seplos
2023-06-05 15:33:19.699226500 10080 Battery Seplos connected to dbus from /dev/ttyUSB4
2023-06-05 15:33:19.699525500 10080 ========== Settings ==========
2023-06-05 15:33:19.699862500 10080 > Connection voltage: 54.34V | Current: 16.33A | SoC: 98.0%
2023-06-05 15:33:19.700226500 10080 > Cell count: 16 | Cells populated: 0
2023-06-05 15:33:19.700521500 10080 > LINEAR LIMITATION ENABLE: True
2023-06-05 15:33:19.700785500 10080 > MAX BATTERY CHARGE CURRENT: 194.0A | MAX BATTERY DISCHARGE CURRENT: 194.0A
2023-06-05 15:33:19.701104500 10080 > CVCM:     True
2023-06-05 15:33:19.701430500 10080 > MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
2023-06-05 15:33:19.701752500 10080 > CCCM CV:  True  | DCCM CV:  True
2023-06-05 15:33:19.702086500 10080 > CCCM T:   False | DCCM T:   True
2023-06-05 15:33:19.702398500 10080 > CCCM SOC: False | DCCM SOC: True
2023-06-05 15:33:19.751576500 10080 DeviceInstance = 4
2023-06-05 15:33:19.752117500 10080 com.victronenergy.battery.ttyUSB4
2023-06-05 15:33:19.768404500 10080 publish config values = 1
2023-06-05 15:33:24.781941500 10080 read status data
2023-06-05 15:33:24.782319500 10080 read serial data seplos
2023-06-05 15:33:24.794308500 10080 wrote 20 bytes to serial port /dev/ttyUSB4, command=b'~20004642E00201FD36\r'
2023-06-05 15:33:25.911059500 10080 return info data of length 150 : b'0000100D730D4A0D6C0D520D760D360D730D690D0F0D750D130D750D1E0D260D0E0D10060C040BF20BF40C030C230C0500D5153F6B4B0A6D6003D46D60001603E8153E000500000BCA0B5C'
2023-06-05 15:33:25.926719500 10080 Voltage cell[0]=3.443V
2023-06-05 15:33:25.926724500 10080 Voltage cell[1]=3.402V

```

schindlerit1984, 2023-06-05:
confused as well : 

did disconnect usb4 earlier

now reconnected it again  - and so far (fingers crossed) it stays up - already for more then 3-4mins

wollew, 2023-06-05:
Well it is definitely possible to get the Seplos BMS into a state where it doesn't reply anymore. Happened a few times during my initial implementation and also just now while fiddling with it. Had to turn it off and on again to make it behave again.

schindlerit1984, 2023-06-05:
the gps dbus tries to connect to the usb devices

schindlerit1984, 2023-06-05:
But - current state is that it works in parallel - i will keep an eye on it and report again. 

gps dbus definitly tries to use the device - saw it in the watch .. command

schindlerit1984, 2023-06-05:
spoke to soon. its pulling data after doing the disconnect reconnect usb thingy - but does not pull cell voltages and some other details ^^ 

```

2023-06-05 18:26:42.888810500 5497 read status data
2023-06-05 18:26:42.890770500 5497 read serial data seplos
2023-06-05 18:26:42.898261500 5497 wrote 20 bytes to serial port /dev/ttyUSB5, command=b'~20004642E00201FD36\r'
2023-06-05 18:26:44.010920500 5497 return info data of length 150 : b'0000100CF10CEE0CF00CEE0CEF0CED0CEF0CED0CF00CF10CE80CF00CEE0CF10CF00CF2060C010BEF0BF20BFF0C230C0BF7C814B25EFA0A6D6003646D60001603E814B3000600000BCA0B6D'
2023-06-05 18:26:44.016149500 5497 Current = -21.04A , Voltage = 52.98V
2023-06-05 18:26:44.016453500 5497 Capacity = 243.14/280.0Ah , SOC = 86.8%
2023-06-05 18:26:44.016716500 5497 Cycles = 22
2023-06-05 18:26:44.017000500 5497 Environment temp = 37.6°C ,  Power temp = 35.2°C
2023-06-05 18:26:44.017260500 5497 HW:Seplos BMS 16 cells
2023-06-05 18:26:44.017608500 5497 read serial data seplos
2023-06-05 18:26:44.027154500 5497 wrote 20 bytes to serial port /dev/ttyUSB5, command=b'~20004644E00201FD34\r'
2023-06-05 18:26:45.113910500 5497 return info data of length 98 : b'00001000000000000000000000000000000000060000000000000000140000000000000300000100000000000000000001'
2023-06-05 18:26:45.120981500 5497 alarm info raw b'00001000000000000000000000000000000000060000000000000000140000000000000300000100000000000000000001'
2023-06-05 18:26:45.121293500 5497 alarm info decoded b'\x00\x00\x10\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x00\x00\x00\x00\x00\x14\x00\x00\x00\x00\x00\x00\x03\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01'
2023-06-05 18:26:45.130799500 5497 logged to dbus [86.8]
2023-06-05 18:26:45.130803500 5497 Cells:
2023-06-05 18:26:47.890170500 5497 read status data
2023-06-05 18:26:47.890551500 5497 read serial data seplos
2023-06-05 18:26:47.902443500 5497 wrote 20 bytes to serial port /dev/ttyUSB5, command=b'~20004642E00201FD36\r'
2023-06-05 18:26:49.014007500 5497 return info data of length 150 : b'0000100CF10CEE0CF00CEE0CEF0CED0CF00CED0CF10CF10CE90CF00CEF0CF10CF10CF2060C010BEF0BF20BFF0C230C0BF7B314B25EF70A6D6003646D60001603E814B3000400000BCA0B6D'

```

wollew, 2023-06-05:
> gps dbus definitly tries to use the device - saw it in the watch .. command

What exactly did you see there? 

> the disconnect reconnect usb thingy 
What exactly do you mean, did you do anything or did it happen automatically? Please post the full log output, ideally including the command that you used to extract that particular output. If you did something in between, please also add what you did. I have to admit it is a little bit hard to follow which of your outputs belong to what action and or file (not that I am currently having a lot of ideas, but still, it doesn't make it easier). 

One more thing: do you have any idea why your USB TTY numbering changes all the time, do you have any other USB devices connected? It _should_ be consistent if I'm not mistaken but appararently it is not.

schindlerit1984, 2023-06-06:
```
{start-gps.sh} /bin/bash /opt/victronenergy/gps-dbus/start-gps.sh ttyUSB5
 /opt/victronenergy/gps-dbus/gps_dbus -v --banner --dbus system --timeout 2 -s /dev/ttyUSB5 -b 38400

```

will try to use the device sometimes  ( thats from 10h ago). 

during the night i could see that almost every 30 minutes something is going on that interferes with the usb / rs485 connection. the driver then goes into the checksum error loop on ttyUSB0  - USB1 i was not pulling data via dbus mqtt . 

usb numbering was changing because the second controller was running via usb hub  - i had changed this after you wrote and both come up as usb0 +1 now.  

i will try to keep an eye on whats going on every 20-30 mins

schindlerit1984, 2023-06-06:
so its running now since ~3h without any problems. however - what i did was : 

/opt/victronenergy/gps-dbus/start-gps.sh modified to args="-v --banner --dbus system --timeout 2 -s /dev/ttyUSB9"   ( had usb7 before which killed the autodetec from one of the victron mppt´s )

also changed inside the bms/seplos.py the parameter **self.poll_interval = 5000** to **self.poll_interval = 15000** ( it think this does not have any impact) 

with the gps change the device has not disappeared anymore . i will keep monitoring and rebooting once in a wile and report back. 

from my perspective the gps process trying out all the usb´s kills it. maybe the serial driver and dbus serial battery is stuck shortly and the device is "free" again to be tried out for all processes that it goes in a weird mode that serial battery cant take it anymore. 
```
 {start-gps.sh} /bin/bash /opt/victronenergy/gps-dbus/start-gps.sh ttyUSB5
 /opt/victronenergy/gps-dbus/gps_dbus -v --banner --dbus system --timeout 2 -s /dev/ttyUSB5 -b 38400

```

mr-manuel, 2023-06-06:
> also changed inside the bms/seplos.py the parameter self.poll_interval = 5000 to self.poll_interval = 15000 ( it think this does not have any impact)

This is the inverval which is used to poll the data. If the interval is to long, then the serial adapter is free and could be used from other drivers. I'm not sure if this is good, that the driver take 15 seconds to refresh all data. This has to be evaluated.

Please try also at least to run the driver without restarting for 2 days. Better would be 7 days.

schindlerit1984, 2023-06-06:
thanks for the info : if its pulling data only every 5 seconds would this mean that cell voltages are only updated every 5 seconds ? because currently i feel like its faster then that.

mr-manuel, 2023-06-06:
Thats correct. Add a message in the log with the battery ID, so you see exactly which battery updates the cell informations with which interval

schindlerit1984, 2023-06-06:
thanks again for all of your and the teams affords . really appreciate it

wollew, 2023-06-06:
@schindlerit1984 After reading the docs I gathered that everything that serial-starter does should be logged, maybe you can watch (and post) the output of 

`tail -F /data/log/serial-starter/current | tai64nlocal`

wollew, 2023-06-06:
That's really weird, as far as I understand after "Start service dbus-serialbattery.ttyUSB1 once" there should *not* be any more entries in serial-starter for that tty, because dbus-serialbattery on ttyUSB1 would not exit thus no more work for serial-starter for that tty. Can you attach/post the logs of dbus-serialbattery.ttyUSB0 and dbus-serialbattery.ttyUSB1 for that same time frame also?

mr-manuel, 2023-06-11:
I added some new documentation regarding the serial starter. See https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#no-log-file

wollew, 2023-06-11:
Thanks for the added documentation. I'll wait and see if @schindlerit1984 comes up with more logs of the startup. As already stated in https://github.com/Louisvdw/dbus-serialbattery/issues/652#issuecomment-1578385594, I currently don't see that this is a problem of the driver but rather something outside of it, possibly handling the serial ttys. But we'll see, without any additional information I won't be able to work on it, though.

schindlerit1984, 2023-06-11:
Will try in the coming days.

mr-manuel, 2023-07-31:
@schindlerit1984 any updates on this?

closed 2023-09-05 by mr-manuel