Louisvdw/dbus-serialbattery #653: Daly bms high voltage alarm

thomasinaz41213, 2023-05-20:
### Describe the bug

Installed the latest nightly 1.0.20230519beta and getting high voltage alarm. everything appears to be functioning properly but the alarm will not clear even though there is not actually a high voltage situation.

### How to reproduce

Steps to reproduce the behavior:
installed nightly 0519 and high voltage alarm appeared. revert back to 0508 and alarm disappeared. reinstall 0519 and alarm reappears.

### Expected behavior

should be no high voltage alarm since the voltage is not above the threshold.

### Driver version

1.0.20230519beta

### Venus OS device type

Cerbo GX

### Venus OS version

V3.00~42

### BMS type

Daly Smart BMS

### Cell count

8

### Connection type

Serial USB adapter to TTL

### Config file

```ini
TEMPERATURE_LIMITS_WHILE_CHARGING = 0,   2,   5,  10,  15, 20, 35,  40, 55
MAX_CHARGE_CURRENT_T_FRACTION     = 0, 0.1, 0.2, 0.4, 0.8,  1,  1, 0.4,  0

TEMPERATURE_LIMITS_WHILE_DISCHARGING = -20,   0,   5,  10, 15, 45, 55
MAX_DISCHARGE_CURRENT_T_FRACTION     =   0, 0.2, 0.3, 0.4,  1,  1,  0

; --------- SOC limitation (affecting CCL/DCL) ---------
; Description: Maximal charge / discharge current will be increased / decreased depending on State of Charge,
;              see CC_SOC_LIMIT1 etc.
; Example: The SoC limit will be monitored to control the currents.
; Charge current control management enable (True/False).
CCCM_SOC_ENABLE = False
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = False

; Charge current soc limits
CC_SOC_LIMIT1 = 98
CC_SOC_LIMIT2 = 95
CC_SOC_LIMIT3 = 91

; Charge current limits
CC_CURRENT_LIMIT1_FRACTION = 0.1
CC_CURRENT_LIMIT2_FRACTION = 0.3
CC_CURRENT_LIMIT3_FRACTION = 0.5

; Discharge current soc limits
DC_SOC_LIMIT1 = 10
DC_SOC_LIMIT2 = 20
DC_SOC_LIMIT3 = 30

; Discharge current limits
DC_CURRENT_LIMIT1_FRACTION = 0.1
DC_CURRENT_LIMIT2_FRACTION = 0.3
DC_CURRENT_LIMIT3_FRACTION = 0.5

; --------- Time-To-Go ---------
; Description: Calculates the time to go shown in the GUI
;              Recalculation is done based on TIME_TO_SOC_RECALCULATE_EVERY
TIME_TO_GO_ENABLE = True

; --------- Time-To-Soc ---------
; Description: Calculates the time to a specific SoC
; Example: TIME_TO_SOC_POINTS = 50, 25, 15, 0
;          6h 24m remaining until 50% SoC
;          17h 36m remaining until 25% SoC
;          22h 5m remaining until 15% SoC
;          28h 48m remaining until 0% SoC
; Set of SoC percentages to report on dbus and MQTT. The more you specify the more it will impact system performance.
; [Valid values 0-100, comma separated list. More that 20 intervals are not recommended]
; Example: TIME_TO_SOC_POINTS = 100, 95, 90, 85, 75, 50, 25, 20, 10, 0
; Leave empty to disable
TIME_TO_SOC_POINTS = 100, 20, 10
; Specify TimeToSoc value type [Valid values 1, 2, 3]
; 1 Seconds
; 2 Time string <days>d <hours>h <minutes>m <seconds>s
; 3 Both seconds and time string "<seconds> [<days>d <hours>h <minutes>m <seconds>s]"
TIME_TO_SOC_VALUE_TYPE = 2
; Specify in seconds how often the TimeToSoc should be recalculated
; Minimum are 5 seconds to prevent CPU overload
TIME_TO_SOC_RECALCULATE_EVERY = 60
; Include TimeToSoC points when moving away from the SoC point [Valid values True, False]
; These will be as negative time. Disabling this improves performance slightly
TIME_TO_SOC_INC_FROM = False

; --------- Additional settings ---------
; Specify only one BMS type to load else leave empty to try to load all availabe
; LltJbd, Ant, Daly, Daly, Jkbms, Lifepower, Renogy, Renogy, Ecs
BMS_TYPE = Daly

; Publish the config settings to the dbus path "/Info/Config/"
PUBLISH_CONFIG_VALUES = 1

; Select the format of cell data presented on dbus [Valid values 0,1,2,3]
; 0 Do not publish all the cells (only the min/max cell data as used by the default GX)
; 1 Format: /Voltages/Cell (also available for display on Remote Console)
; 2 Format: /Cell/#/Volts
; 3 Both formats 1 and 2
BATTERY_CELL_DATA_FORMAT = 1

; Simulate Midpoint graph (True/False).
MIDPOINT_ENABLE = False

; Battery temperature
; Specifiy how the battery temperature is assembled
; 0 Get mean of temperature sensor 1 and temperature sensor 2
; 1 Get only temperature from temperature sensor 1
; 2 Get only temperature from temperature sensor 2
TEMP_BATTERY = 0

; Temperature sensor 1 name
TEMP_1_NAME = Temp 1

; Temperature sensor 2 name
TEMP_2_NAME = Temp 2

; --------- BMS specific settings ---------

; -- LltJbd settings
; SoC low levels
; NOTE: SOC_LOW_WARNING is also used to calculate the Time-To-Go even if you are not using a LltJbd BMS
SOC_LOW_WARNING = 20
SOC_LOW_ALARM   = 10

; -- Daly settings
; Battery capacity (amps), if the BMS does not support reading it
BATTERY_CAPACITY = 560
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = -1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT = 15

; --------- Battery monitor specific settings ---------
; If you are using a SmartShunt or something else as a battery monitor, the battery voltage reported
; from the BMS and SmartShunt could differ. This causes, that the driver never goapplies the float voltage,
; since max voltage is never reached.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 and the battery is now charged until the SmartShunt measures 55.20 V. The BMS
;     now measures 55.05 V since there is a voltage drop of 0.15 V. Since the dbus-serialbattery measures
;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
;     Set VOLTAGE_DROP to 0.15
VOLTAGE_DROP = 0.10
root@einstein:/data/etc/dbus-serialbattery#
```

### Relevant log output

```shell
im not a programmer can't figure this out
```

mr-manuel, 2023-05-20:
In the versions before the alarms were not working for Daly, so it can be that you already had this alarm on your BMS, but never saw it. Since the alarms are triggered by the BMS please check your BMS settings and alarms.

thomasinaz41213, 2023-05-20:
High voltage alarm is set to 29.2 in the bms. In the config.ini file the max voltage is set to 28.6 and float to 27.6. if I plug the bms to the computer with the same uart adapter there is no high voltage alarm present. If I replace the uart adapter with the Bluetooth adapter and connect with the smart bms app same result, no high voltage alarm. DC volts when connected to the cerbo is as expected...increases to 28.6 then slowly falls to 27.6 and stays there. Bms reports the voltage as 0.1 v lower.

thomasinaz41213, 2023-05-21:
A factory reset of the bms fixed the problem. Not sure why no alarms were indicated on the bms software, but issue is fixed. You can close this.

janekx, 2023-06-10:
Now facing same issue with stable build. 3 batts with daly, report high voltage like 55,7V but bms have 58V limit.

thomasinaz41213, 2023-06-10:
Have you added the correct settings to the config.ini file as directed in the install/setup instructions?

janekx, 2023-06-10:
> Have you added the correct settings to the config.ini file as directed in the install/setup instructions?

sorry not sure what you mean i have use config.default.ini and change values for mu setup

mr-manuel, 2023-06-10:
Could you please post a screenshot of your Daly settings?

Which firmware/hardware version does your Daly have?

janekx, 2023-06-10:
[image]

janekx, 2023-06-10:
[image]
[image]
[image]
[image]

mr-manuel, 2023-06-10:
You have to set the error and warning levels via PC software. With the app this is not possible on Daly BMS.

janekx, 2023-06-11:
Thanks will try setup on my netebook and will feedback

closed 2023-06-12 by mr-manuel

mr-manuel, 2023-06-12:
Please provide some screenshots

janekx, 2023-06-12:
[image]
[image]
[image]
[image]
[image]
[image]

janekx, 2023-06-19:
Hi any idea @mr-manuel

mr-manuel, 2023-06-19:
No, since I‘m not using Daly. Maybe @transistorgit @iLeeeZi

teefixx, 2023-06-20:
Hi, 
I had a similar problem spiking one cell in the measurements causing a high voltage alarm. 
Root cause was a loose sense line (cable) which made the value jumping. Every time it jumped, the high voltage error came up. Tighten that line solved the problem.
The spiking was not visible in the Daly App, recording the values with influxDB showed it.

janekx, 2023-06-20:
well seens not your issue as it makes all 3 batts. i must care of better top balancing then. For now how to disable the alarms readings in driver?

mr-manuel, 2023-06-20:
You can change the values `2` and `1` to `0`:

https://github.com/Louisvdw/dbus-serialbattery/blob/86a6b956e15bee21d5a298283c9d121f0eb82a02/etc/dbus-serialbattery/bms/daly.py#L257-L264

janekx, 2023-06-20:
thanks, maybe good to add to config.ini to setup

transistorgit, 2023-06-20:
In the last screenshots (error list) you can see that the error is a high cell voltage and that the cell volts diff is quite high. So you can configure your warning/error level for cell voltages higher or check if your balancing is working as expected. The bms internal balancing is weak, so maybe you need a parallel active balancer.