Louisvdw/dbus-serialbattery #655: Bluetooth connection disrupted

idstein, 2023-05-21:
### Describe the bug

Using a cronjob to restart at 0 and 12 the entire Bluetooth stack is really disruptive. Can we remove that restart line?

Otherwise it will interrupt the BMS communication for several seconds during that period.

JKBMS BLE may require it, but JBD BMS BLE does not and has built in recovery mechanisms. 

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

Bluetooth Stack is not restarted

### Driver version

1.0

### Venus OS device type

Raspberry Pi

### Venus OS version

3.0

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

18

### Connection type

Bluetooth

### Config file

```ini
Default
```

### Relevant log output

```shell
.
```

mr-manuel, 2023-05-21:
The JKBMS also do not require it, but we tracked it down to the bluez stack. It kills the driver, if there is no more answer. We tried the driver also with Raspian and it was no problem.

The bluez stack freezes after 15 hours and 20 minutes and has then to be killed to recover. This issue is known by bluez.

See the links here https://github.com/Louisvdw/dbus-serialbattery/discussions/485

idstein, 2023-05-21:
Super strange. I do not have had these issues before with 2.93 VenusOS and was running my JBD code for 60+ days without intervention.

mr-manuel, 2023-05-21:
We tried with Venus OS 2.92, bleak 5.66 if I remember correctly. Habe to check, if they updated it maybe.

mr-manuel, 2023-05-22:
On what device are you running Venus OS? Are you also using an external Bluetooth adapter?

idstein, 2023-05-22:
Linux raspberrypi4 5.10.110-rpi-venus-3 #1 SMP Fri May 12 18:57:15 UTC 2023 armv7l GNU/Linux

VenusOS 3.0.0~42
Bleak 0.20.2
BlueZ 5.55.

It was running without any bluetooth restart for 24h. I'm now constantly checking the logs (A) dbus-blebattery and (B) messages for something related to bluetooth.

mr-manuel, 2023-05-22:
@baranator could you also try to remove the cronjob with Venus OS `v3.00~42`? I have no more possibility to connect my JKBMS over Bluetooth.

baranator, 2023-05-26:
I Installed v.3.00~42 and dbus-serialbattery master yesterday evening.

My Bluetooth-connection suddenly unstable and reconnects every 20s, even when calling m `jkbms_brn.py` seperately and independently from `dbus-serialbattery` on the device, so i assume there is an error in v3.00~42 :(

mr-manuel, 2023-05-26:
Can you try to install from the dev branch?

baranator, 2023-05-27:
> Can you try to install from the dev branch? Run the `install.sh`, select `3` and then `2`.

got the same behaviour on almost every software-combination; then disabled raspi3 internal bluetooth and went back to external bluetooth-dongle again. That seemed to be the culprit for the 20s-Disconnect. I will let it run now for two days to verify if everything works for now "as it is" and then remove the cron.

mr-manuel, 2023-06-11:
@baranator do you maybe have some news? In the meanwhile an issue https://github.com/Louisvdw/dbus-serialbattery/issues/693 was opened and the driver was updated.

drago-more, 2023-06-13:
The issue also exists for dbus-serialbattery v1.0.20230612beta. The BT connections die at around 00:05 GMT and 14:05 GMT, respectively.

idstein, 2023-06-13:
All logs I've seen so far are related to JKBMS. Could it be an internal problem with JKBMS implementation? Has anyone also the BT connection salvation with JBD BMS?

mr-manuel, 2023-06-13:
> The issue also exists for dbus-serialbattery v1.0.20230612beta. The BT connections die at around 00:05 GMT and 14:05 GMT, respectively.

This is due to the cronjob. The cronjob is removed in the next beta. But anyway the driver should not behave like this. I'm searching for a solution, but have no system to test. Would you be available to contact me on Discord and give me direct SSH access to your GX device to find a solution?

drago-more, 2023-06-13:
@mr-manuel A couple of days ago I removed the cronjob manually, running dbus-serialbattery v1.0.20230610beta at this time. The BT connection broke after about 15 hours ... as you mentioned above. So it seems that the bluez stack bug is still there. Though, happy to test your next beta.

Unfortunately I am not using Discord. You can drop me a mail: rays.chard.0vATicloud.com

drago-more, 2023-06-15:
@mr-manuel I did some further tests. The issue can be reproduced by just restarting the BT stack with `/etc/init.d/bluetooth restart`. After running `bash /data/etc/dbus-serialbattery/reinstall-local.sh` the BT connection is reestablished. So it seems that when restarting BT in cronjob you also have to perform a "reconnect".

P.S. By the way, check your mail if you haven't already.

drago-more, 2023-06-15:
@mr-manuel 
[Data.rtfd.zip](https://github.com/Louisvdw/dbus-serialbattery/files/11756239/Data.rtfd.zip)

mr-manuel, 2023-06-25:
Crodjob was removed. A new issue arised. See https://github.com/Louisvdw/dbus-serialbattery/issues/720

closed 2023-06-25 by mr-manuel