Louisvdw/dbus-serialbattery #660: Spikes with Solarchargers, are they normal?

pos-ei-don, 2023-05-22:
### Describe the bug

Attached is a Screenshot. On all Solarchargers connected to the Victron System, 
I get spikes in the power-values when serialbattery is used.
I don't geht them if i do not use serialbattery, but then the solarchargers work in "stand alone" mode and not in 
network mode.
i am not certain if this is a serialbattery problem or if it is designed like this in "network mode", as i dont have
a CAN-bus battery to check.
I think, maybe these spikes are while the mppt researches for the best voltage, but maybe someone knows
if I should just ignore this.
The blue and olive-green solarcharger is connected voa viczton direct.ve calbe to the venusos.
The red and green sholarcharger is disconnected.

I did reproduce it in a fresh install.

Tnaky you a lot!

[image]

### How to reproduce

see above.

### Expected behavior

in thinx the spikes are not needed.

### Driver version

master

### Venus OS device type

Raspberry Pi

### Venus OS version

2,94

### BMS type

JKBMS / Heltec BMS

### Cell count

10*16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
-
```

### Relevant log output

```shell
-
```

pos-ei-don, 2023-05-28:
OK,  i did not find a solution, but will close this as nö one seems to have an idea why this is happening.

mr-manuel, 2023-05-28:
Hi, why did you not pasted your config file?

What is your driver version? If you opened the issue there was a description where you have to check it.

Could it maybe be, that it‘s temperature related?

seidler2547, 2023-06-05:
Yes, this is partial shading detection. I disabled it on mine and without it looks like this:
[image]

With it enabled it looks like yours:
[image]

pos-ei-don, 2023-06-05:
Very interesting, thank you! Will give it a try on one System of my two, so i can compare it.

mr-manuel, 2023-06-11:
@pos-ei-don can we close this?

pos-ei-don, 2023-06-11:
Yes, its fixed by switch Ing this of.
It is still present with Devices which can not set this Feature because they Don't have Bluetooth and dlthis Feature cannot be set via victron console.
But this is nothing which belongs to this great pice of Software.

Thank you!

closed 2023-06-11 by pos-ei-don