Louisvdw/dbus-serialbattery #665: Permission denied

Seb-Man, 2023-05-23:
### Describe the problem

Hello,
I can not change the DCL value on version V1.0.0.
I got a "Permission denied" message.
I don't understand.

Can you help me?

Thank you so much

### Driver version

1.0.0 (20230508)

### Venus OS device type

Cerbo GX

### Venus OS version

2.94

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
login as: root
root@192.168.1.170's password:
Last login: Tue May 23 18:56:05 2023
root@einstein:~# /data/etc/dbus-serialbattery/reinstall-local.sh
Venus OS v2.92 is older than v3.00~14. Replacing VisibleItemModel with VisualIte                                                                                                                                                             mModel... done.
sed: /var/spool/cron/root: No such file or directory
No Bluetooth battery configuration found in "/data/etc/dbus-serialbattery/config                                                                                                                                                             .ini".
You can ignore this, if you are using only a serial connection.

SERIAL battery connection: The installation is complete. You don't have to do an                                                                                                                                                             ything more.

BLUETOOTH battery connection: There are a few more steps to complete installatio                                                                                                                                                             n.

    1. Please add the Bluetooth BMS to the config file "/data/etc/dbus-serialbat                                                                                                                                                             tery/config.ini" by adding "BLUETOOTH_BMS":
       Example with 1 BMS: BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
       Example with 3 BMS: BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Bl                                                                                                                                                             e C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
       If your Bluetooth BMS are nearby you can show the MAC address with "bluet                                                                                                                                                             oothctl devices".

    2. Make sure to disable Settings -> Bluetooth in the remote console/GUI to p                                                                                                                                                             revent reconnects every minute.

    3. Re-run "/data/etc/dbus-serialbattery/reinstall-local.sh", if the Bluetoot                                                                                                                                                             h BMS were not added to the "config.ini" before.

    ATTENTION!
    If you changed the default connection PIN of your BMS, then you have to pair                                                                                                                                                              the BMS first using OS tools like the "bluetoothctl".
    See https://wiki.debian.org/BluetoothUser#Using_bluetoothctl for more detail                                                                                                                                                             s.

CUSTOM SETTINGS: If you want to add custom settings, then check the settings you                                                                                                                                                              want to change in "/data/etc/dbus-serialbattery/config.default.ini"
                 and add them to "/data/etc/dbus-serialbattery/config.ini" to pe                                                                                                                                                             rsist future driver updates.

root@einstein:~# /data/etc/dbus-serialbattery/config.default.ini
-sh: /data/etc/dbus-serialbattery/config.default.ini: Permission denied
root@einstein:~#
```

### Relevant log output

```shell
login as: root
root@192.168.1.170's password:
Last login: Tue May 23 18:56:05 2023
root@einstein:~# /data/etc/dbus-serialbattery/reinstall-local.sh
Venus OS v2.92 is older than v3.00~14. Replacing VisibleItemModel with VisualIte                                                                                                                                                             mModel... done.
sed: /var/spool/cron/root: No such file or directory
No Bluetooth battery configuration found in "/data/etc/dbus-serialbattery/config                                                                                                                                                             .ini".
You can ignore this, if you are using only a serial connection.

SERIAL battery connection: The installation is complete. You don't have to do an                                                                                                                                                             ything more.

BLUETOOTH battery connection: There are a few more steps to complete installatio                                                                                                                                                             n.

    1. Please add the Bluetooth BMS to the config file "/data/etc/dbus-serialbat                                                                                                                                                             tery/config.ini" by adding "BLUETOOTH_BMS":
       Example with 1 BMS: BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
       Example with 3 BMS: BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Bl                                                                                                                                                             e C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
       If your Bluetooth BMS are nearby you can show the MAC address with "bluet                                                                                                                                                             oothctl devices".

    2. Make sure to disable Settings -> Bluetooth in the remote console/GUI to p                                                                                                                                                             revent reconnects every minute.

    3. Re-run "/data/etc/dbus-serialbattery/reinstall-local.sh", if the Bluetoot                                                                                                                                                             h BMS were not added to the "config.ini" before.

    ATTENTION!
    If you changed the default connection PIN of your BMS, then you have to pair                                                                                                                                                              the BMS first using OS tools like the "bluetoothctl".
    See https://wiki.debian.org/BluetoothUser#Using_bluetoothctl for more detail                                                                                                                                                             s.

CUSTOM SETTINGS: If you want to add custom settings, then check the settings you                                                                                                                                                              want to change in "/data/etc/dbus-serialbattery/config.default.ini"
                 and add them to "/data/etc/dbus-serialbattery/config.ini" to pe                                                                                                                                                             rsist future driver updates.

root@einstein:~# /data/etc/dbus-serialbattery/config.default.ini
-sh: /data/etc/dbus-serialbattery/config.default.ini: Permission denied
root@einstein:~#
```

mr-manuel, 2023-05-23:
Did you check the [FAQ](https://louisvdw.github.io/dbus-serialbattery/faq/)?

Seb-Man, 2023-05-23:
yes several times, but I can't do it

Seb-Man, 2023-05-23:
I had done it on a previous version, but on version V1.0.0 I don't understand the process

Seb-Man, 2023-05-23:
it is not possible to perform the same operation as on the previous versions?
I'm a beginner I have a hard time understanding

mr-manuel, 2023-05-24:
Unfortunately not. Just read correctly the documentation and my post above.

Alone this command `nano /data/etc/dbus-serialbattery/FILE_NAME` shows me that you did not read the documentation, but only copy and pasted the command you found. If you were able to edit the `utils.py` I don't know, why you are not able to edit the `config.ini`...

Here again the links to the documentation. Read all and it will work.

[Settings location/path](https://louisvdw.github.io/dbus-serialbattery/general/install/#settings-locationpath)

[How to edit utils.py or config.ini](https://louisvdw.github.io/dbus-serialbattery/general/install/#how-to-edit-utilspy-or-configini)

Seb-Man, 2023-05-24:
I'm really not good, I spent several hours trying to change the settings but I can't do it...
I don't know if it's a translation problem as I'm French and don't speak English.
Attached is a screenshot, is this the right way?
[image]

mr-manuel, 2023-05-24:
Try deepl.com instead of Google translate. I speak four languages, but not french.

The file should look like this, so you are close.

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

Seb-Man, 2023-05-24:
thank you for your patience
I made an error that bugged the system
I can't get the V1.0.0 version to work 
I can only manage to install v0.14.3
Is this normal?
Thank you

mr-manuel, 2023-05-25:
I think the biggest problem in your case is the language barrier.

You can [uninstall and remove](https://louisvdw.github.io/dbus-serialbattery/general/install#uninstallremove-the-driver) the driver and then reinstall it.

Seb-Man, 2023-05-25:
I uninstalled several and reinstalled like the screenshot and I have version 0.14.3 and not V1.0.0
I do not understand why V1.0.0 is the latest version recommended?
[image]

mr-manuel, 2023-05-25:
> I do not understand why V1.0.0 is the latest version recommended?

Because it contains a lot of fixes and improvements. Shortly it will be released as stable version.

Seb-Man, 2023-05-25:
I don't know what I'm doing wrong?
I've spent a lot of time today trying to understand, but it still doesn't work. My home is really bothered by this lack of energy.
Can you explain the procedure to me differently, perhaps a translation problem.

Thank you very much.
[image]

Seb-Man, 2023-05-27:
hello,
always me with my problems ;-)
I have a new problem, I no longer have the battery percentage displayed on the home screen even though the bms communicates well with the GX.
I've uninstalled the driver and reinstalled several times and it's still the same...
Do you have any idea how to solve my problem?
thank you
[image]

[image]

Seb-Man, 2023-06-01:
Hello,
I have just installed the new version v1.0.20230531.
It's great, many thanks for all the work you've done.
But I still can't modify the CCL and DCL.

I tried with this code but still no change.
I don't understand what I'm doing wrong.

Can you enlighten me?

Thank you

nano /data/etc/dbus-serialbattery
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 120.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; --------- Charge mode ---------
; Choose the mode for voltage / current limitations (True / False)
; False is a step mode: This is the default with limitations on hard boundary steps
; True is a linear mode:
;     For CCL and DCL the values between the steps are calculated for smoother values (by WaldemarFech)
;     For CVL max battery voltage is calculated dynamically in order that the max cell voltage is not exceeded
LINEAR_LIMITATION_ENABLE = True

; Specify in seconds how often the linear values should be recalculated
LINEAR_RECALCULATION_EVERY = 60
; Specify in percent when the linear values should be recalculated immediately
; Example: 5 for a immediate change, when the value changes by more than 5%
LINEAR_RECALCULATION_ON_PERC_CHANGE = 5

; --------- Charge Voltage limitation (affecting CVL) ---------
; Description: Limit max charging voltage (MAX_CELL_VOLTAGE * cell count), switch from max voltage to float
;              voltage (FLOAT_CELL_VOLTAGE * cell count) and back
;     False: Max charging voltage is always kept
;     True: Max charging voltage is reduced based on charge mode
;         Step mode: After max voltage is reached for MAX_VOLTAGE_TIME_SEC it switches to float voltage. After
;                    SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT it switches back to max voltage.
;         Linear mode: After max voltage is reachend and cell voltage difference is smaller or equal to
;                      CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL it switches to float voltage after 300 (fixed)
;                      additional seconds.
;                      After cell voltage difference is greater or equal to CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT
;                      OR
;                      SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT
;                      it switches back to max voltage.
; Example: The battery reached max voltage of 55.2V and hold it for 900 seconds, the the CVL is switched to
;          float voltage of 53.6V to don't stress the batteries. Allow max voltage of 55.2V again, if SoC is
;          once below 90%
;          OR
;          The battery reached max voltage of 55.2V and the max cell difference is 0.010V, then switch to float
;          voltage of 53.6V after 300 additional seconds to don't stress the batteries. Allow max voltage of
;          55.2V again if max cell difference is above 0.080V or SoC below 90%.
; Charge voltage control management enable (True/False).
CVCM_ENABLE = True

; -- CVL reset based on cell voltage diff (linear mode)
; Specify cell voltage diff where CVL limit is kept until diff is equal or lower
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.010
; Specify cell voltage diff where CVL limit is reset to max voltage, if value get above
; the cells are considered as imbalanced, if the cell diff exceeds 5% of the nominal cell voltage
[... 137 lines cut ...]
; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT = 15

; -- HeltecModbus (Heltec SmartBMS/YYBMS) settings
; Set the Modbus addresses from the adapters
; Separate each address to check by a comma like: 1, 2, 3, ...
; factory default address will be 1
HELTEC_MODBUS_ADDR = 1

; --------- Battery monitor specific settings ---------
; If you are using a SmartShunt or something else as a battery monitor, the battery voltage reported
; from the BMS and SmartShunt could differ. This causes, that the driver never goapplies the float voltage,
; since max voltage is never reached.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 and the battery is now charged until the SmartShunt measures 55.20 V. The BMS
;     now measures 55.05 V since there is a voltage drop of 0.15 V. Since the dbus-serialbattery measures
;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
;     Set VOLTAGE_DROP to 0.15
VOLTAGE_DROP = 0.00

Seb-Man, 2023-06-01:
I'm sorry, I selected the wrong notepad.

nano /data/etc/dbus-serialbattery/config.default.ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 120.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; --------- Charge mode ---------
; Choose the mode for voltage / current limitations (True / False)
; False is a step mode: This is the default with limitations on hard boundary steps
; True is a linear mode:
;     For CCL and DCL the values between the steps are calculated for smoother values (by WaldemarFech)
;     For CVL max battery voltage is calculated dynamically in order that the max cell voltage is not exceeded
LINEAR_LIMITATION_ENABLE = True

; Specify in seconds how often the linear values should be recalculated
LINEAR_RECALCULATION_EVERY = 60
; Specify in percent when the linear values should be recalculated immediately
; Example: 5 for a immediate change, when the value changes by more than 5%
LINEAR_RECALCULATION_ON_PERC_CHANGE = 5

; --------- Charge Voltage limitation (affecting CVL) ---------
; Description: Limit max charging voltage (MAX_CELL_VOLTAGE * cell count), switch from max voltage to float
;              voltage (FLOAT_CELL_VOLTAGE * cell count) and back
;     False: Max charging voltage is always kept
;     True: Max charging voltage is reduced based on charge mode
;         Step mode: After max voltage is reached for MAX_VOLTAGE_TIME_SEC it switches to float voltage. After
;                    SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT it switches back to max voltage.
;         Linear mode: After max voltage is reachend and cell voltage difference is smaller or equal to
;                      CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL it switches to float voltage after 300 (fixed)
;                      additional seconds.
;                      After cell voltage difference is greater or equal to CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT
;                      OR
;                      SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT
;                      it switches back to max voltage.
; Example: The battery reached max voltage of 55.2V and hold it for 900 seconds, the the CVL is switched to
;          float voltage of 53.6V to don't stress the batteries. Allow max voltage of 55.2V again, if SoC is
;          once below 90%
;          OR
;          The battery reached max voltage of 55.2V and the max cell difference is 0.010V, then switch to float
;          voltage of 53.6V after 300 additional seconds to don't stress the batteries. Allow max voltage of
;          55.2V again if max cell difference is above 0.080V or SoC below 90%.
; Charge voltage control management enable (True/False).
CVCM_ENABLE = True

; -- CVL reset based on cell voltage diff (linear mode)
; Specify cell voltage diff where CVL limit is kept until diff is equal or lower
CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.010
; Specify cell voltage diff where CVL limit is reset to max voltage, if value get above
; the cells are considered as imbalanced, if the cell diff exceeds 5% of the nominal cell voltage
; e.g. 3.2 V * 5 / 100 = 0.160 V
CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.080

; -- CVL reset based on SoC option (step mode)
; Specify how long the max voltage should be kept, if reached then switch to float voltage
MAX_VOLTAGE_TIME_SEC = 900
; Specify SoC where CVL limit is reset to max voltage, if value gets below
SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 90

; --------- Cell Voltage Current limitation (affecting CCL/DCL) ---------
[... 127 lines cut ...]
; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT = 15

; -- HeltecModbus (Heltec SmartBMS/YYBMS) settings
; Set the Modbus addresses from the adapters
; Separate each address to check by a comma like: 1, 2, 3, ...
; factory default address will be 1
HELTEC_MODBUS_ADDR = 1

; --------- Battery monitor specific settings ---------
; If you are using a SmartShunt or something else as a battery monitor, the battery voltage reported
; from the BMS and SmartShunt could differ. This causes, that the driver never goapplies the float voltage,
; since max voltage is never reached.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 and the battery is now charged until the SmartShunt measures 55.20 V. The BMS
;     now measures 55.05 V since there is a voltage drop of 0.15 V. Since the dbus-serialbattery measures
;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
;     Set VOLTAGE_DROP to 0.15
VOLTAGE_DROP = 0.00

Seb-Man, 2023-06-01:
I copied the code above, then Ctrl O and Ctrl X, then reboot

[image]
[image]

then I can no longer access the bms

[image]

Seb-Man, 2023-06-05:
Hello,
a friend came to help me translate.
I finally managed to change the values.

Thank you for your help, you're doing a great job!

Thank you