Louisvdw/dbus-serialbattery #666: Auto reset soc

thomasinaz41213, 2023-05-23:
### Is your feature request related to a problem? Please describe.

Many bms only reset soc to 100% if a high cell volt condition is reached. I would assume most people would prefer not to charge to this high of a voltage on regular interval.

### Describe the solution you'd like

Current nightly build has added the ability to set the SOC through the battery driver. It would be great if an option could be included, which could be selected from within the config file, that would automatically set the SOC when conditions are met to exit maximum voltage and enter float voltage. 

### Describe alternatives you've considered

It is possible to occasionally manually reset the SOC through the device menu of the battery driver, but this is not ideal.

mr-manuel, 2023-05-24:
Unfortunately this is not possible with every BMS. Which BMS are you using?

thomasinaz41213, 2023-05-24:
Daly smart bms

transistorgit, 2023-05-24:
I just made a suggestion [here](https://github.com/Louisvdw/dbus-serialbattery/discussions/657), please upvote. We can detail the requirements out there

thomasinaz41213, 2023-05-24:
> 

Are you in the USA? I have a 100amp 8s Daly I would be willing to donate and send to you for the cause.

mr-manuel, 2023-05-24:
No, I'm from Italy as you can see in my GitHub profile :-)

transistorgit, 2023-05-24:
You can try with dbus-spy and then implenent in node-red

mr-manuel, 2023-05-24:
@thomasinaz41213 you can subscribe to all MQTT topics and search for ResetSoc. So you can also write to it over MQTT.

thomasinaz41213, 2023-05-24:
I wish I knew how to do any of that, but I am quite illiterate for programming.

transistorgit, 2023-05-25:
nice. then please close this issue (we have also the feature request discussion open)

closed 2023-05-25 by mr-manuel