Louisvdw/dbus-serialbattery #670: JBD BMS Force Charge/Discharge disable controls

subnoize417, 2023-05-25:
### Is your feature request related to a problem? Please describe.

I saw the recent addition of charge/discharge mosfet controls, and would like the feature to be expanded to more BMS's

### Describe the solution you'd like

I was able to get it working with 2 of my batteries, after some trial and error, using code from [batmon-ha](https://github.com/fl4p/batmon-ha), I know just enough python to hit it with a hammer

### Describe alternatives you've considered

Cobbling together some code that is working, but likely not the ideal way to integrate into the project.

### Additional context

I made 2 modifications to [lltjbd.py](https://github.com/Louisvdw/dbus-serialbattery/blob/dev/etc/dbus-serialbattery/bms/lltjbd.py), this bit into the `read_gen_data` function,
```
       switches=dict(
                discharge=fet == 2 or fet == 3,
                charge=fet == 1 or fet == 3,
            )
        self._switches = dict(switches)
```

and this is handling the callbacks
```
    def force_charging_off_callback(self, path, value):
        
        if value is None:
            return False

        if value == 0:
            switch = "charge"
            state = True
            self.set_switch(switch, state)
            return True

        if value == 1:
            logger.info(f"charging callback 1")
            switch = "charge"
            state = False
            self.set_switch(switch, state)
            return True

        return False

    def force_discharging_off_callback(self, path, value):
        if value is None:
            return False

        if value == 0:
            switch = "discharge"
            state = True
            self.set_switch(switch, state)
            return True

        if value == 1:
            switch = "discharge"
            state = False
            self.set_switch(switch, state)
            return True

        return False

    def set_switch(self, switch: str, state: bool):

        assert switch in {"charge", "discharge"}
        
        new_switches = {**self._switches, switch: state}
        switches_sum = sum(new_switches.values())
        if switches_sum == 2:
            tc = 0x00  # all on
        elif switches_sum == 0:
            tc = 0x03  # all off
        elif (switch == "charge" and not state) or (switch == "discharge" and state):
            tc = 0x01  # charge off
        else:
            tc = 0x02  # charge on, discharge off

        mosdata = bytes([0x00, tc])
        #logger.info(f"write fet cmd {REG_CTRL_MOSFET, mosdata}")
        self.read_serial_data_llt(writeCmd(REG_CTRL_MOSFET, mosdata))
```

Most of that was copied directly from [batmon-ha](https://github.com/fl4p/batmon-ha), with a bit from the daly bms file.
I'd create a pull request, but don't really know how

mr-manuel, 2023-05-26:
@idstein maybe you can check this?

idstein, 2023-05-26:
Yes, have a branch on that and will open a PR soon after testing.
I would also like to add balancing control support and reading more internal limits (cell voltages). Plan is to read the entire config once to have access to the values.

mr-manuel, 2023-07-27:
This feature war implemented by @idstein

closed 2023-07-27 by mr-manuel