Louisvdw/dbus-serialbattery #683: CVL reset

Joeknx, 2023-05-31:
### Describe the bug

I noticed that the CVL reset also happens at SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT
even if linear mode is enabled.

### How to reproduce

...

### Expected behavior

only: CVL reset based on cell voltage diff

### Driver version

latest Beta

### Venus OS device type

Raspberry Pi

### Venus OS version

3.0.0-18

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 90.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; Example with 1 BMS: Jkbms_Ble C8:47:8C:00:00:00
; Example with 3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
;              BMS on purpose, then you have to restart the driver/system to reset the block.
; False: Charge and discharge is not blocked on BMS communication loss
; True: Charge and discharge is blocked on BMS communication loss, it's unblocked when connection is established
;       again or the driver/system is restarted
BLOCK_ON_DISCONNECT = True

; --------- Charge mode ---------
; Choose the mode for voltage / current limitations (True / False)
[... 174 lines cut ...]

; -- LltJbd settings
; SoC low levels
; NOTE: SOC_LOW_WARNING is also used to calculate the Time-To-Go even if you are not using a LltJbd BMS
SOC_LOW_WARNING = 20
SOC_LOW_ALARM   = 10

; -- Daly settings
; Battery capacity (amps) if the BMS does not support reading it
BATTERY_CAPACITY = 50
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = 1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT = 15
```

### Relevant log output

```shell
...
```

mr-manuel, 2023-05-31:
The driver has only the data that is provided from the battery. If you select the MultiPlus II as the battery monitor this is only for the Venus OS system.

To solve this, lower this setting in your config.

mr-manuel, 2023-05-31:
I don't think so.

https://github.com/Louisvdw/dbus-serialbattery/blob/81517190c9373b10a45ae1a2f79e14ac59ed5c70/etc/dbus-serialbattery/config.default.ini#L44-L64

mr-manuel, 2023-05-31:
Please always check the `config.default.ini` for descriptions and copy only the values to the `config.ini`, so you check always the description that matches your version :-).