Louisvdw/dbus-serialbattery #691: lifepower driver issue

csloz, 2023-06-05:
### Describe the bug

I see that my (previously working) revov driver has been replaced with the lifepower one;
Unfortunately It _**doesn't**_ work on my older revov tianpower unit though.

@40000000647dc05509b3aa5c INFO:SerialBattery:Testing Lifepower
@40000000647dc05518895964 ERROR:SerialBattery:Unexpected err=IndexError('list index out of range'), type(err)=<class 'IndexError'>

Will do a bit of debugging and see whats going on.  

From a cursory look, I see he's calling 7E vs 7C on the "modbus" call, first stop I guess will be to change to 7C calls see if that resolves, then go from there.

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

Expect working code ;)

### Driver version

latest stable

### Venus OS device type

Raspberry Pi

### Venus OS version

3.0

### BMS type

Revov

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
.
```

### Relevant log output

```shell
.
```

### Any other information that may be helpful

.

csloz, 2023-06-05:
snuck mine back in for some debugging.  Mine definitely prefers the 7C call, not the Lifepower driver 7E modbus call, so ultimately this is going to need a different driver type than the LifePower one unfortunately.

I can hopefully get mine a bit more finished as I can infer a bit more from the Lifepower driver ie. error bit

---

My TianPower firmware version -  TP-BD16S48V150-V3.1.29

With 7C

 tail /data/log/dbus-serialbattery.ttyUSB0/current -F

@40000000647dc72b341a3414 INFO:SerialBattery:Testing Revov
@40000000647dc72c00b64a04 WARNING:SerialBattery:Modbus Address: 124
@40000000647dc72c00bec1fc WARNING:SerialBattery:Modbus Type   : 1
@40000000647dc72c00c6d84c WARNING:SerialBattery:Modbus Command: 51
@40000000647dc72c00ce98ac WARNING:SerialBattery:Modbus PackLen: 22
@40000000647dc72c00d83d6c WARNING:SerialBattery:Modbus Packet : [7C:01:33:16:54:50:2D:42:44:31:36:53:34:38:56:31:35:30:2D:56:33:2E:31:2E:32:39:E2:0D]
@40000000647dc72c00df4a1c WARNING:SerialBattery:== Modbus packet good ==
@40000000647dc72c00e754b4 ERROR:SerialBattery:Revov TP-BD16S48V150-V3.1.29
@40000000647dc72c0403cb3c WARNING:SerialBattery:Modbus Address: 124
@40000000647dc72c040cb864 WARNING:SerialBattery:Modbus Type   : 1
@40000000647dc72c0414de54 WARNING:SerialBattery:Modbus Command: 66
@40000000647dc72c041c48c4 WARNING:SerialBattery:Modbus PackLen: 15
@40000000647dc72c04252264 WARNING:SerialBattery:Modbus Packet : [7C:01:42:0F:54:42:4D:31:39:30:34:31:33:30:30:32:36]
@40000000647dc72c042c235c WARNING:SerialBattery:== Modbus packet good ==
@40000000647dc72c043621f4 ERROR:SerialBattery:Revov ver ( TBM1904130026)
@40000000647dc72c05c75114 WARNING:SerialBattery:Modbus Address: 124
@40000000647dc72c05d06d1c WARNING:SerialBattery:Modbus Type   : 1
@40000000647dc72c05d89ec4 WARNING:SerialBattery:Modbus Command: 6
@40000000647dc72c05e10eec WARNING:SerialBattery:Modbus PackLen: 4
@40000000647dc72c05e8b00c WARNING:SerialBattery:Modbus Packet : [7C:01:06:04:00:00:0E:A6:EC]
@40000000647dc72c05f0f154 WARNING:SerialBattery:== Modbus packet good ==
@40000000647dc72c146b321c WARNING:SerialBattery:Modbus Address: 124
@40000000647dc72c1475c524 WARNING:SerialBattery:Modbus Type   : 1
@40000000647dc72c14888dbc WARNING:SerialBattery:Modbus Command: 1
@40000000647dc72c148ab484 WARNING:SerialBattery:Modbus PackLen: 118
@40000000647dc72c1495ad1c WARNING:SerialBattery:Modbus Packet : [7C:01:01:76:01:10:8D:5E:0D:5F:8D:62:0D:6A:8D:78:0D:5B:8D:C3:0D:59:8D:6B:0D:47:8D:59:0D:47:8D:7E:0D:52:8D:3B:0D:3A:02:01:70:F9:03:01:27:10:04:01:46:50:05:03:00:42:00:42:20:44:06:05:00:00:00:01:00:00:00:00:00:00:07:01:00:C9:08:01:15:67:09:01:27:10:0A:01:00:00:0B:01:00:00:E2:F7:0C:01:00:00:3D:D0:0D:01:00:40:00:00:0E:01:00:40:00:00:0F:01:00:24:EE:9A:10:01:00:09:C6:B9:88:0D]
@40000000647dc72c149e1574 WARNING:SerialBattery:== Modbus packet good ==
@40000000647dc72c22c66994 WARNING:SerialBattery:Modbus Address: 124
@40000000647dc72c22cfbc4c WARNING:SerialBattery:Modbus Type   : 1
@40000000647dc72c22d818ec WARNING:SerialBattery:Modbus Command: 1
@40000000647dc72c22e02384 WARNING:SerialBattery:Modbus PackLen: 118
@40000000647dc72c22e90cc4 WARNING:SerialBattery:Modbus Packet : [7C:01:01:76:01:10:8D:5E:0D:5F:8D:62:0D:6A:8D:78:0D:5B:8D:C3:0D:59:8D:6B:0D:47:8D:59:0D:47:8D:7E:0D:52:8D:3B:0D:3A:02:01:70:ED:03:01:27:10:04:01:46:50:05:03:00:42:00:42:20:44:06:05:00:00:00:01:00:00:00:00:00:00:07:01:00:C9:08:01:15:67:09:01:27:10:0A:01:00:00:0B:01:00:00:E2:F7:0C:01:00:00:3D:D0:0D:01:00:40:00:00:0E:01:00:40:00:00:0F:01:00:24:EE:9A:10:01:00:09:C6:B9:80:0D]
@40000000647dc72c22f0ffec WARNING:SerialBattery:== Modbus packet good ==
@40000000647dc72c22f9b664 WARNING:SerialBattery:Voltage Data: [54.79v]
@40000000647dc72c230216ec WARNING:SerialBattery:Battery Cycles: [201]
@40000000647dc72c230a350c WARNING:SerialBattery:Battery Capacity: [180.0Ah]
@40000000647dc72c2312820c WARNING:SerialBattery:Cell count: [16]
@40000000647dc72c231ad2f4 WARNING:SerialBattery:Raw Cell Data: [8D:5E:0D:5F:8D:62:0D:6A:8D:78:0D:5B:8D:C3:0D:59:8D:6B:0D:47:8D:59:0D:47:8D:7E:0D:52:8D:3B:0D:3A]
@40000000647dc72c2324e12c ERROR:SerialBattery:Unexpected err=IndexError('list index out of range'), type(err)=<class 'IndexError'>

Getting same error now though on my code, so looking through.

For reference a sample packet reply is below;
i.e. a full reply from a 7C:01:01:00:02:0D request - 

[7C:01:01:76:01:10:8D:5E:0D:5F:8D:62:0D:6A:8D:78:0D:5B:8D:C3:0D:59:8D:6B:0D:47:8D:59:0D:47:8D:7E:0D:52:8D:3B:0D:3A:02:01:70:ED:03:01:27:10:04:01:46:50:05:03:00:42:00:42:20:44:06:05:00:00:00:01:00:00:00:00:00:00:07:01:00:C9:08:01:15:67:09:01:27:10:0A:01:00:00:0B:01:00:00:E2:F7:0C:01:00:00:3D:D0:0D:01:00:40:00:00:0E:01:00:40:00:00:0F:01:00:24:EE:9A:10:01:00:09:C6:B9:80:0D]

Breaking that down;
id under 10 are 2 x byte value length, id values > 10 are 4 x byte value length.
in format of     ID: Length: Data

Type, 1,1 length (hex)
7C:01:01:76:

first data type, length (16 x 2 byte value)
01:10
	Data 16 x 2 bytes
	8D:3C:0D:39:8D:41:0D:47:8D:4E:0D:58:8E:3A:0E:76:8D:49:0D:27:8D:65:0D:73:8D:30:0D:48:8D:35:0D:36
02:01
	75:30
03:01
	27:10
04:01 
	46:50
05:03:
	00:42:00:42:20:44
06:05
	00:00:00:00:00:00:00:00:00:01
07:01
	00:C9
08:01
	15:73
09:01
	27:10
0A:01
	00:00
---- From here 0xB onward 01 length indicates 4 bytes not 2!
0B:01
	00:00:E2:F7
0C:01
	00:00:3D:D0
0D:01
	00:40:00:00
0E:01
	00:40:00:00
0F:01
	00:24:EE:9A
10:01
	00:09:C6:B9
END 
7E:0D

mr-manuel, 2023-06-06:
Liked PR https://github.com/Louisvdw/dbus-serialbattery/pull/694

mr-manuel, 2023-07-26:
Any updates on this?

tuxntoast, 2023-08-14:
I had issues getting this LifePower.py driver working with Eg4-LL 12v BMS. Found that my BMS is using different commands, then what was written by other in the Revov or the LifePower Driver. Spent some time writing up something that works on my device. Have issues on the Test.connect still (errors out), but the driver still picks it up, and the daemon is able to pull the stats with out fail. Reboots do not require any extra work to get this driver restarted. 

https://github.com/tuxntoast/eg4-ll/blob/main/egll.py

mr-manuel, 2023-09-26:
@tuxntoast feel free to open a PR for the dev branch.

closed 2023-09-26 by mr-manuel