Louisvdw/dbus-serialbattery #693: Bluetooth with JKBMS doesn't work

simsandmann, 2023-06-05:
### Describe the bug

Bluetooth connection to JKBMS is not working since the new latest version (1.0.20230531)

### How to reproduce

Steps to reproduce the behavior:
1. Reboot Venus OS under 3.0.0

### Expected behavior

Venus OS is connecting to the JKBMS

### Driver version

1.0.20230531

### Venus OS device type

Cerbo GX

### Venus OS version

3.0.0 (newest stable)

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; Example with 1 BMS: Jkbms_Ble C8:47:8C:00:00:00
; Example with 3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E8:04:67, Jkbms_Ble C8:47:8C:E9:03:8B

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
;              BMS on purpose, then you have to restart the driver/system to reset the block.
; False: Charge and discharge is not blocked on BMS communication loss
; True: Charge and discharge is blocked on BMS communication loss, it's unblocked when connection is established
;       again or the driver/system is restarted
BLOCK_ON_DISCONNECT = False

.........
```

### Relevant log output

```shell
JKBMS 1:

2023-06-04 08:38:22.491621500
2023-06-05 14:44:51.436875500 Missing device address argument
2023-06-05 14:44:55.843897500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-05 14:44:55.855365500 INFO:SerialBattery:dbus-serialbattery v1.0.2023053                     1
2023-06-05 14:44:56.879772500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E                     8:04:67
2023-06-05 14:44:56.884387500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E                     8:04:67
2023-06-05 14:44:57.967597500 ERROR:SerialBattery:[org.freedesktop.DBus.Error.Se                     rviceUnknown] The name org.bluez was not provided by any .service files
2023-06-05 14:44:57.968800500 ERROR:SerialBattery:ERROR >>> No battery connectio                     n at Jkbms_Ble

JKBMS 2:

2023-06-04 08:38:22.461507500
2023-06-05 14:44:51.432646500 Missing device address argument
2023-06-05 14:44:55.670973500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-05 14:44:55.677493500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-06-05 14:44:56.645783500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E9:03:8B
2023-06-05 14:44:56.649977500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E9:03:8B
2023-06-05 14:44:57.729882500 ERROR:SerialBattery:[org.freedesktop.DBus.Error.ServiceUnknown] The name org.bluez was not provided by any .service files
2023-06-05 14:44:57.730651500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
```

simsandmann, 2023-06-05:
One JKBMS via Bluetooth is not working for me, but I am also connected with serial connection as my system is productive.
So I am connected to "two" BMS as well.

mr-manuel, 2023-06-05:
Could you try to execute this lines manually and check the results?

```bash
opkg update
opkg install python3-misc python3-pip
pip3 install bleak
```

simsandmann, 2023-06-05:
root@einstein:/data/log# opkg update
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/einstein/Packages.gz.
Updated source 'einstein'.
root@einstein:/data/log#  opkg install python3-misc python3-pip
No packages installed or removed.
root@einstein:/data/log# pip3 install bleak
Requirement already satisfied: bleak in /usr/lib/python3.8/site-packages (0.20.2)
Requirement already satisfied: async-timeout<5,>=3.0.0; python_version < "3.11" in /usr/lib/python3.8/site-packages (from bleak) (4.0.2)
Requirement already satisfied: dbus-fast<2.0.0,>=1.83.0; platform_system == "Linux" in /usr/lib/python3.8/site-packages (from bleak) (1.86.0)

mr-manuel, 2023-06-05:
Could you also post the output of
```bash
/usr/libexec/bluetooth/bluetoothd -v
```

```bash
/usr/bin/bluetoothctl -v
```

```bash
bluetoothctl devices
```

Does it connect, if you restart the bluetooth driver?
```bash
/etc/init.d/bluetooth restart
```

If you revert back to the latest beta all works fine?

simsandmann, 2023-06-05:
```
root@einstein:/data/log# /usr/libexec/bluetooth/bluetoothd -v
5.55

root@einstein:/data/log# /usr/bin/bluetoothctl -v
bluetoothctl: 5.55

root@einstein:/data/log# bluetoothctl devices
Device ED:D5:E6:BC:BF:4E MPPT 1
Device CA:BF:98:7A:39:16 MPPT 2
Device C8:47:8C:E9:03:8B JKBMS 1
Device C8:47:8C:E8:04:67 JKBMS 2
Device 88:BF:E4:CD:2E:8E 88-BF-E4-CD-2E-8E
 ```
 
 /etc/init.d/bluetooth restart --> Still no connection
 
 
At the latest beta version the connection was already unstable. It connected and disconnected around 4 times a day. It worked 6h and then it did not for the next 6 hours. And so on.

I also updated from Venus OS 2.92 to 3.0 trying to fix the issue with the latest version. It did not work.

mr-manuel, 2023-06-05:
You have disabled Bluetooth in the Venus OS settings?

Have you tried an external Bluetooth dongle?

simsandmann, 2023-06-06:
Bluetooth is disabled.

I have no Bluetooth dongle.

Some time later the Bluetooth connection established. Maybe that helps:

JKBMS 1:
`root@einstein:~# tail -n 200 -f /data/log/dbus-blebattery.0/current | tai64nlocal
2023-06-06 00:12:08.061872500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 361, in publish_battery
2023-06-06 00:12:08.061876500     success = self.battery.refresh_data()
2023-06-06 00:12:08.061879500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_ble.py", line 162, in refresh_data
2023-06-06 00:12:08.061883500     self.jk.start_scraping()
2023-06-06 00:12:08.061885500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 366, in start_scraping
2023-06-06 00:12:08.062018500     self.bt_thread.start()
2023-06-06 00:12:08.062022500   File "/usr/lib/python3.8/threading.py", line 848, in start
2023-06-06 00:12:08.062024500     raise RuntimeError("threads can only be started once")
2023-06-06 00:12:08.062027500 RuntimeError: threads can only be started once
2023-06-06 00:12:08.062080500
2023-06-06 00:12:08.062082500 connect and scrape on address: C8:47:8C:E8:04:67
2023-06-06 00:12:08.062084500 btloop
2023-06-06 00:12:08.062086500 reconnect
2023-06-06 00:12:08.062087500
2023-06-06 00:12:08.062088500 btloop
2023-06-06 00:12:08.062089500 reconnect
2023-06-06 00:12:08.062090500 Exiting bt-loop
2023-06-06 00:12:08.769699500 Missing device address argument
2023-06-06 00:12:10.962119500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-06 00:12:10.963453500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-06-06 00:12:11.539063500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E8:04:67
2023-06-06 00:12:11.539869500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E8:04:67
2023-06-06 00:12:21.298406500 INFO:SerialBattery:Jkbms_Ble found!
2023-06-06 00:12:21.299587500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20221217)
2023-06-06 00:12:21.301136500 INFO:SerialBattery:Connection established to Jkbms_Ble
2023-06-06 00:12:21.301873500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce80467
2023-06-06 00:12:21.302468500 INFO:SerialBattery:========== Settings ==========
2023-06-06 00:12:21.303168500 INFO:SerialBattery:> Connection voltage: 53.14V | Current: -1.7A | SoC: 93%
2023-06-06 00:12:21.303762500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2023-06-06 00:12:21.304382500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-06-06 00:12:21.305095500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2023-06-06 00:12:21.306386500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 110.0A | MAX BATTERY DISCHARGE CURRENT: 180.0A (read from BMS)
2023-06-06 00:12:21.308612500 INFO:SerialBattery:> CVCM:     True
2023-06-06 00:12:21.309349500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-06-06 00:12:21.310144500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-06-06 00:12:21.311038500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-06-06 00:12:21.311763500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-06-06 00:12:21.312401500 INFO:SerialBattery:Serial Number/Unique Identifier: 2082316058
2023-06-06 00:12:21.582115500 INFO:SerialBattery:DeviceInstance = 3
2023-06-06 00:12:21.583995500 INFO:SerialBattery:com.victronenergy.battery.c8478ce80467
2023-06-06 00:12:21.585330500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20221217)
2023-06-06 00:12:21.651462500 INFO:SerialBattery:publish config values = 1
2023-06-06 00:13:05.820105500 INFO:SerialBattery:Jkbms_Ble: Bluetooth died
2023-06-06 00:13:05.827462500 INFO:SerialBattery:Reset of Bluetooth triggered
2023-06-06 00:13:05.827471500 INFO:SerialBattery:Scraping ended, issuing sys-commands
2023-06-06 00:13:07.963645500 Starting bluetooth: bluetoothd.
2023-06-06 00:13:07.987368500 INFO:SerialBattery:Bluetooth should have been restarted
2023-06-06 00:13:08.029511500 Traceback (most recent call last):
2023-06-06 00:13:08.029520500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 361, in publish_battery
2023-06-06 00:13:08.029524500     success = self.battery.refresh_data()
2023-06-06 00:13:08.029527500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_ble.py", line 162, in refresh_data
2023-06-06 00:13:08.029530500     self.jk.start_scraping()
2023-06-06 00:13:08.029533500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 366, in start_scraping
2023-06-06 00:13:08.029674500     self.bt_thread.start()
2023-06-06 00:13:08.029677500   File "/usr/lib/python3.8/threading.py", line 848, in start
2023-06-06 00:13:08.029681500     raise RuntimeError("threads can only be started once")
2023-06-06 00:13:08.029683500 RuntimeError: threads can only be started once
2023-06-06 00:13:08.038150500
2023-06-06 00:13:08.038157500 connect and scrape on address: C8:47:8C:E8:04:67
2023-06-06 00:13:08.038159500 btloop
2023-06-06 00:13:08.038160500 reconnect
2023-06-06 00:13:08.038162500
[... 205 lines cut ...]
2023-06-06 00:15:23.011497500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-06 00:15:23.012837500 INFO:SerialBattery:dbus-serialbattery v1.0.2023053                                                                                   1
2023-06-06 00:15:23.393564500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E                                                                                   9:03:8B
[... cut ...]
2023-06-06 00:15:34.546482500 INFO:SerialBattery:Serial Number/Unique Identifier                                                                                   : 2092245065
2023-06-06 00:15:34.832464500 INFO:SerialBattery:DeviceInstance = 4
2023-06-06 00:15:34.836261500 INFO:SerialBattery:com.victronenergy.battery.c8478                                                                                   ce9038b
2023-06-06 00:15:34.838402500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (2022                                                                                   1224)
2023-06-06 00:15:34.907198500 INFO:SerialBattery:publish config values = 1
`

mr-manuel, 2023-06-06:
Please put tripple backticks \`\`\` in the line before and after your code/logs for correct formatting.

This should help, thanks. What is your BMS hardware/firmware version?

simsandmann, 2023-06-06:
Do you mean 11.XW and 10.XW ?
That is at least stated at the last lines in the logs.

sstolk90, 2023-06-06:
I can confirm the same issue with also two JK bms'es. Version is 2.92 and the troubleshooting mentioned in this topic/bug report is the same with mine setup. My Victron also reboots the GX side once in the X hours. My MQTT messages are missing for a couple of minutes and that is the moment the VenusOS reboots. The system also resets the names I've given them and when I check the names are standard. The connection is sometimes restored to both the units and sometimes not. I'm keeping the serial connection to one of the BMS'es live to keep the CCL and DCL accurate with settings.

sstolk90, 2023-06-06:
Restarting the bluetooth connection as mentioned above works: /etc/init.d/bluetooth restart. Both BMS'es are displayed. When the VenusOS reboots it's probally resetted.
[image]

seidler2547, 2023-06-08:
@mr-manuel seems like it happens here, on reconnect: https://github.com/Louisvdw/dbus-serialbattery/blob/86a6b956e15bee21d5a298283c9d121f0eb82a02/etc/dbus-serialbattery/bms/jkbms_ble.py#L154

Do you really think it should do that? If anything, it should call `stop_scraping` first. But I would rather just exit and let a "normal" reconnect happen.

It could also be that with the scanning removed (#695) that the initial connect will be more reliable since there wouldn't be two processes doing exactly the same scan at exactly the same time.

@sstolk90 can you find out the reason for the reboot? There could be a hint in `/var/log/messages` (or `/var/log/messages.0`) on your GX device. For me it was OOM (out of memory) until I put a script in place to fix a memory leak.

mr-manuel, 2023-06-08:
Please execute this commands to apply the potential fixes:

```bash
wget -O /data/etc/dbus-serialbattery/bms/jkbms_ble.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery/dev/etc/dbus-serialbattery/bms/jkbms_ble.py
wget -O /data/etc/dbus-serialbattery/bms/jkbms_brn.py https://raw.githubusercontent.com/mr-manuel/venus-os_dbus-serialbattery/dev/etc/dbus-serialbattery/bms/jkbms_brn.py
bash /data/etc/dbus-serialbattery/reinstall-local.sh
```

Let me know, if this fixes your problem.

sstolk90, 2023-06-08:
> 

I have this in the logs: 
Jun  8 04:27:53 nanopi daemon.debug rngd: [jitter]: xread_jitter requests 2500 bytes from pipe
Jun  8 04:27:53 nanopi daemon.debug rngd: [jitter]: xread_jitter falls back to AES
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 2472/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 2624/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 2762/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 2887/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3000/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3103/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3196/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3280/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3357/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3426/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3489/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3546/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3597/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3644/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3686/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3724/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3759/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3791/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3819/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3845/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3868/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3890/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3909/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3926/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3942/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3956/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3969/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3981/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 3992/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 4001/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 4010/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 4018/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 4025/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Added 4032/4096 bits entropy
Jun  8 04:27:53 nanopi daemon.debug rngd: Pool full at 4096, sleeping!
Jun  8 04:27:53 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 04:27:53 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 1 wakes up for refill
Jun  8 04:27:54 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 0 is 2.727370146900e+01 sec
Jun  8 04:27:54 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 04:27:54 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 04:27:54 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 0 wakes up for refill
Jun  8 04:27:56 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 3 is 2.818484219600e+01 sec
Jun  8 04:27:56 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 04:27:56 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 04:27:56 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 3 wakes up for refill
Jun  8 04:27:57 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 2 is 3.010089842900e+01 sec
Jun  8 04:27:57 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 04:27:57 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 04:27:57 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 2 wakes up for refill
Jun  8 04:28:02 nanopi daemon.info connmand[648]: ntp: time slew +0.980480 s
Jun  8 04:28:08 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 1 is 1.501241921800e+01 sec
Jun  8 04:28:08 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 04:28:08 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 04:28:08 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 1 wakes up for refill
Jun  8 04:28:12 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 0 is 1.933747561700e+01 sec
Jun  8 04:28:12 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 04:28:12 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 04:28:12 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 0 wakes up for refill
Jun  8 04:28:14 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 3 is 1.801609917500e+01 sec
Jun  8 04:28:14 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 04:28:14 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 04:28:14 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 3 wakes up for refill
Jun  8 04:28:20 nanopi daemon.info bluetoothd[581]: Terminating
Jun  8 04:28:20 nanopi daemon.info bluetoothd[581]: Exit
Jun  8 04:28:20 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 2 is 2.439997932700e+01 sec
[... 102 lines cut ...]
Jun  8 08:02:54 nanopi daemon.debug rngd: Added 3746/4096 bits entropy
[... cut ...]
Jun  8 08:02:54 nanopi daemon.debug rngd: Added 4023/4096 bits entropy
Jun  8 08:02:54 nanopi daemon.debug rngd: Added 4030/4096 bits entropy
Jun  8 08:02:54 nanopi daemon.debug rngd: Added 4036/4096 bits entropy
Jun  8 08:02:54 nanopi daemon.debug rngd: Pool full at 4096, sleeping!
Jun  8 08:03:35 nanopi daemon.notice wpa_supplicant[710]: wifi0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Jun  8 08:19:59 nanopi authpriv.notice sudo:     root : TTY=pts/0 ; PWD=/data/home/root ; USER=root ; COMMAND=/bin/cat /var/log/messages

sstolk90, 2023-06-08:
> ```shell
> bash /data/etc/dbus-serialbattery/reinstall-local.sh
> ```

I've installed this and will monitor for issues!

sstolk90, 2023-06-08:
Well, the bluetooth is not coming back after installing the above command's:

`Jun  8 08:23:35 nanopi user.notice shutdown[19486]: shutting down for system reboot
**snip reboot process*
Jun  8 08:24:18 nanopi daemon.info rngd: [jitter]: Initialized
Jun  8 08:24:18 nanopi daemon.debug rngd: Reading entropy from JITTER Entropy generator
Jun  8 08:24:18 nanopi daemon.debug rngd: [jitter]: xread_jitter requests 2500 bytes from pipe
Jun  8 08:24:18 nanopi daemon.debug rngd: [jitter]: xread_jitter falls back to AES
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 2496/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 2646/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 2782/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 2905/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3016/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3118/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3209/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3292/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3367/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3436/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3497/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3553/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3604/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3650/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3692/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3730/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3764/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3795/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3823/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3849/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3872/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3893/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3912/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3929/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3944/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3959/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3971/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3983/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 3993/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 4003/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 4012/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 4019/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 4027/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Added 4033/4096 bits entropy
Jun  8 08:24:18 nanopi daemon.debug rngd: Pool full at 4096, sleeping!
Jun  8 08:24:18 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 08:24:18 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 1 wakes up for refill
Jun  8 08:24:19 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 0 is 2.708381613600e+01 sec
Jun  8 08:24:19 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 08:24:19 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 08:24:19 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 0 wakes up for refill
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 3 is 2.714628865500e+01 sec
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 3 wakes up for refill
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 2 is 2.859560384600e+01 sec
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 08:24:20 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 2 wakes up for refill
Jun  8 08:23:53 nanopi daemon.info dnsmasq[592]: started, version 2.83 cachesize 150
Jun  8 08:24:24 nanopi daemon.warn dnsmasq[592]: overflow: 3 log entries lost
Jun  8 08:24:27 nanopi daemon.info connmand[651]: ntp: time slew +1.014409 s
Jun  8 08:24:34 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 3 is 1.538958903100e+01 sec
Jun  8 08:24:34 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 08:24:34 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 08:24:34 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 3 wakes up for refill
Jun  8 08:24:37 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 0 is 1.803464688300e+01 sec
Jun  8 08:24:37 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 08:24:37 nanopi daemon.debug rngd: [jitter]: DONE Writing to pipe with return 16535
Jun  8 08:24:37 nanopi daemon.debug rngd: [jitter]: JITTER thread on cpu 0 wakes up for refill
Jun  8 08:24:38 nanopi daemon.debug rngd: [jitter]: jent_read_entropy time on cpu 1 is 2.150073990900e+01 sec
Jun  8 08:24:38 nanopi daemon.debug rngd: [jitter]: Writing to pipe
[... 30 lines cut ...]
[... cut ...]
Jun  8 08:24:57 nanopi daemon.debug rngd: [jitter]: Writing to pipe
Jun  8 08:25:57 nanopi daemon.notice wpa_supplicant[710]: wifi0: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
Jun  8 08:26:33 nanopi user.err kernel: [  165.724899] Bluetooth: hci0: unexpected event for opcode 0xfc19
Jun  8 08:26:33 nanopi user.err kernel: [  165.731008] Bluetooth: hci0: unexpected event for opcode 0xfc19
Jun  8 08:26:48 nanopi authpriv.notice sudo:     root : TTY=pts/0 ; PWD=/data/home/root ; USER=root ; COMMAND=/bin/cat /var/log/messages
`

[image]

mr-manuel, 2023-06-08:
What does the driver log say? Did you reboot after install? If not, please reboot.

mr-manuel, 2023-06-08:
In the meanwhile I found this two links regarding `Jun  8 08:26:33 nanopi user.err kernel: [  165.724899] Bluetooth: hci0: unexpected event for opcode 0xfc19`:

* https://community.nxp.com/t5/i-MX-Processors/Bluez-BLE-auto-disconnect-after-connected-a-few-seconds/m-p/1063455/highlight/true
* https://software-dl.ti.com/simplelink/esd/simplelink_cc2640r2_sdk/5.30.00.03/exports/docs/ble5stack/vendor_specific_guide/BLE_Vendor_Specific_HCI_Guide/hci_interface.html
  HCI_EXT_HaltDuringRfCm -> 0xFC19

@seidler2547 do you also see this in your logs with the `LltJbd_Ble` driver?

sstolk90, 2023-06-08:
> What does the driver log say?
> 
> ~Did you reboot after install? If not, please reboot.~
> 
> The logs say it...
> 
> > ```
> > Jun  8 08:23:35 nanopi user.notice shutdown[19486]: shutting down for system reboot
> > **snip reboot process*
> > ```

```

root@nanopi:~# tail -n 200 -f /data/log/dbus-blebattery.0/current | tai64nlocal
2023-06-08 08:37:17.157119500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:37:17.157634500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:37:36.712378500 ERROR:SerialBattery:No BMS found at C8:47:8C:EC:C0:B5
2023-06-08 08:37:36.712929500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-06-08 08:37:36.713172500
2023-06-08 08:37:36.713174500 connect and scrape on address: C8:47:8C:EC:C0:B5
2023-06-08 08:37:36.713177500 btloop
2023-06-08 08:37:36.713178500 reconnect
2023-06-08 08:37:39.931032500 Exiting bt-loop
2023-06-08 08:37:40.348123500 Missing device address argument
2023-06-08 08:37:40.348133500 [CHG] Device 64:24:24:D2:5B:B2 RSSI: -86
2023-06-08 08:37:40.348135500 [CHG] Device 53:5F:79:C4:C5:5F RSSI: -86
2023-06-08 08:37:41.429974500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-08 08:37:41.431079500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-06-08 08:37:41.745620500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:37:41.746134500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:37:55.379084500 ERROR:SerialBattery:No BMS found at C8:47:8C:EC:C0:B5
2023-06-08 08:37:55.379768500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-06-08 08:37:55.380031500
2023-06-08 08:37:55.380033500 connect and scrape on address: C8:47:8C:EC:C0:B5
2023-06-08 08:37:55.380035500 btloop
2023-06-08 08:37:55.380036500 reconnect
2023-06-08 08:37:55.380038500 Exiting bt-loop
2023-06-08 08:37:55.772616500 Missing device address argument
2023-06-08 08:37:56.866450500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-08 08:37:56.867572500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-06-08 08:37:57.213077500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:37:57.213592500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:38:16.766025500 ERROR:SerialBattery:No BMS found at C8:47:8C:EC:C0:B5
2023-06-08 08:38:16.766577500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-06-08 08:38:16.766821500
2023-06-08 08:38:16.766823500 connect and scrape on address: C8:47:8C:EC:C0:B5
2023-06-08 08:38:16.766826500 btloop
2023-06-08 08:38:16.766827500 reconnect
2023-06-08 08:38:17.272814500 Exiting bt-loop
2023-06-08 08:38:17.679314500 Missing device address argument
2023-06-08 08:38:18.810272500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-08 08:38:18.811441500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-06-08 08:38:19.153269500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:38:19.153784500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:38:38.705242500 ERROR:SerialBattery:No BMS found at C8:47:8C:EC:C0:B5
2023-06-08 08:38:38.705828500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-06-08 08:38:38.706118500
2023-06-08 08:38:38.706121500 connect and scrape on address: C8:47:8C:EC:C0:B5
2023-06-08 08:38:38.706124500 btloop
2023-06-08 08:38:38.706125500 reconnect
2023-06-08 08:38:41.924633500 Exiting bt-loop
2023-06-08 08:38:42.358526500 Missing device address argument
2023-06-08 08:38:42.358535500 [CHG] Device C8:47:8C:E5:01:1B RSSI: -72
2023-06-08 08:38:43.467436500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-08 08:38:43.468483500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-06-08 08:38:43.779671500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:38:43.780220500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EC:C0:B5
2023-06-08 08:39:01.033969500 ERROR:SerialBattery:No BMS found at C8:47:8C:EC:C0:B5
2023-06-08 08:39:01.036566500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-06-08 08:39:01.036579500
[... 126 lines cut ...]
2023-06-08 08:41:35.354509500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (20230504)
2023-06-08 08:41:35.355559500 INFO:SerialBattery:Connection established to Jkbms_Ble
2023-06-08 08:41:35.357001500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478cecc0b5
2023-06-08 08:41:35.357435500 INFO:SerialBattery:========== Settings ==========
2023-06-08 08:41:35.358025500 INFO:SerialBattery:> Connection voltage: 53.57V | Current: 34.7A | SoC: 58%
2023-06-08 08:41:35.358412500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2023-06-08 08:41:35.358809500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-06-08 08:41:35.359218500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2023-06-08 08:41:35.359780500 INFO:SerialBattery:> CVCM:     True
2023-06-08 08:41:35.360215500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-06-08 08:41:35.360614500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-06-08 08:41:35.360984500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-06-08 08:41:35.361346500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-06-08 08:41:35.361700500 INFO:SerialBattery:Serial Number/Unique Identifier: 2120146523
2023-06-08 08:41:35.415809500 INFO:SerialBattery:DeviceInstance = 4
2023-06-08 08:41:35.416358500 INFO:SerialBattery:com.victronenergy.battery.c8478cecc0b5
2023-06-08 08:41:35.417060500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (20230504)
2023-06-08 08:41:35.441925500 INFO:SerialBattery:publish config values = 1

^C
root@nanopi:~#

```

After the reboot it detects 1 JK BMS over Bluetooth and the other one on serial. I think when I reboot the bluetooth process it will reconnect to the second one. Will test it now

sstolk90, 2023-06-08:
After entering the command /etc/init.d/bluetooth restart it shows this. Correct in my case but not sure if it will reboot. I will rename one of the BLE names to something own and when it's rebooted I will notice the name change.

[image]

sstolk90, 2023-06-08:
After renaming all three it reconnects both bluetooth and resets their name after a while. The unit has not rebooted since the Batterij Onder (eng: Battery Below) has kept his name:
[image]

mr-manuel, 2023-06-08:
Ok, this means, that the driver creashes and restarts. Can you contact me on Discord?

seidler2547, 2023-06-08:
> `hci0: unexpected event for opcode 0xfc19`
I think I've seen this too, occasionally. I don't think it's a problem.

@sstolk90 is the BT dongle far away from the batteries or is there a lot of interference, e.g. From WiFi?

Also, you did disable Bluetooth in the GUI, didn't you?

Right now from the logs it seems that the connection doesn't work because the communication between the GX and the BMS is interrupted. I don't immediately see how the driver could work around that, but it's hard to say.

mr-manuel, 2023-06-08:
I made some changes in my `dev` branch. The log will display the signal strenght of a Bluetooth connections in the logs, this will help troubleshooting.

mr-manuel, 2023-06-10:
This is solved with https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.0.20230610beta which gets merged with the `dev` branch of this repo later today. Then it can be installed using the install script by selecting `nightly` and then `dev`.

closed 2023-06-10 by mr-manuel