Louisvdw/dbus-serialbattery #703: JK BMS - SoC daily decreasing by 2%

Dighy, 2023-06-11:
### Describe the problem

Hello everybody,

I would like to start telling that I'm just beginer and I'm following all the instruction provided.
I was replacing my old battery with 24V/280Ah lifepo4 battery monitored by JK Bms (2A8S20P) in a Victron Easy solar 24/3000 with CCGX system.
Initially I let it running by setting only the VE Bus System (multiplus) and Solar charger (70A/150V type) with settings like CVL-27.2V and CCL - 50A and I checked the battery behavior at high discharge rate (120A). Everything fine Soc was going from 100% to 99%(float mode) and 68-74% in the morning as per consumption.
After few days, I installed the driver version **[v1.0.20230508beta]** with every setting from default, I set JK-BMS serial battery as master (external control) and behavior was the same like without driver from SoC calculation point of view.
Once the latest version was released **[v1.0.20230531]**, I update it by USB in my CCGX and updated the value of CCL to 70A and DCL to 100A through SSL client Putty and reboot the GX.
That I noticed is that in the first 2 days SoC was not going to 100% and only to 99% and after that every day maximum was 2% less compare with previous day (eq. 97%-95%-93%......) and now I reached 89% and as per logic CVL is going to 27.6V, then reduced to 27V (float) and after increasing again to 27.6V due to the SoC < 90%.
Today, I came back to CCL - 50A and all the others the same as before to see what is going on...
I would be so greatful for any support on what is the issue and how to correct it.

Thanks.

### Driver version

v1.0.20230531

### Venus OS device type

CCGX

### Venus OS version

2.94

### BMS type

JKBMS / Heltec BMS

### Cell count

8

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; Example with 1 BMS: Jkbms_Ble C8:47:8C:00:00:00
; Example with 3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
;              BMS on purpose, then you have to restart the driver/system to reset the block.
; False: Charge and discharge is not blocked on BMS communication loss
; True: Charge and discharge is blocked on BMS communication loss, it's unblocked when connection is established
;       again or the driver/system is restarted
BLOCK_ON_DISCONNECT = False

[... 281 lines cut ...]
2023-06-11 05:54:29.989622500 INFO:SerialBattery:Testing Ecs
2023-06-11 05:54:30.519377500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-11 05:54:30.544188500 INFO:SerialBattery:Testing HeltecModbus
2023-06-11 05:54:35.136473500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-11 05:54:35.149535500 INFO:SerialBattery:Testing HLPdataBMS4S
2023-06-11 05:54:39.718108500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-11 05:54:39.794737500 INFO:SerialBattery:Testing Jkbms
2023-06-11 05:54:40.544066500 INFO:SerialBattery:Connection established to Jkbms
2023-06-11 05:54:40.615324500 INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB0
2023-06-11 05:54:40.735716500 INFO:SerialBattery:========== Settings ==========
2023-06-11 05:54:40.862151500 INFO:SerialBattery:> Connection voltage: 26.92V | Current: 10.51A | SoC: 83%
2023-06-11 05:54:40.864012500 INFO:SerialBattery:> Cell count: 8 | Cells populated: 0
2023-06-11 05:54:40.993437500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-06-11 05:54:40.995360500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2023-06-11 05:54:41.124907500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 75.0A | MAX BATTERY DISCHARGE CURRENT: 140.0A (read from BMS)
2023-06-11 05:54:41.235411500 INFO:SerialBattery:> CVCM:     True
2023-06-11 05:54:41.307494500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-06-11 05:54:41.400694500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-06-11 05:54:41.476469500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-06-11 05:54:41.613432500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-06-11 05:54:41.676390500 INFO:SerialBattery:Serial Number/Unique Identifier: Ghy Baterie 7.2K
2023-06-11 05:54:42.644408500 INFO:SerialBattery:DeviceInstance = 1
2023-06-11 05:54:42.659331500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2023-06-11 05:54:43.215483500 INFO:SerialBattery:publish config values = 1
```

jdeus, 2023-06-11:
Same issue. The issue seems that this project does not implement scheduled charge

mr-manuel, 2023-06-11:
The JKBMS resets the SoC to 95%, when all cell voltages are greater or equal to 3.500 V. Please try to increase `MAX_CELL_VOLTAGE` to `3.50` or `3.51`.

Dighy, 2023-06-11:
> The JKBMS resets the SoC to 95%, when all cell voltages are greater or equal to 3.500 V. Please try to increase `MAX_CELL_VOLTAGE` to `3.50` or `3.51`.

I will try this and leave the feedback.
My concern is why with previous version there was no issue by using the same max cell voltage 3.45V. The only thing was changed is CCL and DCL. But today I came back to default CCL (50A).
Let's see how it behave.
Thanks

mr-manuel, 2023-06-11:
Probably because before the CVL was not as accurate as it is now.

mr-manuel, 2023-06-11:
> Same issue. The issue seems that this project does not implement scheduled charge

@jdeus can you please explain better with an example?

mr-manuel, 2023-06-12:
@Dighy @TazerReloaded, could you try to lower the `OVP` in the BMS settings to `3.48` volts and the OVPR to `3.45`. Then change the `MAX_CELL_VOLTAGE` in the `config.ini` to `3.48` and see, if this resets the SoC to 100%.

If this works you could set the OVP values to the values you need/want, e.g. OVP to `3.45` and OVPR to `3.40`. But you have to try what works best. Please report back.

Dighy, 2023-06-12:
Today I've tried your sugestion with max. cell voltage 3.51V but still the same. Soc not sereted.
See the pictures.
![Console](https://github.com/Louisvdw/dbus-serialbattery/assets/127723834/51ecafe0-bddd-4283-b62a-4d
[image]
312937ab46)

I will check also OVP 3.48V and OVPR 3.45V in BMS settings and let you know.

jdeus, 2023-06-12:
@mr-manuel Due to the voltage-SOC relationship on LFP, the only way to ensure batteries stay balanced is to periodically increase the voltage such as all cells reach 3.6V. Once or twice a month seems reasonable. Currently, this option is missing from this project.

We should be able to define the frequency of this top-up.

Dighy, 2023-06-12:
> @Dighy @TazerReloaded, could you try to lower the `OVP` in the BMS settings to `3.48` volts and the OVPR to `3.45`. Then change the `MAX_CELL_VOLTAGE` in the `config.ini` to `3.48` and see, if this [...]

Feedback: I was in settings proposed for reseting SoC (max. voltage 3.51V each cell) and then I've changed back to max. 3.45V as default limit and reboot the CCGX. After, I changed in the BMS settings the OVPR to 3.45V and OVP to 3.48V and suddenly SoC increased from 86% to 95%. That time cells voltage was 3.45V
After going back to float (3.75V/cell), I've changed the OVP to 3.46V and OVPR to 3.43V and tomorow I will see what SoC will be coming. :)

mr-manuel, 2023-06-12:
@jdeus please open a feature request for this. This is not related to this issue.

@Dighy it will be interesting, if the JKBMS resets the SoC to 100%, if OVP is hit. Please check also, if this is needed for all cells or only one. Currently it seems that when OVPR is reachet the BMS resets to 95% and probably when OVP is reached the BMS resets to 100%.

mr-manuel, 2023-06-12:
We cannot change what the BMS firmware does with this SOC. You can try to write to the JKBMS developers, but I had no luck since months to receive valid answers...

TazerReloaded, 2023-06-12:
Setting the OVP to a value lower than the highest cell resets the SoC to 98% on my JKs, but also disconnects charging and raises a lot of alarms. It's an emergency function after all, to protect the cells if something in the charging logic goes wrong.
I had better luck completely ignoring the BMS SoC and just using it for cell protection. Victrons own coulomb counter seems to be pretty accurate, and it resets when absorption voltage is reached (55.2V in my setup).
The JKs tend to drift downwards over time, but not to the same amount, one BMS is only 6% lower than the actual SoC at the moment, the other one a whole 30%.
I did a current and voltage calibration with the same meter on both, but that did not improve things, they were already calibrated reasonably well from the factory. They were also bought together, same exact type, wiring and configuration on both.

Dighy, 2023-06-13:
> @jdeus please open a feature request for this. This is not related to this issue.
>
> @Dighy it will be interesting, if the JKBMS resets the SoC to 100%, if OVP is hit. Please check also, if this is [...]

Feedback: With CVL=3.45V as per driver defaut settings and OVP=3.46V in the BMS settings, The Soc was reseted to 100% once alarm os **high cell voltage** was triggered by BMS. This alarm was coming in cascade until I set back the OVP to higher value in order to let the controller judge the situation and reduce the current to stabilize at 3.45V.
This is only a solution if you are closer to intervene immediately.
My oppinion: During the charging, let's say, it is reached 3.45V and it waiting for cells balance and SoC  is 100%.  
Until evening, controller is trying to maintain the foalt voltage of 3.375V and when this value is increasing start discharging the battery or viceversa. At the and of the day, battery SoC will ve 2% max 3% lower, which means 98-97%. I think BMS is not counting this and next day is calculating the energy consumed during the night and it will charge up to 97-98% only and like this BMS SoC will be decreasing daily some value. 
I will follow this for next days to see if its confirmed.

ArendsM, 2023-06-15:
Hi all,

I’m owning a JK BMS as well and having the same issue. I would like to share my idea with you:

After or just before the MAX_VOLTAGE_TIME_SEC time is over, perform the following algorithm:

•	Rember actual BMS values for OVP and OVPR
•	Supress “High voltage alarm” in BMS readout loop by setting a filter
•	Change OVPR value to the value of the highest cell voltage – 0.5 by modifying the matching BMS register
•	Change OVP value to the value of the highest cell voltage – 0.1 by modifying the matching BMS register

That should trigger an OVP alert (at least for the BMS) and resets the SoC to 100%

•	Reset OVP and OVPR values back to the previous settings
•	Remove the filter to supress the “High voltage alarm”

What do you think?

Best regards
Meik

mr-manuel, 2023-06-17:
I had a similar idea and it should work, but is not that easy to implement. Anyway that would be a great feature!

It has to work with serial and Bluetooth connection. For the Bluetooth connection there is already some code to change settings.

Surpressing the BMS alert is the easiest thing, changing the `OVPR` and `OVP` values the trickiest one. Feel free to open a PR for this!

josmeijer, 2023-06-19:
I have the same (or comparable) problem with my DALY smart BMS. SOC was gradually decreasing and I reset it a couple of times (by means of the Daly app) to 100% but due to busy times I did not attend the instalation for a while to find out that the max SOX is 85% by now. I can find no anomalies (no reports of bad cells preventin it from full charge) 
If someone is in the opportunity to have a look and see if I overlooked something essential, I will set the VRM link to open for a while  https://vrm.victronenergy.com/installation/31525/share/c43e4823

thomasinaz41213, 2023-06-19:
> I have the same (or comparable) problem with my DALY smart BMS. SOC was gradually decreasing and I reset it a couple of times (by means of the Daly app) to 100% but due to busy times I did not [...]

This is a pretty common issue with Daly and some of the other bms. It just doesn't do a good job of calculating SOC. This is why I have a flow in node red to compare the bms SOC with the smart shunt SOC at intervals and when they are out of sync by a few % I use mqtt to set the bms SOC to match the smart shunt. Works great.

mr-manuel, 2023-06-19:
@josmeijer check https://github.com/Louisvdw/dbus-serialbattery/discussions/657 For Daly

ogurevich, 2023-06-21:
This variant looks like a plan. What will happen if the existing loads (e.g. electric stove or washing machine) are suddenly switched off at this very moment. Does BMS then disconnect the battery due to over voltage protection?

ArendsM, 2023-06-23:
> I had a similar idea and it should work, but is not that easy to implement. Anyway that would be a great feature!
>
> It has to work with serial and Bluetooth connection. For the Bluetooth connection [...]

Hi Manuel, I have some working code now to modify the `OVP` and `OVPR` registers in jkbms_brn.py. I'll let you know when I've included that functionality triggered by the `MAX_VOLTAGE_TIME_SEC` timeout. I'll create a PR as soon I feel comfortable with the results ;-)

ArendsM, 2023-06-23:
> This variant looks like a plan. What will happen if the existing loads (e.g. electric stove or washing machine) are suddenly switched off at this very moment. Does BMS then disconnect the battery due to over voltage protection?

Hi, only the charging MOSFETs will be switched off until `OVPR` voltage is reached again. Discharging is not affected. High loads during that time might affect the reset to 100% in case the higest battery voltage will drop under OVP to fast. Let's see ...

ogurevich, 2023-06-25:
Another control question: What about several batteries connected in parallel? The SOC is updated for one battery, but not for the others. And that first battery is almost always the same.

ArendsM, 2023-06-27:
> Another control question: What about several batteries connected in parallel? The SOC is updated for one battery, but not for the others. And that first battery is almost always the same.

Hi Oleg, with the new code submitted, all batteries should have their own instance of dbus-serialbattery and the SOC reset should work for all of them when the requirements are met.

ArendsM, 2023-06-27:
> Hi Manuel, I have some working code now to modify the `OVP` and `OVPR` registers in jkbms_brn.py. I'll let you know when I've included that functionality triggered by the `MAX_VOLTAGE_TIME_SEC` [...]

Hi Manuel, I've created PR #736. Please review my changes. I'm happy to get your feedback.

mr-manuel, 2023-06-28:
It seems that the JKBMS over the serial connection is read only. I now spend about 50 hours to search a solution and writing with the developers, but they answer with all other suff other than the answers to my questions...

The only workaround would be to use the Bluetooth connection (which I dont want) or to set the OVPR to the `MAX_CELL_VOLTAGE`. This resets the SoC at least to 95%.

Otherwise I can add an option to disable high voltage warning and you can set OVP to 3.48 V.

jdeus, 2023-06-30:
So it was in the end the correct answer. However, the proposed solution is problematic when batteries are in parallel (i.e more than one BMS). This bulk period should be determined by the controller aggregating multiple batteries and not on a per-battery (i.e per-BMS) basis

drago-more, 2023-06-30:
> The only workaround would be to use the Bluetooth connection (which I dont want) or to set the OVPR to the `MAX_CELL_VOLTAGE`. This resets the SoC at least to 95%.
> 

@mr-manuel Just fyi. This is not working for the JKBMS 4/8s, FW 11.XW. I just tested it with my two JKBMS'. The SOC is not reset to 95% even if all cells are exceeding OVPR.

mr-manuel, 2023-06-30:
> @mr-manuel Just fyi. This is not working for the JKBMS 4/8s, FW 11.XW. I just tested it with my two JKBMS'. The SOC is not reset to 95% even if all cells are exceeding OVPR.

I have to say, that I also tried it again and it do not work anymore with the OVPR voltage. If I check the stats I see that sometime it works and sometimes not.

ogurevich, 2023-07-01:
to get SOC recalculated try to change OVPR and then change capacity, for example from 280 to 281 it triggers recalculation in JKBMS. (remind to restore changed values) :)

mr-manuel, 2023-07-01:
I think when you change the capacity you also loose the history of the battery cycles.

drago-more, 2023-07-01:
The only confirmed and repeatable feature in the JKBMS rule set is the reset to 100% when exceeding OVP. Everything else seems to be dependent on FW version and various other factors which we don‘t know for sure.

Dighy, 2023-07-02:
Hi all,
for me this issue seems to be solved. The last reset of SOC I did about one week ago and since then my SOC is going daily up to 99% even 100% for a while, which is waht I need.
I would like to share with you my settings, this may work/help somebody to solve this issue. 
In BMS, I set the OVP at 3.5V permanently and 3.46V for reseting the SOC. OVPR at 3.4V to avoid alarms and restarting charging at higher cell voltage.
In the driver only thing I changed is float voltage reduce from 3.375V to 3.35V (26.8 in my 24V system).
It needs patience, from SOC reset until see the result. In my case only after third reset the things were going well. Before, reseting SOC, after 2-3days with 99% SOC, again start decreasing 1-2% daily (more or less). 
I attached my system overvie, where you can see the battery status. Only one day SOC gone up to 95% due to cloudy weather.
[image]

[image]

Valdiralita, 2023-07-14:
For me hitting OVP resets soc to 100%, currently i get alarms every day for high voltage which are expected. Thats not really desireable because everything freaks out and i get warnings in VRM, the remove console and the app.
Would either of these be possible?

- Disable high voltage alarm (i don't really like this but its the same as unplugging the bms and using a smart shunt, but at least we would get all other values from the bms)
- Disable high voltage alarm unless one cell voltage is over a defined limit in config.ini (for example OVP is 3.6V and alarm mute threshold is 3.61v)

mr-manuel, 2023-07-14:
It‘s already done. I‘m still testing it. You can download from my repository the dev branch. There you have a bulk voltage, after how many days the bulk voltage should be applied again and a muting of the high voltage alarm in the bulk voltage phase.

joseal, 2023-07-14:
@mr-manuel is it only working for the bluetooth connected jk BMSs? Or did you managed to make it work for the rs485 ?

mr-manuel, 2023-07-15:
The bulk voltage is working for all BMS that do not support the SoC reset directly.

joseal, 2023-07-15:
Thanks @mr-manuel ! I will test it and provide you some feedback.

Valdiralita, 2023-07-15:
> It‘s already done. I‘m still testing it. You can download from my repository the dev branch. There you have a bulk voltage, after how many days the bulk voltage should be applied again and a muting of the high voltage alarm in the bulk voltage phase.

> It was merged yesterday into the dev branch so you can install the nightly and select dev.

Using install.sh it installed 1.0.20230711dev today (option nightly -> dev) and set `BULK_CELL_VOLTAGE  = 3.610` and still getting high voltage alerts at 57.65V (16S).
With the version number im not sure if its the wrong version or wrong settings.

mr-manuel, 2023-07-15:
@Valdiralita what do you see in the parameter page in the remote console/GUI -> SerialBattery -> Parameters? Please post a screenshot.

Valdiralita, 2023-07-15:
> @Valdiralita what do you see in the parameter page in the remote console/GUI -> SerialBattery -> Parameters? Please post a screenshot. What is your max cell voltage and to which values did you set OVP and OVPR in the JKBMS app.

[image]

`MIN_CELL_VOLTAGE   = 2.750
MAX_CELL_VOLTAGE   = 3.600
FLOAT_CELL_VOLTAGE = 3.375
BULK_CELL_VOLTAGE  = 3.610`

The JKBMS has 3.6V OVP and 3.55V OVPR.

mr-manuel, 2023-07-15:
Max cell voltage and OVP should not be the same. Therefore you have the warning.

Set max voltage to 3.45 and try bulk voltage to 3.56. If it‘s not resetting to 100% SoC try to increase bulk voltage by 0.01 per test.

Valdiralita, 2023-07-15:
> Max cell voltage and OVP should not be the same. Therefore you have the warning.
> 
> Set max voltage to 3.45 and try bulk voltage to 3.56. If it‘s not resetting to 100% SoC try to increase bulk voltage by 0.01 per test.

I thought I need to exceed the OVP in the jkbms for it to set soc to 100%, which means the bulk voltage should be higher than the OVP?

mr-manuel, 2023-07-15:
If you have OVP set to 3.60 you won‘t be able to go higher. One cell has to reach OVP to set SoC to 100%. Therefore you can set bulk voltage a bit lower because of cell voltage difference.

joseal, 2023-07-18:
I'm doing something wrong for sure, but with this Dev build and the bellow settings, I'm stuck in Absorption (Linear Mode) all the time (I don't see bulk anymore, even when bellow 40%) and SoC is not occuring...

[image]

My current settings:
```
MAX_BATTERY_CHARGE_CURRENT = 60.0
MAX_BATTERY_DISCHARGE_CURRENT = 80.0

BULK_CELL_VOLTAGE  = 3.56

AUTO_RESET_SOC = True
JK_BMS_AUTO_RESET_SOC = True

```

[image]

mr-manuel, 2023-07-18:
Please read the `config.default.ini` again.

https://github.com/Louisvdw/dbus-serialbattery/blob/a24d6ef458239a1bc70399f774ae71348e48bf9d/etc/dbus-serialbattery/config.default.ini#L16-L28

joseal, 2023-07-18:
> Please read the `config.default.ini` again.
> 
> https://github.com/Louisvdw/dbus-serialbattery/blob/a24d6ef458239a1bc70399f774ae71348e48bf9d/etc/dbus-serialbattery/config.default.ini#L16-L28

Got it... thank you! Testing...

[image]

mr-manuel, 2023-07-18:
> Reached the 56.96v (BULK_CELL_VOLTAGE x 16) and moved to Float, but no SoC reset:

Just read again https://github.com/Louisvdw/dbus-serialbattery/issues/703#issuecomment-1636759210

> Doubt: do we need AUTO_RESET_SOC and JK_BMS_AUTO_RESET_SOC?

No, where does `JK_BMS_AUTO_RESET_SOC` come from?

joseal, 2023-07-20:
Hi @mr-manuel,

I'm sorry, you were more than right... I had a wrong understanding of this feature (I had the impression that you were temporarely adjusting the OVP), reason why I was so lost....

After review with more calm all the thread and look at the source, I've now a much better understanding os the issue and the challanages behind it.

I can confirm that the feature works fine (still need to do some fine tuning, to find the best values for me, but it is working):

[image]

It's not so good as being able to play with the OVP value, but it is good workaround!

Thanks again for the good work and patience.

mr-manuel, 2023-07-20:
> I'm sorry, you were more than right... I had a wrong understanding of this feature (I had the impression that you were temporarely adjusting the OVP), reason why I was so lost....

Unfortunately this is not possible via serial connection. I also wrote with the JKBMS developers and they were not really cooperative. They just told me that it's not possible via serial connection. With the Bluetooth connection it works without problems.

Neverthless, the workaround works for all BMS where you cannot reset the SOC via software.

Pengo84, 2023-07-20:
Hello mr-manuel,
does it mean when I connect via Bluetooth the JK-Bms with VenusOS the SoC will be set right when the voltage of 3.45V is reached?

mr-manuel, 2023-07-20:
The SoC of the JKBMS connected via Bluetooth will set to 100% when switching to float mode. PR is still open: https://github.com/Louisvdw/dbus-serialbattery/pull/736

Pengo84, 2023-07-20:
Do I have to set anything for this or is everything already set automatically?

mr-manuel, 2023-07-20:
As soon as it is merged with the dev branch I will post an instruction here.

Valdiralita, 2023-07-21:
Ive tested the dev version the previous days and it works really well with the following settings:

OVP: 3,6V
OVPR 3.55V
MIN_CELL_VOLTAGE   = 2.750
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375
BULK_CELL_VOLTAGE  = 3.601
BULK_AFTER_DAYS = 1

I'll continue using the dev branch until this is merged.

Pengo84, 2023-07-23:
Hello, another question about this marked value, what is it and what is it good for?
[image]

mr-manuel, 2023-07-23:
> Hello, another question about this marked value, what is it and what is it good for?

https://github.com/Louisvdw/dbus-serialbattery/blob/a24d6ef458239a1bc70399f774ae71348e48bf9d/etc/dbus-serialbattery/config.default.ini#L290-L302

Joeknx, 2023-08-04:
Thanks for the bulk voltage hack. 
It works with the JKBMS, bulk (3,535) slightly under OPV (3,54).
I have Grade A LF280K from EVE and they stay very balanced. 

Some questions/suggestions:
Can the tDiff be changed? 900 is pretty long for bulk.
When i "reinstall local" to start the bulk voltage for testing, the battery is charged with full speed (in my case 70A). 5A or so would be better in my opinion.

mr-manuel, 2023-08-04:
If you are using the latest nightly from the dev branch, then `MAX_VOLTAGE_TIME_SEC` is used.

maxx8888, 2023-08-27:
I'm not sure if i understand this Solution correctly.
Basically it means i need to run Batteries into OverVoltageProject from BMS every few days?
Or just short before disconnect? How is this working on multiple Batteries? Aggregate Batteries will run all of them up until they disconnect? If the Packs do not align, Pack1 would make reset on Day1, Pack2 reset on Day2 and so on?
Worried about Battery life and important disabled Alarm which might stay disabled.

I've exactly this issue on 5x Batteries with JKBMS.
What i've tried on one of them is to calibrate BMS on discharge to measure around 0,5 Amps less than what the actual current is. That means during discharging it measures to less and SOC stays higher than it really is. On the next re-charge the BMS hits 100% as the battery still absorbs more than what the BMS is expecting. This seems to work for a few days now... need to monitor. This Solution might also make troubles when Batteries get empty before SOC shows that...

PR #736 (2023-09-17): Introduction of automatic SOC reset for JK BMS Bluetooth only (HW Version 11)
This will add a feature to automatically reset SOC of JK BMS based on the `manage_charge_voltage` calculations.

**Warning:** This will modify `OVP` and `OVPR` settings of your BMS when enabled. **Use it at your own risk!**

The feature is disabled by default. `AUTO_RESET_SOC` and `JK_BMS_AUTO_RESET_SOC` have to be enabled in your config to make it work.

Probably fixes #703