Louisvdw/dbus-serialbattery #707: Battery does not charge until cutoff 55,2v anymore

dersch81, 2023-06-12:
### Describe the bug

After uprading to the latest release the battery stopps charging at ~55,12 v but the end voltage should be 55,2 v.

[image]

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

Charge until 55.2 overall voltage

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v2,94

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
cat /data/etc/dbus-serialbattery/config.default.ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 60.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example:
;     1 BMS: Jkbms_Ble C8:47:8C:00:00:00
;     3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =

; --------- BMS disconnect behaviour ---------
[... 385 lines cut ...]
@40000000647f03522a06433c WARNING:SerialBattery:Error reading settings from BMS, retry (10/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
@40000000647f03522a11dbfc INFO:SerialBattery:Testing HLPdataBMS4S
@40000000647f03570d232ab4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000647f03570d2b19f4 INFO:SerialBattery:Testing Jkbms
@40000000647f03571e9c7aac ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000647f03571ee7cfd4 INFO:SerialBattery:Testing Lifepower
@40000000647f0357303fdc2c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000647f035730954b44 INFO:SerialBattery:Testing LltJbd
@40000000647f035809be7414 INFO:SerialBattery:Connection established to LltJbd
@40000000647f035809d77e3c INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
@40000000647f03580a97057c INFO:SerialBattery:========== Settings ==========
@40000000647f03580a9724bc INFO:SerialBattery:> Connection voltage: 55.31V | Current: 3.12A | SoC: 99.96%
@40000000647f03580a973844 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@40000000647f03580a9743fc INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@40000000647f03580a97539c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@40000000647f03580a976724 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 300.0A | MAX BATTERY DISCHARGE CURRENT: 300.0A (read from BMS)
@40000000647f03580a988c1c INFO:SerialBattery:> CVCM:     False
@40000000647f03580a9897d4 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.7V | MAX CELL VOLTAGE: 3.47V
@40000000647f03580a98ab5c INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
@40000000647f03580a98b714 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
@40000000647f03580a98c6b4 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@40000000647f03580e5f0de4 INFO:SerialBattery:DeviceInstance = 1
@40000000647f03580e6c506c INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@40000000647f035819b0e58c INFO:SerialBattery:publish config values = 1
```

dersch81, 2023-06-12:
It's telling me CVL of 55.52v? From where is that value coming from?

[image]

It stopped at 55.1v like every day. 

[image]

To aggregate i'm using https://github.com/Dr-Gigavolt/dbus-aggregate-batteries and this is also selected as the main BMS

I'm totally lost why i can't reach the 55.2v anymore since i have upgraded serialbattery to the latest release.

mr-manuel, 2023-06-12:
Looking at all the data you provided it does not match. Can you contact me on Discord?

dersch81, 2023-06-12:
Do you mean this command? tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal 

i never got any output with it. 

CVCM has been changed and i did not restart after it. That mismatched the config and log. 

I have changed the MAX_BATTERY_CURRENT in the config.ini now. 

I will try the nightly

mr-manuel, 2023-06-12:
I will update it in the next hours, then it will work. If you don't want to wait copy the code from my `dev` branch.

dersch81, 2023-06-12:
Ok i will try this. Is there any change regarding the  initial symptom?

mr-manuel, 2023-06-12:
Since it's not clear what the problem is, no.

mr-manuel, 2023-06-12:
The changes are now in the `dev` branch and you can reinstall now.

dersch81, 2023-06-13:
No, doesn't start

```
root@VenusOS:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyACM0/current | tai64nlocal
2023-06-13 09:47:33.218853500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-06-13 09:47:33.218862500 INFO:SerialBattery:Testing Daly at address "\x40"
2023-06-13 09:47:33.327004500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:47:33.327718500 INFO:SerialBattery:Testing Daly at address "\x80"
2023-06-13 09:47:33.457481500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:47:33.458742500 INFO:SerialBattery:Testing Ecs
2023-06-13 09:47:33.525954500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:47:33.526662500 INFO:SerialBattery:Testing HeltecModbus
2023-06-13 09:47:34.061621500 WARNING:SerialBattery:Error reading settings from BMS, retry (1/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:34.478721500 WARNING:SerialBattery:Error reading settings from BMS, retry (2/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:34.916760500 WARNING:SerialBattery:Error reading settings from BMS, retry (3/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:35.332263500 WARNING:SerialBattery:Error reading settings from BMS, retry (4/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:35.749523500 WARNING:SerialBattery:Error reading settings from BMS, retry (5/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:36.160952500 WARNING:SerialBattery:Error reading settings from BMS, retry (6/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:36.575141500 WARNING:SerialBattery:Error reading settings from BMS, retry (7/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:37.010297500 WARNING:SerialBattery:Error reading settings from BMS, retry (8/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:37.422952500 WARNING:SerialBattery:Error reading settings from BMS, retry (9/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:37.833808500 WARNING:SerialBattery:Error reading settings from BMS, retry (10/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:47:37.834504500 INFO:SerialBattery:Testing HLPdataBMS4S
2023-06-13 09:48:02.079537500 INFO:SerialBattery:
2023-06-13 09:48:02.115787500 INFO:SerialBattery:Starting dbus-serialbattery
2023-06-13 09:48:02.115795500 INFO:SerialBattery:dbus-serialbattery v1.0.20230612dev
2023-06-13 09:48:02.115799500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-06-13 09:48:02.115802500 INFO:SerialBattery:Testing Daly at address "\x40"
2023-06-13 09:48:02.212532500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:48:02.212541500 INFO:SerialBattery:Testing Daly at address "\x80"
2023-06-13 09:48:02.339062500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:48:02.340041500 INFO:SerialBattery:Testing Ecs
2023-06-13 09:48:02.403686500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:48:02.403693500 INFO:SerialBattery:Testing HeltecModbus
2023-06-13 09:48:02.906838500 WARNING:SerialBattery:Error reading settings from BMS, retry (1/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:03.318891500 WARNING:SerialBattery:Error reading settings from BMS, retry (2/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:03.731123500 WARNING:SerialBattery:Error reading settings from BMS, retry (3/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:04.150973500 WARNING:SerialBattery:Error reading settings from BMS, retry (4/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:04.577395500 WARNING:SerialBattery:Error reading settings from BMS, retry (5/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:05.005766500 WARNING:SerialBattery:Error reading settings from BMS, retry (6/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:05.417443500 WARNING:SerialBattery:Error reading settings from BMS, retry (7/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:05.827416500 WARNING:SerialBattery:Error reading settings from BMS, retry (8/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:06.239153500 WARNING:SerialBattery:Error reading settings from BMS, retry (9/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:06.668891500 WARNING:SerialBattery:Error reading settings from BMS, retry (10/10): Wrong functioncode: 1 instead of 3. The response is: '\x01\x01Áà'
2023-06-13 09:48:06.673334500 INFO:SerialBattery:Testing HLPdataBMS4S
2023-06-13 09:48:11.203267500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:48:11.203740500 INFO:SerialBattery:Testing Jkbms
2023-06-13 09:48:11.483296500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:48:11.491744500 INFO:SerialBattery:Testing Lifepower
2023-06-13 09:48:11.819337500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-13 09:48:11.822662500 INFO:SerialBattery:Testing LltJbd
[... cut ...]
2023-06-13 09:48:41.372689500   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 1067, in log_settings
2023-06-13 09:48:41.372694500     logger.info(f"Serial Number/Unique Identifier: {self.unique_identifier()}")
2023-06-13 09:48:41.372896500   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 144, in unique_identifier
2023-06-13 09:48:41.372901500     "".join(filter(str.isalnum, self.hardware_version)) + "_"
2023-06-13 09:48:41.372905500 TypeError: 'float' object is not iterable
```

mr-manuel, 2023-06-13:
Thanks for reporting. The JBD BMS is reporting this value in another way. Could you please try with [`v1.0.20231306beta`](https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.0.20230613beta)?

dersch81, 2023-06-13:
[image]

dersch81, 2023-06-14:
But also with the new beta the charging is stopping at ~ 55,12 v

mr-manuel, 2023-06-14:
Please provide config, logs and screenshot without changing the config and all in the smallest timespan possible.

mr-manuel, 2023-06-29:
Please also measure with a multimeter the voltage difference between the charger output and the battery terminals. Since with `v1.0.0` the logic is different, you can configure a voltage drop of `0.08` or `0.10`.

closed 2023-06-29 by mr-manuel