Louisvdw/dbus-serialbattery #709: Writing SOC to the BMS?

Markus1301, 2023-06-13:
### Is your feature request related to a problem? Please describe.

Hi,
 have a 16S battery with Daly BMS. Because the current is not measured 100% accurately, I have a SOC drift. A little bit every day. This is a problem because I only want to discharge to 40% to have a reserve in case of a power outage. My idea is now: I charge up to 55.1V. If you set the SOC to e.g. 90% when 55.1V is reached, then it would help me. I've now checked whether I can put something like this in dbus-serialbattery. 

I don't understand the complete code, but I saw that there are functions to set the SOC. Is there such a function already or is it possible that you build something like that? That would be great.

### Describe the solution you'd like

Just something like: If Voltage = 55,1V then set SOC = 90%

mr-manuel, 2023-06-13:
Please see https://github.com/Louisvdw/dbus-serialbattery/discussions/657 anf join the discussion. Currently you can do this via Node-RED whis is included in Venus OS large.

Markus1301, 2023-06-13:
Thank you very much.

closed 2023-06-13 by mr-manuel