Louisvdw/dbus-serialbattery #710: Max_Charge_Current nur auf Laden!!

Micha854, 2023-06-13:
### Is your feature request related to a problem? Please describe.

Hallo,

Toller Treiber, Danke dafür!

Ich habe bei mir das Phänomen dass AC Verbraucher das Laden enorm einschränken!
Beispiel:

CVL Charge Current ist bei 15A weil Akku schon fast voll ist.
Wenn ich einen AC Verbraucher habe welcher 10A zieht wird nur noch mit 5A geladen

Anderes Beispiel:
ein AC Verbrauch von 20A sorgt dafür das die Batterie mit 5A entladen wird obwohl diese nicht einmal Voll war.

Ist das so gewollt?
Müssen die aktuellen Verbräuche bei der Max_Charge_Current nicht mit berücksichtigt werden?

mr-manuel, 2023-06-13:
Please provide more informations, since this is not enough to fully understand your problem.

1. How is your system made?
2. What devices do you have in your system?
3. Are you using ESS? If yes what settings do you have applied?

Please write in English as requested, when you open an issue. In the case you have difficulties to write in english, try the [deepl.com](https://www.deepl.com/) translator.

[image]

shabbab, 2023-07-19:
Maybe I can add more context as I have a similar issue.

My boat's setup has a 12v LiFePo with JKBMS, 3 Victron MPPTs, Victron SmartShunt, Victron Phoenix inverter, all connected to an RPi running VenusOS Large. ESS is disabled.

Victron console shows the following values:
BMS => Parameters: CVL 14V and CCL 20A, limited by SoC (>95%)
Inverter: 5.9A DC
MPPT => Networked Operation: Charge Voltage 14V, Charge Current 25.9A (total of all MPPTs unless the strongest sees enough sun to produce the 25.9A)

If I switch off the inverter the MPPTs show a total CCL of 20A requested. 

The Amps requested from all MPPTs are calculated based on CCL plus inverter draw.  

But there are other DC consumers connected to the battery, drawing roughly 15A which are _not_ taken into account. 

My MPPTs produce the 25.9A as requested, the DC consumers draw 15A and the battery only receives whatever is left over (5-6 A).
 
As the battery approaches 100% the CCL is dropped further. At some point the CCL barely matches the DC loads and starts to discharge.
The battery never reaches 100%. Depending on DC loads (5- 35A constantly) it fluctuates between 90-97%.

Current behavior: Amps requested from MPPTs = CCL + inverter draw.
Expected behavior: Amps requested from MPPTs = CCL + inverter draw + DC draw 

Basically the Amps requested from the MPPTs have to be constantly adjusted so BMS truly receives the Amps required to meet the CCL.

But I have no clue if this is a Victron issue or a driver issue.

mr-manuel, 2023-07-19:
Please ask this in the Victron community and post the link here.

shabbab, 2023-07-19:
After a few hours of digging through the Victron forum and manuals:
The Victron system considers only Victron inverters and Victron chargers for it's DVCC calculations. It does _not_ take any other DC loads nor DC chargers into consideration, so this is indeed intended behaviour  

Workaround is to wire & configure a separate SmartShunt to measure all non-Victron DC loads / chargers. I have to look into this in more detail to understand how that works.

shabbab, 2023-07-22:
To wrap this up:

Problem description:
With external DC loads the battery never sees full charge current, with external DC chargers the battery receives too much current.

Root cause:
Victron DVCC adjusts charge current only for Victron inverter loads and Victron chargers. Any Non-Victron DC devices are ignored, as are some Victron devices like my "Smart" Orion DC-DC charger which sucks. 

Official Solution: 
A SmartShunt wired between the DC loads / chargers and configured to become a measuring point for a DC system. 

Result:
Victron's DVCC now adjusts for all currents (Victron inverters & chargers as well as other DC loads & chargers as measured by the SmartShunt). The battery is charged with the current & voltage as calculated by the driver.

Alternative solution idea:
The driver could adjust the charge current limit reported to DVCC based on the actual current measured by the BMS. 
For example if the calculated charge limit is 20A but the BMS measures only 15A then the driver could increase the published charge limit to 25A. 
This obviously needs some more thinking, like sensible ramp up / down speeds & limits. Otherwise the driver would adjust multiple times for 5A missing, not realizing it's not caused by some unknown DC load but just the lack of sunshine.

Sounds a bit akward but saves 130 Euro for another Smartshunt.

mr-manuel, 2023-09-26:
Another solution might be a MQTT SmartShunt. Then you can calculate the other loads how you want with Node-RED or other tools.

shabbab, 2023-09-28:
The production & consumption data is already available in the system, it's just not correctly used by Victron.

mr-manuel, 2023-09-29:
I close this, since it's not a driver issue.

closed 2023-09-29 by mr-manuel