Louisvdw/dbus-serialbattery #716: Two JK BMS'ses with different hardware firmware and no MOSFET Temp

sstolk90, 2023-06-16:
### Describe the bug

I use the nightly build DEV branch with both JK BMS'ses on bluetooth. So far they connect good and all the values are displayed.
Exccept for the MOSFET Temp of one of the newer JK BMS'ses.
I have 1 BMS with firmware version 10.XW 16 cells and 1 with firmware version 11.XW. The one with 11.ZW does display the MOSFET temp in the JKBMS app but not in this driver. The MOSFET temp of the 10.XW is displayed in the app and also this driver.
Maybe the driver gets the information of the 11.XW on a different scheme so that the values are misplaced.

### How to reproduce

Always not displayed

### Expected behavior

MOSFET Temp not 0 degrees in the Remote Console / MQTT values

### Driver version

1.0.20230613dev

### Venus OS device type

Venus_GX

### Venus OS version

2.92

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; Jkbms_Ble C8:47:8C:E5:01:1B
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:EC:C0:B5, Jkbms_Ble C8:47:8C:E5:01:1B
BLOCK_ON_DISCONNECT = False
TIME_TO_GO_ENABLE = True

BMS_TYPE = Jkbms

; CUSTOM_BATTERY_NAMES = c8478ce5011b:Batterij Onder

; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.555
FLOAT_CELL_VOLTAGE = 3.535

; Set steps to reduce battery current
; The current will be changed linear between those steps if LINEAR_LIMITATION_ENABLE is set to True
CELL_VOLTAGES_WHILE_CHARGING   = 3.56, 3.50, 3.45, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.05,  0.5,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.70, 2.80, 2.90, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0,  0.1,  0.5,    1
```

### Relevant log output

```shell
==> /data/log/dbus-blebattery.0/current <==
2023-06-16 10:20:46.092590500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-06-16 10:20:46.092603500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-06-16 10:20:46.092606500 INFO:SerialBattery:Serial Number/Unique Identifier: 2120146523
2023-06-16 10:20:46.264484500 INFO:SerialBattery:DeviceInstance = 4
2023-06-16 10:20:46.266055500 INFO:SerialBattery:com.victronenergy.battery.c8478cecc0b5
2023-06-16 10:20:46.272138500 INFO:SerialBattery:BAT: JKBMS 11.XW 16 cells (20230504)
2023-06-16 10:20:46.350782500 INFO:SerialBattery:publish config values = 1
2023-06-16 10:25:30.836464500 [CHG] Device C8:47:8C:EC:C0:B5 RSSI: -50
2023-06-16 10:25:30.836476500 [CHG] Device C8:47:8C:EC:C0:B5 RSSI: -50

==> /data/log/dbus-blebattery.1/current <==
2023-06-16 10:20:45.454636500 INFO:SerialBattery:Serial Number/Unique Identifier: 2052302250
2023-06-16 10:20:45.506025500 INFO:SerialBattery:DeviceInstance = 5
2023-06-16 10:20:45.506767500 INFO:SerialBattery:com.victronenergy.battery.c8478ce5011b
2023-06-16 10:20:45.507499500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20221223)
2023-06-16 10:20:45.533133500 INFO:SerialBattery:publish config values = 1
```

PeerD, 2023-06-23:
I have a similar issue (But with a serial connection). But in my case it will not even connect.

The UI shows me a JK BMS has been found:
[image]

But it does not connect. The other JK (Hardware version 10.XW) does work correctly.

In the log I get the following line:
```
@40000000649572b800b3a63c ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000649572b800b841ec INFO:SerialBattery:Testing HLPdataBMS4S
@40000000649572bc1f4b19cc ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000649572bc1f501b0c INFO:SerialBattery:Testing Jkbms
@40000000649572cc1dc9d4e4 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:243/3262]
@40000000649572cc1dd1627c INFO:SerialBattery:Testing Lifepower
@40000000649572cc2dd72544 ERROR:SerialBattery:>>> ERROR: No reply - returning
@40000000649572cc2dde9784 INFO:SerialBattery:Testing LltJbd
@40000000649572cd027d90cc ERROR:SerialBattery:>>> ERROR: No reply - returning
```

And in the (official) App there are some changes aswell:

[image]
[image]
[image]

The JK BMS was discovered while it was in CAN mode. But the logs show in this case:
```
@4000000064957ac6249df854 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
...
@4000000064957acf0736a824 INFO:SerialBattery:Testing Jkbms
@4000000064957acf10093b74 ERROR:SerialBattery:>>> ERROR: No reply - returning [len:0]
```

In another discovery run I got:
```
@4000000064957b673585782c INFO:SerialBattery:Testing Jkbms
@4000000064957b673866492c ERROR:SerialBattery:CRC checksum mismatch: Expected 0x4f59, Got 0x4e92
```

Switching the Temp Sensor does not make a difference (just for completness sake)

I assume with Hardware Version 11.XW they changed the serial protocol. That there is an additional field in the config called "Device Address" suggests this even more.

PeerD, 2023-06-23:
Please disregard my previous comment for now. I had a serial node in my node-red which interfered with the connection.

I changed to a different (hardware 10) JK BMS and had the same issue. Then I found the node-red node. I will check with the hardware version 11 JK BMS again and report back!

drago-more, 2023-06-28:
The same here. In serial-battery the MOSFET temperatures of my two JKBMS' (4S, FW 11.XW, connected via BT) are shown as 0° independently from the real temperature. In the JKBMS app the values are correct.

fah, 2023-08-27:
I see the all the temperatures, in both JK BMS I own, via BT. 

Version 3.01 is used.

JKBMS :
FW: 1.0.20230531  
HW: JKBMS 10.XW 16 cells (20220808) 

Other JK BMS the same but other HW
HW: JKBMS 10.XW 16 cells (20220925)

mr-manuel, 2023-09-04:
The problem resides only with the 11.XW hardware version. Someone with such a BMS can contact me on Discrod, so I can take a look?

drago-more, 2023-09-04:
@mr-manuel temp_23.09.04

mr-manuel, 2023-09-04:
Does the `JKBMS 11.XW` only have one external temperature sensor?

Temp 1 is `24.4` which is equal to `244` (*10, decimal) or `F4` (hex)
Temp 2 is `24.6` which is equal to `246` (*10, decimal) or `F6` (hex)

In the logs you see `..., 244, 0, 246, 0, 0, 0, ...` where it should be `..., temp 1, 0, temp2, 0, mos temp, 0, ...` or `...\xf4\x00\xf6\x00\x00\x00...` for `..\temp1\x00\temp2\x00\mos temp\x00`.

<details><summary>Logs</summary>

```
2023-09-04 15:08:32.638047500 INFO:SerialBattery:
2023-09-04 15:08:32.638056500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-04 15:08:32.639599500 INFO:SerialBattery:dbus-serialbattery v1.0.20230902dev
2023-09-04 15:08:33.029785500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E2:00
2023-09-04 15:08:33.030507500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E2:00
tail: /var/log/dbus-blebattery.0/current has been replaced; following end of new file
2023-09-04 15:08:36.975458500 ERROR:root:--> NEW PACKAGE! lenght:  22
2023-09-04 15:08:36.976211500 ERROR:root:ncallback(): data
2023-09-04 15:08:36.976865500 ERROR:root:bytearray(b'\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00')
2023-09-04 15:08:36.977491500 ERROR:root:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  128
2023-09-04 15:08:36.978538500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0]
2023-09-04 15:08:36.979307500 ERROR:root:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  150
2023-09-04 15:08:36.980299500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
2023-09-04 15:08:37.025636500 ERROR:root:--> NEW PACKAGE! lenght:  128
2023-09-04 15:08:37.026347500 ERROR:root:ncallback(): data
2023-09-04 15:08:37.027054500 ERROR:root:bytearray(b'|\xf8\xff\xf3\x0f\x00\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-09-04 15:08:37.027720500 ERROR:root:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  150
2023-09-04 15:08:37.028945500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
2023-09-04 15:08:37.029748500 ERROR:root:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  278
2023-09-04 15:08:37.030917500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 124, 248, 255, 243, 15, 0, 0, 0, 0, 0, 0, 0, 144, 15, 0, 0, 0, 0, 192, 216, 3, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-09-04 15:08:37.064764500 ERROR:root:--> NEW PACKAGE! lenght:  22
2023-09-04 15:08:37.065479500 ERROR:root:ncallback(): data
2023-09-04 15:08:37.066117500 ERROR:root:bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02')
2023-09-04 15:08:37.066736500 ERROR:root:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  278
[... cut ...]
        await self.write_register(0x1E, b"\x01\x00\x00\x00", 4, c)
        await self.write_register(0x1F, b"\x01\x00\x00\x00", 4, c)
        await self.write_register(0x40, b"\x01\x00\x00\x00", 4, c)

if __name__ == "__main__":
    import sys

    jk = Jkbms_Brn(sys.argv[1])
    if not jk.test_connection():
        logging.error(">>> ERROR: Unable to connect")
    else:
        jk.start_scraping()
        while True:
            logging.debug(jk.get_status())
            sleep(5)

```

</details>

drago-more, 2023-09-04:
Each JKBMS has two external temperature sensors plus the internal one for the MOS. See the attached screen shots.

[PastedGraphic-1.png][PastedGraphic-2.png]

On 4 Sep 2023, at 17:26, Manuel ***@***.***> wrote:

Does the JKBMS 11.XW only have one external temperature sensor?

Temp 1 is 24.4 which is equal to 244 (*10, decimal) or F4 (hex)
Temp 2 is 24.6 which is equal to 246 (*10, decimal) or F6 (hex)

In the logs you see ..., 244, 0, 246, 0, 0, 0, ... where it should be ..., temp 1, 0, temp2, 0, mos temp, 0, ... or ...\xf4\x00\xf6\x00\x00\x00... for ..\temp1\x00\temp2\x00\mos temp\x00.

Logs

2023-09-04 15:08:32.638047500 INFO:SerialBattery:
2023-09-04 15:08:32.638056500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-04 15:08:32.639599500 INFO:SerialBattery:dbus-serialbattery v1.0.20230902dev
2023-09-04 15:08:33.029785500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E2:00
2023-09-04 15:08:33.030507500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E2:00
tail: /var/log/dbus-blebattery.0/current has been replaced; following end of new file
2023-09-04 15:08:36.975458500 ERROR:root:--> NEW PACKAGE! lenght:  22
2023-09-04 15:08:36.976211500 ERROR:root:ncallback(): data
2023-09-04 15:08:36.976865500 ERROR:root:bytearray(b'\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00')
2023-09-04 15:08:36.977491500 ERROR:root:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  128
2023-09-04 15:08:36.978538500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0]
2023-09-04 15:08:36.979307500 ERROR:root:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  150
2023-09-04 15:08:36.980299500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
2023-09-04 15:08:37.025636500 ERROR:root:--> NEW PACKAGE! lenght:  128
2023-09-04 15:08:37.026347500 ERROR:root:ncallback(): data
2023-09-04 15:08:37.027054500 ERROR:root:bytearray(b'|\xf8\xff\xf3\x0f\x00\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-09-04 15:08:37.027720500 ERROR:root:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  150
2023-09-04 15:08:37.028945500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
2023-09-04 15:08:37.029748500 ERROR:root:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  278
2023-09-04 15:08:37.030917500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 124, 248, 255, 243, 15, 0, 0, 0, 0, 0, 0, 0, 144, 15, 0, 0, 0, 0, 192, 216, 3, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-09-04 15:08:37.064764500 ERROR:root:--> NEW PACKAGE! lenght:  22
2023-09-04 15:08:37.065479500 ERROR:root:ncallback(): data
2023-09-04 15:08:37.066117500 ERROR:root:bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe/\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02')
[... cut ...]
2023-09-04 15:08:43.095604500 ERROR:root:decode_cellinfo_jk02(): self.frame_buffer
2023-09-04 15:08:43.102184500 ERROR:root:[85, 170, 235, 144, 2, 48, 52, 13, 51, 13, 52, 13, 51, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 15, 0, 0, 0, 52, 13, 2, 0, 0, 1, 43, 0, 42, 0, 40, 0, 38, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0

drago-more, 2023-09-04:
Latest TCP address ...

[PastedGraphic-1.png]

On 4 Sep 2023, at 17:39, Tempuno ***@***.***> wrote:

I am only at my camper for the next 10 minutes ... let me know if you need anything else which is only visible in the app via Blutetooth.

On 4 Sep 2023, at 17:36, Tempuno ***@***.***> wrote:

Each JKBMS has two external temperature sensors plus the internal one for the MOS. See the attached screen shots.

<PastedGraphic-1.png><PastedGraphic-2.png>

On 4 Sep 2023, at 17:26, Manuel ***@***.***> wrote:

Does the JKBMS 11.XW only have one external temperature sensor?

Temp 1 is 24.4 which is equal to 244 (*10, decimal) or F4 (hex)
Temp 2 is 24.6 which is equal to 246 (*10, decimal) or F6 (hex)

In the logs you see ..., 244, 0, 246, 0, 0, 0, ... where it should be ..., temp 1, 0, temp2, 0, mos temp, 0, ... or ...\xf4\x00\xf6\x00\x00\x00... for ..\temp1\x00\temp2\x00\mos temp\x00.

Logs

2023-09-04 15:08:32.638047500 INFO:SerialBattery:
2023-09-04 15:08:32.638056500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-04 15:08:32.639599500 INFO:SerialBattery:dbus-serialbattery v1.0.20230902dev
2023-09-04 15:08:33.029785500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E2:00
2023-09-04 15:08:33.030507500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E2:00
tail: /var/log/dbus-blebattery.0/current has been replaced; following end of new file
2023-09-04 15:08:36.975458500 ERROR:root:--> NEW PACKAGE! lenght:  22
2023-09-04 15:08:36.976211500 ERROR:root:ncallback(): data
2023-09-04 15:08:36.976865500 ERROR:root:bytearray(b'\x00\x00\x00\x00\x00\x00Input Userdata\x00\x00')
2023-09-04 15:08:36.977491500 ERROR:root:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  128
2023-09-04 15:08:36.978538500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0]
2023-09-04 15:08:36.979307500 ERROR:root:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  150
2023-09-04 15:08:36.980299500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
2023-09-04 15:08:37.025636500 ERROR:root:--> NEW PACKAGE! lenght:  128
2023-09-04 15:08:37.026347500 ERROR:root:ncallback(): data
2023-09-04 15:08:37.027054500 ERROR:root:bytearray(b'|\xf8\xff\xf3\x0f\x00\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-09-04 15:08:37.027720500 ERROR:root:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  150
2023-09-04 15:08:37.028945500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
2023-09-04 15:08:37.029748500 ERROR:root:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  278
2023-09-04 15:08:37.030917500 ERROR:root:[85, 170, 235, 144, 3, 40, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 240, 124, 131, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 124, 248, 255, 243, 15, 0, 0, 0, 0, 0, 0, 0, 144, 15, 0, 0, 0, 0, 192, 216, 3, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,  [... cut ...]  5, 0, 9, 0, 138, 128, 64, 64, 0, 0, 0, 0, 71, 5, 163, 29, 0, 1, 0, 1, 0, 5, 0, 0, 212, 185, 137, 4, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 254, 255, 127, 220, 31, 1, 0, 0, 0, 0, 0, 0, 246]
2023-09-04 15:08:43.095604500 ERROR:root:decode_cellinfo_jk02(): self.frame_buffer
2023-09-04 15:08:43.102184500 ERROR:root:[85, 170, 235, 144, 2, 48, 52, 13, 51, 13, 52, 13, 51, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0

mr-manuel, 2023-09-04:
Unfortunately the pictures were not attached to GitHub via mail.

drago-more, 2023-09-04:
Resent it to your other e-mail address.

mr-manuel, 2023-09-04:
I searched for temperature values from `25.0` to `27.9` in the raw data, but there isn't any. Maybe @baranator can help us?

rver2024, 2023-10-29:
Any progress on fixing this issue?

I am putting together a battery system together for a RV build and just found out this awesome project, installed it on the Cerbo GX and everything seems to be working, but I have a jk-b2a8s20p 11.XW BMS which has this MOSFET temperature reading 0 issue.

Coding is not really my forte since I am a hardware guy, but I know my way around the Linux, I can help troubleshooting if needed, just need the direction. Thanks!

jensbehrens, 2023-11-01:
Hi, just found following:
https://github.com/syssi/esphome-jk-bms/issues/326
could this be of help?

jensbehrens, 2023-11-01:
Changed to 112 instead of 134 for temperature_mos…
That did the trick!

rver2024, 2023-11-02:
Thank you! @jensbehrens I followed your lead, changed 134 to 112 in jkbms_brn.py, and voila, the MOSFET temperature shows up.

mr-manuel, 2023-11-02:
Would be interesting of this works also for hardware version 10. I will add this in the meanwhile for version 11.

jensbehrens, 2023-11-02:
In https://github.com/syssi/esphome-jk-bms/blob/main/components/jk_bms_ble/jk_bms_ble.cpp#LL555C1-L555C119 they do a check for 32S frame (line 554) and use 112, in other cases (line 587) 134 is used for the mos temp.

mr-manuel, 2023-11-02:
Could you please try with this file? https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/dev/etc/dbus-serialbattery/bms/jkbms_brn.py

mr-manuel, 2023-11-02:
I found another bug in the code where for some 11.* JKBMS no values were shown. Since I have to change the check if the data structure is 16s or 32s I need someone with a 10.* JKBMS connected over Bluetooth. Can someone with this requisite contact me on Discord?

mr-manuel, 2023-11-02:
The issue was fixed in `v1.0.20231102dev` which is the nightly of today.

closed 2023-11-02 by mr-manuel

rver2024, 2023-11-03:
Hi @mr-manuel, I installed the v1.0.20231102dev nightly to my Cerbo GX, now the driver doesn’t read most of the parameters, soc, voltage, current, temperatures, etc. All the individual cell voltages do show up in “Cell Voltages” submenu though. I am not sure what’s going on.

mr-manuel, 2023-11-03:
Can you please enable debug logging by adding this in the `config.ini`

~~~ini
LOGGING = DEBUG
~~~

then execute `reinstall-local.sh` wait some time and then send me as much logs as possible (about 100 lines needed at least)?

rver2024, 2023-11-03:
Hi @mr-manuel, here is the log from my Cerbo GX. Thank you very much for your help!

2023-11-03 13:14:35.500031500 DEBUG:SerialBattery:logged to dbus [0]
2023-11-03 13:14:35.502175500 DEBUG:SerialBattery:Cells:[1]3.339V [2]3.338V [3]3.34V [4]3.342V [5]3.34V [6]3.34V [7]3.339V [8]3.339V [9]NoneV [10]NoneV [11]NoneV [12]NoneV [13]NoneV [14]NoneV [15]NoneV [16]NoneV 
2023-11-03 13:14:35.878436500 DEBUG:SerialBattery:--> NEW PACKAGE! lenght:  128
2023-11-03 13:14:35.879111500 DEBUG:SerialBattery:ncallback(): data
2023-11-03 13:14:35.879827500 DEBUG:SerialBattery:bytearray(b'U\xaa\xeb\x90\x02"\x0b\r\x0b\r\x0c\r\x0f\r\x0c\r\x0b\r\x0c\r\x0c\r\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x00\x00\x0c\r\x03\x00\x03\x003\x002\x002\x003\x003\x003\x003\x005\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-11-03 13:14:35.880387500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  0
2023-11-03 13:14:35.880929500 DEBUG:SerialBattery:[]
2023-11-03 13:14:35.881579500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  128
2023-11-03 13:14:35.882388500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 34, 11, 13, 11, 13, 12, 13, 15, 13, 12, 13, 11, 13, 12, 13, 12, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 12, 13, 3, 0, 3, 0, 51, 0, 50, 0, 50, 0, 51, 0, 51, 0, 51, 0, 51, 0, 53, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-11-03 13:14:36.458988500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  0
2023-11-03 13:14:36.458997500 DEBUG:SerialBattery:[]
2023-11-03 13:14:36.458999500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  128
2023-11-03 13:14:36.459003500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 35, 11, 13, 11, 13, 12, 13, 14, 13, 12, 13, 12, 13, 12, 13, 12, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 12, 13, 3, 0, 3, 0, 51, 0, 50, 0, 50, 0, 51, 0, 51, 0, 51, 0, 51, 0, 53, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-11-03 13:14:36.459141500 DEBUG:SerialBattery:--> NEW PACKAGE! lenght:  22
2023-11-03 13:14:36.459167500 DEBUG:SerialBattery:ncallback(): data
2023-11-03 13:14:36.459169500 DEBUG:SerialBattery:bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe6\x00\x00\x00\x00\x00')
2023-11-03 13:14:36.459174500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  128
2023-11-03 13:14:36.459178500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 35, 11, 13, 11, 13, 12, 13, 14, 13, 12, 13, 12, 13, 12, 13, 12, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 12, 13, 3, 0, 3, 0, 51, 0, 50, 0, 50, 0, 51, 0, 51, 0, 51, 0, 51, 0, 53, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-11-03 13:14:36.459229500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  150
2023-11-03 13:14:36.459250500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 35, 11, 13, 11, 13, 12, 13, 14, 13, 12, 13, 12, 13, 12, 13, 12, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 12, 13, 3, 0, 3, 0, 51, 0, 50, 0, 50, 0, 51, 0, 51, 0, 51, 0, 51, 0, 53, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 230, 0, 0, 0, 0, 0]
2023-11-03 13:14:36.479469500 DEBUG:SerialBattery:--> NEW PACKAGE! lenght:  128
2023-11-03 13:14:36.480118500 DEBUG:SerialBattery:ncallback(): data
2023-11-03 13:14:36.480759500 DEBUG:SerialBattery:bytearray(b'dh\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xdb\x00\xdb\x00\x00\x00\x00\x00\x00\x00\x00b\xf6\x8d\x04\x00\xb0\x9b\x04\x00\x01\x00\x00\x00\xa4^\x07\x00d\x00\x00\x00Q\xa9:\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x00\x01\x00\x00\x00\xa1\x03\x00\x00\x00\x00\x929>@\x00\x00\x00\x00p\n\x1e\x1c\x00\x01\x00\x01\x00\x05\x00\x00\\/\x14\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2023-11-03 13:14:36.481386500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  150
2023-11-03 13:14:36.482275500 DEBUG:SerialBattery:[85, 170, 235, 144, 2, 35, 11, 13, 11, 13, 12, 13, 14, 13, 12, 13, 12, 13, 12, 13, 12, 13, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 255, 0, 0, 0, 12, 13, 3, 0, 3, 0, 51, 0, 50, 0, 50, 0, 51, 0, 51, 0, 51, 0, 51, 0, 53, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 230, 0, 0, 0, 0, 0]
[... cut ...] l_count_wrong': False, 'charge_overtemp': False, 'charge_undertemp': False, 'discharge_overtemp': False, 'cell_overvoltage': False, 'cell_undervoltage': False, 'charge_overcurrent': False, 'discharge_overcurrent': False}}
2023-11-03 13:14:38.740261500 DEBUG:SerialBattery:logged to dbus [0]
2023-11-03 13:14:38.741487500 DEBUG:SerialBattery:Cells:[1]3.339V [2]3.339V [3]3.34V [4]3.343V [5]3.34V [6]3.34V [7]3.34V [8]3.34V [9]NoneV [10]NoneV [11]NoneV [12]NoneV [13]NoneV [14]NoneV [15]NoneV [16]NoneV

mr-manuel, 2023-11-03:
Please change

https://github.com/Louisvdw/dbus-serialbattery/blob/a6d45e6776077ec7ed4989584108b5a6737dda0a/etc/dbus-serialbattery/bms/jkbms_brn.py#L169

to

```python
if fb[70] == 255: 
```

and then run `reinstall-local.sh`

rver2024, 2023-11-03:
Made the change per your instruction and it seems the problem is fixed. Thank you very much!

rver2024, 2023-11-03:
My hardware version is JKBMS 11.XW 8 Cells (20221225)

rver2024, 2023-11-03:
Can this dbus driver be used to turn the JKBMS’s charge and discharge on and off, through the remote console or the config.ini file? Or I will have to do it on JKBMS’ app?

Sorry I understand this is more of a support question and I am hijacking the thread here, but I would be very much appreciated if someone knowledgeable can answer it quickly. Thanks!

mr-manuel, 2023-11-03:
Tecnically it's possible over Bluetooth but not implemented yet.

baranator, 2023-11-03:
Sorry that i couldn't be helpful in this issue - I'm too busy at the moment. But it's very nice to see that you got this fixed! 

@mr-manuel your enhancements of the bluetooth-functionality and the dbus-serialbattery brought great progress! From this perspective i'm now quite happy, that i killed the wired interface of my jkbms not even a year ago :D

mr-manuel, 2023-11-06:
Yeah, all started with your defective serial port. Great that it happened :-D

rver2024, 2023-11-06:
Hi @mr-manuel, is this issue considered fixed and closed? Or you will further improve it to accommodate different versions of JKBMS, 8 cells, 16 cells, etc. As you know, the v1.0.20231102dev nightly doesn’t work on my 8 cells JK-B2A8S20P BMS without the fix in jkbms_brn.py. Thank you very much!

mr-manuel, 2023-11-06:
The fix for your BMS will be included in the next merge.

drago-more, 2023-11-08:
@mr-manuel I also tested the latest nightly build **v1.0.20231103dev** with my **JKBMS 11.XW 4 cells (20230527)**. Line 168 in jkbms_brn.py is changed to `if fb[70] == 255:`  Still, most of the parameters (volt, current, SOC etc.) are not showing up in serial driver. The standard log does not show any anomalies at start but ends with zero division errors ... I assume due to the missing parameters.

```
2023-11-08 12:34:39.511688500 INFO:SerialBattery:
2023-11-08 12:34:39.512974500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-08 12:34:39.514936500 INFO:SerialBattery:dbus-serialbattery v1.0.20231103dev
2023-11-08 12:34:40.149216500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E2:00
2023-11-08 12:34:40.150944500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E2:00
2023-11-08 12:34:45.180733500 INFO:SerialBattery:BAT: JKBMS 11.XW 4 cells (20230527)
2023-11-08 12:34:45.182091500 INFO:SerialBattery:Connection established to Jkbms_Ble
2023-11-08 12:34:45.183230500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce6e200
2023-11-08 12:34:45.183900500 INFO:SerialBattery:========== Settings ==========
2023-11-08 12:34:45.184576500 INFO:SerialBattery:> Connection voltage: 0.0V | Current: 0.0A | SoC: 0%
2023-11-08 12:34:45.185424500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2023-11-08 12:34:45.186277500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2023-11-08 12:34:45.188492500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 40.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2023-11-08 12:34:45.189530500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS)
2023-11-08 12:34:45.189905500 INFO:SerialBattery:> CVCM:     True
2023-11-08 12:34:45.190643500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-11-08 12:34:45.191341500 INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
2023-11-08 12:34:45.192189500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2023-11-08 12:34:45.193134500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2023-11-08 12:34:45.194018500 INFO:SerialBattery:Serial Number/Unique Identifier: 2081215375
2023-11-08 12:34:45.294446500 INFO:SerialBattery:DeviceInstance = 1
2023-11-08 12:34:45.295554500 INFO:SerialBattery:com.victronenergy.battery.c8478ce6e200
2023-11-08 12:34:45.296531500 INFO:SerialBattery:BAT: JKBMS 11.XW 4 cells (20230527)
2023-11-08 12:34:45.346034500 INFO:SerialBattery:publish config values = 1
2023-11-08 12:34:45.934202500 ERROR:SerialBattery:Exception occurred: ZeroDivisionError('float division by zero') of type <class 'ZeroDivisionError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #638
2023-11-08 12:35:45.441425500 ERROR:SerialBattery:Exception occurred: ZeroDivisionError('float division by zero') of type <class 'ZeroDivisionError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #638
2023-11-08 12:36:45.285741500 ERROR:SerialBattery:Exception occurred: ZeroDivisionError('float division by zero') of type <class 'ZeroDivisionError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #638
2023-11-08 12:37:45.221731500 ERROR:SerialBattery:Exception occurred: ZeroDivisionError('float division by zero') of type <class 'ZeroDivisionError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #638
2023-11-08 12:38:45.073800500 ERROR:SerialBattery:Exception occurred: ZeroDivisionError('float division by zero') of type <class 'ZeroDivisionError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #638 
```

mr-manuel, 2023-11-08:
Please set the logging to debug and post the logs again.

drago-more, 2023-11-08:
@mr-manuel Here are the first 10 seconds after driver start:

<details><summary>Log</summary>
<p>

````2023-11-08 20:34:54.221429500 INFO:SerialBattery:
2023-11-08 20:34:54.222644500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-08 20:34:54.224323500 INFO:SerialBattery:dbus-serialbattery v1.0.20231103dev
2023-11-08 20:34:54.868147500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E2:00
2023-11-08 20:34:54.875962500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E2:00
2023-11-08 20:34:54.879981500 DEBUG:SerialBattery:scraping thread started -> main thread id: 3070177296 scraping thread: 3051521120
2023-11-08 20:34:54.883073500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): Connect and scrape on address: C8:47:8C:E6:E2:00
2023-11-08 20:34:55.364801500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): btloop
2023-11-08 20:34:55.365678500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): reconnect
tail: /data/log/dbus-blebattery.0/current has been replaced; following end of new file
2023-11-08 20:34:57.895864500 DEBUG:SerialBattery:ncallback(): \x55\xaa\xeb\x90\x03\xc1\x4a\x4b\x5f\x42\x32\x41\x38\x53\x32\x30\x50\x00\x00\x00\x00\x00\x31\x31\x2e\x58\x57\x00\x00\x00\x31\x31\x2e\x31\x37\x00\x00\x00\x9c\x7a\xd9\x00\x16\x00\x00\x00\x42\x61\x74\x74\x2d\x31\x20\x53\x42\x00\x00\x00\x00\x00\x00\x00\x31\x32\x33\x34\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x32\x33\x30\x35\x32\x37\x00\x00\x32\x30\x38\x31\x32\x31\x35\x33\x37\x35\x00\x30\x30\x30\x30\x00\x49\x6e\x70\x75\x74\x20\x55\x73\x65\x72\x64\x61\x74\x61\x00\x00\x31\x34\x33\x35\x31\x31\x00\x00\x00\x00
2023-11-08 20:34:57.898913500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  0
2023-11-08 20:34:57.902224500 DEBUG:SerialBattery:bytearray(b'')
2023-11-08 20:34:57.903611500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  128
2023-11-08 20:34:57.907294500 DEBUG:SerialBattery:[85, 170, 235, 144, 3, 193, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 156, 122, 217, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0]
2023-11-08 20:34:57.910826500 DEBUG:SerialBattery:--> NEW PACKAGE! lenght:  22
2023-11-08 20:34:57.914653500 DEBUG:SerialBattery:ncallback(): \x00\x00\x00\x00\x00\x00\x49\x6e\x70\x75\x74\x20\x55\x73\x65\x72\x64\x61\x74\x61\x00\x00
2023-11-08 20:34:57.915860500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  128
2023-11-08 20:34:57.918045500 DEBUG:SerialBattery:[85, 170, 235, 144, 3, 193, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 156, 122, 217, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0]
2023-11-08 20:34:57.921355500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (after extend) -> lenght:  150
2023-11-08 20:34:57.924685500 DEBUG:SerialBattery:[85, 170, 235, 144, 3, 193, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 156, 122, 217, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
2023-11-08 20:34:57.947814500 DEBUG:SerialBattery:--> NEW PACKAGE! lenght:  128
2023-11-08 20:34:57.950342500 DEBUG:SerialBattery:ncallback(): \x7c\xf8\xff\xf3\x0f\x00\x00\x00\x00\x00\x00\x00\x90\x0f\x00\x00\x00\x00\xc0\xd8\x03\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00
2023-11-08 20:34:57.951569500 DEBUG:SerialBattery:--> assemble_frame() -> self.frame_buffer (before extend) -> lenght:  150
2023-11-08 20:34:57.954662500 DEBUG:SerialBattery:[85, 170, 235, 144, 3, 193, 74, 75, 95, 66, 50, 65, 56, 83, 50, 48, 80, 0, 0, 0, 0, 0, 49, 49, 46, 88, 87, 0, 0, 0, 49, 49, 46, 49, 55, 0, 0, 0, 156, 122, 217, 0, 22, 0, 0, 0, 66, 97, 116, 116, 45, 49, 32, 83, 66, 0, 0, 0, 0, 0, 0, 0, 49, 50, 51, 52, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 50, 51, 48, 53, 50, 55, 0, 0, 50, 48, 56, 49, 50, 49, 53, 51, 55, 53, 0, 48, 48, 48, 48, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0, 49, 52, 51, 53, 49, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 73, 110, 112, 117, 116, 32, 85, 115, 101, 114, 100, 97, 116, 97, 0, 0]
[... cut ...] scharging_switch_enabled': False, 'balancing_active': False, 'error_bitmask_16': '0x0', 'error_bitmask_2': '0000000000000000', 'power': 0.0}, 'warnings': {'resistance_too_high': False, 'cell_count_wrong': False, 'charge_overtemp': False, 'charge_undertemp': False, 'discharge_overtemp': False, 'cell_overvoltage': False, 'cell_undervoltage': False, 'charge_overcurrent': False, 'discharge_overcurrent': False}}
2023-11-08 20:35:04.919817500 DEBUG:SerialBattery:logged to dbus [0] ````
</p>
</details>

bjoernh, 2023-11-11:
Today I had the same Problem that all main Parameters are Zeros. 
[image]

So I installed the Nightly Build and this fixed my problem too.

[image]

What I also observed (before the update), the cycles count was very high (54000) and not stable it goes up and down all the time.

I think the bug will happened, because my Battery is in the garage and now the temperatures outside going down. So this will happen by other JK-BMS Bluetooth users as well.

@mr-manuel thank you very much for fixing this issue 😃

sstolk90, 2023-11-12:
The reason why you cycles count was high because it is displaying the
voltage there.
I had the same issue if you look at my screenshots.

drago-more, 2023-11-17:
@mr-manuel Could you find anything in my log?

mr-manuel, 2023-11-17:
@drago-more sorry I'm pretty busy at the moment. In the meanwhile could you try to change

https://github.com/Louisvdw/dbus-serialbattery/blob/a3fe5ebc093b290336ad73f2faada258915758d1/etc/dbus-serialbattery/bms/jkbms_brn.py#L173

to

```python
if 1 == 1:
```

then check, if now all data is shown correctly, After change it to

```python
if 1 != 1:
```

Please report back which one works for you. You have to run `/data/etc/dbus-serialbattery/reinstall-local.sh` after each change or better reboot the system.

drago-more, 2023-11-17:
@mr-manuel Easy, no stress! 😉 

What works for me is 
```
        self.bms_max_cell_count = 32
        self.translate_cell_info = TRANSLATE_CELL_INFO_32S
```
It shows all values incl. the MOSFET temperature. Thus the issue seems to be with the IF condition `fb[70] == 255`. In my case (JKBMS 11.XW, 4 cells) it results to `FALSE` instead to `TRUE`.

mr-manuel, 2023-11-17:
Could you all please test the latest nightly and check, if it works for you?

drago-more, 2023-11-17:
Works for JKBMS 11.XW, 4 cells. 👍

rver2024, 2023-11-18:
What is the proper way to update to the latest nightly? I installed the latest nightly on my Cerbo GX without uninstalling the existing driver first, it didn’t work, I had to uninstall the existing driver then install the new one to make it work, is it the correct procedure?

Anyway, the latest nightly works for my 8-cell jkbms with all parameters correctly displayed, will need to let it run for longer to check the Bluetooth connection stability. I do have one issue on this version that clicking “Menu” on the console screen doesn’t bring up the menu, it will return to the main screen if I am in a submenu, but nothing happens if clicking it on main screen.

mr-manuel, 2023-11-18:
You can just execute the `install.sh` and select to install the nightly build and then select the dev branch. No uninstall needed, only a reboot.

rver2024, 2023-11-18:
Hi @mr-manuel, I had to reboot twice for my jkbms to display all the parameters in my case, the first reboot only shows the voltage I remember. I has since done an uninstall first and then the install, then everything seems to work fine.

One additional question, is there option to set balancing start voltage in this driver? Say, I want the jkbms to balance cells only above certain voltage. Thanks!

mr-manuel, 2023-11-18:
No, you can only set the balancing voltage through the JKBMS app, since this is not something you need at a daily base. You set it once and forget about it, until you optimized your system.

rver2024, 2023-11-18:
Got it. Thank you! @mr-manuel

rver2024, 2023-12-08:
Hi @mr-manuel, has this fix been committed to the latest release in the main branch? I lost track of it since I accidentally damage my JKBMS, I since bought a new one, exactly the same one from JK, installed the latest driver by selecting option 1, but the “MOSFET temperature” is still showing 0. I tried to modify the jkbms_brn.py the way you told me before, but I couldn’t find the line with code “ if fb[70] == 255 and fb[71] == 255: ” anymore. Any inside on how the fix the issue? My new JKBMS is also a 8-cell with 11.XW hardware version, exactly the same as my old BMS. Thank you very much!

mr-manuel, 2023-12-08:
Install the latest `nightly` of the `dev` branch.

rver2024, 2023-12-08:
> Install the latest `nightly` of the `dev` branch.

Yep, the latest dev version of the nightly works. Thanks!

commit 04c74562 (2023-11-17): fixes #716
https://github.com/Louisvdw/dbus-serialbattery/issues/716