Louisvdw/dbus-serialbattery #720: Driver gets unresponsive, if connection is lost on Bluetooth

mr-manuel, 2023-06-17:
### Describe the bug

See https://github.com/Louisvdw/dbus-serialbattery/pull/689#issuecomment-1592977482

### How to reproduce

Steps to reproduce the behavior:
1. Use a BMS that uses `callback` instead of `polling` (e.g. JKBMS_BLE)
2. Disconnect the BMS physically

### Expected behavior

Driver should restart

### Driver version

v1.0.20230617beta

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.10~8

### BMS type

JKBMS / Heltec BMS

### Cell count

18

### Connection type

Bluetooth

### Config file

```ini
Not needed
```

### Relevant log output

```shell
2023-06-15 12:41:45.074095500 INFO:SerialBattery:processing frame with battery cell info
2023-06-15 12:41:45.080366500 INFO:SerialBattery:--> DBusHelper: publish_battery()
2023-06-15 12:41:45.081041500 INFO:SerialBattery:--> JKBMS_BLE: refresh_data()
2023-06-15 12:41:45.081637500 INFO:SerialBattery:--> JKBMS_BRN: get_status()
2023-06-15 12:41:45.613783500 INFO:SerialBattery:processing frame with battery cell info
2023-06-15 12:41:45.620835500 INFO:SerialBattery:--> DBusHelper: publish_battery()
2023-06-15 12:41:45.621594500 INFO:SerialBattery:--> JKBMS_BLE: refresh_data()
2023-06-15 12:41:45.622258500 INFO:SerialBattery:--> JKBMS_BRN: get_status()
2023-06-15 12:41:46.154417500 INFO:SerialBattery:processing frame with battery cell info
2023-06-15 12:41:46.160787500 INFO:SerialBattery:--> DBusHelper: publish_battery()
2023-06-15 12:41:46.161481500 INFO:SerialBattery:--> JKBMS_BLE: refresh_data()
2023-06-15 12:41:46.162093500 INFO:SerialBattery:--> JKBMS_BRN: get_status()
2023-06-15 12:41:46.709338500 INFO:SerialBattery:processing frame with battery cell info
2023-06-15 12:41:46.720693500 INFO:SerialBattery:--> DBusHelper: publish_battery()
2023-06-15 12:41:46.721490500 INFO:SerialBattery:--> JKBMS_BLE: refresh_data()
2023-06-15 12:41:46.722210500 INFO:SerialBattery:--> JKBMS_BRN: get_status()
2023-06-15 12:41:46.897818500 INFO:SerialBattery:--> JKBMS_BRN: asy_connect_and_scrape(): Ended
2023-06-15 12:41:46.899188500 INFO:SerialBattery:--> self.main_thread.is_alive(): True
2023-06-15 12:41:47.233798500 [CHG] Device C8:47:8C:E6:E2:00 RSSI: -48
2023-06-15 12:41:47.233807500 [CHG] Device C8:47:8C:E6:E2:00 RSSI: -58
2023-06-15 12:41:47.233810500 [CHG] Device C8:47:8C:E6:E2:00 RSSI: -48
2023-06-15 12:41:47.233812500 [CHG] Device C8:47:8C:E6:E2:00 RSSI is nil
```

mr-manuel, 2023-06-25:
@seidler2547 maybe you already have an idea how to fix that?

djkakadu, 2023-12-29:
I have a new insight into this problem. It only happens when an external bluetooth dongle is used. When I removed it from the USB and restarted the Cerbo GX, it no longer disconnects the batteries and has been working stable for 3 days now.

closed 2024-02-16 by mr-manuel