Louisvdw/dbus-serialbattery #721: DALY Inverted_current_meassurement not (properly) working

josmeijer, 2023-06-18:
### Describe the problem

I did insert the line "Inverted_current_meassurement = -1 " in a config.ini file (after loading the latest build of the driver file) to find that the charging is more or less behaving like before finding out about this parameter and the need to use it.
Funny thing is, that replacing it with (positive) 1 makes no visible difference.
[image]

https://github.com/Louisvdw/dbus-serialbattery/assets/35534038/7282e632-2341-4272-994a-ade6cb72eb58

### Driver version

1.0.20230531

### Venus OS device type

CCGX

### Venus OS version

3.10~8

### BMS type

Daly Smart BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60
INVERT_CURRENT_MEASUREMENT = -1
```

### Relevant log output

```shell
no time now
```

### Any other information that may be helpful

no

thomasinaz41213, 2023-06-18:
Did you restart the driver? Invert current works for me.

josmeijer, 2023-06-18:
Lots of times.

mr-manuel, 2023-06-21:
Please check the documentation. There was also a link on how to get the logfiles when you opened the issue.