Louisvdw/dbus-serialbattery #723: High Voltage in the Night, Everyday

Micha854, 2023-06-19:
### Describe the bug

i use venus os 3.00 release and the latest nightly dev version of the serial driver.

I have almost every after the message "high voltage"

my daly bms is max cell voltage set to 3.65V and max voltage to 28.40V

### How to reproduce

i dont no

### Expected behavior

[image]
[image]

### Driver version

1.0.20230613dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.00

### BMS type

Daly Smart BMS

### Cell count

8

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0
TIME_TO_SOC_VALUE_TYPE = 2
BMS_TYPE = Daly
BATTERY_CAPACITY = 105
INVERT_CURRENT_MEASUREMENT = -1
```

### Relevant log output

```shell
no
```

teefixx, 2023-06-19:
Hi, not sure if that helps, but I had exactly the same issue.

root cause was a loose sense wire for a cell. This caused a frequent jumping of a cell voltage and the BMS threw out the exact same error you have. I was very surprised, that not a cell voltage error was thrown out, but anyway. 
The jump was not visible in the VenusOS console, but with 1s interval  recordings of the values in influxDB.

mr-manuel, 2023-06-25:
Did you searched first the existing and closed issues? Please see https://github.com/Louisvdw/dbus-serialbattery/issues/653

mr-manuel, 2023-06-29:
The alarm is triggered by the BMS, so this driver only displays it. Please check it with the PC software, since with the app you don't see everything.

closed 2023-06-29 by mr-manuel