Louisvdw/dbus-serialbattery #724: JKBMS

polocup100, 2023-06-19:
### Describe the problem

Hello i have a JKBMS JK-B2A24S15P-C used for a lifepo 16S 280Ah Battery.

Charged by a Mppt Easun IGrid 5600, with an ESS System Victron multiplus 5000.

The JK BMS is connected via Bluetooth to the Venus GX and it is working fine. When the battery is charged or discharged everything is ok, but when it is charged by the Mppt and discharged by the multiplus at the same time , then the current measurements becomes 0A , 0W for along time, and of course the SOC did not change. If the loads are getting smaller 400W or less the measurements starts again but the SOC are wrong.

Has someone any idea?

Best regards

Sascha

### Driver version

latest 

### Venus OS device type

Raspberry Pi

### Venus OS version

latest 

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
No
```

### Relevant log output

```shell
No
```

### Any other information that may be helpful

No

pos-ei-don, 2023-06-19:
Adding a victron smart shunt device solved the Problem here.
The JKBMS just is not good in measuring current.

mr-manuel, 2023-06-19:
Please do not write latest in the version you use. Today one version is the latest tomorrow another. Maybe you are using also the latest beta. Specify please the exact version number.

polocup100, 2023-06-19:
Thanks for your fast reply,

that is exactly what I did.
But then the General consumption in the Dashboard was just calculating from
the Battery current and the Grid consumption. Everything from my MPPT was
not calculating. With the JKBMS as the Battery monitor the calculation was
correct.

pos-ei-don, 2023-06-19:
Where dit you connect the mppt ? on the left side or on the right side of the shunt?

polocup100, 2023-06-19:
The same as the Multiplus is connected, then the shunt and then the
battery. In serial to the JKBMS

Otherwise the SOC would not be calculated correctly.

pos-ei-don, 2023-06-19:
So the shunt does not see the power from the mppt. and the mppt is not connected to the victron gx device.
This has nothing to do with this driver.

Can you read the Values from the Easun  somehow? maybe you can add it as a Device to the Victron System via mqtt?!?
maybe you can use something like 
https://github.com/freakent/dbus-mqtt-devices

polocup100, 2023-06-19:
Thanks for helping,

but when i use only the JKBMS the measurement is correct.
What is the difference?

mr-manuel, 2023-06-20:
Maybe also this could interest you: [venus-os_dbus-mqtt-solar-charger](https://github.com/mr-manuel/venus-os_dbus-mqtt-solar-charger)

closed 2023-06-29 by mr-manuel