Louisvdw/dbus-serialbattery #725: Daly BMS - 'Allow to balance' = No

mrgithub, 2023-06-20:
### Describe the bug

Is the Daly BMS balancing deactivated? If so how to activate as I can't seem to figure it out.

[image]

### How to reproduce

On touchpanel

### Expected behavior

Should be enabled.

### Driver version

1.03

### Venus OS device type

Raspberry Pi

### Venus OS version

v3

### BMS type

Daly Smart BMS

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = -1

; Daly BMS is 125A max charge 250A max discharge and EVE 304A batteries and Multi 48/5000-70
MAX_BATTERY_CHARGE_CURRENT = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 250.0
```

### Relevant log output

```shell
.
```

mr-manuel, 2023-06-20:
Try to turn `Turn balancing off` on and then off again. It that don‘t work you have to check the Daly mobile or PC app.

transistorgit, 2023-06-20:
Hi,
the balancing is controlled by the BMS and will start above a configured voltage level, like 3.3V (I think this is the default in Daly BMS) and if the cell diff is greater than a configured value (0.05V default).

The allow Balance should automatically switch on as soon as both requirements are met.

(if you have an active balancer in parallel, make shure to check the values there also, so that the gui reflects the real balancing behaviour)

mrgithub, 2023-06-20:
Is there a way to check and set the balancing parameters via the driver?

mr-manuel, 2023-06-20:
> The allow Balance should automatically switch on as soon as both requirements are met.

Oh ok. Didn't know that. On the JKBMS it's always on :-)

transistorgit, 2023-06-20:
> Is there a way to check and set the balancing parameters via the driver?
> 

No, thats not implemented. you have to use the daly sw or app.

mrgithub, 2023-06-20:
When is the Daly BMS settings due to be implemented?

transistorgit, 2023-06-20:
Probably never, as you just need it once or at least very seldom

closed 2023-06-21 by mr-manuel