Louisvdw/dbus-serialbattery #728: Jkbms_Ble: Bluetooth died

itsdaveit, 2023-06-22:
### Describe the problem

Hi, i have 2 JKBMS connected via Bluetooth, the initial setup worked well.
After a few hours i frequently lose connection to the BMS.
The devices are just a meter apart.

### Driver version

v1.0.20230531

### Venus OS device type

Cerbo GX

### Venus OS version

V3.00

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 150
MAX_BATTERY_DISCHARGE_CURRENT = 150
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E5:C5:66, Jkbms_Ble C8:47:8C:E6:E8:32
```

### Relevant log output

```shell
@400000006492e8921f483f54 device_info
@400000006492e8921fc9eaa4 device_info
@400000006492e8921fca09e4 Exiting bt-loop
@400000006492e89239e45dac Missing device address argument
@400000006492e894147cf4fc INFO:SerialBattery:Starting dbus-serialbattery
@400000006492e8941491d0d4 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@400000006492e8942fec3234 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E5:C5:66
@400000006492e8942ff7b76c INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E5:C5:66
@400000006492e89f1a1f2434 INFO:SerialBattery:Jkbms_Ble found!
@400000006492e89f1a4132f4 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20221009)
@400000006492e89f1a6ab7dc INFO:SerialBattery:Connection established to Jkbms_Ble
@400000006492e89f1a7b3e54 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce5c566
@400000006492e89f1a8bb914 INFO:SerialBattery:========== Settings ==========
@400000006492e89f1acd03ec INFO:SerialBattery:> Connection voltage: 55.12V | Current: 1.7A | SoC: 99%
@400000006492e89f1acd2714 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@400000006492e89f1b481ba4 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@400000006492e89f1b54c1ec INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A
@400000006492e89f1b772e6c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS)
@400000006492e89f1b83a9bc INFO:SerialBattery:> CVCM:     True
@400000006492e89f1b911354 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@400000006492e89f1b9f480c INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@400000006492e89f1bad599c INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@400000006492e89f1bb861d4 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@400000006492e89f1bc23d44 INFO:SerialBattery:Serial Number/Unique Identifier: 2080303198
@400000006492e89f24b1f96c INFO:SerialBattery:DeviceInstance = 2
@400000006492e89f24bde81c INFO:SerialBattery:com.victronenergy.battery.c8478ce5c566
@400000006492e89f24cd804c INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20221009)
@400000006492e89f27e3b92c INFO:SerialBattery:publish config values = 1
@4000000064935562214ee8ac INFO:SerialBattery:Jkbms_Ble: Bluetooth died
@4000000064935562214f0bd4 INFO:SerialBattery:Reset of Bluetooth triggered
@40000000649355622155f944 INFO:SerialBattery:Scraping ended, issuing sys-commands
@40000000649355642f1f974c Starting bluetooth: bluetoothd.
@40000000649355643088379c INFO:SerialBattery:Bluetooth should have been restarted
@4000000064935566313cf524 INFO:SerialBattery:Jkbms_Ble: Bluetooth died
@40000000649355673157eb7c INFO:SerialBattery:Jkbms_Ble: Bluetooth died
@4000000064935568316cc754 INFO:SerialBattery:Jkbms_Ble: Bluetooth died
@400000006493556931807e34 INFO:SerialBattery:Jkbms_Ble: Bluetooth died
@400000006493556a319b2e3c INFO:SerialBattery:Jkbms_Ble: Bluetooth died
@400000006493556b31c29814 INFO:SerialBattery:Jkbms_Ble: Bluetooth died
```

### Any other information that may be helpful

I modified the current settings, after generating the log. (thats why it says: @400000006492e89f1b54c1ec INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A
@400000006492e89f1b772e6c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A (read from BMS))

mr-manuel, 2023-06-22:
Please install the latest `nightly` version of the `dev` branch.

itsdaveit, 2023-06-23:
Thanks for your response. Just did it, now i`m at v1.0.20230620dev. After the first reboot, only one battery connected. The second one did this in a loop:

```shell
@40000000649562e91d364774 INFO:SerialBattery:Starting dbus-serialbattery
@40000000649562e91d4b522c INFO:SerialBattery:dbus-serialbattery v1.0.20230620dev
@40000000649562e938da11cc INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E8:32
@40000000649562e938ebc50c INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E8:32
@40000000649562f52f273c54 ERROR:SerialBattery:No BMS found at C8:47:8C:E6:E8:32
@40000000649562f52f33e684 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@40000000649562f612db81cc
@40000000649562f612f5b4d4 INFO:Bluetooth details
@40000000649562f63465cf8c Attempting to disconnect from C8:47:8C:E6:E8:32
@40000000649562f63465f2b4 [^AESC[0;93m^BCHG^AESC[0m^B] Device CC:6E:A4:D0:31:75 RSSI is nil
@40000000649562f634660254 [^AESC[0;91m^BDEL^AESC[0m^B] Device CC:6E:A4:D0:31:75 CC-6E-A4-D0-31-75
@40000000649562f634660e0c [^AESC[0;93m^BCHG^AESC[0m^B] Device 00:04:4B:F3:94:C6 RSSI is nil
@40000000649562f6346619c4 [^AESC[0;93m^BCHG^AESC[0m^B] Device CE:74:DD:51:70:CC RSSI is nil
@40000000649562f634662194 [^AESC[0;93m^BCHG^AESC[0m^B] Controller F0:B0:40:FD:5F:47 Discovering: no
@40000000649562f634662d4c Successful disconnected
@40000000649562fc06ca942c Device C8:47:8C:E6:E8:32 (public)
@40000000649562fc06cab36c       Alias: BAT02
@40000000649562fc06cabb3c       Paired: no
@40000000649562fc06cabf24       Trusted: no
@40000000649562fc06cac6f4       Blocked: no
@40000000649562fc06cacadc       Connected: no
@40000000649562fc06cacec4       LegacyPairing: no
@40000000649562fc06cad694       UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@40000000649562fc06d80594
@40000000649562fd26eb3414 INFO:SerialBattery:
@40000000649562fd26f5a00c INFO:SerialBattery:Starting dbus-serialbattery
@40000000649562fd2709d7d4 INFO:SerialBattery:dbus-serialbattery v1.0.20230620dev
@40000000649562fe06c30694 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E8:32
@40000000649562fe06cf4364 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E8:32
@40000000649563083826dae4 ERROR:SerialBattery:No BMS found at C8:47:8C:E6:E8:32
@40000000649563083826fe0c ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@40000000649563090f033d24
@40000000649563090f0caf1c INFO:Bluetooth details
@40000000649563092421e2b4 Attempting to disconnect from C8:47:8C:E6:E8:32
@4000000064956309242205dc [^AESC[0;93m^BCHG^AESC[0m^B] Device CE:74:DD:51:70:CC RSSI is nil
@40000000649563092422157c [^AESC[0;93m^BCHG^AESC[0m^B] Controller F0:B0:40:FD:5F:47 Discovering: no
@400000006495630924222134 Successful disconnected
@400000006495630e3553a644 Device C8:47:8C:E6:E8:32 (public)
@400000006495630e3553cd54       Alias: BAT02
@400000006495630e3553d524       Paired: no
@400000006495630e3553dcf4       Trusted: no
@400000006495630e3553e0dc       Blocked: no
@400000006495630e3553e4c4       Connected: no
@400000006495630e3553ec94       LegacyPairing: no
@400000006495630e3553f07c       UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@400000006495630e357635ec
@40000000649563101d98e1a4 INFO:SerialBattery:
@40000000649563101da3bafc INFO:SerialBattery:Starting dbus-serialbattery
@40000000649563101db625d4 INFO:SerialBattery:dbus-serialbattery v1.0.20230620dev
@400000006495631100e30ef4 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E6:E8:32
@400000006495631100fe6ec4 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E6:E8:32
```

After another reboot, the second battery started also. I will watch, if the initial problem occurs again.

mr-manuel, 2023-06-23:
You installed with the install script or by uploading via FTP? This looks like a line endling problem.

itsdaveit, 2023-06-24:
I Installed with:

```shell
wget -O /tmp/install.sh https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/install.sh

bash /tmp/install.sh
```
Also switched to nightly/dev with this method.

But: No crashes since yesterday. It looks promissing. :-)

closed 2023-06-29 by mr-manuel

apharcha, 2024-11-04:
I am getting a very similar problem where 2 of my JKBMS's Connect fine at setup first during install but become unresponsive over sometime 5-7hours later. I just updated the latest branch of code from MrManuels last night update. Any idea what this might be?

==> /data/log/dbus-blebattery.1/current <==
2024-11-04 01:03:43.846236500 Device C8:47:80:07:A7:35 (public)
2024-11-04 01:03:43.846242500   Alias: JK3101A16S15P-02
2024-11-04 01:03:43.846245500   Paired: no
2024-11-04 01:03:43.846247500   Trusted: no
2024-11-04 01:03:43.846249500   Blocked: no
2024-11-04 01:03:43.846251500   Connected: no
2024-11-04 01:03:43.846254500   LegacyPairing: no
2024-11-04 01:03:43.846256500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-11-04 01:03:43.846259500   RSSI: -59
2024-11-04 01:03:43.846570500
2024-11-04 01:03:45.746110500 INFO:SerialBattery:
2024-11-04 01:03:45.746578500 INFO:SerialBattery:Starting dbus-serialbattery
2024-11-04 01:03:45.747663500 INFO:SerialBattery:Venus OS v3.50~34
2024-11-04 01:03:45.748055500 INFO:SerialBattery:dbus-serialbattery v1.5.20241031dev
2024-11-04 01:03:56.601713500 *** CCGX booted (0) ***
2024-11-04 01:04:52.992728500
2024-11-04 01:04:52.992732500
2024-11-04 01:04:52.992734500 INFO:Preparing Bluetooth for connection to BMS
2024-11-04 01:04:52.992737500 INFO:Bluetooth details
2024-11-04 01:04:53.219318500 Attempting to disconnect from C8:47:80:07:A7:35
2024-11-04 01:04:53.219323500 Successful disconnected
2024-11-04 01:04:58.319776500 Device C8:47:80:07:A7:35 (public)
2024-11-04 01:04:58.319782500   Alias: JK3101A16S15P-02
2024-11-04 01:04:58.319785500   Paired: no
2024-11-04 01:04:58.319787500   Trusted: no
2024-11-04 01:04:58.319789500   Blocked: no
2024-11-04 01:04:58.319791500   Connected: no
2024-11-04 01:04:58.319794500   LegacyPairing: no
2024-11-04 01:04:58.319796500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-11-04 01:04:58.319800500   RSSI: -59
2024-11-04 01:04:58.329713500
2024-11-04 01:05:02.018473500 INFO:SerialBattery:
2024-11-04 01:05:02.018758500 INFO:SerialBattery:Starting dbus-serialbattery
2024-11-04 01:05:02.019830500 INFO:SerialBattery:Venus OS v3.50~34
2024-11-04 01:05:02.020148500 INFO:SerialBattery:dbus-serialbattery v1.5.20241031dev
2024-11-04 01:05:18.322690500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:07:A7:35
2024-11-04 01:05:18.322696500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:07:A7:35
2024-11-04 01:05:30.273422500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-11-04 01:05:30.344116500 ERROR:SerialBattery:No BMS found at C8:47:80:07:A7:35
2024-11-04 01:05:30.344499500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-11-04 01:05:30.711670500
2024-11-04 01:05:30.711808500
2024-11-04 01:05:30.711877500 INFO:Preparing Bluetooth for connection to BMS
2024-11-04 01:05:30.711936500 INFO:Bluetooth details
2024-11-04 01:05:30.804448500 [CHG] Device C8:47:80:07:A7:35 RSSI: -59
2024-11-04 01:05:30.804454500 [CHG] Device C8:47:80:07:A7:35 RSSI: -58
2024-11-04 01:05:30.914157500 Attempting to disconnect from C8:47:80:07:A7:35
2024-11-04 01:05:30.914162500 [CHG] Device C9:F7:EE:2A:2B:77 RSSI is nil
2024-11-04 01:05:30.914168500 [CHG] Device F6:EF:F9:33:1A:D7 RSSI is nil
2024-11-04 01:05:30.914174500 [CHG] Device C8:69:CD:75:4E:60 TxPower is nil
2024-11-04 01:05:30.914181500 [CHG] Device C8:69:CD:75:4E:60 RSSI is nil
2024-11-04 01:05:30.914187500 [CHG] Device 31:8D:35:46:79:91 RSSI is nil
2024-11-04 01:05:30.914193500 [CHG] Device FF:FF:DE:02:21:90 RSSI is nil
2024-11-04 01:05:30.914272500 [CHG] Device FF:FF:DE:01:CD:08 RSSI is nil
2024-11-04 01:05:30.914278500 [CHG] Device FF:FF:30:06:66:A5 RSSI is nil
2024-11-04 01:05:30.914283500 [CHG] Device 4F:77:55:64:73:4A TxPower is nil
2024-11-04 01:05:30.914289500 [CHG] Device 4F:77:55:64:73:4A RSSI is nil
2024-11-04 01:05:30.914294500 [CHG] Device 35:94:FA:17:5A:11 RSSI is nil
2024-11-04 01:05:30.914299500 [CHG] Device 7B:74:85:16:69:DB RSSI is nil
2024-11-04 01:05:30.914342500 [CHG] Device 6C:88:E6:1F:31:B4 TxPower is nil
2024-11-04 01:05:30.914347500 [CHG] Device 6C:88:E6:1F:31:B4 RSSI is nil
2024-11-04 01:05:30.914351500 [CHG] Device C8:47:80:07:A6:7E RSSI is nil
2024-11-04 01:05:30.914354500 [CHG] Device 9C:20:7B:B1:B5:93 RSSI is nil
2024-11-04 01:05:30.914357500 [CHG] Device C8:47:80:07:A7:35 RSSI is nil
2024-11-04 01:05:30.914433500 [CHG] Device 04:21:4C:00:7F:28 RSSI is nil
2024-11-04 01:05:30.914437500 [CHG] Controller D8:3A:DD:9F:B5:70 Discovering: no
2024-11-04 01:05:30.914441500 Successful disconnected
[... 35 lines cut ...]
2024-11-04 01:06:06.854074500 [CHG] Device 6C:88:E6:1F:31:B4 TxPower is nil
2024-11-04 01:06:06.854078500 [CHG] Device 6C:88:E6:1F:31:B4 RSSI is nil
2024-11-04 01:06:06.854081500 [CHG] Device 7B:74:85:16:69:DB RSSI is nil
2024-11-04 01:06:06.854124500 [CHG] Device C8:47:80:07:A6:7E RSSI is nil
2024-11-04 01:06:06.854128500 [CHG] Device 9C:20:7B:B1:B5:93 RSSI is nil
2024-11-04 01:06:06.854131500 [CHG] Device C8:47:80:07:A7:35 RSSI is nil
2024-11-04 01:06:06.854134500 [CHG] Device 04:21:4C:00:7F:28 RSSI is nil
2024-11-04 01:06:06.854137500 [CHG] Device C9:F7:EE:2A:2B:77 RSSI is nil
2024-11-04 01:06:06.854201500 [CHG] Device F6:EF:F9:33:1A:D7 RSSI is nil
2024-11-04 01:06:06.854204500 [CHG] Controller D8:3A:DD:9F:B5:70 Discovering: no
2024-11-04 01:06:06.854208500 Successful disconnected
2024-11-04 01:06:12.002126500 Device C8:47:80:07:A7:35 (public)
[... cut ...]
2024-11-04 01:06:12.002143500   LegacyPairing: no
2024-11-04 01:06:12.002145500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-11-04 01:06:12.002149500   RSSI: -57
2024-11-04 01:06:12.003411500
2024-11-04 01:06:13.984152500 INFO:SerialBattery:
2024-11-04 01:06:13.984658500 INFO:SerialBattery:Starting dbus-serialbattery
2024-11-04 01:06:13.985928500 INFO:SerialBattery:Venus OS v3.50~34
2024-11-04 01:06:13.986432500 INFO:SerialBattery:dbus-serialbattery v1.5.20241031dev

apharcha, 2024-11-04:
More logs as the problem re-appears : 
2024-11-04 01:11:34.388641500 INFO:SerialBattery:Starting dbus-serialbattery
2024-11-04 01:11:34.389625500 INFO:SerialBattery:Venus OS v3.50~34
2024-11-04 01:11:34.389955500 INFO:SerialBattery:dbus-serialbattery v1.5.20241031dev
2024-11-04 01:11:50.671779500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:80:07:A6:7E
2024-11-04 01:11:50.672089500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:80:07:A6:7E
2024-11-04 01:11:54.181391500 INFO:SerialBattery:BAT: JKBMS 14.XA 16S (20240505)
2024-11-04 01:11:54.182141500 INFO:SerialBattery:Connection established to Jkbms_Ble
2024-11-04 01:11:54.512660500 INFO:SerialBattery:Found existing battery with DeviceInstance = 1
2024-11-04 01:11:54.646397500 INFO:SerialBattery:DeviceInstance = 1
2024-11-04 01:11:54.647538500 INFO:SerialBattery:PID file created successfully: /var/tmp/dbus-serialbattery_1.pid
2024-11-04 01:11:54.648422500 INFO:SerialBattery:Used DeviceInstances = ['1', '2']
2024-11-04 01:11:54.648841500 INFO:SerialBattery:com.victronenergy.battery.ble_c8478007a67e
2024-11-04 01:11:54.669799500 INFO:SerialBattery:publish config values = False
2024-11-04 01:11:54.675233500 INFO:SerialBattery:Battery JKBMS BLE connected to dbus from ble_c8478007a67e
2024-11-04 01:11:54.675239500 INFO:SerialBattery:========== Settings ==========
2024-11-04 01:11:54.675242500 INFO:SerialBattery:> Connection voltage: 54.26 V | Current: 8.9 A | SoC: 93%
2024-11-04 01:11:54.675246500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2024-11-04 01:11:54.676017500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2024-11-04 01:11:54.676510500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.900 V | MAX CELL VOLTAGE: 3.450 V| FLOAT CELL VOLTAGE: 3.375 V
2024-11-04 01:11:54.676942500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0 A | MAX BATTERY DISCHARGE CURRENT: 60.0 A
2024-11-04 01:11:54.677317500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0 A | MAX BATTERY DISCHARGE CURRENT: 150.0 A (read from BMS)
2024-11-04 01:11:54.677640500 INFO:SerialBattery:> CVCM:     True
2024-11-04 01:11:54.678007500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2024-11-04 01:11:54.678347500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2024-11-04 01:11:54.678673500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2024-11-04 01:11:54.679003500 INFO:SerialBattery:> CHARGE FET: True | DISCHARGE FET: True | BALANCE FET: True
2024-11-04 01:11:54.679338500 INFO:SerialBattery:Serial Number/Unique Identifier: 3092570117
2024-11-04 01:13:56.084737500
2024-11-04 01:13:56.084741500
2024-11-04 01:13:56.084743500 INFO:Preparing Bluetooth for connection to BMS
2024-11-04 01:13:56.085100500 INFO:Bluetooth details
2024-11-04 01:13:56.136722500 [CHG] Device C8:47:80:07:A6:7E RSSI: -57
2024-11-04 01:13:56.136729500 [CHG] Device C8:47:80:07:A6:7E RSSI: -56
2024-11-04 01:13:58.202492500 Attempting to disconnect from C8:47:80:07:A6:7E
2024-11-04 01:13:58.202498500 [CHG] Device AE:A3:76:B2:0D:B5 RSSI is nil
2024-11-04 01:13:58.202501500 [CHG] Device FF:FF:DE:02:21:90 RSSI is nil
2024-11-04 01:13:58.202505500 [CHG] Device C8:69:CD:75:4E:60 TxPower is nil
2024-11-04 01:13:58.202508500 [CHG] Device C8:69:CD:75:4E:60 RSSI is nil
2024-11-04 01:13:58.202511500 [CHG] Device 5F:C7:23:CF:95:8B RSSI is nil
2024-11-04 01:13:58.202514500 [CHG] Device C9:F7:EE:2A:2B:77 RSSI is nil
2024-11-04 01:13:58.202588500 [CHG] Device FF:FF:30:06:66:A5 RSSI is nil
2024-11-04 01:13:58.202591500 [CHG] Device 66:10:1A:14:FC:91 TxPower is nil
2024-11-04 01:13:58.202595500 [CHG] Device 66:10:1A:14:FC:91 RSSI is nil
2024-11-04 01:13:58.202598500 [CHG] Device 35:94:FA:17:5A:11 RSSI is nil
2024-11-04 01:13:58.202601500 [CHG] Device C8:47:80:07:A6:7E RSSI is nil
2024-11-04 01:13:58.202604500 [CHG] Device 9C:20:7B:B1:B5:93 RSSI is nil
2024-11-04 01:13:58.202632500 [CHG] Device 7C:C1:1B:7B:22:B2 TxPower is nil
2024-11-04 01:13:58.202636500 [CHG] Device 7C:C1:1B:7B:22:B2 RSSI is nil
2024-11-04 01:13:58.202639500 [CHG] Device F6:EF:F9:33:1A:D7 RSSI is nil
2024-11-04 01:13:58.202642500 [CHG] Device 04:21:4C:00:7F:28 RSSI is nil
2024-11-04 01:13:58.202645500 [CHG] Controller D8:3A:DD:9F:B5:70 Discovering: no
2024-11-04 01:13:58.202766500 [CHG] Device C8:47:80:07:A6:7E ServicesResolved: no
2024-11-04 01:13:58.202770500 Successful disconnected
2024-11-04 01:13:58.202772500 [CHG] Device C8:47:80:07:A6:7E Connected: no
2024-11-04 01:14:03.349988500 Device C8:47:80:07:A6:7E (public)
2024-11-04 01:14:03.349993500   Alias: JK2801A16S15P-01
2024-11-04 01:14:03.349995500   Paired: no
2024-11-04 01:14:03.349998500   Trusted: no
2024-11-04 01:14:03.350000500   Blocked: no
2024-11-04 01:14:03.350002500   Connected: no
2024-11-04 01:14:03.350004500   LegacyPairing: no
2024-11-04 01:14:03.350007500   UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2024-11-04 01:14:03.350010500   RSSI: -55
2024-11-04 01:14:03.353654500
[image]
[image]