Louisvdw/dbus-serialbattery #729: JBD-BMS, SOC different in Xiaoxiang app and dbus-serialbattery

OutOfSync1, 2023-06-22:
### Describe the bug

The new SOC calculation leads to different SOC values shown in the Xiaoxiang app (via bluetooth on iOS) and in dbus-serialbattery. 
https://github.com/Louisvdw/dbus-serialbattery/blob/86a6b956e15bee21d5a298283c9d121f0eb82a02/etc/dbus-serialbattery/bms/lltjbd.py#L405
I think the issue here is that two values can be set for the capacity in the BMS: "total battery capacity" (in my case 280 Ah) and "total cycle capacity" (in my case 260 Ah). The JBD-BMS uses the cycle capacity as 100% SOC value, e.g. when 260 Ah is the remaining capacity and 260 Ah is set as "total cycle capacity" then the SOC is 100%. The calculation in serialbattery however uses the total capacity value as 100% SOC value, in this case 280 Ah. As a result the calculation in the example is 260 Ah/280 Ah which leads to 92.7% SOC. In this case 100% is never reached. This can lead to multiple problems, e.g. charge current is not reduced although battery is already full etc.

Unfortunately the serial data read via command_general only includes "total battery capacity" and not "total cycle capacity". A workaround may be to set both total and cycle capacity to the same value in the BMS.

I suggest to revert this change and use the soc reported by the BMS although it is an integer value. Otherwise provide a recommendation to set cycle and total capacity to the same value.

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

SOC reported via BMS and calculated in serial-battery should be the same

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

3.00

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting)
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.470 
; was 3.450
FLOAT_CELL_VOLTAGE = 3.45
; was 3.375

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
[... 259 lines cut ...]
@4000000064943c1239e1d154 WARNING:SerialBattery:Error reading settings from BMS, retry (10/10): Checksum error in rtu mode: '\x12a' instead of 'a\x91' . The response is: '\x01\x01\x03\x12a' (plain response: '\x01\x01\x03\x12a')
@4000000064943c1239edeee4 INFO:SerialBattery:Testing HLPdataBMS4S
@4000000064943c171d6d802c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064943c171d6da73c INFO:SerialBattery:Testing Jkbms
@4000000064943c172dcb2ec4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064943c172e011b74 INFO:SerialBattery:Testing Lifepower
@4000000064943c18027da06c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064943c1802b759c4 INFO:SerialBattery:Testing LltJbd
@4000000064943c182d496434 INFO:SerialBattery:Connection established to LltJbd
@4000000064943c182d5454fc INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
@4000000064943c182d5d6934 INFO:SerialBattery:========== Settings ==========
@4000000064943c182d67c974 INFO:SerialBattery:> Connection voltage: 55.67V | Current: 0.0A | SoC: 92.72%
@4000000064943c182d70b2b4 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
@4000000064943c182d79da74 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000064943c182d832944 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
@4000000064943c182d8e50bc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A (read from BMS)
@4000000064943c182d986e94 INFO:SerialBattery:> CVCM:     True
@4000000064943c182da226dc INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.47V
@4000000064943c182dab5284 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000064943c182db4477c INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000064943c182dbd1d34 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@4000000064943c1830f30f2c INFO:SerialBattery:DeviceInstance = 1
@4000000064943c183102de0c INFO:SerialBattery:com.victronenergy.battery.ttyACM0
@4000000064943c190e141e4c INFO:SerialBattery:publish config values = 1
```

mr-manuel, 2023-06-29:
@idstein maybe you can take a look at this?

idstein, 2023-06-29:
Currently, I’m very busy, but the cycle capacity can be read via the factory mode at the beginning as we read the max current.

mr-manuel, 2023-06-29:
There is no rush. If you have some spare time in the next weeks :-)

idstein, 2023-07-27:
This should be resolved by the open PR.

mr-manuel, 2023-07-27:
The PR was merged.

closed 2023-07-27 by mr-manuel