Louisvdw/dbus-serialbattery #730: JBD BMS - unstable communication

Project51At, 2023-06-23:
### Describe the bug

My Setup:
	- JBD-AP20S006
	- JBD-UART-TOOLS-RS485 (Rev. 1.7)
	
The setup works well withh Venus OS V3.0 and dbus-serialbattery v0.14.3.

After a dbus-serial battery update (or a fresh install) to v1.0.20230531 the communication is unstable or cannot be established.

### How to reproduce

Do a update to v1.0.20230531

### Expected behavior

well communication

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

V3.0

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
default config
```

### Relevant log output

```shell
$	ls -l /dev/ttyACM*
	crw-rw----    1 root     dialout   166,   0 Jun 19 18:00 /dev/ttyACM0

$	head /data/var/lib/serial-starter/*
	==> /data/var/lib/serial-starter/ttyACM0 <==
	sbattery

	==> /data/var/lib/serial-starter/ttyUSB0 <==
	cgwacs

	==> /data/var/lib/serial-starter/ttyUSB1 <==
	mkx

$	tail -F -n 100 /data/log/dbus-serialbattery.ttyACM0/current | tai64nlocal
2023-06-23 11:23:46.397021500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:23:47.397040500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:23:47.405336500
2023-06-23 11:23:47.405341500 #0
2023-06-23 11:23:47.405343500 #1
[... 397 lines cut ...]
2023-06-23 11:24:36.012616500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:36.018291500 INFO:SerialBattery:Connection established to LltJbd
2023-06-23 11:24:36.018767500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
2023-06-23 11:24:36.019141500 INFO:SerialBattery:========== Settings ==========
2023-06-23 11:24:36.019558500 INFO:SerialBattery:> Connection voltage: 53.1V | Current: 1.98A | SoC: 77.89%
2023-06-23 11:24:36.019936500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2023-06-23 11:24:36.020388500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-06-23 11:24:36.020776500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 50.0A | MAX BATTERY DISCHARGE CURRENT: 60.0A
2023-06-23 11:24:36.021148500 INFO:SerialBattery:> CVCM:     True
2023-06-23 11:24:36.021534500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-06-23 11:24:36.021922500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-06-23 11:24:36.022305500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-06-23 11:24:36.022678500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-06-23 11:24:36.054965500 INFO:SerialBattery:DeviceInstance = 1
2023-06-23 11:24:36.054971500 INFO:SerialBattery:com.victronenergy.battery.ttyACM0
2023-06-23 11:24:36.554328500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:36.824112500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:37.095356500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:37.114681500 INFO:SerialBattery:publish config values = 1
2023-06-23 11:24:38.399594500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:39.396989500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:40.630047500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:42.400253500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-23 11:24:43.400996500 ERROR:SerialBattery:>>> ERROR: No reply - returning
```

mr-manuel, 2023-06-23:
Could you try to install the nightly from the dev branch?

Project51At, 2023-06-24:
behaviour keeps the same
after 

2023-06-24 13:11:15.159839500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2023-06-24 13:11:15.429825500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:15.436864500 INFO:SerialBattery:Testing Seplos
2023-06-24 13:11:16.447544500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:16.948778500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-06-24 13:11:16.950076500 INFO:SerialBattery:Testing Daly at address "\x40"
2023-06-24 13:11:17.081893500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:17.082378500 INFO:SerialBattery:Testing Daly at address "\x80"
2023-06-24 13:11:17.210667500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:17.211121500 INFO:SerialBattery:Testing Ecs
2023-06-24 13:11:17.618432500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:17.618949500 INFO:SerialBattery:Testing HeltecModbus
2023-06-24 13:11:21.326398500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:21.326865500 INFO:SerialBattery:Testing HLPdataBMS4S
2023-06-24 13:11:22.140681500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:22.141124500 INFO:SerialBattery:Testing Jkbms
2023-06-24 13:11:22.411912500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:22.417614500 INFO:SerialBattery:Testing Lifepower
2023-06-24 13:11:22.695876500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:22.701256500 INFO:SerialBattery:Testing LltJbd
2023-06-24 13:11:22.966470500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:22.971996500 INFO:SerialBattery:Testing Renogy at address "\x30"
2023-06-24 13:11:23.240518500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:23.246210500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2023-06-24 13:11:23.512271500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:23.518817500 INFO:SerialBattery:Testing Seplos
2023-06-24 13:11:24.534828500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:25.036886500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-06-24 13:11:25.036891500 INFO:SerialBattery:Testing Daly at address "\x40"
2023-06-24 13:11:25.167332500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:25.167808500 INFO:SerialBattery:Testing Daly at address "\x80"
2023-06-24 13:11:25.295963500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:25.296417500 INFO:SerialBattery:Testing Ecs
2023-06-24 13:11:25.704062500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:25.704067500 INFO:SerialBattery:Testing HeltecModbus
2023-06-24 13:11:29.409067500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:29.409525500 INFO:SerialBattery:Testing HLPdataBMS4S
2023-06-24 13:11:30.222604500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:30.222609500 INFO:SerialBattery:Testing Jkbms
2023-06-24 13:11:30.498210500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:30.503729500 INFO:SerialBattery:Testing Lifepower
2023-06-24 13:11:30.779778500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:30.785165500 INFO:SerialBattery:Testing LltJbd
2023-06-24 13:11:31.052265500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:31.058006500 INFO:SerialBattery:Testing Renogy at address "\x30"
2023-06-24 13:11:31.334848500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:31.340467500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2023-06-24 13:11:31.616701500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:31.622231500 INFO:SerialBattery:Testing Seplos
2023-06-24 13:11:32.637825500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:33.142950500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM0
tail: /data/log/dbus-serialbattery.ttyACM0/current has been replaced; following end of new file
2023-06-24 13:11:47.746566500 INFO:SerialBattery:Testing HeltecModbus
2023-06-24 13:11:51.449152500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:51.449641500 INFO:SerialBattery:Testing HLPdataBMS4S
2023-06-24 13:11:52.262940500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:52.262945500 INFO:SerialBattery:Testing Jkbms
2023-06-24 13:11:52.529586500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:52.537635500 INFO:SerialBattery:Testing Lifepower
2023-06-24 13:11:52.809394500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:52.814708500 INFO:SerialBattery:Testing LltJbd
2023-06-24 13:11:53.081399500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:53.086782500 INFO:SerialBattery:Testing Renogy at address "\x30"
2023-06-24 13:11:53.352924500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:53.358520500 INFO:SerialBattery:Testing Renogy at address "\xf7"
2023-06-24 13:11:53.630270500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:11:53.635795500 INFO:SerialBattery:Testing Seplos
[... 137 lines cut ...]
2023-06-24 13:14:40.340906500 INFO:SerialBattery:Testing Jkbms
2023-06-24 13:14:40.610715500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:14:40.616152500 INFO:SerialBattery:Testing Lifepower
[... cut ...]
2023-06-24 13:14:41.625982500 INFO:SerialBattery:Serial Number/Unique Identifier: 40_280.0Ah
2023-06-24 13:14:41.665663500 INFO:SerialBattery:DeviceInstance = 1
2023-06-24 13:14:41.666022500 INFO:SerialBattery:com.victronenergy.battery.ttyACM0
2023-06-24 13:14:42.049969500 INFO:SerialBattery:publish config values = 1
2023-06-24 13:14:51.346850500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:14:52.345942500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-06-24 13:14:53.347333500 ERROR:SerialBattery:>>> ERROR: No reply - returning

mr-manuel, 2023-06-24:
Did you troubleshoot like described in the docs? Did you check the FAQ? Please describe your setup and post pictures of the cabling.

Project51At, 2023-06-25:
After a lot of investigations, I was able to identify the BMS as the issue itself. 
After the exchange, the latest DEV branch runs without any errors.

The problem already existed in v0.14.3., but I never noticed it.
The error 'ERROR: No reply - returning' was published each second after some minutes, but did not affect the GUI (the error was not visible).
The new version seems to be more sensitive to connection errors.

Thx for supporting

mr-manuel, 2023-06-25:
Hi, the new version has a better error log therefore you now probably see already existing errors that you did not saw before.

Taranis-Energy, 2023-08-01:
I realize this is close but. I am able to replicate this exactly using the Jiabaida SP16S020 bms. I used 3 different ones so its not  a defective BMS. 
Any idea why its ttyACM instead of ttyUSB?    Could this be the RS485 to USB converter?

Project51At, 2023-08-01:
> Could this be the RS485 to USB converter?

I tried different USB converters from different vendors, the issue still occurred.

Marvo2011, 2023-08-12:
I think I have the same trouble with JBD BMS over Bluetooth randomly it disconnects.

mr-manuel, 2023-09-04:
Probably connected issues:
https://github.com/Louisvdw/dbus-serialbattery/issues/769
https://github.com/Louisvdw/dbus-serialbattery/issues/777
https://github.com/Louisvdw/dbus-serialbattery/issues/778

Possible relevant info https://github.com/Louisvdw/dbus-serialbattery/issues/769#issuecomment-1657227203:
> I also have the same issue (identical logs) on all four of my LltJbd BMS with the latest in dev branch (https://github.com/Louisvdw/dbus-serialbattery/pull/763) from July 27. However, using the latest tagged release (https://github.com/Louisvdw/dbus-serialbattery/pull/757) from July 23 works fine.

mr-manuel, 2023-09-04:
@Taranis-Energy @Project51At @Marvo2011 could you please provide fresh logs? If you don't see the `Starting dbus-serialbattery` line in your logs, please reboot the Venus OS device or reinstall the driver once.

mr-manuel, 2023-09-04:
Please try to install the `nightly` from the `dev` branch. I just pushed some changes.

Project51At, 2023-09-09:
Today I installed the latest nightly from the dev branch, but no connection to the jbd bms could be established.

root@raspberrypi4:~# tail -F -n 100 /data/log/dbus-serialbattery.ttyACM0/current | tai64nlocal
2023-09-09 12:56:10.521740500 #6
2023-09-09 12:56:10.521742500 #7
2023-09-09 12:56:10.521744500 #8
2023-09-09 12:56:10.521746500 #9
2023-09-09 12:56:10.521748500 #10
2023-09-09 12:56:10.521750500 #11
2023-09-09 12:56:10.521752500 #12
2023-09-09 12:56:10.521754500 #13
2023-09-09 12:56:10.521755500 #14
2023-09-09 12:56:10.521757500 #15
2023-09-09 12:56:10.521759500 #0
2023-09-09 12:56:10.521761500 #1
2023-09-09 12:56:10.521788500 #2
2023-09-09 12:56:10.521790500 #3
2023-09-09 12:56:10.521792500 #4
2023-09-09 12:56:10.521794500 #5
2023-09-09 12:56:10.521796500 #6
2023-09-09 12:56:10.521798500 #7
2023-09-09 12:56:10.521800500 #8
2023-09-09 12:56:10.521801500 #9
2023-09-09 12:56:10.521803500 #10
2023-09-09 12:56:10.521805500 #11
2023-09-09 12:56:10.521807500 #12
2023-09-09 12:56:10.521809500 #13
2023-09-09 12:56:10.521811500 #14
2023-09-09 12:56:10.521813500 #15
2023-09-09 12:56:10.521814500 #0
2023-09-09 12:56:10.521816500 #1
2023-09-09 12:56:10.521818500 #2
2023-09-09 12:56:10.521843500 #3
2023-09-09 12:56:10.521846500 #4
2023-09-09 12:56:10.521847500 #5
2023-09-09 12:56:10.521849500 #6
2023-09-09 12:56:10.521851500 #7
2023-09-09 12:56:10.521853500 #8
2023-09-09 12:56:10.521855500 #9
2023-09-09 12:56:10.521857500 #10
2023-09-09 12:56:10.521858500 #11
2023-09-09 12:56:10.521860500 #12
2023-09-09 12:56:10.521862500 #13
2023-09-09 12:56:10.521864500 #14
2023-09-09 12:56:10.521866500 #15
2023-09-09 12:56:25.818725500 INFO:SerialBattery:
2023-09-09 12:56:25.819241500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-09 12:56:25.825587500 INFO:SerialBattery:dbus-serialbattery v1.0.20230905dev
2023-09-09 12:56:25.825592500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-09-09 12:56:25.825595500 INFO:SerialBattery:Testing LltJbd
2023-09-09 12:56:27.209444500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-09-09 12:56:27.209881500 INFO:SerialBattery:Testing LltJbd
2023-09-09 12:56:28.596300500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-09-09 12:56:28.596843500 INFO:SerialBattery:Testing LltJbd
2023-09-09 12:56:29.995606500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyACM0
2023-09-09 12:56:29.995711500 #0
2023-09-09 12:56:29.995714500 #1
2023-09-09 12:56:29.995716500 #2
2023-09-09 12:56:29.995719500 #3
2023-09-09 12:56:29.995721500 #4
2023-09-09 12:56:29.995722500 #5
2023-09-09 12:56:29.995724500 #6
2023-09-09 12:56:29.995726500 #7
2023-09-09 12:56:29.995728500 #8
2023-09-09 12:56:29.995730500 #9
2023-09-09 12:56:29.995731500 #10
2023-09-09 12:56:29.995733500 #11
2023-09-09 12:56:29.995735500 #12
2023-09-09 12:56:29.995737500 #13
2023-09-09 12:56:29.995739500 #14
2023-09-09 12:56:29.995741500 #15
2023-09-09 12:56:29.995743500 #0
2023-09-09 12:56:29.995745500 #1
2023-09-09 12:56:29.995772500 #2
2023-09-09 12:56:29.995774500 #3
2023-09-09 12:56:29.995776500 #4
2023-09-09 12:56:29.995778500 #5
2023-09-09 12:56:29.995780500 #6
2023-09-09 12:56:29.995781500 #7
2023-09-09 12:56:29.995783500 #8
2023-09-09 12:56:29.995785500 #9
2023-09-09 12:56:29.995787500 #10
2023-09-09 12:56:29.995789500 #11
2023-09-09 12:56:29.995791500 #12
2023-09-09 12:56:29.995792500 #13
2023-09-09 12:56:29.995794500 #14
2023-09-09 12:56:29.995796500 #15
2023-09-09 12:56:29.995798500 #0
2023-09-09 12:56:29.995800500 #1
2023-09-09 12:56:29.995802500 #2
2023-09-09 12:56:29.995827500 #3
2023-09-09 12:56:29.995830500 #4
2023-09-09 12:56:29.995831500 #5
2023-09-09 12:56:29.995833500 #6
2023-09-09 12:56:29.995835500 #7
2023-09-09 12:56:29.995837500 #8
2023-09-09 12:56:29.995839500 #9
2023-09-09 12:56:29.995840500 #10
2023-09-09 12:56:29.995842500 #11
2023-09-09 12:56:29.995844500 #12
2023-09-09 12:56:29.995846500 #13
2023-09-09 12:56:29.995848500 #14
2023-09-09 12:56:29.995850500 #15

mr-manuel, 2023-09-09:
@idstein could you open a PR to troubleshoot this better?

mr-manuel, 2023-09-25:
@Project51At could you please try with the latest `nightly` of the `dev` branch? I just merged another fix for LLT/JBD BMS.

Project51At, 2023-09-27:
@mr-manuel 
I updated to the recommended version and it is working. I will further monitor the behaviour

mr-manuel, 2023-09-29:
Have you an update?

Project51At, 2023-09-29:
I did not notice any issues. So far, so good

closed 2023-09-29 by mr-manuel