Louisvdw/dbus-serialbattery #733: permanently in absorption mode

Micha854, 2023-06-24:
### Describe the problem

I have two different installations.
in both I've recently noticed that it no longer comes into float mode. the voltage of the battery is at max_cell_voltage * cells and the maximum cell drift is 0.002V - 0.007V

what can I do?

I first switched to step mode but would like to continue linearly

thanks for the support

### Driver version

newest on dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.00

### BMS type

Daly Smart BMS

### Cell count

15

### Connection type

Serial USB adapter to RS485

### Config file

```ini
default config, no changes
```

### Relevant log output

```shell
im not have input
```

thomasinaz41213, 2023-06-24:
You say default config? No changes? You need to make settings in the config according to your battery bank. One of the settings should be cell deviation. Possibly you are not going below the cell deviation requirement for float voltage. Setup instructions will tell you how.

Micha854, 2023-06-24:
yes default config. my English is not so good. sorry, which setting is that exactly?

thomasinaz41213, 2023-06-24:
At the least you should configure absorption and float voltages, and cell deviation. If cell deviation setting is below what the bms is set for then you will never enter float when in linear mode. I can't remember exactly how it acts in step mode. When you open the config file every setting has explanation for how to set it and how it behaves.

mr-manuel, 2023-06-24:
@ogurevich could you please check? I'm seeing the same since your last PR.

ogurevich, 2023-06-24:
The mode change worked perfectly for me and the cause of the problem described could be different settings.

I compare the settings in my setup with default and I suspect that the value VOLTAGE_DROP 0 - in default and 0.025 in my setup can lead to this effect 

the related code is in lines 269 -283
```
if self.max_voltage_start_time is None:
                # start timer, if max voltage is reached and cells are balanced
                if (
                    self.max_battery_voltage - utils.VOLTAGE_DROP <= voltageSum
                    and voltageDiff <= utils.CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL
                    and self.allow_max_voltage
                ):
                    self.max_voltage_start_time = int(time())

                # allow max voltage again, if cells are unbalanced or SoC threshold is reached
                elif (
                    utils.SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT > self.soc
                    or voltageDiff >= utils.CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT
                ) and not self.allow_max_voltage:
                    self.allow_max_voltage = True
```
Would you please try and also set this value to 0.03 or 0.025 volts. This has the following meaning (according to code logic). If the battery reports a small but small penalty, the driver reduces CVL. This in turn leads to a reduction in Voltage, e.g. no longer 55.2, but to 55.18, for example. This causes us to switch back to Builk from absorption mode. VOLTAGE_DROP has a slightly different meaning here. The value indicates the tolerance limit at which we remain in absorption mode. To see the life you can uncomment 313-327 and see it in dbus-spy.

[image]

for 3 days:
[image]

my current config.ini 
```
[DEFAULT]

; If you want to add custom settings, then check the settings you want to change in "config.default.ini"
; and add them below to persist future driver updates.

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
;MAX_CELL_VOLTAGE   = 3.45625
MAX_CELL_VOLTAGE   = 3.45
FLOAT_CELL_VOLTAGE = 3.425
;FLOAT_CELL_VOLTAGE = 3.382
TIME_TO_GO_ENABLE = False

; --------- Charge Voltage limitation (affecting CVL) ---------
; Description: Limit max charging voltage (MAX_CELL_VOLTAGE * cell count), switch from max voltage to float
;              voltage (FLOAT_CELL_VOLTAGE * cell count) and back
;     False: Max charging voltage is always kept
;     True: Max charging voltage is reduced based on charge mode
;         Step mode: After max voltage is reached for MAX_VOLTAGE_TIME_SEC it switches to float voltage. After
;                    SoC is below SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT it switches back to max voltage.
;         Linear mode: After max voltage is reachend and cell voltage difference is smaller or equal to
;                      CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL it switches to float voltage after 300 (fixed)
;                      additional seconds. After cell voltage difference is greater or equal to
;                      CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT it switches back to max voltage.
; Example: The battery reached max voltage of 55.2V and hold it for 900 seconds, the the CVL is switched to
;          float voltage of 53.6V to don't stress the batteries. Allow max voltage of 55.2V again, if SoC is
;          once below 90%
;          OR
;          The battery reached max voltage of 55.2V and the max cell difference is 0.010V, then switch to float
;          voltage of 53.6V after 300 additional seconds to don't stress the batteries. Allow max voltage of
;          55.2V again if max cell difference is above 0.050V
; Charge voltage control management enable (True/False).
[... 29 lines cut ...]
;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
;     Set VOLTAGE_DROP to 0.15
VOLTAGE_DROP = 0.025

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; Example with 1 BMS: Jkbms_Ble C8:47:8C:00:00:00
; Example with 3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
;
;C8:47:8C:E1:FB:B9 JK-gurgur2
;C8:47:8C:E5:50:4C JK-gurgur1

;BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E5:50:4C, Jkbms_Ble C8:47:8C:E1:FB:B9

; --------- Additional settings ---------
; Specify only one BMS type to load else leave empty to try to load all availabe
; LltJbd, Ant, Daly, Daly, Jkbms, Lifepower, Renogy, Renogy, Ecs
BMS_TYPE = Jkbms

; Specify in seconds how often the linear values should be recalculated
LINEAR_RECALCULATION_EVERY = 10

CUSTOM_BATTERY_NAMES = /dev/ttyUSB2:SB1(Jkbms)

```

ogurevich, 2023-06-24:
Logs in details:
[image]

I'm still trying to figure out what else could be the cause.

ogurevich, 2023-06-24:
> @ogurevich could you please check? I'm seeing the same since your last PR.
> 
> [image]

i see in the graph (max charge voltage) it goes into float as well (last graph) the question for me why the driver left float mode and goes into bulk again. .... strange ....

ogurevich, 2023-06-24:
> [Micha854](/Micha854)
could you send please from VRM portal these three graphs of your battery:
[image]

mr-manuel, 2023-06-25:
> > @ogurevich could you please check? I'm seeing the same since your last PR.
> > [image]
> 
> i see in the graph (max charge voltage) it goes into float as well (last graph) the question for me why the driver left float mode and goes into bulk again. .... strange ....

The first 3 days are before applying the PR. Float is left because heavy loads are turned on. I will try with the voltage diff for testing

ogurevich, 2023-06-25:
> The first 3 days are before applying the PR. Float is left because heavy loads are turned on. I will try with the voltage diff for testing

Ok, thanks. In your constellation, the graphic below shows that even after using the pull request, the algorithm behaves as intended.

Micha854, 2023-06-25:
Hello guys,

here my screens of:

Limits:
[image]

Min / Max Cell Voltage
[image]

Voltage and Current
[image]

i hope this help

mr-manuel, 2023-06-25:
Found the error in my case. I set the reset SOC to 90%. Since the JKBMS has a great SOC drift it was reset immediately to absorption again. Maybe it would be better to use the voltage instead?

@Micha854 what is your SoC, when the battery reached the max voltage?

Micha854, 2023-06-25:
soc of daly bms is arround 99%

mr-manuel, 2023-06-25:
Can you write me on Discord?

ogurevich, 2023-06-25:
@Micha854 
> Hello guys,
> 
> here my screens of:
> 
.....
> 
> i hope this help
could you set please in config.ini

VOLTAGE_DROP = 0.125

and run.  ./reinstall-local.sh

-- it will let drive to ignore these voltage deviation and keeps Absorption mode at least 300 sec.
[image]

ogurevich, 2023-06-25:
> Found the error in my case. I set the reset SOC to 90%. Since the JKBMS has a great SOC drift it was reset immediately to absorption again. Maybe it would be better to use the voltage instead?

a little all-clear is already very good :)
with heavy loads the voltage goes down even though the battery is almost full. 

Since I cannot rely on the SOC values of my JK-BMSs, I seriously thought whether the SOC value of SmartShunt (if available) 
for example an VeDbusItemImport in dbushelper.py could provide this value to battery. And "Battery"  #727 cann better decide when to leave float mode. 

I'm not sure yet if that's actually the driver's job to rely on values other than its own battery.
👍

ogurevich, 2023-06-25:
sorry for off topic:

it is interesting to see how the system works in combination with multiple batteries. Depending on the "health" of the battery, the right settings can enable the desired behavior. In this example, reducing VOLTAGE_DROP to 0.02 will "prevent" the first battery from jumping into float mode so early.
[image]

ogurevich, 2023-06-25:
> Found the error in my case. I set the reset SOC to 90%. Since the JKBMS has a great SOC drift it was reset immediately to absorption again. Maybe it would be better to use the voltage instead?

Another possibility of what to do at this point. From the moment we switch from absorption to float mode, we can ourselves calculate the amount of energy the battery is giving off. When this amount exceeds a threshold, that is the moment for MAX_VOLTAGE to be allowed again. We can rely on voltage and current values. What do you make of it ?

mr-manuel, 2023-06-25:
This would cause, that if you have a spike in drawing the power from the battery, that the max voltage is allowed again, even if it was only for some seconds. I would keep that in mind and try to fix the SoC problem first. If that is solved, than we can think about it again.

ogurevich, 2023-06-25:
We would count the amount of energy per time of course. Like a shunt it does. The spikes don‘t affect the calculations. EnergyAmount += v * i * secsSincePreviousStep

towatai, 2023-06-26:
I have the same problem here. The float voltage set in the driver is used, but the display does not change and remains on Absorbtion. It would be great if you could select in the config whether the SOC should be set to 100% when switching to float mode or alternatively when a certain voltage is reached. Since we LiFePo4-users have no real benefit from the float mode, you then have the choice (depending on the battery type used)

ogurevich, 2023-06-26:
Does it make sense to adjust the default value of VOLTAGE_DROP to 0.3 or 0.5 volts?
PS VOLTAGE_DROP in terms of (VOLTAGE_MARGIN_BOTTOM_TO_STAY_IN_ABSORPTION_MODE)

mr-manuel, 2023-06-26:
@towatai the display will never change, since it's controlled by the MutliPlus itself. In this case the display is useless and should show external control. Since this is very complex and takes a lot of time, effort and money it will take a while. I'm also in contact with the JK developers to find a solution. I would appreciate it, if you donate something for all the effort we are making.

> Does it make sense to adjust the default value of VOLTAGE_DROP to 0.3 or 0.5 volts?
> PS VOLTAGE_DROP in terms of (VOLTAGE_MARGIN_BOTTOM_TO_STAY_IN_ABSORPTION_MODE)

@ogurevich I renamed your variable to avoid the double usage: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/1fffdafcda0026c07b39bdb484ec8c1cff153668/etc/dbus-serialbattery/battery.py#L340-L366

I do not fully understand where you want to decrease the voltage.

towatai, 2023-06-26:
> @towatai the display will never change, since it's controlled by the MutliPlus itself. In this case the display is useless and should show external control. Since this is very complex and takes a [...]

Thank you for your insightful reply. I have already donated something to you via paypal on May 5 ;-)

ogurevich, 2023-06-26:
> @ogurevich I renamed your variable to avoid the double usage: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/1fffdafcda0026c07b39bdb484ec8c1cff153668/etc/dbus-serialbattery/battery.py#L340-L366
> 
> I do not fully understand where you want to decrease the voltage.

sorry, ... had unhappily printed my thoughts
i don't mean to decrease the voltage
https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/1fffdafcda0026c07b39bdb484ec8c1cff153668/etc/dbus-serialbattery/battery.py#L306-L307

These conditions resets absorption time if voltage slightly under max_battery_voltage 
But this condition is needed and is on the right place. Here we use as margin some delta. probably on this place is an other constant value.

mr-manuel, 2023-06-26:
This `VOLTAGE_DROP` is only needed, if you have a voltage drop on the cable. E.g. if the BMS measures 55.10 V, the max voltage is 55.20V and the charger is providing already the max 55.20V. So you have a voltage drop on the cable of 0.10V that you need to configure. Else it will never switch to float.

Default is 0 so there is no delta. In 95% of the cases this is not needed to configure, only if you have a very bad and/or long cable to the charger.

ogurevich, 2023-06-26:
This purpose is undisputed.

Algorithm sees 2 conditions for absorption mode which are:

a - Cells Voltage is close to each other (voltageDiff <= utils.CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL)
and
b - Battery voltage reached max_battery_voltage
self.max_battery_voltage - utils.VOLTAGE_DROP <= voltageSum
                    
Let's look at Voltage Graph of @Micha854
in his case max_battery_voltage = 51,75 but under real conditions the voltage is not constant it goes up and down.
[image]

Each time the voltage slightly goes down condition not met, and we are again in bulk mode. We need this voltage level within 300 seconds. That is what i mean. If voltage get to 51,73 volts we should stay in the absoption. It works really fine if
Condition self.max_battery_voltage - 0.03 <= voltageSum

That is the cause for this topic

mr-manuel, 2023-06-29:
@Micha854 could you try the lastes nightly from the dev branch? This should be solved now.

Micha854, 2023-07-01:
> @Micha854 could you try the lastes nightly from the dev branch? This should be solved now.

yes it has been working great for 1 week! Thanks very much

PR #735 (2023-06-27): fix proposal for #733
allow "Measurement" und "Variation Tolerance" to keep in absorption mode within MAX_VOLTAGE_TIME_SEC. Fixes #733
closed 2023-07-01 by Micha854