Louisvdw/dbus-serialbattery #738: VRM instance changes sporadically

Micha854, 2023-07-02:
### Describe the bug

Hello,

I have two installations with the Serial Driver and with both I observe the following:

If you start the Raspberry again, the VRM Instance sometimes changes.

This is bad because my graphs then remain empty in the expanded view...

Would it be possible to tell the driver (config.ini) VRM Instance = 1? That would solve the problem

Even if the PI is freshly installed, this problem occurs...

in both installations the battery is connected to an RS485 adapter

### How to reproduce

when restarting the Pi, sporadically

### Expected behavior

..

### Driver version

newest dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.00

### BMS type

Daly Smart BMS

### Cell count

15

### Connection type

Serial USB adapter to RS485

### Config file

```ini
..
```

### Relevant log output

```shell
..
```

transistorgit, 2023-07-02:
Probably duplicate of #718

mr-manuel, 2023-07-02:
@Micha854 if you use the search you will find a lot of issues and discussions about that.

closed 2023-07-02 by mr-manuel