Louisvdw/dbus-serialbattery #743: RS485 Inccorect reply, unpack_from buffer error

Michal22569, 2023-07-09:
### Describe the problem

The rs485 communication doesnt want to work with raspberry or venus gx tried both. The bluetooth communication works but is unstable it drops out twice a day and VRM portal often goes dead with warning "Instalation busy realtime link disabled". So this is the reason i want to switch on RS485 it may be less stressful for venus than BLE. Any idea whats wrong?

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.00

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0
BMS_TYPE = Jkbms
```

### Relevant log output

```shell
tail -F -n 100 /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal
reports : 
2023-07-09 13:04:25.854286500 INFO:SerialBattery:Starting dbus-serialbattery
2023-07-09 13:04:25.855026500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-07-09 13:04:25.855423500 INFO:SerialBattery:Testing Jkbms
2023-07-09 13:04:26.071627500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 13:04:26.572796500 INFO:SerialBattery:Testing Jkbms
2023-07-09 13:04:26.592485500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 13:04:27.093577500 INFO:SerialBattery:Testing Jkbms
2023-07-09 13:04:27.112357500 ERROR:SerialBattery:Unexpected err=error('unpack_from requires a buffer of at least 5 bytes for unpacking 5 bytes at offset 0 (actual buffer size is 4)'), type(err)=<class 'struct.error'>
2023-07-09 13:04:27.613526500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2023-07-09 13:04:27.613775500
2023-07-09 13:04:48.999887500 INFO:SerialBattery:Starting dbus-serialbattery
2023-07-09 13:04:49.000626500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2023-07-09 13:04:49.001023500 INFO:SerialBattery:Testing Jkbms
2023-07-09 13:04:49.024131500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 13:04:49.525895500 INFO:SerialBattery:Testing Jkbms
2023-07-09 13:04:49.544066500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 13:04:50.045028500 INFO:SerialBattery:Testing Jkbms
2023-07-09 13:04:50.063848500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 13:04:50.564963500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
```

### Any other information that may be helpful

BMS model JKBMS B5A25S60P this one has intergrated chip for CAN/RS485. I also tried on Venus GX but i get the same error.
i tried these RS485 to USB converters
https://www.aliexpress.com/item/33017179197.html?spm=a2g0o.order_list.order_list_main.11.22561802B6ZogF
https://www.aliexpress.com/item/1005001621816794.html?spm=a2g0o.detail.0.0.302dnLbInLbIeG&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40050.281175.0&scm_id=1007.40050.281175.0&scm-url=1007.40050.281175.0&pvid=9d80c9b6-5e0f-47b7-b352-3b909c66a44b&_t=gps-id:pcDetailTopMoreOtherSeller,scm-url:1007.40050.281175.0,pvid:9d80c9b6-5e0f-47b7-b352-3b909c66a44b,tpp_buckets:668%232846%238107%231934&pdp_npi=3%40dis%21EUR%210.92%210.73%21%21%21%21%21%402103253416889080386512436ebaf5%2112000016846594795%21rec%21SK%213712598953

mr-manuel, 2023-07-09:
Please install the latest nighly from the dev branch.

Michal22569, 2023-07-09:
installed nightly version dev branch still the same response
2023-07-09 15:04:30.324041500 INFO:SerialBattery:
2023-07-09 15:04:30.324496500 INFO:SerialBattery:Starting dbus-serialbattery
2023-07-09 15:04:30.325163500 INFO:SerialBattery:dbus-serialbattery v1.0.20230625dev
2023-07-09 15:04:30.325564500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-07-09 15:04:30.325928500 INFO:SerialBattery:Testing Jkbms
2023-07-09 15:04:30.352855500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 15:04:30.854031500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-07-09 15:04:30.854448500 INFO:SerialBattery:Testing Jkbms
2023-07-09 15:04:30.873470500 ERROR:SerialBattery:Unexpected err=error('unpack_from requires a buffer of at least 5 bytes for unpacking 5 bytes at offset 0 (actual buffer size is 4)'), type(err)=<class 'struct.error'>
2023-07-09 15:04:31.374629500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-07-09 15:04:31.375055500 INFO:SerialBattery:Testing Jkbms
2023-07-09 15:04:31.395350500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 15:04:31.896496500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
2023-07-09 15:04:53.396409500 INFO:SerialBattery:
2023-07-09 15:04:53.396691500 INFO:SerialBattery:Starting dbus-serialbattery
2023-07-09 15:04:53.397310500 INFO:SerialBattery:dbus-serialbattery v1.0.20230625dev
2023-07-09 15:04:53.397605500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-07-09 15:04:53.397873500 INFO:SerialBattery:Testing Jkbms
2023-07-09 15:04:53.434658500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 15:04:53.935684500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-07-09 15:04:53.935955500 INFO:SerialBattery:Testing Jkbms
2023-07-09 15:04:54.099407500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 15:04:54.600464500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-07-09 15:04:54.600741500 INFO:SerialBattery:Testing Jkbms
2023-07-09 15:04:54.630966500 ERROR:SerialBattery:>>> ERROR: Incorrect Reply
2023-07-09 15:04:55.131973500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0

mr-manuel, 2023-07-09:
The nightly from the dev branch is only for better BLE connection. Probably you have connected the BMS and adapter wrong. Have you checked the FAQ?

Check also https://github.com/Louisvdw/dbus-serialbattery/discussions/250

post also detailed pictures where we can see how every PIN is connected.

Michal22569, 2023-07-10:
[image]
[image]
[image]
[image]
[image]
Also the nightly dev build didnt improve the BLE connection it seems worse now, the installation busy error in vrm seems to appear more often.

mr-manuel, 2023-07-10:
Did you read https://github.com/Louisvdw/dbus-serialbattery/discussions/250? Normally the labeling of the JKBMS is wrong. Try to connect a TTL adapter to the GPS ports like on the other BMS. Messure the voltage on the PINs first. 99% the cabling is wrong.

Michal22569, 2023-07-19:
ok I finally got the conector for GPS port connected the middle pins RX TX to A B on TTL adapter, i tried both ways RX to A and TX to B and vica versa.
both tries result in this. the deamon is created and started against USBtty0
2023-07-19 17:46:49.178618500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-07-19 17:46:49.178883500 INFO:SerialBattery:Testing Jkbms
2023-07-19 17:46:49.449713500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                     g
2023-07-19 17:46:49.951808500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-07-19 17:46:49.952083500 INFO:SerialBattery:Testing Jkbms
2023-07-19 17:46:50.221765500 ERROR:SerialBattery:>>> ERROR: No reply - returnin                                     g
2023-07-19 17:46:50.723782500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-07-19 17:46:50.724052500 INFO:SerialBattery:Testing Jkbms
2023-07-19 17:46:50.993545500 ERROR:SerialBattery:>>> ERROR: No reply - returnin

mr-manuel, 2023-07-31:
> Did you read #250? Normally the labeling of the JKBMS is wrong. Try to connect a TTL adapter to the GPS ports like on the other BMS. Messure the voltage on the PINs first. 99% the cabling is wrong.

If there is.no voltage on the middle pins probably the BMS is damaged. Normally there should be 3.3 volts on one pin and a little more than 0 on the other. If you read #250 you will find a lot of informations.

Michal22569, 2023-07-31:
Well thanks for help seems like I`ll be stuck on BT connection. I read that article and it was helpfull a little bit, I was confused by different model of my bms which stated that has integrated 485chip. 
Few questions off topic:
Is there a way how to slow down refresh rate of BT comm? 
Is there a way how to start comm against the CAN/RS485 port since the default GPS seems broken.? my BMS doesnt offer the RS485 dongle.

mr-manuel, 2023-09-04:
What is your system load? You can check it by executing the `uptime` or `top` (exit again by pressing `CTRL + C`) command over SSH.

> Is there a way how to slow down refresh rate of BT comm?

You can try to manually change the code. Change

https://github.com/Louisvdw/dbus-serialbattery/blob/f07d9533b8cc759a135017bf39bd715d298521f7/etc/dbus-serialbattery/dbus-serialbattery.py#L195-L198

to 

```python
    # try using active callback on this battery
    # if not battery.use_callback(lambda: poll_battery(mainloop)):
    # if not possible, poll the battery every poll_interval milliseconds
    gobject.timeout_add(battery.poll_interval, lambda: poll_battery(mainloop))
```

The default value for `poll_interval` is `1000` (miliseconds).

https://github.com/Louisvdw/dbus-serialbattery/blob/f07d9533b8cc759a135017bf39bd715d298521f7/etc/dbus-serialbattery/battery.py#L65

> Is there a way how to start comm against the CAN/RS485 port since the default GPS seems broken.? my BMS doesnt offer the RS485 dongle.

Reviewing your logs it seems that the driver is getting some data from the RS485 port. Contact me on Discord so I can take directly a look.

Michal22569, 2023-09-04:
top command resulted in : total 60-62%cpu usage

 2435  2413 root     S    46232   9%  13% python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:47:8C:F2:86:56
 1927  1917 root     S    39152   8%   9% {dbus_mqtt.py} /usr/bin/python3 -u /opt/victronenergy/dbus-mqtt/dbus_mqtt.py --init-broker
  880   851 root     S    21340   4%   6% {dbus_systemcalc} /usr/bin/python3 -u /opt/victronenergy/dbus-systemcalc-py/dbus_systemcalc.py
  805   792 root     S    37448   7%   4% {vrmlogger.py} /usr/bin/python3 -u /opt/victronenergy/vrmlogger/vrmlogger.py
 1318   845 root     S<   77380  15%   3% /opt/victronenergy/gui/gui -nomouse -display VNC:size=480x272:depth=32:passwordFile=/data/conf/vncpas
  543   541 messageb S     3664   1%   3% dbus-daemon --system --nofork
  892   861 root     S    21060   4%   2% {dbus_generator.} /usr/bin/python3 -u /opt/victronenergy/dbus-generator-starter/dbus_generator.py
 1935  1915 mosquitt S    10032   2%   2% /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf
 1708  1652 root     S     3620   1%   2% /opt/victronenergy/mk2-dbus/mk2-dbus --log-before 25 --log-after 25 --banner -w -s /dev/ttyO5 -i -t m
  874   843 root     S     8628   2%   1% /opt/victronenergy/hub4control/hub4control
  894   863 root     S    11184   2%   1% /opt/victronenergy/dbus-fronius/dbus-fronius
  878   847 root     S    19532   4%   1% {dbus_vebus_to_p} /usr/bin/python3 -u /opt/victronenergy/dbus-vebus-to-pvinverter/dbus_vebus_to_pvinv
 1931  1919 root     S     9568   2%   1% /opt/victronenergy/dbus-modbustcp/dbus-modbustcp
  625   623 root     S     5080   1%   1% connmand --nodnsproxy --nodaemon
 2415  2413 root     S     4556   1%   1% bluetoothctl scan on
 1638  1630 root     S     3360   1%   1% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -
 3429  3427 root     S     3360   1%   1% /opt/victronenergy/vedirect-interface/vedirect-dbus -v --log-before 25 --log-after 25 -t 0 --banner -
 5300  5289 root     R     2800   1%   1% top

- the mqtt broker seems to chug from 9-23% of cpu power even though the mqtt is disabled in settings.
-  I managed to get cpu load under 40% by increasing the poll_interval to 15000. At 5000 i still had 50-60%cpu usage and often Installation busy in VRM portal. At 15000 it seems fine. No matter the value of poll_interval, the dbus-serialbattery.py process was chugging constant 12-13% of cpu. But it still affected CPU load. 
- after resetting the whole BMS and letting it sit without power overnight I measured on middle pins of gps port 47V and 50V. But I didnt even try to connect the rs485 usb since this voltage would prob fry that.
I will contact u on discord in near future for additional testing

mr-manuel, 2023-09-22:
You could check the CPU time. See https://github.com/Louisvdw/dbus-serialbattery/issues/808

mr-manuel, 2023-09-26:
> Is there a way how to start comm against the CAN/RS485 port since the default GPS seems broken.? my BMS doesnt offer the RS485 dongle.

The latest nightly supports CAN for JKBMS and others.

mr-manuel, 2023-09-27:
Directly for CAN port

closed 2023-09-29 by mr-manuel

gyrex, 2024-07-26:
Did you ever get this BMS to work? I'm thinking of buying the JK-B2A25S-RP - similar BMS but with a 2A balancer.

Michal22569, 2024-07-27:
> Did you ever get this BMS to work? I'm thinking of buying the JK-B2A25S-RP - similar BMS but with a 2A balancer.
No i ended up with rewiring it for 2 wire BMS the bms had 3 output relays for "allow disharge" "allow charge" "allow precharge" and configured it in victron. But i would not recommend trying to implement this communication if u have **venus gx**. its pain.... it often froze up, or was incredibly delayed... since then i already switched to new Cerbo but i have no energy to tinker with this again.