Louisvdw/dbus-serialbattery #752: Incorrect display of Min/Max Cell Voltages

centro-max, 2023-07-21:
### Describe the bug

For me, only the 1st cell is shown in color, not the critical cell. I've noticed this for many driver versions.
[image]

### How to reproduce

Steps to reproduce the behavior:
1. Go to "Cell-Voltages"
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

It would be nice if the minimum and maximum cell voltages were displayed

### Driver version

1.0.20230711beta

### Venus OS device type

Raspberry Pi

### Venus OS version

3.01

### BMS type

Daly Smart BMS

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 175.0
MAX_BATTERY_DISCHARGE_CURRENT = 350.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage (can seen as absorption voltage)
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.375

; Bulk voltage (may be needed to reset the SoC to 100% once in a while for some BMS)
; Has to be higher as the MAX_CELL_VOLTAGE
BULK_CELL_VOLTAGE  = 3.650
; Specify after how many days the bulk voltage should be reached again
; The timer is reset when the bulk voltage is reached
; Leave empty if you don't want to use this
; Example: Value is set to 15
; day 1: bulk reached once
; day 16: bulk reached twice
; day 31: bulk not reached since it's very cloudy
; day 34: bulk reached since the sun came out
[... 530 lines cut ...]
@4000000064b8fbec1253fe64 INFO: Start service dbus-cgwacs.ttyUSB4 once
@4000000064b8fbf12800fd5c INFO: Start service gps-dbus.ttyUSB4 once
@4000000064b8fbfb0ba2e79c INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064b8fbfe1d229094 INFO: Start service dbus-serialbattery.ttyUSB4 once
@4000000064b8fc0332b24b5c INFO: Start service dbus-cgwacs.ttyUSB4 once
@4000000064b8fc090d9bd16c INFO: Start service gps-dbus.ttyUSB4 once
@4000000064b8fc122bb191dc INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064b8fc1602015fd4 INFO: Start service dbus-serialbattery.ttyUSB4 once
@4000000064b8fc1b1823022c INFO: Start service dbus-cgwacs.ttyUSB4 once
@4000000064b8fc202e56f01c INFO: Start service gps-dbus.ttyUSB4 once
@4000000064b8fc2a114ea424 INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064b8fc2d237d1f04 INFO: Start service dbus-serialbattery.ttyUSB4 once
@4000000064b8fc340e2738ec INFO: Start service dbus-cgwacs.ttyUSB4 once
@4000000064b8fc392b3d540c INFO: Start service gps-dbus.ttyUSB4 once
@4000000064b8fc430eb59e0c INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064b8fc462001988c INFO: Start service dbus-serialbattery.ttyUSB4 once
@4000000064b8fc4b377ab87c INFO: Start service dbus-cgwacs.ttyUSB4 once
@4000000064b8fc511281925c INFO: Start service gps-dbus.ttyUSB4 once
@4000000064b8fc5a31b8e79c INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064b8fc5e08a91ffc INFO: Start service dbus-serialbattery.ttyUSB4 once
@4000000064b8fc631df14e5c INFO: Start service dbus-cgwacs.ttyUSB4 once
@4000000064b8fc68345bc154 INFO: Start service gps-dbus.ttyUSB4 once
@4000000064b8fc7217f579ec INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064b8fc752aac9d2c INFO: Start service dbus-serialbattery.ttyUSB4 once
```

centro-max, 2023-07-21:
[image]

I have 4 Baly-BMS connected and always the same behaviour.

mr-manuel, 2023-07-22:
Please post the log of the driver itself: https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#datalogdbus-serialbatteryttyusbcurrent-or-datalogdbus-serialbatteryttyama0current

centro-max, 2023-07-22:
There seems to be a problem here that I don't understand. All 4 BMS report the same error in the LOG file, but the display of the BMS in the Victron is fine except for the red display of the min/max values. The data (voltage, current, SOC etc.) are displayed correctly by all 4 BMS.

/data/log/dbus-serialbattery.ttyUSB*/current
...
@4000000064bb8a7f22ed9104 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8a802544bbbc INFO:SerialBattery:refresh_data: result: False
@4000000064bb8ab125f36694 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8b2930f0406c INFO:SerialBattery:refresh_data: result: False
@4000000064bb8c23023d8e14 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8c8b08665e24 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8c8c099577e4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8d2a11fe3574 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8d2b14c72144 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8e7a27b8c8fc INFO:SerialBattery:refresh_data: result: False
@4000000064bb8ec82dce72b4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8f1030ef0404 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8f273228a8fc INFO:SerialBattery:refresh_data: result: False
@4000000064bb8f8d384e63fc INFO:SerialBattery:refresh_data: result: False
@4000000064bb8f8e3a18d35c INFO:SerialBattery:refresh_data: result: False
@4000000064bb8fc0035b0c04 INFO:SerialBattery:refresh_data: result: False
@4000000064bb90c01992516c INFO:SerialBattery:refresh_data: result: False
@4000000064bb90f412cbf554 INFO:SerialBattery:refresh_data: result: False
@4000000064bb91441a313d04 INFO:SerialBattery:refresh_data: result: False

mr-manuel, 2023-07-22:
Why did you not execute the whole command?

centro-max, 2023-07-22:
I do not understand, here a screenshot what i did:

[image]

centro-max, 2023-07-22:
The entire "current" file from the USB4 port to which my Daly BMS No. 2 is connected:

@4000000064bb6b5a19a86d94 INFO:SerialBattery:refresh_data: result: False
@4000000064bb6bbd1e04edcc INFO:SerialBattery:refresh_data: result: False
@4000000064bb6bd72109f83c INFO:SerialBattery:refresh_data: result: False
@4000000064bb6cda2d6fcc3c INFO:SerialBattery:refresh_data: result: False
@4000000064bb6d0c2fe859d4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb6d273553d13c INFO:SerialBattery:refresh_data: result: False
@4000000064bb6d5a37ce1c24 INFO:SerialBattery:refresh_data: result: False
@4000000064bb6d8e3b029b04 INFO:SerialBattery:refresh_data: result: False
@4000000064bb6dc103403cbc INFO:SerialBattery:refresh_data: result: False
@4000000064bb6ddc04c31bcc INFO:SerialBattery:refresh_data: result: False
@4000000064bb6e3e092291d4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb6eed15606b74 INFO:SerialBattery:refresh_data: result: False
@4000000064bb6f07156d8ebc INFO:SerialBattery:refresh_data: result: False
@4000000064bb6f3d191abc24 INFO:SerialBattery:refresh_data: result: False
@4000000064bb6f3e1b0f641c INFO:SerialBattery:refresh_data: result: False
@4000000064bb6fa322475a3c INFO:SerialBattery:refresh_data: result: False
@4000000064bb6ff6255c8594 INFO:SerialBattery:refresh_data: result: False
@4000000064bb700f2729f294 INFO:SerialBattery:refresh_data: result: False
@4000000064bb702829a32ff4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7112368ed1dc INFO:SerialBattery:refresh_data: result: False
@4000000064bb716300790a54 INFO:SerialBattery:refresh_data: result: False
@4000000064bb721b0c20ca2c INFO:SerialBattery:refresh_data: result: False
@4000000064bb72340d64a46c INFO:SerialBattery:refresh_data: result: False
@4000000064bb72650f43c894 INFO:SerialBattery:refresh_data: result: False
@4000000064bb72eb1909d01c INFO:SerialBattery:refresh_data: result: False
@4000000064bb740229e8a534 INFO:SerialBattery:refresh_data: result: False
@4000000064bb747f31071bfc INFO:SerialBattery:refresh_data: result: False
@4000000064bb75620422ad7c INFO:SerialBattery:refresh_data: result: False
@4000000064bb757c0393468c INFO:SerialBattery:refresh_data: result: False
@4000000064bb757d05f7a814 INFO:SerialBattery:refresh_data: result: False
@4000000064bb75e40c3da8cc INFO:SerialBattery:refresh_data: result: False
@4000000064bb75fc0d5b78c4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb76300c83f4e4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb76310ffd8b44 INFO:SerialBattery:refresh_data: result: False
@4000000064bb764e108dc13c INFO:SerialBattery:refresh_data: result: False
@4000000064bb766a133409dc INFO:SerialBattery:refresh_data: result: False
@4000000064bb769f16976ca4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb76b8190572ec INFO:SerialBattery:refresh_data: result: False
@4000000064bb77221f7b8c24 INFO:SerialBattery:refresh_data: result: False
@4000000064bb77e92bd0b684 INFO:SerialBattery:refresh_data: result: False
@4000000064bb78032c1bd8e4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb784b3100ea0c INFO:SerialBattery:refresh_data: result: False
@4000000064bb78653409e44c INFO:SerialBattery:refresh_data: result: False
@4000000064bb7a020e218bcc INFO:SerialBattery:refresh_data: result: False
@4000000064bb7a030fd6b1f4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7a04110d594c INFO:SerialBattery:refresh_data: result: False
@4000000064bb7ab81ae794b4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7ad41d591d6c INFO:SerialBattery:refresh_data: result: False
@4000000064bb7aed1c2ebab4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7aee1e912454 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7b5223837804 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7b5326bcd114 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7b8624ffdaec INFO:SerialBattery:refresh_data: result: False
@4000000064bb7ba229090eec INFO:SerialBattery:refresh_data: result: False
@4000000064bb7c372fe928dc INFO:SerialBattery:refresh_data: result: False
@4000000064bb7c38323bcf54 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7c3934a29624 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7c51337a3af4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7c6b3420a86c INFO:SerialBattery:refresh_data: result: False
@4000000064bb7c9e38fd3db4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7dd612e03104 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7e731a41a43c INFO:SerialBattery:refresh_data: result: False
@4000000064bb7ea61da754dc INFO:SerialBattery:refresh_data: result: False
@4000000064bb7eda22d20254 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7ef42074ebfc INFO:SerialBattery:refresh_data: result: False
@4000000064bb7ef52316fa94 INFO:SerialBattery:refresh_data: result: False
@4000000064bb7f8b29ba55bc INFO:SerialBattery:refresh_data: result: False
@4000000064bb800d32d8da74 INFO:SerialBattery:refresh_data: result: False
[... 18 lines cut ...]
@4000000064bb880838f6542c INFO:SerialBattery:refresh_data: result: False
@4000000064bb88230079642c INFO:SerialBattery:refresh_data: result: False
@4000000064bb88a505fffce4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8a141c687fc4 INFO:SerialBattery:refresh_data: result: False
@4000000064bb8a151de94f7c INFO:SerialBattery:refresh_data: result: False
[... cut ...]
@4000000064bb8f273228a8fc INFO:SerialBattery:refresh_data: result: False
@4000000064bb8f8d384e63fc INFO:SerialBattery:refresh_data: result: False
@4000000064bb8f8e3a18d35c INFO:SerialBattery:refresh_data: result: False
@4000000064bb8fc0035b0c04 INFO:SerialBattery:refresh_data: result: False
@4000000064bb90c01992516c INFO:SerialBattery:refresh_data: result: False
@4000000064bb90f412cbf554 INFO:SerialBattery:refresh_data: result: False
@4000000064bb91441a313d04 INFO:SerialBattery:refresh_data: result: False

mr-manuel, 2023-07-22:
The logs in the screenshot and the logs you posted are different. In the screenshot (correct) you see the timestamp and in the logs you posted (wrong) an encoded string.

centro-max, 2023-07-22:
Is that what you need The logs just tell me that something is wrong...

mr-manuel, 2023-07-22:
We always need the timestamp, but we have it in the screenshot.

mr-manuel, 2023-07-22:
Please also post the exact model of the BMS you are using as also the soft and hardware version of the BMS.

centro-max, 2023-07-22:
Here the screenshot (BMS Nr. 2, USB4) from the Daly-App and a photo from my reserve BMS which is the same type.
[image]
[image]

mr-manuel, 2023-07-22:
@transistorgit if you have some spare time could you check this?

transistorgit, 2023-07-22:
Im on vacation right now, but I will check it I find some time.

1st impression:
On my 2 Dalys, it works as expected: random cells get highlighted. The cells are indicated by bits from the daly directly, thats nothing that I calculate. And this Bit is a Fault flag, so the highlight doesnt mean that they are the min/max cells.

I suggest a bms firmware update 

Brgds
Bernd

PS: adding the cell index to the min/max voltage fields may be a meaningful feature request

mr-manuel, 2023-07-22:
> PS: adding the cell index to the min/max voltage fields may be a meaningful feature request

They are on the "Details" page. Are they also needed on the cells page?

centro-max, 2023-07-31:
I have updated the firmware of my Daly BMS to the latest version from 25.03.2023 (31_230325_001T, obtained directly from the manufacturer/supplier). Unfortunately without success, the error persists.

While researching I found out that there are different hardware versions of Daly, the MCU (ST103 or ST030 or GD230 - first Number of Firmware) and the AFE (309E or 303E or DVCE or 9818 - second Number of Firmware) are different. Maybe this helps.

I have GD230 and 309E.

[image]

[image]

mr-manuel, 2023-09-04:
After reviewing this issue I'm the opinion that this is not a problem of incorrect min/max cell voltage dislay, but only a problem of incorrect cell balancing display. Since the values in `Cells (Min/Max/Diff)` are always correct in the screenshots.

The red background is displaying which cell is currently being balanced. 

> [image]

In this screenshot I see that you have disabled the active cell balancing. What happens if you turn it on?

centro-max, 2023-09-04:
The "active cells balacimg" switch is automatically activated by the system and indicates whether balancing is active or not.

the cell voltages are displayed correctly, the red background should always be on the cell with the highest or lowest voltage, the first cell is always displayed, regardless of the voltage of the first cell. this is definitely not correct!

mr-manuel, 2023-09-04:
Do you see in the app which cells are being balanced or do you see only that balancing is active?

centro-max, 2023-09-04:
> the App shows always the highest cell in red and the lowest cell in green

[image]

centro-max, 2023-09-04:
the App shows always the highest cell in red and the lowest cell in green

[image]

mr-manuel, 2023-09-26:
@transistorgit if you find some time can you help me with that?

transistorgit, 2023-10-08:
Hi,

today I looked into this issue. Unfortunately, I have the hardware ST103 which seems different to @centro-max hw. It would be interesting, if any other G230 user could acknowledge the problem.

In the meantime, my only suggestion would be to make a config entry to disable the balance status for these users.

best regards

mr-manuel, 2024-02-16:
I don't think that the red and blue colors indicate that these cells are balancing. It's showing only the highes and lowest cell voltage, since the cell balancing is not active in your screenshot.

@centro-max can you confirm that the one cell has only a red background, if the balancing is active?