Louisvdw/dbus-serialbattery #753: Battery synced to 100% on alarm

Micha854, 2023-07-21:
### Describe the bug

today my daly bms reported a (high voltage) alarm

Immediately afterwards the battery was synchronized to 100% although the actual state of charge according to smartshunt was only about 78%...

I suspect that the driver was to blame

### How to reproduce

on bms alarm

### Expected behavior

...

### Driver version

newest dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.01

### BMS type

Daly Smart BMS

### Cell count

8

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):

MAX_BATTERY_CHARGE_CURRENT    = 50.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0
TIME_TO_GO_ENABLE = False
TIME_TO_SOC_VALUE_TYPE = 2
BMS_TYPE = Daly
BATTERY_CAPACITY = 95
INVERT_CURRENT_MEASUREMENT = -1

#VOLTAGE_DROP = 0.125
```

### Relevant log output

```shell
dont no
```

### Any other information that may be helpful

[image]

teefixx, 2023-07-22:
Hi

I had a similar problem. I raised bot upper alarm levels in the daly BMS via the PC software. It was set too low and threw out an error...

Micha854, 2023-07-24:
I have a smart shunt in my system and temporarily solved the problem with a small bash script which is called every 15 minutes by a cron job

what does the script do?
as soon as the soc difference between BMS and shunt is higher than 10%, the soc of the BMS is adjusted to the soc of the shunt and charging continues

Line 6 may need to be adjusted. with me the shunt is on USB0

```
#!/bin/bash

X=$(dbus -y com.victronenergy.system /Dc/Battery/Soc GetValue)
socShunt=${X%.*}

Y=$(dbus -y com.victronenergy.battery.ttyUSB0 /Soc GetValue)
socDaly=${Y%.*}

datum=$(TZ=Europe/Berlin date +'%Y-%m-%d %H:%M:%S')

difference=$(($socDaly-$socShunt))

if [[ "$difference" -ge 10 ]]; then
    echo "SoC Difference bei $difference% | synchronisiere Ladezustand Neu!"
    echo "$(dbus -y com.victronenergy.battery.ttyUSB0 /Settings/ResetSoc SetValue %$socShunt)"
    echo "$datum: SoC Difference bei $difference% | synchronisiere Ladezustand Neu!" >> log
fi
```

mr-manuel, 2023-07-24:
You could do this in Node-RED.

closed 2023-07-31 by mr-manuel