Louisvdw/dbus-serialbattery #755: Before reaching 55.2V - charge current drops to 5A

exenbema, 2023-07-22:
### Describe the problem

hello! thx for this driver, i use it since the beginning - now i update and have a question to the max cell voltage of 55,2 (my setting daly, lifepo4, MP2 and see below)

With the old version i reached the 55,2 - then it dropped to 53,5V. but now:
i reach 54.4 but then driver reduces to 5A charge current and sometimes it dont allow charging at all
around  13:30 it reached 55.2V but then it dont reduce to float voltage (every day like that)

if there is load, the PV has enogh power for 2-3000Watts, so this is not the limitation..

Daly 100% (i dont trust it)
shunt 100%

thx for helping me out! thx markus

### Driver version

1

### Venus OS device type

Cerbo GX

### Venus OS version

2.93

### BMS type

Daly Smart BMS

### Cell count

16

### Connection type

Please select

### Config file

```ini
my changed settings in the config file
70
90

LINEAR_LIMITATION_ENABLE = False (step mode)
CVCM_enable: True
MAX_VOLTAGE_TIME_SEC = 180

CELL_VOLTAGES_WHILE_CHARGING = 3.5, 3.47, 3.40, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION = 0, 0.05, 0.5, 1

CELL_VOLTAGES_WHILE_DISCHARGING = 2.90, 2.95, 3.00, 3.10
MAX_DISCHARGE_CURRENT_CV_FRACTION = 0, 0.1, 0.5, 1

CCCM_T_ENABLE = False
; Charge current control management referring to temperature enable (True/False).
DCCM_T_ENABLE = False

CCCM_SOC_ENABLE = False
; Discharge current control management enable (True/False).
DCCM_SOC_ENABLE = False

BATTERY_CAPACITY = 230
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = -1
```

### Relevant log output

```shell
cant create log file
error no reply - i dont know how to do it
```

mr-manuel, 2023-07-22:
Unfortunately you did not correctly filled the fields that were asked. Every field had a very detailed description on how to get that informations.

exenbema, 2023-07-22:
i did the command tail -F -n 100 /data/log/serial-starter/current | grep dbus-serialbattery | tai64nlocal
but i just get a list of errors

i checked i have this log directory...is there another way to get the logfiles? i am the only one with trouble to get the log files? (i m superadmin via ssh)

mr-manuel, 2023-07-22:
Wrong fields.

> ### Driver version
> 1

> ### Connection type
> Please select

A config file like that would never work.

> ### Config file
>
> ```ini
> my changed settings in the config file
> 70
> 90
> 
> LINEAR_LIMITATION_ENABLE = False (step mode)
> CVCM_enable: True
> ```

Just check [Serial BMS connection](https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#serial-bms-connection)

exenbema, 2023-07-24:
i try to get the log file, but i cant . here is my problem, i get a "no reply" (as you described in your info site) - but i dont know how to go on. the cerbo gx is connected via USB Hub to the 2 BMS (2 batteries) - the cables and connection are ok, otherwise i would not function (i checked with a second installation - same here). the driver is working, doing his job for 1 year by now - but i just get you the log file:(

No reply in log file
Check your cable connections, if the log file shows ERROR: No reply - returning from the battery.

The RX/TX lights should both flash as data is transfered. If only one flashes then your RX/TX might be swapped.

mr-manuel, 2023-07-24:
Sorry but I cannot understand you. How many problems do you have now? Please list one after another. There seems two problems in the first post, but now you also have no connection? How can you then have the problem in the first post? Also I don‘t understand when you are using which version of the driver with which problem. You are writing with the old version and now. Nobody except you can know which versions you mean.

You have to explain all with much details possible else we have to imagine the rest :-)

exenbema, 2023-07-24:
thx that you helping me and sorry for confusion

lets stick to my first problem: in the morning solarcharger goes (as expected) to bulk loading and i reach relative quick the 54.4 V but then driver reduces to 5A charge current, so it slowes down way to early. slowly it charges, but not with full power, ...with reduced power. much later than around 13:30 it reached 55.2V

i cant find any explanation for that behavior - because i set every other charging method to false (so its not a mix of differnent modes)

probably my question is: how i can set the same as in version 0.14.3 where you can set everything to false exept the part with the cell voltages - and the reason is that daly is soo bad in SOC counting, but i cant change that. so i excluded the temp and SOC charging as much as possible and let only the VOLTAGE based charging True. But in the new driver this is much more complex or is there a way to just reduce it to voltage based? i think that would solve my problem

big thx for your time!!

mr-manuel, 2023-07-25:
Please read
https://github.com/Louisvdw/dbus-serialbattery/blob/86a6b956e15bee21d5a298283c9d121f0eb82a02/etc/dbus-serialbattery/config.default.ini#L86-L104

Additionally check the Parameters pahe in the remote console/GUI under Serialbattery. There you see the limiting component of the driver.

exenbema, 2023-07-26:
hi manuel
i set everything back to default and rebootet - but same behavior, in the parameter of the serial driver he limits to 5A.

the only section is true is this you copied  ; --------- Cell Voltage Current limitation (affecting CCL/DCL) --------- 

and we tried the charge mode true and false setting - no difference

.......
for me it seems to look into the daly SOC and there is 100% and so the driver limits the current to 5 A - but with the driver version 0.14.3. this is irrelevant, because its only voltage based and not soc

mr-manuel, 2023-07-26:
Yes, because this is enabled... you have to change the parameters or to disable it.

> Additionally check the Parameters page in the remote console/GUI under Serialbattery. There you see the limiting component of the driver.

Did you do that? It will tell you all you need.

exenbema, 2023-07-26:
yes i did all
1) under parameters i see CCL to 5 A (the component what causes this i don t see)
2) i set every section to FALSE (SOC, TEMP and so on) but not the section with voltage part - i think therefore i have disabled it.

mr-manuel, 2023-07-26:
Check the FAQ there you find all. I recommend updating to >v1.0.0 in order to see what is limiting the current.

exenbema, 2023-07-26:
we made the update last week and choose 1 in the update process, so its the new driver.BUT strange that i dont see the limiting component in the parameters. also strange that the serialdriver shows stil version 0.12 in the GUI overview of the cerbo.

could it be that i have a mix of the new and old driver? if so, should i reinstall? did you ever saw a thing like that or am i the only and first one with this? 

thx!!

exenbema, 2023-07-26:
ok i will do, thx

exenbema, 2023-07-27:
today i solved the problem:)

- i wanted to go back to 0.14.3 - but after pasting the link it destroyed the OS of venus
- so we reinstalled complete new version 3 of venus OS
- then installed new serial Version 1.... 
- now its ok - i see now in the parameter the settings, much more as before
- now the settings are ok - we go to 55.2V immediatly 

tomorrow we will observe

i think the problem was in the update process between 0.12 to version 1. because not everything was updated obvisly. i told you the version nummer remains 0.12 and i didnt see the change in the parameters - i dont know what was the problem - anyway its solved!!

thx for helping!!

kind regards markus

closed 2023-07-27 by mr-manuel