Louisvdw/dbus-serialbattery #764: Problem with the charging current limitation

flyer9999, 2023-07-29:
### Describe the bug

The charge current limit oscillates between the setpoint of: MAX_CHARGE_CURRENT_CV_FRACTION and MAX_CHARGE_CURRENT_T_FRACTION ...
Visible under: Device List -> SerialBattery(LLT/JBD) -> Parameters -> Charge Limitation 

Although the value (limitation) of MAX_CHARGE_CURRENT_CV_FRACTION is SMALLER than the value of MAX_CHARGE_CURRENT_T_FRACTION, the charge current limitation is constantly changed or increased to the value of MAX_CHARGE_CURRENT_T_FRACTION. 
Shouldn't the smallest value for the charge current limitation be decisive here ?

Translated with www.DeepL.com/Translator (free version)

### How to reproduce

[image]

[image]

[image]

### Expected behavior

In my opinion, the lowest calculated charge current limit should apply ...

### Driver version

1.0.20230531

### Venus OS device type

Please select

### Venus OS version

v3.01

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

8

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 20.0
MAX_BATTERY_DISCHARGE_CURRENT = 0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.500
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example:
;     1 BMS: Jkbms_Ble C8:47:8C:00:00:00
;     3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
;              BMS on purpose, then you have to restart the driver/system to reset the block.
; False: Charge and discharge is not blocked on BMS communication loss
; True: Charge and discharge is blocked on BMS communication loss, it's unblocked when connection is established
;       again or the driver/system is restarted
BLOCK_ON_DISCONNECT = False

; --------- Charge mode ---------
; Choose the mode for voltage / current limitations (True / False)
; False is a step mode: This is the default with limitations on hard boundary steps
; True is a linear mode:
;     For CCL and DCL the values between the steps are calculated for smoother values (by WaldemarFech)
;     For CVL max battery voltage is calculated dynamically in order that the max cell voltage is not exceeded
LINEAR_LIMITATION_ENABLE = False

; Specify in seconds how often the linear values should be recalculated
LINEAR_RECALCULATION_EVERY = 60
; Specify in percent when the linear values should be recalculated immediately
; Example: 5 for a immediate change, when the value changes by more than 5%
LINEAR_RECALCULATION_ON_PERC_CHANGE = 5
```

### Relevant log output

```shell
[DEFAULT]
```

mr-manuel, 2023-07-29:
Thanks for reporting. This could be indeed a bug. I will look into it after next week.

mr-manuel, 2023-07-29:
Scrolling through the code I did not found what I thought the problem could be. Could you please also post a screenrecording of this behaviour? You can use [ScreenToGif](https://www.screentogif.com/) on Windows. There is also a portable version where you don't have to install anything.

flyer9999, 2023-08-01:
Sorry mr-Manuel, I can't recreate the problem currently. After I noticed that after several changes in the config.default.ini with a restart each of my changes no longer showed effect I reinstalled with your installer --> option 1 (Latest Release), from then on the adjusted values in the config.default.ini were adopted again. I could not reproduce the described problem.

Fact is that after long testing (and thus changing values in the config.default.ini with a reboot each time) I need a new installation because the values are no longer taken over by the system....... After a new installation and adjusting the config.default.ini the values are immediately taken over again.

mr-manuel, 2023-08-02:
You should not change the `config.default.ini`. Please read carefully the notes you see after the installation.

flyer9999, 2023-08-04:
so, have now reinstalled and only edited the /data/etc/dbus-serialbattery/config.ini ....

[image]

[image]

mr-manuel, 2023-08-04:
1. What is the reason to limit the charging current at lower voltage: `3.25`, `3.00`
2. What is the reason to limit the discharging current at higher voltage: `3.35`, `3.40`, `3.45`

This makes no sense to me and is very likely the reason of the issue. Please change your configuration that `CELL_VOLTAGES_WHILE_CHARGING` and `CELL_VOLTAGES_WHILE_DISCHARGING` are not overlapping.

Also try with the default config for this values.

flyer9999, 2023-08-04:
The background for this gradation is the following: When the battery is empty, there is usually a difference in cell voltage, e.g. 2.95V and 3.05. Now I do not want to start charging with full current, full current only when one cell has 3.3V (then the other cells are all together again).
The same principle for discharging ...

mr-manuel, 2023-08-04:
Never saw this and it does not make any sense to me. Try the default configuration and check if the same issue is still present.

flyer9999, 2023-08-26:
First: With the default settings there is no problem ...
But well, so I think that it is more gentle for the cells in the empty state not to charge them immediately with the full current. There are also topics that deal with the problem of dentrieden ...
Regardless of that, there should be a clear definition and the software should not jump back and forth between 2 different limits ? Something must / should have yes Prio, for me actually the smallest current specification.

Do not get me wrong, I think your work is great and always like to be taught better :-)

Translated with www.DeepL.com/Translator (free version)

mr-manuel, 2023-12-04:
I'm of the same opinion of @dropax. I will close this, since it's not planned to apply lower voltage limits for charging and upper limits for discharging. The oscillating is only due to your configuration which is not supported.

closed 2023-12-04 by mr-manuel