Louisvdw/dbus-serialbattery #767: Max Voltage Reset not respected

optio50, 2023-07-29:
### Describe the bug

Driver Version 1.0.20230723dev

modified config.ini
MAX_CELL_VOLTAGE   = 3.55
FLOAT_CELL_VOLTAGE = 3.45

The default config suggests it resets when SOC reaches 90%
I have a SOC of 83% and max charge voltage has not yet reset (still shows Float voltage of 13.8V instead of 14.2V)

### How to reproduce

Steps to reproduce the behavior:

Use modified config.ini 
MAX_CELL_VOLTAGE   = 3.55
FLOAT_CELL_VOLTAGE = 3.45

Fully charge battery (14.2)
correctly sets Float to 13.8 after 15 min.
discharge below 90%
Max charge voltage does not return to 14.2

### Expected behavior

Max charge voltage reset to Bulk/Absorption voltage (MAX_CELL_VOLTAGE)

### Driver version

1.0.20230723dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.10~18

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

MAX_BATTERY_CHARGE_CURRENT    = 20.0
MAX_BATTERY_DISCHARGE_CURRENT = 40.0
MAX_CELL_VOLTAGE   = 3.55
FLOAT_CELL_VOLTAGE = 3.45

TIME_TO_SOC_VALUE_TYPE = 3
TIME_TO_SOC_RECALCULATE_EVERY = 10
TIME_TO_SOC_INC_FROM = True
TIME_TO_SOC_POINTS = 100, 95, 90, 85, 75, 50, 25, 20, 10, 0
```

### Relevant log output

```shell
.
```

optio50, 2023-07-29:
Reset finally occurred at 79.9%

mr-manuel, 2023-07-29:
This was changed with the latest dev because of SoC drift.

https://github.com/Louisvdw/dbus-serialbattery/blob/b4b9552ca603683ca62eef7197f6dfb01b4791a9/etc/dbus-serialbattery/config.default.ini#L99-L102

closed 2023-07-29 by mr-manuel

optio50, 2023-07-30:
Sorry
Dont know How I missed that.
This part of the documentation in config.default.ini threw me off.

The battery reached max voltage of 55.2V and hold it for 900 seconds, the the CVL is switched to
 float voltage of 53.6V to don't stress the batteries. Allow max voltage of 55.2V again, if SoC is
 once below 90%
 OR
 The battery reached max voltage of 55.2V and the max cell difference is 0.010V, then switch to float
 voltage of 53.6V after 300 additional seconds to don't stress the batteries. Allow max voltage of
 55.2V again if max cell difference is above 0.080V or SoC below 90%.

mr-manuel, 2023-07-30:
Thanks. I will correct it.