Louisvdw/dbus-serialbattery #768: JKBMS BLE unstable

githubfrance, 2023-07-30:
### Describe the bug

Hi,
"With a USB connection, everything is OK. Thank you again for this excellent work.
However, the number of USB ports is not sufficient to my project, so I have to use the BLE connection for the batteries. As you can see below, the Raspberry Pi detects both batteries. I have added the two MAC addresses in config.ini and reinstalled the latest version.
The battery data is collected once, but then nothing happens. Did I forget something?"

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

Same behavior as connected to USB

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

V3.0.1

### BMS type

JKBMS / Heltec BMS

### Cell count

4

### Connection type

Please select

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 60.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.550
FLOAT_CELL_VOLTAGE = 3.450

;Device C8:47:8C:ED:06:11 JKBMS-Heating
;Device C8:47:8C:EE:29:AF JKBMS-DISPLAY

;Discovered device  c8:47:8c:ed:06:11
;Discovered device  c8:47:8c:ee:29:af

BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:ED:06:11, Jkbms_Ble C8:47:8C:EE:29:AF
BLOCK_ON_DISCONNECT = False
TIME_TO_GO_ENABLE = True
BMS_TYPE = Jkbms
```
[... 127 lines cut ...]
@4000000064c61267178bb7b4 reconnect
@4000000064c61267178bbf84 Exiting bt-loop
@4000000064c612672bce1a8c Missing device address argument
@4000000064c6126836da6d54 INFO:SerialBattery:Starting dbus-serialbattery
@4000000064c6126836ede1cc INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@4000000064c6126910312634 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:ED:06:11
@4000000064c6126910389874 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:ED:06:11
@4000000064c61322095e3f2c ERROR:SerialBattery:[org.freedesktop.DBus.Error.NoReply] Message recipient disconnected from message bus without replying
@4000000064c6132209667c8c ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@4000000064c61322096a2ddc 
@4000000064c613c722918e54 Missing device address argument
@4000000064c613c829eb4514 INFO:SerialBattery:Starting dbus-serialbattery
@4000000064c613c829f9fe9c INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@4000000064c613c9024a6b0c INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:ED:06:11
@4000000064c613c902513554 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:ED:06:11
@4000000064c6184b398113b4 Terminated
@4000000064c6184c09bf6a2c Missing device address argument
@4000000064c6184d15fe41b4 INFO:SerialBattery:Starting dbus-serialbattery
@4000000064c6184d160e723c INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@4000000064c6184d2b3f115c INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:ED:06:11
@4000000064c6184d2b47c3ec INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:ED:06:11
@4000000064c6185108cf60f4 ERROR:SerialBattery:[org.freedesktop.DBus.Error.NoReply] Message recipient disconnected from message bus without replying
@4000000064c6185108d82324 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@4000000064c6185108dbcca4
```

mr-manuel, 2023-07-30:
> provide the logs in the correct form

See https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#serial-bms-connection, I copied the important notice over here for you:

💡 If you are opening an issue or posting your logs somewhere please make sure you execute the complete commands to get the logs, including `tai64nlocal`. Without readable timestamps we cannot help you.

> Have you tried the latest nightly from the dev branch?

See https://github.com/Louisvdw/dbus-serialbattery/issues/765#issuecomment-1656802054

mr-manuel, 2023-07-30:
By the way: You do‘t have to create a new issue and close this one. You can edit it or add the correct information as a comment.

githubfrance, 2023-07-30:
Hi,
Perhaps I found a way to get the JKBMS connected via BLE.
I tuned off the BMSs and the Raspberry then started again the tow batteries, and the Raspberry.
This time, I could get the batteries on the venus.
[image]

After a while the RUUVY came back to the Venus.
[image]

After one hour run the batteries remains active on Venus OS
Will reproduce this procedure to see it is the right way.
Bests
Patrick

githubfrance, 2023-07-30:
Bad luck, the batteries were unconnected after a while without any action on the batteries nor the VenosOS.
[image]

githubfrance, 2023-07-30:
Hi Mr-Manuel
Attached all files you requested

[JKBMS-VenusOS.zip](https://github.com/Louisvdw/dbus-serialbattery/files/12209243/JKBMS-VenusOS.zip)

mr-manuel, 2023-07-30:
Unfortunately I don't know why you don't get it, but this is not that what was requested. Probably a language barrier. Try to read again https://github.com/Louisvdw/dbus-serialbattery/issues/768#issuecomment-1657097367.

I also don't know where you picked up `tail -F -n 100 /data/log/dbus-ble-sensors/current | grep dbus-ble-sensors | tai64nlocal`. This has nothing to do with this driver.

Anyway only one Bluetooth device can be connected at the same time to the BMS.

githubfrance, 2023-07-30:
Hi, this time the log file are as you wan, with the time stamp :-)

[bat1.log](https://github.com/Louisvdw/dbus-serialbattery/files/12209390/bat1.log)
[bat0.log](https://github.com/Louisvdw/dbus-serialbattery/files/12209391/bat0.log)

mr-manuel, 2023-07-31:
I still don't know, if you tried the nightly from the dev branch...

Somehow the service files are not generated correctly. Did you get any error message on installing?

Post the output of `cat /service/dbus-blebattery.*/run`. Maybe it's services instead of service, can't check right now.

githubfrance, 2023-07-31:
Hi I had no errors on the installation 
root@raspberrypi2:~# cat /service/dbus-blebattery.0/run
#!/bin/sh
exec 2>&1
bluetoothctl disconnect
python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:47:8C:ED:06:11

root@raspberrypi2:~# cat /service/dbus-blebattery.1/run
#!/bin/sh
exec 2>&1
bluetoothctl disconnect
python /opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py Jkbms_Ble C8:47:8C:EE:29:AF
root@raspberrypi2:~# /etc/init.d/bluetooth start
Starting bluetooth: bluetoothd.

githubfrance, 2023-07-31:
I just reinstalled it 👍  wget -O /tmp/install.sh https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-                                                           serialbattery/install.sh
--2023-07-31 08:49:14--  https://raw.githubusercontent.com/Louisvdw/dbus-serialbattery/master/etc/dbus-serialbattery/instal                                                           l.sh
Resolving raw.githubusercontent.com... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4889 (4.8K) [text/plain]
Saving to: ‘/tmp/install.sh’

/tmp/install.sh                100%[===================================================>]   4.77K  --.-KB/s    in 0.002s

2023-07-31 08:49:15 (3.01 MB/s) - ‘/tmp/install.sh’ saved [4889/4889]

root@raspberrypi2:~#
root@raspberrypi2:~# bash /tmp/install.sh

1) latest release (recommended)  3) nightly build                 5) quit
2) specific version              4) local tar file
Select which version you want to install and enter the corresponding number [1]: 1
Selected: latest release (recommended)

Found 2 Bluetooth BMS in the config file!

Installing required packages...
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/raspberrypi2/Packages.gz.
Updated source 'raspberrypi2'.
No packages installed or removed.
Requirement already satisfied: bleak in /usr/lib/python3.8/site-packages (0.20.2)
Requirement already satisfied: async-timeout<5,>=3.0.0; python_version < "3.11" in /usr/lib/python3.8/site-packages (from bleak) (4.0.2)
Requirement already satisfied: dbus-fast<2.0.0,>=1.83.0; platform_system == "Linux" in /usr/lib/python3.8/site-packages (from bleak) (1.87.5)
Packages installed.

Installing Jkbms_Ble C8:47:8C:ED:06:11 as dbus-blebattery.0
Installing Jkbms_Ble C8:47:8C:EE:29:AF as dbus-blebattery.1
Stopping bluetooth: bluetoothd.
Starting bluetooth: bluetoothd.

SERIAL battery connection: The installation is complete. You don't have to do anything more.

BLUETOOTH battery connection: There are a few more steps to complete installation.

    1. Please add the Bluetooth BMS to the config file "/data/etc/dbus-serialbattery/config.ini" by adding "BLUETOOTH_BMS":
       Example with 1 BMS: BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00
       Example with 3 BMS: BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
       If your Bluetooth BMS are nearby you can show the MAC address with "bluetoothctl devices".

    2. Make sure to disable Settings -> Bluetooth in the remote console/GUI to prevent reconnects every minute.

    3. Re-run "/data/etc/dbus-serialbattery/reinstall-local.sh", if the Bluetooth BMS were not added to the "config.ini" before.

    ATTENTION!
    If you changed the default connection PIN of your BMS, then you have to pair the BMS first using OS tools like the "bluetoothctl".
    See https://wiki.debian.org/BluetoothUser#Using_bluetoothctl for more details.

CUSTOM SETTINGS: If you want to add custom settings, then check the settings you want to change in "/data/etc/dbus-serialbattery/config.default.ini"
                 and add them to "/data/etc/dbus-serialbattery/config.ini" to persist future driver updates.

githubfrance, 2023-07-31:
Batteries are back again
[image]

githubfrance, 2023-07-31:
Strange,  the Ruuvy Tag are disconnected a short time after the re-installation. Perhaps compatibility issues between the Ruuvy BLE program and yours

mr-manuel, 2023-07-31:
Why are you installing the latest release, if I told you over 10x to install the nightly version from the dev branch?

Did you read the after install notes?

> 2. Make sure to disable Settings -> Bluetooth in the remote console/GUI to prevent reconnects every minute.

Therefore it may not work woth your sensors. @

githubfrance, 2023-07-31:
Hi, 
Bluetooth was always disabled
[image]

I just installed the dev branch 
[image]

Both JKBMSs and RUUVIs  are back again.
I will keep you informed in case any instabilities
Many thanks
I already support $$ you and will do it again right now, Despite the occasionally "aggressive" tone used, sometimes the end user may not be very attentive to the instructions :-) and this has nothing to do with the language barrier :-)

mr-manuel, 2023-07-31:
Sometimes it's really annoing, if you spend 100+ hours for a good driver documentation and nobody is reading it. Thank you for your appreciation!

closed 2023-08-05 by githubfrance

mr-manuel, 2023-08-05:
It‘s running fine now?

githubfrance, 2023-08-05:
Hi, Thank you for asking, 
yes every thing is OK, stable and works fine.
Grat Work.

Baxter117, 2023-09-19:
Hi,
Are the fixes which maked the connection stable in the dev-nightly already dropped into the master branch ?

in early august I had the same issue as here discussed with stability problems in the master, which fixed also by using the dev-nightly branch. But I hoped I could turn back to master some day :)

Thank you for the impressive work !

mr-manuel, 2023-09-19:
Hi,
no the changes are not merged into the master yet. First we have to fix some issues.