Louisvdw/dbus-serialbattery #769: No connection zo batteries with v1.0.20230724beta

beliffm, 2023-07-30:
### Describe the problem

I updated to the latest dev-Version and rebooted the GX. It never established the serial connection again. After returning to the latest Version of the master branch it worked instantly again.

I tried the dev-Version because i have intermittent drops of the serial connection with my two BMSes. They are both V2 LLT/JBD. 
This is a different issue though.

### Driver version

1.0.20230723dev

### Venus OS device type

GX Card (integrated in Victron device)

### Venus OS version

V3.01

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 120.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0
BMS_TYPE = LltJbd
```

### Relevant log output

```shell
@4000000064c62d040d64564c #1
@4000000064c62d040db6fe74 #2
@4000000064c62d040db719cc #3
@4000000064c62d040db7219c #4
@4000000064c62d040db72584 #5
@4000000064c62d040db7296c #6
@4000000064c62d040db72d54 #7
@4000000064c62d040db7313c #8
@4000000064c62d040db73524 #9
@4000000064c62d040db7390c #10
@4000000064c62d040db73cf4 #11
@4000000064c62d040db740dc #12
@4000000064c62d040db744c4 #13
@4000000064c62d040db748ac #14
@4000000064c62d040db74c94 #15
@4000000064c62d040db7507c #0
@4000000064c62d040db75464 #1
@4000000064c62d040db7584c #2
@4000000064c62d040dba61a4 #3
@4000000064c62d040dba658c #4
@4000000064c62d040dba6974 #5
@4000000064c62d040dba6d5c #6
@4000000064c62d040dba7144 #7
@4000000064c62d040dba752c #8
@4000000064c62d040dba7914 #9
@4000000064c62d040dba7cfc #10
@4000000064c62d040dba80e4 #11
@4000000064c62d040dba84cc #12
@4000000064c62d040dba88b4 #13
@4000000064c62d040dba8c9c #14
@4000000064c62d040dba9084 #15
@4000000064c62d27210934ec INFO:SerialBattery:
@4000000064c62d272117bf94 INFO:SerialBattery:Starting dbus-serialbattery
@4000000064c62d27212e9f0c INFO:SerialBattery:dbus-serialbattery v1.0.20230723dev
@4000000064c62d27213ad40c INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
@4000000064c62d2721453064 INFO:SerialBattery:Testing LltJbd
@4000000064c62d28301cac5c INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
@4000000064c62d283024842c INFO:SerialBattery:Testing LltJbd
@4000000064c62d2a015c4634 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
@4000000064c62d2a0164c9e4 INFO:SerialBattery:Testing LltJbd
@4000000064c62d2b0fe67134 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB1
@4000000064c62d2b0fe9f3a4 #0
@4000000064c62d2b0fe9ff5c #1
@4000000064c62d2b0fea0344 #2
@4000000064c62d2b0fea072c #3
@4000000064c62d2b0fea0b14 #4
@4000000064c62d2b0fea0efc #5
@4000000064c62d2b0fea12e4 #6
@4000000064c62d2b0fea16cc #7
@4000000064c62d2b0fea1ab4 #8
@4000000064c62d2b0fea1ab4 #9

This is repeating endlessly
```

### Any other information that may be helpful

The BMSes are Version 2 LLT ones The have only one external temperature sensor and two internal ones. The sensor counting is different to the Version 3 BMSes, so the driver shows the FET temperature as battery temperature.

proptied, 2023-07-30:
I also have the same issue (identical logs) on all four of my `LltJbd` BMS with the latest in `dev` branch from July 27. However, using the latest tagged release from July 23 works fine.

beliffm, 2023-07-30:
Thanks for pointing me in the right direction. I mistook the latest dev-version with the prerelese in the master branch. This is working for me too.

beliffm, 2023-07-30:
And there we go.....

optio50, 2023-07-31:
It is looking at /dev/ttyUSB1
Whats connected to /dev/ttyUSB0

beliffm, 2023-08-01:
I have two identical BMSes connected to the GX, one via USB0 the other via USB1. Both logs were also identical.

iLeeeZi, 2023-08-03:
Can confirm that the latest JBD changes broke something

`root@raspberrypi2:~# tail -f /data/log/dbus-serialbattery.ttyUSB3/current | tai64nlocal
2023-08-03 08:06:43.921188500 #2
2023-08-03 08:06:43.921190500 #3
2023-08-03 08:06:43.921191500 #0
2023-08-03 08:06:43.921192500 #1
2023-08-03 08:06:43.921193500 #2
2023-08-03 08:06:43.921195500 #3
2023-08-03 08:06:43.921196500 #0
2023-08-03 08:06:43.921197500 #1
2023-08-03 08:06:43.921199500 #2
2023-08-03 08:06:43.921200500 #3
2023-08-03 08:07:18.148439500 INFO:SerialBattery:
2023-08-03 08:07:18.149973500 INFO:SerialBattery:Starting dbus-serialbattery
2023-08-03 08:07:18.151511500 INFO:SerialBattery:dbus-serialbattery v1.0.20230723dev
2023-08-03 08:07:18.152720500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-08-03 08:07:18.153915500 INFO:SerialBattery:Testing LltJbd
2023-08-03 08:07:19.393235500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-08-03 08:07:19.393244500 INFO:SerialBattery:Testing LltJbd
2023-08-03 08:07:20.501858500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-08-03 08:07:20.501864500 INFO:SerialBattery:Testing LltJbd
2023-08-03 08:07:21.790061500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB3
2023-08-03 08:07:21.790213500 #0
2023-08-03 08:07:21.790215500 #1
2023-08-03 08:07:21.790217500 #2
2023-08-03 08:07:21.790218500 #3
2023-08-03 08:07:21.790219500 #0
2023-08-03 08:07:21.790220500 #1
2023-08-03 08:07:21.790222500 #2
2023-08-03 08:07:21.790223500 #3
2023-08-03 08:07:21.790224500 #0
2023-08-03 08:07:21.790226500 #1
2023-08-03 08:07:21.790227500 #2
2023-08-03 08:07:21.790228500 #3`

mr-manuel, 2023-08-03:
@idstein can you take a look?

idstein, 2023-08-03:
I don’t have access to those JBD BMS versions. I need to implement more verbose logging.

idstein, 2023-08-07:
@iLeeeZi @optio50 @proptied can you post which BMS model your are using from JBD and the firmware version?

iLeeeZi, 2023-08-08:
I have a JBD-SP04S034 4S 200A, need to check the firmware later

wherzig, 2023-08-11:
My JBD BMS SP22S003B (16S) also does not work with Version 1.0x . It is not detected any longer. It works fine with Version 0.14.3 . Is is recognized as hardware version "SP25S003-L16S-100A-B-U-R" by the old driver.

What have been changed to the initialization of the driver that it no longer works? Why have it been changed anyway?

optio50, 2023-08-14:
I have a JBD Version JBD-SP04S020A-L4S-100A-B-U
nightly dev build 1.0.20230723dev
It seems to be working without issue.

tuxntoast, 2023-08-14:
v1.0.20230724beta seems to have an issue with connecting to BMS via Serial. Working on a new BMS driver for EG4-LL (using Revov). Works fine in release [v1.0.20230531] on OS 3.00 or 2.94, however when the Dbus-serialbattery is updated to v1.0.20230724beta  it is suddenly unable to communicate to the ttyUSB0 (or actually is seeming like it is unable to view that this tty device is even connected. Removing this version, and stepping down to DBus-SerialBattery v1.0.20230531 fixes this issue. Have been unable to track down in code the case of this error yet.

mr-manuel, 2023-09-04:
@beliffm @iLeeeZi @wherzig can anyone with the problem contact me on Discord? This would help me to find relevant informations.

mr-manuel, 2023-09-04:
I got this logs from the system of @iLeeeZi. There seems to be a bug (?) in [dbus-fast](https://github.com/Bluetooth-Devices/dbus-fast) additionally. I tried downgrading and then the connection worked again. @idstein  the strange thing is, that this does not seem to happen with the JKBMS_Ble driver.

```bash
2023-09-04 17:54:15.746142500 INFO:SerialBattery:
2023-09-04 17:54:15.746659500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-04 17:54:15.747858500 INFO:SerialBattery:dbus-serialbattery v1.0.20230902dev
2023-09-04 17:54:16.115121500 INFO:SerialBattery:Init of LltJbd_Ble at E0:9F:2A:xx:xx:xx
2023-09-04 17:54:16.115755500 INFO:SerialBattery:Test of LltJbd_Ble at E0:9F:2A:xx:xx:xx
2023-09-04 17:54:21.127748500 --- Logging error ---
2023-09-04 17:54:21.127753500 Traceback (most recent call last):
2023-09-04 17:54:21.127756500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 55, in bt_main_loop
2023-09-04 17:54:21.127761500     self.device = await BleakScanner.find_device_by_address(
2023-09-04 17:54:21.127765500   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 310, in find_device_by_address
2023-09-04 17:54:21.127771500     return await cls.find_device_by_filter(
2023-09-04 17:54:21.127774500   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 369, in find_device_by_filter
2023-09-04 17:54:21.127881500     async with cls(detection_callback=apply_filter, **kwargs):
2023-09-04 17:54:21.127885500   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 139, in __init__
2023-09-04 17:54:21.127890500     get_platform_scanner_backend_type() if backend is None else backend
2023-09-04 17:54:21.127894500   File "/usr/lib/python3.8/site-packages/bleak/backends/scanner.py", line 253, in get_platform_scanner_backend_type
2023-09-04 17:54:21.127988500     from bleak.backends.bluezdbus.scanner import BleakScannerBlueZDBus
2023-09-04 17:54:21.127993500   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/scanner.py", line 6, in <module>
2023-09-04 17:54:21.127998500     from dbus_fast import Variant
2023-09-04 17:54:21.128001500   File "/usr/lib/python3.8/site-packages/dbus_fast/__init__.py", line 1, in <module>
2023-09-04 17:54:21.128070500     from . import aio, glib, introspection, message_bus, proxy_object, service
2023-09-04 17:54:21.128075500   File "/usr/lib/python3.8/site-packages/dbus_fast/aio/__init__.py", line 1, in <module>
2023-09-04 17:54:21.128080500     from .message_bus import MessageBus
2023-09-04 17:54:21.128083500   File "/usr/lib/python3.8/site-packages/dbus_fast/aio/message_bus.py", line 9, in <module>
2023-09-04 17:54:21.128110500     from .. import introspection as intr
2023-09-04 17:54:21.128113500   File "/usr/lib/python3.8/site-packages/dbus_fast/introspection.py", line 5, in <module>
2023-09-04 17:54:21.128118500     from .errors import InvalidIntrospectionError
2023-09-04 17:54:21.128121500   File "/usr/lib/python3.8/site-packages/dbus_fast/errors.py", line 57, in <module>
2023-09-04 17:54:21.128126500     from .message import Message
2023-09-04 17:54:21.128179500   File "src/dbus_fast/message.py", line 1, in init dbus_fast.message
2023-09-04 17:54:21.128184500 ValueError: dbus_fast._private.marshaller.Marshaller size changed, may indicate binary incompatibility. Expected 24 from C header, got 20 from PyObject
2023-09-04 17:54:21.128192500
2023-09-04 17:54:21.128193500 During handling of the above exception, another exception occurred:
2023-09-04 17:54:21.128197500
2023-09-04 17:54:21.128198500 Traceback (most recent call last):
2023-09-04 17:54:21.128223500   File "/usr/lib/python3.8/logging/__init__.py", line 1085, in emit
2023-09-04 17:54:21.128268500     msg = self.format(record)
2023-09-04 17:54:21.128270500   File "/usr/lib/python3.8/logging/__init__.py", line 929, in format
2023-09-04 17:54:21.128274500     return fmt.format(record)
2023-09-04 17:54:21.128277500   File "/usr/lib/python3.8/logging/__init__.py", line 668, in format
2023-09-04 17:54:21.128281500     record.message = record.getMessage()
2023-09-04 17:54:21.128369500   File "/usr/lib/python3.8/logging/__init__.py", line 373, in getMessage
2023-09-04 17:54:21.128399500     msg = msg % self.args
2023-09-04 17:54:21.128402500 TypeError: not all arguments converted during string formatting
2023-09-04 17:54:21.128406500 Call stack:
2023-09-04 17:54:21.128408500   File "/usr/lib/python3.8/threading.py", line 890, in _bootstrap
2023-09-04 17:54:21.128431500     self._bootstrap_inner()
2023-09-04 17:54:21.128433500   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
2023-09-04 17:54:21.128460500     self.run()
2023-09-04 17:54:21.128462500   File "/usr/lib/python3.8/threading.py", line 870, in run
2023-09-04 17:54:21.128466500     self._target(*self._args, **self._kwargs)
2023-09-04 17:54:21.128469500   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 80, in background_loop
2023-09-04 17:54:21.128474500     asyncio.run(self.bt_main_loop())
2023-09-04 17:54:21.128477500   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
2023-09-04 17:54:21.128502500     return loop.run_until_complete(main)
2023-09-04 17:54:21.128505500   File "/usr/lib/python3.8/asyncio/base_events.py", line 603, in run_until_complete
2023-09-04 17:54:21.128557500     self.run_forever()
2023-09-04 17:54:21.128559500   File "/usr/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
[... cut ...]
2023-09-04 17:54:21.128697500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-09-04 17:54:21.130092500 ERROR:SerialBattery:No BMS found at E0:9F:2A:xx:xx:xx
2023-09-04 17:54:21.130567500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-09-04 17:54:21.391216500
2023-09-04 17:54:21.391515500 INFO:Bluetooth details
2023-09-04 17:54:21.447466500 Device E0:9F:2A:xx:xx:xx not available
2023-09-04 17:54:26.512120500 Device E0:9F:2A:xx:xx:xx not available
```

mr-manuel, 2023-09-04:
Please try to install the `nightly` from the `dev` branch. I just pushed some changes.

tuxntoast, 2023-09-06:
got a build running off of the changes you just made. Seems to be running, will check back in a few days to see if the drive is able to remain stable running.

mr-manuel, 2023-09-12:
@tuxntoast were you able to check the logs? Can you give me a short update?

wherzig, 2023-09-12:
> @beliffm @iLeeeZi @wherzig can anyone with the problem contact me on Discord? This would help me to find relevant informations.

Sorry, I can not help to get additional information at this time. I have changed back to 0.14.3 . This works very well.

In about two weeks: I will remove my old JBD-BMS and replace it with a new JK BMS (with display and 1A active balancer). At the changing date, I can try the new version before the swap with my JBD BMS and give feedback.

beliffm, 2023-09-12:
Sorry you guys, i did not have the time to look further into the matter until just now.
Still no dice with v1.0.20230905dev. After install the driver loops without finding either of my two BMSes.
So i reverted to v1.0.20230531 and the BMSes were found again instantly.
Ich will attach the log files to this message.
Both BMSes are attached via DC-isolated serial adapters. 
There is another issue with my installation that i allready mentioned earlier. After some minutes the serial connection is lost independantly on both BMSes but they are reconnected at once.
My guess is, that my version of the LLtJbd utilizes a somewhat different message scheme and that might lead to loss of synchronisation. But that is just a guess.

Again, appologies for reacting so late.

[masterUSB1.log](https://github.com/Louisvdw/dbus-serialbattery/files/12590183/masterUSB1.log)
[devUSB2.log](https://github.com/Louisvdw/dbus-serialbattery/files/12590184/devUSB2.log)
[masterUSB2.log](https://github.com/Louisvdw/dbus-serialbattery/files/12590185/masterUSB2.log)
[devUSB1.log](https://github.com/Louisvdw/dbus-serialbattery/files/12590186/devUSB1.log)

mr-manuel, 2023-09-12:
In this case we have to wait for this https://github.com/Louisvdw/dbus-serialbattery/issues/730#issuecomment-1712522694

iLeeeZi, 2023-09-15:
@idstein @mr-manuel I found the cause of my problem. It is because of PR #760, and differences in jbd bms firmwares. There is two settings in jbd for battery capacity, total capacity and cycle capacity. For me total capacity is 272ah and cycle capacity 217,6ah so 80% of total capacity (I usually set cycle capacity to 272ah but for some reason the bms resets it sometimes back to 217,6ah).

It seems that some bms versions use cycle capacity to calculate the soc internally, but mine uses total capacity. After that PR the driver can show soc values over 100%, so in my case if bms has 100% soc then the driver shows it as 125%. In current state the driver fails to start if its calculated soc is over 100%.

Maybe the easiest way to detect which capacity the bms uses for soc, is to use the soc from bms when driver connects and after that test if it matches to soc calculated from cycle capacity, or soc calculated from total capacity

mr-manuel, 2023-09-21:
@iLeeeZi can you try to change

https://github.com/Louisvdw/dbus-serialbattery/blob/fea0effc7baa31122055572ae53789dcab2a7230/etc/dbus-serialbattery/bms/lltjbd.py#L555

to

```python
        if not self.cycle_capacity or self.cycle_capacity < capacity_remain:
```

mr-manuel, 2023-09-25:
@beliffm thats for your feedback. I merged the fix.

commit 60bdde9a (2023-09-25): Merge pull request #821 from mr-manuel/dev

fixes #769
closed 2023-09-25 by mr-manuel