Louisvdw/dbus-serialbattery #777: JBD BMS very unstable bluetooth connection

Marvo2011, 2023-08-13:
### Describe the bug

When I connect a JBD BMS over bluetooth the connection drops randomly, all charger and loads stoped working.

### How to reproduce

See log....

### Expected behavior

BMS / driver should reconnect, for the moment I have to restart the driver.

### Driver version

1.0.20230531 (also latest dev nightly)

### Venus OS device type

Raspberry Pi

### Venus OS version

3.10~2

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

BLUETOOTH_BMS = LltJbd_Ble A4:C1:37:21:62:96
BLOCK_ON_DISCONNECT = False
```

### Relevant log output

```shell
@4000000064d75d423321dddc   File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 497, in run
@4000000064d75d4233e93904     super(MainLoop, self).run()
@4000000064d75d433387854c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 155, in <lambda>
@4000000064d75d433387a874     gobject.timeout_add(battery.poll_interval, lambda: poll_battery(mainloop))
@4000000064d75d433387b814   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 67, in poll_battery
@4000000064d75d4333880e04     helper.publish_battery(loop)
@4000000064d75d43338819bc   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 361, in publish_battery
@4000000064d75d433389717c     success = self.battery.refresh_data()
@4000000064d75d433389811c   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 292, in refresh_data
@4000000064d75d43338994a4     result = self.read_gen_data()
@4000000064d75d4333899c74   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 382, in read_gen_data
@4000000064d75d433389affc     gen_data = self.read_serial_data_llt(self.command_general)
@4000000064d75d43338a158c   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 165, in read_serial_data_llt
@4000000064d75d43338a2cfc     data = asyncio.run(self.async_read_serial_data_llt(command))
@4000000064d75d43338a3c9c   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
@4000000064d75d43338a4c3c     return loop.run_until_complete(main)
@4000000064d75d43338a57f4   File "/usr/lib/python3.8/asyncio/base_events.py", line 603, in run_until_complete
@4000000064d75d43338b175c     self.run_forever()
@4000000064d75d43338b1f2c   File "/usr/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
@4000000064d75d43338b32b4     self._run_once()
@4000000064d75d43338b3a84   File "/usr/lib/python3.8/asyncio/base_events.py", line 1859, in _run_once
@4000000064d75d43338b4a24     handle._run()
@4000000064d75d43338b51f4   File "/usr/lib/python3.8/asyncio/events.py", line 81, in _run
@4000000064d75d43338bb39c     self._context.run(self._callback, *self._args)
@4000000064d75d43338bc33c   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 158, in async_read_serial_data_llt
@4000000064d75d43338bdaac     logger.error(">>> ERROR: No reply - returning", e)
@4000000064d75d43338be664 Message: '>>> ERROR: No reply - returning'
@4000000064d75d43338bf604 Arguments: (BleakDBusError('org.bluez.Error.InProgress', 'In Progress'),)
@4000000064d75d43338c53c4 --- Logging error ---
@4000000064d75d43338c5f7c Traceback (most recent call last):
@4000000064d75d43338c674c   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 152, in async_read_serial_data_llt
@4000000064d75d43338c7ebc     result = await asyncio.wait_for(asyncio.wrap_future(bt_task), 20)
@4000000064d75d43338c8e5c   File "/usr/lib/python3.8/asyncio/tasks.py", line 494, in wait_for
@4000000064d75d43338cf004     return fut.result()
@4000000064d75d43338cf7d4   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 138, in send_command
@4000000064d75d43338da3b4     await self.bt_client.start_notify(BLE_CHARACTERISTICS_RX_UUID, rx_collector)
@4000000064d75d43338db73c   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 717, in start_notify
@4000000064d75d43338dcac4     await self._backend.start_notify(characteristic, wrapped_callback, **kwargs)
@4000000064d75d43338dde4c   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 1003, in start_notify
@4000000064d75d43338e47c4     assert_reply(reply)
@4000000064d75d43338e537c   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/utils.py", line 22, in assert_reply
@4000000064d75d43338e6704     raise BleakDBusError(reply.error_name, reply.body)
@4000000064d75d43338e76a4 bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] In Progress
@4000000064d75d43338e8644 
@4000000064d75d43338e8a2c During handling of the above exception, another exception occurred:
@4000000064d75d43338eebd4 
@4000000064d75d43338eefbc Traceback (most recent call last):
@4000000064d75d43338efb74   File "/usr/lib/python3.8/logging/__init__.py", line 1085, in emit
@4000000064d75d43338f0b14     msg = self.format(record)
@4000000064d75d43338f16cc   File "/usr/lib/python3.8/logging/__init__.py", line 929, in format
@4000000064d75d43338f266c     return fmt.format(record)
@4000000064d75d43338f2e3c   File "/usr/lib/python3.8/logging/__init__.py", line 668, in format
@4000000064d75d43339089e4     record.message = record.getMessage()
@4000000064d75d433391417c   File "/usr/lib/python3.8/logging/__init__.py", line 373, in getMessage
@4000000064d75d433391511c     msg = msg % self.args
@4000000064d75d4333915cd4 TypeError: not all arguments converted during string formatting
@4000000064d75d433391688c Call stack:
@4000000064d75d433391705c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 163, in <module>
@4000000064d75d43339183e4     main()
@4000000064d75d4333918bb4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 157, in main
@4000000064d75d433391f144     mainloop.run()
@4000000064d75d433391fcfc   File "/usr/lib/python3.8/site-packages/gi/overrides/GLib.py", line 497, in run
@4000000064d75d4333920c9c     super(MainLoop, self).run()
@4000000064d75d4333921854   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 155, in <lambda>
@4000000064d75d4333922bdc     gobject.timeout_add(battery.poll_interval, lambda: poll_battery(mainloop))
@4000000064d75d433392899c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 67, in poll_battery
@4000000064d75d433392a10c     helper.publish_battery(loop)
@4000000064d75d433392acc4   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 361, in publish_battery
@4000000064d75d433392c04c     success = self.battery.refresh_data()
@4000000064d75d433392cc04   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 292, in refresh_data
@4000000064d75d43339377e4     result = self.read_gen_data()
@4000000064d75d433393839c   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd.py", line 382, in read_gen_data
@4000000064d75d4333939724     gen_data = self.read_serial_data_llt(self.command_general)
@4000000064d75d433393a6c4   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 165, in read_serial_data_llt
@4000000064d75d433393ba4c     data = asyncio.run(self.async_read_serial_data_llt(command))
@4000000064d75d4333941bf4   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
@4000000064d75d4333942b94     return loop.run_until_complete(main)
@4000000064d75d433394374c   File "/usr/lib/python3.8/asyncio/base_events.py", line 603, in run_until_complete
@4000000064d75d4333944ad4     self.run_forever()
@4000000064d75d43339452a4   File "/usr/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
@4000000064d75d433394662c     self._run_once()
@4000000064d75d433394b834   File "/usr/lib/python3.8/asyncio/base_events.py", line 1859, in _run_once
@4000000064d75d433394cbbc     handle._run()
@4000000064d75d433394d38c   File "/usr/lib/python3.8/asyncio/events.py", line 81, in _run
@4000000064d75d433394e32c     self._context.run(self._callback, *self._args)
@4000000064d75d433394eee4   File "/opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py", line 158, in async_read_serial_data_llt
@4000000064d75d4333950654     logger.error(">>> ERROR: No reply - returning", e)
@4000000064d75d4333960054 Message: '>>> ERROR: No reply - returning'
@4000000064d75d4333960c0c Arguments: (BleakDBusError('org.bluez.Error.InProgress', 'In Progress'),)
@4000000064d75d4333994c14 
@4000000064d75d43339953e4 #0
@4000000064d75d43339957cc #1
@4000000064d75d4333995bb4 #2
@4000000064d75d4333996384 #3
@4000000064d75d433399676c #4
@4000000064d75d4333996b54 #5
@4000000064d75d4333996f3c #6
@4000000064d75d433399770c #7
```

mr-manuel, 2023-08-13:
Please try also the nightly build from the dev branch.

Marvo2011, 2023-08-13:
@mr-manuel I have tried it yesterday. Also the same result.

mr-manuel, 2023-08-13:
Also identical logs?

Marvo2011, 2023-08-13:
Yes log output is identical.

mr-manuel, 2023-09-04:
Please provide the logs in the correct format as documented and requested. This way we cannot help you.

mr-manuel, 2023-09-04:
Probably related to https://github.com/Louisvdw/dbus-serialbattery/issues/720. @idstein maybe you can also take a look at this, if you find time?

mr-manuel, 2023-09-04:
Please try to install the `nightly` from the `dev` branch. I just pushed some changes.

mr-manuel, 2023-09-05:
Some more logs for this issue:

```bash
2023-09-05 08:33:34.177824500 INFO:SerialBattery:
2023-09-05 08:33:34.177829500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-05 08:33:34.179688500 INFO:SerialBattery:dbus-serialbattery v1.0.20230905dev
2023-09-05 08:33:34.548151500 INFO:SerialBattery:Init of LltJbd_Ble at E0:9F:2A:xx:xx:xx
2023-09-05 08:33:34.548601500 INFO:SerialBattery:Test of LltJbd_Ble at E0:9F:2A:xx:xx:xx
2023-09-05 08:33:40.794466500 INFO:SerialBattery:Connection established to LltJbd_Ble
2023-09-05 08:33:40.794948500 INFO:SerialBattery:Battery LltJbd_Ble connected to dbus from blee09f2axxxxxx
2023-09-05 08:33:40.795353500 INFO:SerialBattery:========== Settings ==========
2023-09-05 08:33:40.795831500 INFO:SerialBattery:> Connection voltage: 13.07V | Current: -2.72A | SoC: 17.06%
2023-09-05 08:33:40.796247500 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
2023-09-05 08:33:40.796667500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-09-05 08:33:40.797549500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 75.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2023-09-05 08:33:40.798006500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 230.0A | MAX BATTERY DISCHARGE CURRENT: 230.0A (read from BMS)
2023-09-05 08:33:40.798421500 INFO:SerialBattery:> CVCM:     True
2023-09-05 08:33:40.798858500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.5V
2023-09-05 08:33:40.799289500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-09-05 08:33:40.799713500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-09-05 08:33:40.800232500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-09-05 08:33:40.800740500 INFO:SerialBattery:Serial Number/Unique Identifier: 80_272.0Ah
2023-09-05 08:33:40.853056500 INFO:SerialBattery:DeviceInstance = 6
2023-09-05 08:33:40.853660500 INFO:SerialBattery:com.victronenergy.battery.blee09f2axxxxxx
2023-09-05 08:33:41.661966500 INFO:SerialBattery:publish config values = 0
2023-09-05 09:18:06.212244500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-09-05 09:18:15.706683500 ERROR:SerialBattery:Exception occurred: BleakDBusError('org.freedesktop.DBus.Error.NoReply', 'Message recipient disconnected from message bus without replying') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #176
2023-09-05 09:18:15.752904500 ERROR:SerialBattery:Exception occurred: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #176
2023-09-05 09:18:15.783531500 [CHG] Device E0:9F:2A:xx:xx:xx RSSI: -53
2023-09-05 09:18:15.783539500 [CHG] Device E0:9F:2A:xx:xx:xx RSSI: -53
2023-09-05 09:18:15.783543500 [CHG] Device E0:9F:2A:xx:xx:xx RSSI: -53
2023-09-05 09:18:16.730540500 ERROR:SerialBattery:Exception occurred: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #176
2023-09-05 09:18:17.730947500 ERROR:SerialBattery:Exception occurred: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #176
2023-09-05 09:18:18.731835500 ERROR:SerialBattery:Exception occurred: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #176
2023-09-05 09:18:19.732333500 ERROR:SerialBattery:Exception occurred: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #176
2023-09-05 09:18:20.737443500 ERROR:SerialBattery:Exception occurred: BleakDBusError('org.freedesktop.DBus.Error.ServiceUnknown', 'The name org.bluez was not provided by any .service files') of type <class 'bleak.exc.BleakDBusError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #176
```

After that I have to run this commands to get it working again:

```bash
# stop bluetooth system service
/etc/init.d/bluetooth stop

# kill all blebattery processes
pkill -f "supervise dbus-blebattery.*"
pkill -f "multilog .* /var/log/dbus-blebattery.*"
pkill -f "python .*/dbus-serialbattery.py .*_Ble"

# start bluetooth system service
/etc/init.d/bluetooth start
```

mr-manuel, 2023-09-05:
Unfortunately not. I can only wait for it to happen and then add some code for error handling.

idstein, 2023-09-06:
Okay, so I now experienced also a very unpleasant connection issue with the BMS. Looking and hcidump output and a simple `hcitool lecc <mac>` (= BLE connect), I can now state that the JBDBT02 device it self seems to have some sort of blacklistening or block mechanism. I could not re-connect from my VenusOS to the device, but could without issues use from another device a BT BMS app.

So I tried to change the MAC address of the hci0 adapter in my RPI and surprise, it can connect again.

BTW unplugging the dongle from the BMS board or rebooting the entire BMS board does not seem to resolve the issue. There seems to be a hardcoded blacklist timeout of at least 24 hours.

Marvo2011, 2023-09-14:
> Okay, so I now experienced also a very unpleasant connection issue with the BMS. Looking and hcidump output and a simple `hcitool lecc <mac>` (= BLE connect), I can now state that the JBDBT02 device [...]

That sounds strange, for me a simple restart of the driver makes a working reconnect.

pieterjanvz, 2023-09-17:
below the BLE log from my JBD BMS:

2023-09-17 10:55:46.391618500
2023-09-17 10:55:46.391623500 INFO:Bluetooth details
2023-09-17 10:55:46.426172500 Device A4:C1:38:C0:53:90 not available
2023-09-17 10:55:51.493111500 Device A4:C1:38:C0:53:90 (public)
2023-09-17 10:55:51.493118500   Alias: Accu 1
2023-09-17 10:55:51.493120500   Paired: no
2023-09-17 10:55:51.493122500   Trusted: no
2023-09-17 10:55:51.493123500   Blocked: no
2023-09-17 10:55:51.493125500   Connected: no
2023-09-17 10:55:51.493127500   LegacyPairing: no
2023-09-17 10:55:51.493129500   RSSI: -61
2023-09-17 10:55:51.494420500
2023-09-17 10:55:53.289267500 INFO:SerialBattery:
2023-09-17 10:55:53.289687500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-17 10:55:53.290617500 INFO:SerialBattery:dbus-serialbattery v1.0.20230905dev
2023-09-17 10:55:53.732126500 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:38:C0:53:90
2023-09-17 10:55:53.732546500 INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:38:C0:53:90
2023-09-17 10:55:53.882705500 ERROR:SerialBattery:BleakScanner(): Exception occurred: ValueError('dbus_fast._private.marshaller.Marshaller size changed, may indicate binary incompatibility. Expected 24 from C header, got 20 from PyObject') of type <class 'ValueError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
2023-09-17 10:55:58.743707500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-09-17 10:55:58.746020500 ERROR:SerialBattery:No BMS found at A4:C1:38:C0:53:90
2023-09-17 10:55:58.746445500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-09-17 10:55:59.636611500
2023-09-17 10:55:59.636744500 INFO:Bluetooth details
2023-09-17 10:55:59.719706500 Attempting to disconnect from A4:C1:38:C0:53:90
2023-09-17 10:55:59.719714500 [CHG] Device FE:7C:93:F9:42:6C RSSI is nil
2023-09-17 10:55:59.719718500 [CHG] Device CD:26:4D:CE:82:AE RSSI is nil
2023-09-17 10:55:59.719721500 [CHG] Device A4:C1:37:10:26:0D RSSI is nil
2023-09-17 10:55:59.719725500 [CHG] Device D4:F1:2E:C1:77:33 RSSI is nil
2023-09-17 10:55:59.719729500 [CHG] Device FD:44:C9:E4:C6:E5 RSSI is nil
2023-09-17 10:55:59.719732500 [CHG] Device A4:C1:38:C0:53:90 RSSI is nil
2023-09-17 10:55:59.719819500 [CHG] Device A4:C1:38:E4:CF:B1 RSSI is nil
2023-09-17 10:55:59.719823500 [CHG] Device A4:C1:38:00:2F:44 RSSI is nil
2023-09-17 10:55:59.719827500 [CHG] Device F5:47:C8:BF:C3:69 RSSI is nil
2023-09-17 10:55:59.719830500 [CHG] Controller B8:27:EB:0E:BC:1D Discovering: no
2023-09-17 10:55:59.719834500 Successful disconnected

mr-manuel, 2023-09-17:
@pieterjanvz can you try to change this line

https://github.com/Louisvdw/dbus-serialbattery/blob/7c5f1c774a014c8bdb1f595754e7a9b8ac834410/etc/dbus-serialbattery/reinstall-local.sh#L190

to

```bash
    pip3 install bleak==0.20.2
```

and then run `reinstall-local.sh` again?

If you still have the same error add

```bash
    pip3 install dbus-fast==1.87.0
```

below

https://github.com/Louisvdw/dbus-serialbattery/blob/7c5f1c774a014c8bdb1f595754e7a9b8ac834410/etc/dbus-serialbattery/reinstall-local.sh#L190

and then run `reinstall-local.sh` again.

pieterjanvz, 2023-09-17:
Thanks , that fixed it

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows

https://github.com/Louisvdw/dbus-serialbattery/blob/7c5f1c774a014c8bdb1f595754e7a9b8ac834410/etc/dbus-serialbattery/reinstall-local.sh#L190

https://github.com/Louisvdw/dbus-serialbattery/blob/7c5f1c774a014c8bdb1f595754e7a9b8ac834410/etc/dbus-serialbattery/reinstall-local.sh#L190

pieterjanvz, 2023-09-17:
Thanks I now can connect , but as above its not stable , hope this wil be solved soon

mr-manuel, 2023-09-17:
Did you downgrade only `bleak` or also `dbus-fast`? This is or a bug in the code or a bug in the module.

Marvo2011, 2023-09-19:
Today I have received my repaired MultiPlus, I have installed the nightly dev. I try to reproduce the issue and will post the logs.

Marvo2011, 2023-09-19:
Here the actual log output:

```
root@raspberrypi2:/data/log/dbus-blebattery.0# cat current 
@400000006509a8a62c16c034 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006509a8a62c3661dc ERROR:SerialBattery:No BMS found at A4:C1:37:21:62:96
@400000006509a8a62c3f81cc ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@400000006509a8a735a7215c 
@400000006509a8a735aba984 INFO:Bluetooth details
@400000006509a8a7375b8c04 Attempting to disconnect from A4:C1:37:21:62:96
@400000006509a8a7375ba75c Successful disconnected
@400000006509a8ac3a1fd83c Device A4:C1:37:21:62:96 (public)
@400000006509a8ac3a1ffb64 	Alias: Wohnwagen
@400000006509a8ac3a200eec 	Paired: no
@400000006509a8ac3a201aa4 	Trusted: yes
@400000006509a8ac3a202274 	Blocked: no
@400000006509a8ac3a202a44 	Connected: no
@400000006509a8ac3a203214 	LegacyPairing: no
@400000006509a8ac3a2039e4 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@400000006509a8ac3a45d6cc 
@400000006509a8ae0317d4fc INFO:SerialBattery:
@400000006509a8ae032146f4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006509a8ae033069f4 INFO:SerialBattery:dbus-serialbattery v1.0.20230919dev
@400000006509a8ae1937d944 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:21:62:96
@400000006509a8ae1941995c INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:37:21:62:96
@400000006509a8ae2e8d790c ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@400000006509a8b31a1aea2c ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006509a8b31a1b0584 ERROR:SerialBattery:No BMS found at A4:C1:37:21:62:96
@400000006509a8b31a1b1524 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@400000006509a8b4207f6794 
@400000006509a8b4208372bc INFO:Bluetooth details
@400000006509a8b4224a1d44 Attempting to disconnect from A4:C1:37:21:62:96
@400000006509a8b4224a389c Successful disconnected
@400000006509a8b924d538dc Device A4:C1:37:21:62:96 (public)
@400000006509a8b924d55434 	Alias: Wohnwagen
@400000006509a8b924d55c04 	Paired: no
@400000006509a8b924d563d4 	Trusted: yes
@400000006509a8b924d567bc 	Blocked: no
@400000006509a8b924d56f8c 	Connected: no
@400000006509a8b924d5775c 	LegacyPairing: no
@400000006509a8b924d57f2c 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@400000006509a8b9250af6ac 
@400000006509a8ba306d6854 INFO:SerialBattery:
@400000006509a8ba30770544 INFO:SerialBattery:Starting dbus-serialbattery
@400000006509a8ba30864f54 INFO:SerialBattery:dbus-serialbattery v1.0.20230919dev
@400000006509a8bb0b489684 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:21:62:96
@400000006509a8bb0b53392c INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:37:21:62:96
@400000006509a8bb22c396ec ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
@400000006509a8c00c306a2c ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
@400000006509a8c00c308584 ERROR:SerialBattery:No BMS found at A4:C1:37:21:62:96
@400000006509a8c00c30913c ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
@400000006509a8c1151d20e4 
@400000006509a8c1152137c4 INFO:Bluetooth details
@400000006509a8c116dd4f44 Attempting to disconnect from A4:C1:37:21:62:96
@400000006509a8c116dd6a9c Successful disconnected
@400000006509a8c6199b65a4 Device A4:C1:37:21:62:96 (public)
@400000006509a8c6199b7d14 	Alias: Wohnwagen
@400000006509a8c6199b84e4 	Paired: no
@400000006509a8c6199b8cb4 	Trusted: yes
@400000006509a8c6199b909c 	Blocked: no
@400000006509a8c6199b986c 	Connected: no
@400000006509a8c6199ba03c 	LegacyPairing: no
@400000006509a8c6199ba80c 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
@400000006509a8c619c0fea4 
@400000006509a8c72825560c INFO:SerialBattery:
@400000006509a8c72834cb14 INFO:SerialBattery:Starting dbus-serialbattery
@400000006509a8c72849aebc INFO:SerialBattery:dbus-serialbattery v1.0.20230919dev
@400000006509a8c805cfa1fc INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:21:62:96
@400000006509a8c805da8edc INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:37:21:62:96
@400000006509a8c81c23163c ERROR:SerialBattery:BleakScanner(): Exception occurred: BleakError('No powered Bluetooth adapters found.') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/lltjbd_ble.py line #59
```

Marvo2011, 2023-09-19:
Yes after a reboot, it works a few hours than the issue comes back.

Marvo2011, 2023-09-20:
Some interesting details:

After reboot:
```
root@raspberrypi2:~# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: 43:45:C0:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:6576 acl:145 sco:0 events:255 errors:0
	TX bytes:3985 acl:110 sco:0 commands:154 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'Raspberry Pi 3 Model B Plus Rev 1.3 HQ0000AAAAA'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 5.0 (0x9)  Revision: 0x156
	LMP Version: 5.0 (0x9)  Subversion: 0x6119
	Manufacturer: Cypress Semiconductor (305)
```

After first crash:
```
root@raspberrypi2:~# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: 43:45:C0:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:1471 acl:0 sco:0 events:83 errors:0
	TX bytes:1228 acl:0 sco:0 commands:83 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
```

I think it's related to the hardware -> Raspberry Pi 3 Model B Plus Rev 1.3

mr-manuel, 2023-09-20:
We had this already in the past. Can you try this routine, if it happen again and see, it it helps without rebooting?

```bash
# stop bluetooth system service
/etc/init.d/bluetooth stop
pkill -f "bluetoothd"

# kill bluetooth system stack
rfkill block bluetooth
rfkill unblock bluetooth

# kill all blebattery processes
pkill -f "supervise dbus-blebattery.*"
pkill -f "multilog .* /var/log/dbus-blebattery.*"
pkill -f "python .*/dbus-serialbattery.py .*_Ble.*"

# start bluetooth system service
/etc/init.d/bluetooth start
```

Marvo2011, 2023-09-20:
Does not work, only the full reboot.

Marvo2011, 2023-09-21:
I find out the following works without a reboot:

```
# stop bluetooth system service
/etc/init.d/bluetooth stop
pkill -f "bluetoothd"

# kill hciattach
pkill -f "hciattach"

# kill bluetooth system stack
rfkill block bluetooth
rfkill unblock bluetooth

# kill all blebattery processes
pkill -f "supervise dbus-blebattery.*"
pkill -f "multilog .* /var/log/dbus-blebattery.*"
pkill -f "python .*/dbus-serialbattery.py .*_Ble.*"

# restart kernel module
rmmod hci_uart
rmmod btbcm
modprobe hci_uart
modprobe btbcm

# start bluetooth system service
/etc/init.d/bluetooth start

# start hciattach
/usr/bin/hciattach -t 10 /dev/ttyS0 bcm43xx 921600 noflow - &
```

I will test again if a hciattach restart is sufficient.

mr-manuel, 2023-09-21:
Sounds great! So we can integrate this in the driver that it restarts the stack in the case of a crash.

Marvo2011, 2023-09-21:
```
# kill hciattach
pkill -f "hciattach"

# restart kernel module
rmmod hci_uart
rmmod btbcm
modprobe hci_uart
modprobe btbcm

# start hciattach
/usr/bin/hciattach -t 10 /dev/ttyS0 bcm43xx 921600 noflow - &
```

Is sufficient, also brings the device in dbus-serialbattery back. 
I think its good to search before with `ps | grep hciattach` to got the correct restart command.

mr-manuel, 2023-09-21:
Could you script that?

Marvo2011, 2023-09-21:
Yes, I will create a PR

randomname32, 2023-11-14:
> ```
> # kill hciattach
> pkill -f "hciattach"
>
> # restart kernel module
> rmmod hci_uart
> rmmod btbcm
> modprobe hci_uart
> modprobe btbcm
>
> # start hciattach
> /usr/bin/hciattach -t 10 /dev/ttyS0 bcm43xx 921600 [...]

I have the same issue with Bluetooth randomly being down on my Raspberry Pi 3 Model B Plus Rev 1.3. 
This solution worked for me. Thanks!

idstein, 2023-12-04:
Tried that solution on my RPI4 fails unfortunately. The hciattach fails for me with a timeout.

Marvo2011, 2023-12-05:
I think it's hardware / raspberry firmware related. I will try to use a usb bluetooth device for a better connection.

mr-manuel, 2023-12-05:
I think it's a software bug of bleak. We had this a year ago and it doesn't changed on using an external USB adapter.

mr-manuel, 2023-12-05:
Here you can find more detailed info https://github.com/Louisvdw/dbus-serialbattery/pull/372

Marvo2011, 2023-12-11:
I got it working with an USB bluetooth module. There are some needed requirements, the usb module must support BLE (bluetooth version > 4.0). Then some changes to bluetooth setting from the pi.

```
# hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: XX:XX:XX:XX:XX:XX  ACL MTU: 1021:5  SCO MTU: 255:11
	UP RUNNING INQUIRY 
	RX bytes:11233162 acl:498440 sco:0 events:350109 errors:0
	TX bytes:5036196 acl:332323 sco:0 commands:14255 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'BlueZ 5.55'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 5.1 (0xa)  Revision: 0xca9
	LMP Version: 5.1 (0xa)  Subversion: 0x8a6b
	Manufacturer: Realtek Semiconductor Corporation (93)
```

I will try it over some days, when it's stable I will create some code which can setup the needed requirements to the pi settings by enabled it in the config.ini.

Marvo2011, 2023-12-13:
@mr-manuel please see pull request #890 
----
@idstein can you try if it also stable for you? 
@randomname32 can you try if it also stable for you?

randomname32, 2023-12-13:
I just did a manual test by running
```
root@raspberrypi2:~# hciconfig hci0 down
root@raspberrypi2:~# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: 43:45:C0:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:165339 acl:5681 sco:0 events:3952 errors:0
	TX bytes:63034 acl:3431 sco:0 commands:933 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
```
And then waiting for the code from @Marvo2011 to bring BT up again. 
BT comes back up indeed. (Raspberry 3B+ with internal Bluetooth)

But the connection to the JBD doesn't get established directly afterwards. So the general problem of dropped connection still persists :-/
The connection came back 13 minutes later.

Here's the log of the manual test with hciconfig hci0 down. Please note that I modified the python code to output more logging (all the lines starting with ####). And I put more sleep commands between the shell commands. So the line references you see in the logfile don't match the code here on github.

Around 22:03:57 it looks like there are 2 threads running at the same time, where one is doing the reset and the other is trying to connect to the BMS. This doesn't seem to be useful I guess.

<details>

<summary>output of "tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal"</summary>

```
2023-12-13 22:03:17.466078500 Device A4:C1:37:52:3E:9D (public)
2023-12-13 22:03:17.466084500 	Alias: JBD-AP21S002
2023-12-13 22:03:17.466087500 	Paired: no
2023-12-13 22:03:17.466088500 	Trusted: yes
2023-12-13 22:03:17.466090500 	Blocked: no
2023-12-13 22:03:17.466092500 	Connected: no
2023-12-13 22:03:17.466094500 	LegacyPairing: no
2023-12-13 22:03:17.466096500 	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
2023-12-13 22:03:17.472243500 
2023-12-13 22:03:18.626321500 INFO:SerialBattery:
2023-12-13 22:03:18.627048500 INFO:SerialBattery:Starting dbus-serialbattery
2023-12-13 22:03:18.628251500 INFO:SerialBattery:dbus-serialbattery v1.0.20230921dev
2023-12-13 22:03:19.006293500 ERROR:SerialBattery:###### reading hciattach via ps -ww
2023-12-13 22:03:19.085133500 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:52:3E:9D
2023-12-13 22:03:19.085140500 INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:37:52:3E:9D
2023-12-13 22:03:19.470260500 ERROR:SerialBattery:###### Reset of hci_uart stack... Reconnecting to: A4:C1:37:52:3E:9D
2023-12-13 22:03:19.470968500 ERROR:SerialBattery:###### pkill -f hciattach
2023-12-13 22:03:24.094955500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
2023-12-13 22:03:24.097285500 ERROR:SerialBattery:No BMS found at A4:C1:37:52:3E:9D
2023-12-13 22:03:24.097726500 ERROR:SerialBattery:ERROR >>> No battery connection at LltJbd_Ble
2023-12-13 22:03:24.532606500 ERROR:SerialBattery:###### rmmod
2023-12-13 22:03:29.671563500 ERROR:SerialBattery:###### modprobe
2023-12-13 22:03:34.735265500 ERROR:SerialBattery:###### exit
2023-12-13 22:03:35.002376500 
2023-12-13 22:03:35.002640500 INFO:Bluetooth details
2023-12-13 22:03:35.023819500 No default controller available
2023-12-13 22:03:40.070866500 
2023-12-13 22:03:41.171141500 INFO:SerialBattery:
2023-12-13 22:03:41.171825500 INFO:SerialBattery:Starting dbus-serialbattery
2023-12-13 22:03:41.172934500 INFO:SerialBattery:dbus-serialbattery v1.0.20230921dev
2023-12-13 22:03:41.558386500 ERROR:SerialBattery:###### reading hciattach via ps -ww
2023-12-13 22:03:41.631735500 ERROR:SerialBattery:###### reading hciattach did not succeed. Executing from file
2023-12-13 22:03:41.631743500 ERROR:SerialBattery:###### execute: /usr/bin/hciattach -t 10 /dev/ttyS0 bcm43xx 921600 noflow -
2023-12-13 22:03:51.641486500 Initialization timed out.
2023-12-13 22:03:51.641491500 bcm43xx_init
2023-12-13 22:03:51.642708500 ERROR:SerialBattery:###### sleep 5 seconds
2023-12-13 22:03:56.651875500 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:52:3E:9D
2023-12-13 22:03:56.651882500 INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:37:52:3E:9D
2023-12-13 22:03:57.009526500 ERROR:SerialBattery:###### Reset of hci_uart stack... Reconnecting to: A4:C1:37:52:3E:9D
2023-12-13 22:03:57.010289500 ERROR:SerialBattery:###### pkill -f hciattach
2023-12-13 22:04:01.664373500 ERROR:SerialBattery:>>> ERROR: Unable to connect with BLE device
[... 71 lines cut ...]
2023-12-13 22:16:50.551060500 INFO:SerialBattery:
2023-12-13 22:16:50.551471500 INFO:SerialBattery:Starting dbus-serialbattery
2023-12-13 22:16:50.552380500 INFO:SerialBattery:dbus-serialbattery v1.0.20230921dev
2023-12-13 22:16:50.935373500 ERROR:SerialBattery:###### reading hciattach via ps -ww
2023-12-13 22:16:51.006741500 INFO:SerialBattery:Init of LltJbd_Ble at A4:C1:37:52:3E:9D
2023-12-13 22:16:51.007212500 INFO:SerialBattery:Test of LltJbd_Ble at A4:C1:37:52:3E:9D
2023-12-13 22:16:58.876022500 INFO:SerialBattery:Connection established to LltJbd_Ble
2023-12-13 22:16:58.876495500 INFO:SerialBattery:Battery LltJbd_Ble connected to dbus from blea4c137523e9d
2023-12-13 22:16:58.876896500 INFO:SerialBattery:========== Settings ==========
2023-12-13 22:16:58.877372500 INFO:SerialBattery:> Connection voltage: 57.77V | Current: 0.0A | SoC: 12.46%
2023-12-13 22:16:58.877787500 INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
2023-12-13 22:16:58.878223500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
[... cut ...]
2023-12-13 22:16:58.881986500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-12-13 22:16:58.882602500 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
2023-12-13 22:16:58.883180500 INFO:SerialBattery:Serial Number/Unique Identifier: 52_280.0Ah
2023-12-13 22:16:58.930026500 INFO:SerialBattery:DeviceInstance = 2
2023-12-13 22:16:58.930662500 INFO:SerialBattery:com.victronenergy.battery.blea4c137523e9d
2023-12-13 22:17:00.699714500 INFO:SerialBattery:publish config values = 1
```
</details>

Marvo2011, 2023-12-14:
@randomname32 Yes we now it takes a lot of time to bring it back. I pushed code for usage of an usb module, can you test this? https://github.com/Louisvdw/dbus-serialbattery/pull/890

mr-manuel, 2023-12-14:
@randomname32 I have a PR for this in the queue, where the battery doesn't disconnect it the cell voltages are in the middle area.

randomname32, 2023-12-14:
> @randomname32 Yes we now it takes a lot of time to bring it back. I pushed code for usage of an usb module, can you test this? #890

I now tested a Bluetooth USB dongle on the RPi. I did not use the code from your PR, but I did the required changes manually (dtoverlay=disable-bt in u-boot/config.txt). My dbus-serialbattery version is v1.0.20231128dev (did a fresh install before testing).

I noticed 2 things:
- the connection to the BMS was established immediately. With the internal BT it took a lot of tries until the connection succeeded
- After I ran `hciconfig hci0 down` and a few seconds later `hciconfig hci0 up`, the connection recovered within less than minute, without any failed attempts.

I will have an eye on this and see if I get random connection issues or the bt device being down.

Marvo2011, 2023-12-14:
> I now tested a Bluetooth USB dongle on the RPi. I did not use the code from your PR, but I did the required changes manually (dtoverlay=disable-bt in u-boot/config.txt). My dbus-serialbattery [...]

Nice sounds good, for me it's stable for 5 days now.

Marvo2011, 2023-12-17:
#890 with usb bluetooth module fixed it.

closed 2023-12-17 by Marvo2011

randomname32, 2023-12-19:
For me, the bluetooth connection has also been stable for 5 days now.
I just got 2 error messages in the logfile, that are probably unrelated to bluetooth:
```
2023-12-16 01:34:21.947465500 ERROR:SerialBattery:Exception occurred: ZeroDivisionError('float division by zero') of type <class 'ZeroDivisionError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #894
2023-12-17 12:48:35.014719500 ERROR:SerialBattery:Exception occurred: ZeroDivisionError('float division by zero') of type <class 'ZeroDivisionError'> in /opt/victronenergy/dbus-serialbattery/dbushelper.py line #894
```

PR #890 (2023-12-17): Add setting and install logic for usb bluetooth module
Some users have trouble with the integrated bluetooth in Raspberry Pi's, here I add a setting and install logic for an usb bluetooth module.

Usage: 
- buy a usb bluetooth module with BLE support (bluetooth version > 4.0)
- add `BLUETOOTH_USE_USB = True` to config.ini
- run `/data/etc/dbus-serialbattery/reinstall-local.sh`
- do a `reboot`! 

Should fix: #777 
Should supersede: #830
PR #830 (2023-12-23): LLT/JBD BMS - Some changes for lost bluetooth connection / hci_uart stack restart
Fixes #777