Louisvdw/dbus-serialbattery #778: BMS with LLT app support connection issue/fix

nzlysenko, 2023-08-13:
### Describe the problem

I have this LiFePO4 battery [Volt Polska](https://voltpolska.pl/akumulatory-i-prostowniki/akumulator-lifepo4-12v-150ah-150a-bms.html). It has its own vendor baked BMS app, which doesn't expose any information about BMS model. But I've found that LLT app also works for the battery. In LLT app parameters the BMS manufacturer and model are identified as:
* manufacturer: DR
* device model: DR21S04100-101-V1
* barcode: 12V150A00684

When I run a configuration with LltJbd_Ble it gives a connection error. After some debug session I found the [condition which fails connection test](https://github.com/Louisvdw/dbus-serialbattery/blob/86a6b956e15bee21d5a298283c9d121f0eb82a02/etc/dbus-serialbattery/bms/lltjbd.py#L384). In this part `len(gen_data) < 27` my `gen_data` has length 25. After updating the condition with `len(gen_data) < 25` the connection works and pulled data also look similar to the values which I have in BMS app. 
So, could the LltJbd_Ble BMS module be updated to support this type of BMS and are there some points I need to check to confirm the module completely supports my battery BMS? 

### Driver version

 v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.01

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 70.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0
BLUETOOTH_BMS = LltJbd_Ble EA:61:4A:23:82:81
```

### Relevant log output

```shell
@4000000064d8c62e25fdd674 INFO:SerialBattery:Starting dbus-serialbattery
@4000000064d8c62e260be804 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@4000000064d8c62f0870dda4 INFO:SerialBattery:Init of LltJbd_Ble at EA:61:4A:23:82:81
@4000000064d8c62f0877cefc INFO:SerialBattery:Test of LltJbd_Ble at EA:61:4A:23:82:81
@4000000064d8c63606d5fa24 INFO:SerialBattery:Connection established to LltJbd_Ble
@4000000064d8c63606de8d74 INFO:SerialBattery:Battery LltJbd_Ble connected to dbus from bleea614a238281
@4000000064d8c63606e4cf04 INFO:SerialBattery:========== Settings ==========
@4000000064d8c63606ebf70c INFO:SerialBattery:> Connection voltage: 13.71V | Current: 12.39A | SoC: 99.2%
@4000000064d8c63606f253f4 INFO:SerialBattery:> Cell count: 4 | Cells populated: 4
@4000000064d8c63606f8b8ac INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000064d8c63606ff5be4 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 70.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
@4000000064d8c63607060ebc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 165.0A | MAX BATTERY DISCHARGE CURRENT: -170.0A (read from BMS)
@4000000064d8c636070c5434 INFO:SerialBattery:> CVCM:     True
@4000000064d8c63607130324 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@4000000064d8c636071992d4 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000064d8c63607200efc INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000064d8c636072692f4 INFO:SerialBattery:> CCCM SOC: True  | DCCM SOC: True
@4000000064d8c6360948656c INFO:SerialBattery:DeviceInstance = 1
@4000000064d8c6360952d934 INFO:SerialBattery:com.victronenergy.battery.bleea614a238281
@4000000064d8c637135ab44c INFO:SerialBattery:publish config values = 1
```

idstein, 2023-09-02:
Indeed that check assumes there exist at least 2 temperature sensors!

It can even be changed to 23 to have the bare minimum information. For every temperature sensor the length increments by 2 bytes.

mr-manuel, 2023-09-02:
@idstein would it be possible to change this, so that the number of temperature sensors is automatically adjusted? If you find some time in the future :-)

mr-manuel, 2023-09-04:
Please try to install the `nightly` from the `dev` branch. I just pushed some changes.

mr-manuel, 2023-09-12:
This was fixed with the latest `nightly` build from the `dev` branch and will be available in the next release and pre-release.

closed 2023-09-12 by mr-manuel