Louisvdw/dbus-serialbattery #783: Venus OS crash after adding mqtt dbus device

Jacke87, 2023-08-23:
### Describe the problem

Hello after adding a PV Inverter to Venus OS on AC1 Out my System crash.
I use Venus OS 3.01 with louis dbus serial JKBms.
Both devices get the VRM instance 1. i don´t know if this is the problem.
my code worked for 3 mOnths untill yesterday. My SD Card was corrupted and i instal a fresh Venus an know i Have the issues.
[image]
[image]
[image]
[image]

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

3.01

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
 MAX_BATTERY_CHARGE_CURRENT = 70.0
 MAX_BATTERY_DISCHARGE_CURRENT = 60.0
```

### Relevant log output

```shell
@4000000064e62ada39ccc69c INFO: Start service vedirect-interface.ttyUSB5 once
@4000000064e62add0c134d0c INFO: Start service vedirect-interface.ttyUSB6 once
@4000000064e62b08233417b4 INFO: Create daemontools service gps-dbus.ttyUSB3
@4000000064e62b092d977494 INFO: Create daemontools service gps-dbus.ttyUSB4
@4000000064e62b0e260089dc INFO: Start service gps-dbus.ttyUSB3 once
@4000000064e62b0f300cd994 INFO: Start service gps-dbus.ttyUSB4 once
@4000000064e62b171499e33c INFO: Create daemontools service vedirect-interface.ttyUSB3
@4000000064e62b1820fce87c INFO: Create daemontools service vedirect-interface.ttyUSB4
@4000000064e62b1d174ce5ac INFO: Start service vedirect-interface.ttyUSB3 once
@4000000064e62b1e2375bc64 INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064e62b21389a41dc INFO: Start service dbus-serialbattery.ttyUSB3 once
@4000000064e62b230d289d3c INFO: Start service dbus-serialbattery.ttyUSB4 once
@4000000064e62b4716fca6b4 INFO: Start service gps-dbus.ttyUSB3 once
@4000000064e62b482260aab4 INFO: Start service gps-dbus.ttyUSB4 once
@4000000064e62b4f3612f2ec INFO: Start service vedirect-interface.ttyUSB3 once
@4000000064e62b5105fc380c INFO: Start service vedirect-interface.ttyUSB4 once
@4000000064e62b5416b0ff84 INFO: Start service dbus-serialbattery.ttyUSB3 once
@4000000064e62b5522a81bc4 INFO: Start service dbus-serialbattery.ttyUSB4 once
@4000000064e62b792d116ffc INFO: Start service gps-dbus.ttyUSB3 once
[... 370 lines cut ...]
@4000000064e631d6165bafac INFO:SerialBattery:Testing Jkbms
@4000000064e631d626ceedcc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631d626e3ba04 INFO:SerialBattery:Testing Lifepower
@4000000064e631d637460064 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631d6376173bc INFO:SerialBattery:Testing LltJbd
@4000000064e631d70c02eda4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631d70c2703ec INFO:SerialBattery:Testing Renogy
@4000000064e631d71c7c07c4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631d71c8f68b4 INFO:SerialBattery:Testing Renogy
@4000000064e631d72cf938c4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631d72d15d8e4 INFO:SerialBattery:Testing Seplos
@4000000064e631d82d76b9ac ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631d91066ad54 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB4
@4000000064e631d91066d07c 
@4000000064e631e9275f009c INFO:SerialBattery:Starting dbus-serialbattery
@4000000064e631e92770179c INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@4000000064e631e92777eb84 INFO:SerialBattery:Testing Daly
@4000000064e631e92ef969dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631e92f01b2f4 INFO:SerialBattery:Testing Daly
@4000000064e631e936761da4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631e9367eff14 INFO:SerialBattery:Testing Ecs
@4000000064e631e939fdb9dc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631e93a04b304 INFO:SerialBattery:Testing HeltecModbus
@4000000064e631ed264a6034 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000064e631ed2652535c INFO:SerialBattery:Testing HLPdataBMS4S

mr-manuel, 2023-08-23:
What is your code doing?

Check the all logs under /var/log/*/current

Jacke87, 2023-08-25:
I use the [https://github.com/freakent/dbus-mqtt-devices](url) driver to add my PV inverter I push the values via Mqtt in Node red on a other device.
I tested a lot, my GUI seems to freeze. If I only have the battery driver running, it stocks in between. If I reload the page via ESC, it works. If the battery driver comes along, the dropouts get worse and I lose the connection to the BMS. Before I reinstalled it I used the trieber version <1, now I'm >1. As if the load is too big for the Raspberry?
I've looked at various logs but can't do anything with them, I don't know what to look for.

pos-ei-don, 2023-08-25:
Jacke87, I had similar Problems with the mqtt driver. Altough I really like its General concept, I stopped using it for Performance reasons. Maybe, there is some kind of Loop in there? If i remember correct we also had a Loop within this driver while writing to dbus.

Jacke87, 2023-08-25:
Before the broken SD card is used Louis dB in 0.14 and mqtt 0.5.2 I think and all was well.
After reinstall I used Louis 1.0.2 and mqtt 0.6 am it crashes. After I tried a knew installation with mqtt 0.5.2 the same result. 
My third attempt was to use only Louis dB dbus driver in version 1. But the GUI is also laggy.
I think I try 0.14 again but I like the changes in version 1

mr-manuel, 2023-08-25:
Try to use the mqtt drivers from my repsoitory.

Jacke87, 2023-08-26:
ok i will give it a try.
thanks for support and your great work

mr-manuel, 2023-09-04:
@Jacke87 any updates?

closed 2023-09-12 by mr-manuel