Louisvdw/dbus-serialbattery #784: 2 Daly BMS - Only one is working the second doesn´t follow values

DIVERSDIVERS, 2023-08-24:
### Describe the problem

Hi,

One of my Batteries is correctly working with values other one not, is there a trick?

thx

Iceman

### Driver version

no Version is shown

### Venus OS device type

Cerbo GX

### Venus OS version

v3.01

### BMS type

Daly Smart BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 250.0
MAX_BATTERY_DISCHARGE_CURRENT = 500.0

MIN_CELL_VOLTAGE   = 2.500
MAX_CELL_VOLTAGE   = 3.625
FLOAT_CELL_VOLTAGE = 3.400

BLOCK_ON_DISCONNECT = True

LINEAR_LIMITATION_ENABLE = True

CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.020

CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT = 0.060

SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT = 97

CCCM_CV_ENABLE = True

DCCM_CV_ENABLE = True

CELL_VOLTAGES_WHILE_CHARGING   = 3.63, 3.625, 3.62, 3.61, 3.60, 3.58, 3.55, 3.50
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.008, 0.02,  0.1,  0.2,  0.5, 0.75,    1

CELL_VOLTAGES_WHILE_DISCHARGING   = 2.60, 2.65, 2.80, 2.90, 3.00
MAX_DISCHARGE_CURRENT_CV_FRACTION =    0, 0.02, 0.25,  0.5,    1

[... 253 lines cut ...]
@4000000064e5ff8e0dd73874 INFO: Start service dbus-cgwacs.ttyUSB1 once
@4000000064e5ff931f2edf3c INFO: Start service gps-dbus.ttyUSB1 once
@4000000064e5ff9c30d4a21c INFO: Start service vedirect-interface.ttyUSB1 once
@4000000064e5ffa208f96e94 INFO: Start service dbus-serialbattery.ttyUSB1 once
@4000000064e5ffa71a14df4c INFO: Start service dbus-cgwacs.ttyUSB1 once
@4000000064e5ffac2888b774 INFO: Start service gps-dbus.ttyUSB1 once
@4000000064e5ffb602942224 INFO: Start service vedirect-interface.ttyUSB1 once
@4000000064e5ffbb1175e304 INFO: Start service dbus-serialbattery.ttyUSB1 once
@4000000064e5ffc02440bd24 INFO: Start service dbus-cgwacs.ttyUSB1 once
@4000000064e5ffc5320e04ac INFO: Start service gps-dbus.ttyUSB1 once
@4000000064e5ffcf0b5690a4 INFO: Start service vedirect-interface.ttyUSB1 once
@4000000064e5ffd41da9ae6c INFO: Start service dbus-serialbattery.ttyUSB1 once
@4000000064e5ffd92daee87c INFO: Start service dbus-cgwacs.ttyUSB1 once
@4000000064e5ffdf0500fba4 INFO: Start service gps-dbus.ttyUSB1 once
@4000000064e5ffe816e6637c INFO: Start service vedirect-interface.ttyUSB1 once
@4000000064e5ffed26ed6a7c INFO: Start service dbus-serialbattery.ttyUSB1 once
@4000000064e5fff237407284 INFO: Start service dbus-cgwacs.ttyUSB1 once
@4000000064e5fff80b71fc2c INFO: Start service gps-dbus.ttyUSB1 once
@4000000064e600011fd5121c INFO: Start service vedirect-interface.ttyUSB1 once
@4000000064e600062e21bb04 INFO: Start service dbus-serialbattery.ttyUSB1 once
@4000000064e6000c09d61eac INFO: Start service dbus-cgwacs.ttyUSB1 once
@4000000064e6001118d3c044 INFO: Start service gps-dbus.ttyUSB1 once
@4000000064e6001a2e0b9af4 INFO: Start service vedirect-interface.ttyUSB1 once
@4000000064e600200ad9f24c INFO: Start service dbus-serialbattery.ttyUSB1 once
```

mr-manuel, 2023-09-04:
Please provide the logs in the correct format as documented and requested. This way we cannot help you.

mr-manuel, 2023-09-04:
Please also describe better your problem since it's not possible to understand what the problem is.

closed 2023-09-09 by DIVERSDIVERS