Louisvdw/dbus-serialbattery #785: JKBMS BLE reinstall-local.sh: Building wheel for dbus-fast won't finish

motti15, 2023-08-24:
### Describe the bug

I tried to connect two JKBMS via bluetooth on an freshly installed venosOS, but the RP 2 w i'm using keeps getting stuck while "building wheels for dbus-fast .." and seems to crash after a while. Even manuell installing of bleak via "pip3 install bleak -no-cache-dir" fails while "building wheels for dbus-fast" and seems to show the same behavior. I even tried to reinstall venusOS, but i keep getting the same behavior. Any ideas where to find the root of the problem?

### How to reproduce

Steps to reproduce the behavior:
1. install venusOS large on RasperryPi 2 w
2. install dbus-serialbattery via SSH
3. reboot RP
4. Add config.ini
5. do "bash /data/etc/dbus-serialbattery/reinstall-local.sh"

### Expected behavior

installing bleak

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.01

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example:
;     1 BMS: Jkbms_Ble C8:47:8C:00:00:00
;     3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
;BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:E9:14:83, Jkbms_Ble C8:47:8C:E9:3D:04

; --------- Time-To-Soc ---------
; Description: Calculates the time to a specific SoC
; Example: TIME_TO_SOC_POINTS = 50, 25, 15, 0
;          6h 24m remaining until 50% SoC
;          17h 36m remaining until 25% SoC
;          22h 5m remaining until 15% SoC
;          28h 48m remaining until 0% SoC
; Set of SoC percentages to report on dbus and MQTT. The more you specify the more it will impact system performance.
; [Valid values 0-100, comma separated list. More that 20 intervals are not recommended]
; Example: TIME_TO_SOC_POINTS = 100, 95, 90, 85, 75, 50, 25, 20, 10, 0
; Leave empty to disable
[... 359 lines cut ...]
[    6.018362] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    6.112497] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[    6.119742] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[    6.329095] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    6.343232] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    6.356328] usbcore: registered new interface driver brcmfmac
[    6.366859] Bluetooth: hci0: BCM: chip id 94
[    6.371991] Bluetooth: hci0: BCM: features 0x2e
[    6.382665] Bluetooth: hci0: BCM43430A1
[    6.386586] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0000
[    6.395213] Bluetooth: hci0: BCM43430A1 'brcm/BCM43430A1.hcd' Patch
[    6.547501] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    6.556462] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    6.565337] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    6.577116] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Feb  5 2021 12:49:07 version 7.45.96 (r745790) FWID 01-71817851 es7
[    7.130109] random: crng init done
[    7.133527] random: 2 urandom warning(s) missed due to ratelimiting
[    7.731838] Bluetooth: hci0: BCM43438A1 37.4MHz Raspberry Pi 3-0062
[    7.738164] Bluetooth: hci0: BCM43430A1 (001.002.009) build 0360
[    7.982297] EXT4-fs (mmcblk0p3): re-mounted. Opts: (null)
```

### Any other information that may be helpful

There are two MPPTs connected via USB and an MK3 with 3 Multiplus-II 3000. The Batteries are close to the RP but the wifi signal is weak (only two bars are shown on the remote-console.)

motti15, 2023-08-24:
I tried the same steps on an fresh venusOS installation on my RaspberryPi 4. It seems to work without problems but the step "Building wheel for dbus-fast" took very long. Could any time-out be a problem with the RaspberryPi 2w? Because they are much less powerfull.

motti15, 2023-08-25:
Trying a fresh start with the RaspberryPi 2w worked today. The only difference i could find was no USB-devices plugged into the RP 2w. I won't do any further testing for now, because the RP 4 is currently used for the victron-system and the system is not installed at my home (but within my family). I'll also close the issue for now.

mr-manuel, 2023-08-26:
I had the same behaviour on my MultiPlus GX, but did not find a solution so I copied over the dbus-fast from my Raspberry Pi 4.

mr-manuel, 2023-09-05:
Now I found a solution. See https://github.com/Bluetooth-Devices/dbus-fast/issues/237

This will be implemented in the next version of this driver. For older drivers you have to run

```bash
export SKIP_CYTHON
```

before running the `install.sh` or `reinstall-local.sh`.

closed 2023-09-05 by mr-manuel

fableman73, 2024-03-09:
This problem still exists.

motti15, 2024-03-09:
@fableman73 did you also use a RaspberryPi 2 w? Whats the driver version you tried and did you try the solutions from above? I haven't experienced the problem since then and had to install the latest stable release from mr-manuel today because of the appearance of another bug that was reported in nov. 2023 and seems to be fix in the current stable.