Louisvdw/dbus-serialbattery #792: Key Error charge_limits["tmp"] != "Max Battery Charge Current" (battery.py line 579)

maxx8888, 2023-09-01:
### Describe the problem

Traceback immidiatly on startup with nightly Dev Version:

@4000000064f20d6a3268f5ec Traceback (most recent call last):
@4000000064f20d6a32690974   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 389, in publish_battery
@4000000064f20d6a32691914     self.battery.manage_charge_current()
@4000000064f20d6a326924cc   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 579, in manage_charge_current
@4000000064f20d6a3269346c     if charge_limits["tmp"] != "Max Battery Charge Current":
@4000000064f20d6a32694024 KeyError: 'tmp'

### Driver version

nighly Dev

### Venus OS device type

Raspberry Pi

### Venus OS version

3.01

### BMS type

JKBMS / Heltec BMS

### Cell count

18

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 115.0
MAX_BATTERY_DISCHARGE_CURRENT = 140.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
; Float voltage (can be seen as resting voltage)
FLOAT_CELL_VOLTAGE = 3.340

;Maxx: Settings GAS
;MAX_CELL_VOLTAGE    = 3.450
;FLOAT_CELL_VOLTAGE  = 3.450

; Bulk voltage (may be needed to reset the SoC to 100% once in a while for some BMS)
; Has to be higher as the MAX_CELL_VOLTAGE
BULK_CELL_VOLTAGE  = 3.650
; Specify after how many days the bulk voltage should be reached again
; The timer is reset when the bulk voltage is reached
; Leave empty if you don't want to use this
; Example: Value is set to 15
; day 1: bulk reached once
; day 16: bulk reached twice
; day 31: bulk not reached since it's very cloudy
; day 34: bulk reached since the sun came out
; day 49: bulk reached again, since last time it took 3 days to reach bulk voltage
BULK_AFTER_DAYS =
[... 263 lines cut ...]
@4000000064f20d6922794f4c INFO:SerialBattery:Testing Jkbms
@4000000064f20d69259e0eec INFO:SerialBattery:Connection established to Jkbms
@4000000064f20d6925a682fc INFO:SerialBattery:Battery Jkbms connected to dbus from /dev/ttyUSB1
@4000000064f20d6925ac478c INFO:SerialBattery:========== Settings ==========
@4000000064f20d6925b3d90c INFO:SerialBattery:> Connection voltage: 60.22V | Current: 1.05A | SoC: 99%
@4000000064f20d6925b995cc INFO:SerialBattery:> Cell count: 18 | Cells populated: 0
@4000000064f20d6925c06bcc INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
@4000000064f20d6925cb02bc INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 115.0A | MAX BATTERY DISCHARGE CURRENT: 140.0A
@4000000064f20d6925d13c7c INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 175.0A | MAX BATTERY DISCHARGE CURRENT: 175.0A (read from BMS)
@4000000064f20d6925d6f93c INFO:SerialBattery:> CVCM:     True
@4000000064f20d6925dce0f4 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
@4000000064f20d6925e38bfc INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
@4000000064f20d6925e948bc INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
@4000000064f20d6925ef0194 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
@4000000064f20d6925f47fd4 INFO:SerialBattery:Serial Number/Unique Identifier: P1_P1
@4000000064f20d69286122a4 INFO:SerialBattery:DeviceInstance = 1
@4000000064f20d6928664324 INFO:SerialBattery:com.victronenergy.battery.ttyUSB1
@4000000064f20d6929723cb4 INFO:SerialBattery:publish config values = 1
@4000000064f20d6a3268f5ec Traceback (most recent call last):
@4000000064f20d6a32690974   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 389, in publish_battery
@4000000064f20d6a32691914     self.battery.manage_charge_current()
@4000000064f20d6a326924cc   File "/opt/victronenergy/dbus-serialbattery/battery.py", line 579, in manage_charge_current
@4000000064f20d6a3269346c     if charge_limits["tmp"] != "Max Battery Charge Current":
@4000000064f20d6a32694024 KeyError: 'tmp'
```

mr-manuel, 2023-09-01:
Thanks for reporting, I will fix that on monday.

mr-manuel, 2023-09-02:
The fix was merged with the dev branch.

closed 2023-09-02 by mr-manuel