Louisvdw/dbus-serialbattery #793: BMS name cannot be permanently set

lfhoward, 2023-09-02:
### Describe the bug

Dear Louis,

This is a fantastic piece of software! Thank you so much for creating this. I installed it on my Venus system this week and it worked right away with no modifications. I am running an Overkill BMS with my battery, and running Venus on a Raspberry Pi 3b.

I want to give my BMS a custom name because the default "SerialBattery(LTT/JBD)" is long and shorter names are easier to display on the GUI. I can rename it to "BMS" using "Device" and "Name" in the settings. However, the name change does not stick, and the name defaults back to "SerialBattery(LTT/JBD)" when Venus OS is rebooted. 

Can we get the BMS name change (and possibly other SerialBattery settings) to survive a reboot?  Has anyone else noticed other customized properties that revert to default when rebooting Venus?

Thanks,
Lauren

### How to reproduce

Steps to reproduce the behavior:
1. In the Device menu, change the Name of the new device "SerialBattery(BMStype)" to a custom name, like "BMS"
2. Go back to the Pages screen and verify that the name change is being displayed properly.
3. Reboot Venus OS.
4. After the system boots up, the custom name change is gone and the device name is back to the default "SerialBattery(BMStype)".

### Expected behavior

Name changes made on the Device menu in Venus should survive a reboot of the system and not have to be set again.

### Driver version

1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

3.01

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.defau>
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
MAX_BATTERY_CHARGE_CURRENT = 60.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0
```

### Relevant log output

```shell
serial connection works fine
```

mr-manuel, 2023-09-02:
Duplicate of https://github.com/Louisvdw/dbus-serialbattery/issues/100.

Try the latest `nightly` from the `dev` branch and it will work.

closed 2023-09-02 by mr-manuel

Baxter117, 2023-10-17:
i am using nightly dev v1.0.20231010dev and sadly it does not work.
after renaming all and then rebooting the original name is restored.

mr-manuel, 2023-10-18:
Please post the content of your `config.ini`