Louisvdw/dbus-serialbattery #805: Battery keeps charing and discharging all the time and never rests

Zahnpasta83, 2023-09-17:
### Describe the problem

I have got a 3 phase Multiplus 2 5000 and a 2 batterie 18S system.

As soon as the set voltage is reached and DC feed in is active, the power constantly fluctuates between approx. 300w charging and then 300w discharging again. the battery never goes to rest, is that normal? There is a constant fluctuation between charging and discharging.

When the set voltage is more like 0.1 - 0.2V above the set MAX_CELL_VOLTAGE, Victrons start to discharge ramping up to almost full power depending on voltage above the set value.

[image]
[image]

### Driver version

latest

### Venus OS device type

Raspberry Pi

### Venus OS version

latest

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

2 x 18

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 150.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.450

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example:
;     1 BMS: Jkbms_Ble C8:47:8C:00:00:00
;     3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
;              BMS on purpose, then you have to restart the driver/system to reset the block.
; False: Charge and discharge is not blocked on BMS communication loss
; True: Charge and discharge is blocked on BMS communication loss, it's unblocked when connection is established
;	again or the driver/system is restarted
BLOCK_ON_DISCONNECT = True

; --------- Charge mode ---------
[... 232 lines cut ...]
2023-08-26 09:48:48.620379500 *** CCGX booted (0) ***
2023-08-26 10:34:04.387938500 *** CCGX booted (0) ***
2023-08-26 12:43:25.794557500 *** CCGX booted (0) ***
2023-08-26 12:50:27.804425500 *** CCGX booted (0) ***
2023-08-26 12:54:11.630219500 *** CCGX booted (0) ***
2023-08-26 13:06:17.618655500 *** CCGX booted (0) ***
2023-08-26 13:12:27.707344500 *** CCGX booted (0) ***
2023-08-26 13:16:03.638566500 *** CCGX booted (0) ***
2023-08-26 13:16:03.592161500 *** CCGX booted (0) ***
2023-08-26 13:16:03.596242500 *** CCGX booted (0) ***
2023-09-04 11:57:08.656663500 *** CCGX booted (0) ***
2023-09-04 11:57:08.554005500 *** CCGX booted (0) ***
2023-09-09 14:49:12.659293500 *** CCGX booted (0) ***
2023-09-10 16:54:00.729330500 *** CCGX booted (0) ***
2023-09-10 16:58:11.732578500 *** CCGX booted (0) ***
2023-09-11 13:08:19.668278500 *** CCGX booted (0) ***
2023-09-11 13:45:40.738699500 *** CCGX booted (0) ***
2023-09-11 16:39:48.698981500 *** CCGX booted (0) ***
2023-09-12 11:58:19.813199500 *** CCGX booted (0) ***
2023-09-12 12:05:44.631340500 *** CCGX booted (0) ***
2023-09-12 12:11:08.606165500 *** CCGX booted (0) ***
2023-09-12 12:11:08.503162500 *** CCGX booted (0) ***
2023-09-12 12:11:08.530343500 *** CCGX booted (0) ***
2023-09-12 14:34:17.594341500 *** CCGX booted (0) ***
```

Zahnpasta83, 2023-09-18:
No one who can help me? 😟

mr-manuel, 2023-09-18:
This is normal and needed for system stability as of Victron. Unfortunately I cannot find the question in the Victron Community anymore. If you want further details ask this question there and post the link here, so I can link it in the FAQ.

mr-manuel, 2023-09-19:
Because system stability has a higher priority. 

> If you want further details ask this question there and post the link here, so I can link it in the FAQ.

closed 2023-09-19 by mr-manuel