Louisvdw/dbus-serialbattery #811: JKBMS not reporting the right values

sstolk90, 2023-09-19:
### Describe the bug

I have a Victron GX MP-II 48/5000/70 with two JKBMS'ses 304 Ah so around ~29 kWh.
The version of the Venus OS is 2.92 and I'm using the latest dev nighly build of the driver

One of my JKBMS'es reports false number/stats and settings.
the BMS shows the right values when connected to the phone app but the driver does not report the correct values.

The only thing that has changed but I do not think what can cause an issue is I installed 4 BLE transmitters and a ESPresence in the same shed where my batteries are.

[image]
[image]

### How to reproduce

Steps to reproduce the behavior:
Rebooting the JKBMS by hand, removing the cables, disconnecting the battery, rebooting the MultiPlus by button or software or disconnecting the battery.

### Expected behavior

Getting the correct values

### Driver version

1.0.20230919dev

### Venus OS device type

Venus_GX

### Venus OS version

2.92

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; Jkbms_Ble C8:47:8C:E5:01:1B
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:EC:C0:B5, Jkbms_Ble C8:47:8C:E5:01:1B
BLOCK_ON_DISCONNECT = False
TIME_TO_GO_ENABLE = True

BMS_TYPE = Jkbms

; CUSTOM_BATTERY_NAMES = C8:47:8C:E5:01:1B:Batterij Onder,C8:47:8C:EC:C0:B5:Batterij Boven

; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.555
FLOAT_CELL_VOLTAGE = 3.535
[... 102 lines cut ...]
2023-09-19 11:06:56.228123500 INFO:SerialBattery:dbus-serialbattery v1.0.20230919dev
2023-09-19 11:06:56.642561500 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:E5:01:1B
2023-09-19 11:06:56.643140500 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:E5:01:1B
2023-09-19 11:07:01.658784500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20221223)
2023-09-19 11:07:01.660014500 INFO:SerialBattery:Connection established to Jkbms_Ble
2023-09-19 11:07:01.661479500 INFO:SerialBattery:Battery Jkbms_Ble connected to dbus from c8478ce5011b
2023-09-19 11:07:01.664933500 INFO:SerialBattery:========== Settings ==========
2023-09-19 11:07:01.666687500 INFO:SerialBattery:> Connection voltage: 53.19V | Current: 31.0A | SoC: 99%
2023-09-19 11:07:01.668166500 INFO:SerialBattery:> Cell count: 16 | Cells populated: 16
2023-09-19 11:07:01.669912500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: False
2023-09-19 11:07:01.673566500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 100.0A | MAX BATTERY DISCHARGE CURRENT: 100.0A
2023-09-19 11:07:01.677865500 INFO:SerialBattery:> CVCM:     True
2023-09-19 11:07:01.689804500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.555V
2023-09-19 11:07:01.691208500 INFO:SerialBattery:> CCCM CV:  False | DCCM CV:  False
2023-09-19 11:07:01.693125500 INFO:SerialBattery:> CCCM T:   False | DCCM T:   False
2023-09-19 11:07:01.694772500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2023-09-19 11:07:01.696421500 INFO:SerialBattery:Serial Number/Unique Identifier: 2052302250
2023-09-19 11:07:02.173584500 INFO:SerialBattery:DeviceInstance = 5
2023-09-19 11:07:02.174578500 INFO:SerialBattery:com.victronenergy.battery.c8478ce5011b
2023-09-19 11:07:02.175711500 INFO:SerialBattery:BAT: JKBMS 10.XW 16 cells (20221223)
2023-09-19 11:07:02.265307500 INFO:SerialBattery:publish config values = 1
2023-09-19 11:11:49.567075500 [CHG] Device C8:47:8C:E5:01:1B RSSI: -74
2023-09-19 11:11:49.567088500 [CHG] Device C8:47:8C:E5:01:1B RSSI: -68
`
```

sstolk90, 2023-09-19:
[image]
[image]

Those are the values in the VRM/ Remote Console where the driver acts on. So when the values are 0 C or soc is 0 the battery is not drained or charged. In the log you can see the same values which are displayed in the log above.

sstolk90, 2023-09-19:
Not quite sure if my BMS is corrupt but I checked my logs and it happend aroun 23:02:31 last night
[image]
When I set Grafana 1 second later this is the output:
[image]

The cycles are reporting the voltage in some what way. Everything else like amps, soc and temp go haywire.
I did not reboot or changed anything at this time since I was asleep. the victron was set on Grid 0 mode so just maintaining the basic house load.

sstolk90, 2023-09-19:
The values in this Grafana / Influx DB are coming from the MQTT queries against the Victron. So the Victron is "feeding" the database. When I disconnect the Victron BLE towards my BMS which is at fault the app displays the correct information.
I already installed the new driver as mentioned above but this does not resolve the problem.

mr-manuel, 2023-09-26:
Can you try an older build? https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/

mr-manuel, 2023-11-02:
Please try the `nightly` of the `dev` branch from today. There are some fixes included for the correct JKBMS BLE recognition.

closed 2023-11-03 by mr-manuel

sstolk90, 2023-11-07:
Confirmed that it working also! All the values are showing the right things!
Thanks for this fix!