Louisvdw/dbus-serialbattery #819: JkBms Bluetooth not connecting

itgenmar, 2023-09-23:
### Describe the problem

**Jk BMS : 
JK_bd4a24s4p

Serial number 
Hardware v11.XW
SOFTWARE V11.25
POWER ON 13 TIMES
TOTL TIMES 1D14H30M0S
FIST ON DATE : 2023-04024
VERSION V4.15.7**

**Getting the following error in the logs, no connection to venus os.**

@40000000650f32362410d384
@40000000650f32362410db54 connect and scrape on address: 20:22:08:25:17:41
@40000000650f32362410eaf4 btloop
@40000000650f32362410eedc reconnect
@40000000650f32362410f6ac Exiting bt-loop
@40000000650f323634403e5c Missing device address argument
@40000000650f323737c83084 INFO:SerialBattery:Starting dbus-serialbattery
@40000000650f323737d7b914 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@40000000650f3238126df2ec INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
@40000000650f323812745f74 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
@40000000650f324723e42dd4 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@40000000650f324723e7a874
@40000000650f324723e7b42c connect and scrape on address: 20:22:08:25:17:41
@40000000650f324723e7bfe4 btloop
@40000000650f324723e7c7b4 reconnect
@40000000650f324723e7cb9c Exiting bt-loop
@40000000650f324733b0a88c Missing device address argument
@40000000650f324837f02ecc INFO:SerialBattery:Starting dbus-serialbattery
@40000000650f324837fed0e4 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@40000000650f3249137efd5c INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
@40000000650f324913853b04 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
@40000000650f325823de6d2c ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@

### Driver version

latest

### Venus OS device type

Raspberry Pi

### Venus OS version

latest

### BMS type

JKBMS / Heltec BMS

### Cell count

7

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; If you want to add custom values/settings, then check the values/settings you want to change in "config.default.ini"
; and insert them below to persist future driver updates.

; Example (remove the semicolon ";" to uncomment and activate the value/setting):
; MAX_BATTERY_CHARGE_CURRENT = 50.0
; MAX_BATTERY_DISCHARGE_CURRENT = 60.0

BLUETOOTH_BMS = Jkbms_Ble 20:22:08:25:17:41

BMS_TYPE = Jkbms
```

### Relevant log output

```shell
@40000000650f32362410d384
@40000000650f32362410db54 connect and scrape on address: 20:22:08:25:17:41
@40000000650f32362410eaf4 btloop
@40000000650f32362410eedc reconnect
@40000000650f32362410f6ac Exiting bt-loop
@40000000650f323634403e5c Missing device address argument
@40000000650f323737c83084 INFO:SerialBattery:Starting dbus-serialbattery
@40000000650f323737d7b914 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@40000000650f3238126df2ec INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
@40000000650f323812745f74 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
@40000000650f324723e42dd4 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@40000000650f324723e7a874
```

mr-manuel, 2023-09-25:
Try to remove `BMS_TYPE = Jkbms` from the `config.ini`. How far is your BMS from the Venus OS device?

Please provide the logfiles as requested. Please do not use `latest` as version and specify the exact version number.

itgenmar, 2023-09-25:
I have tried with and without the BMS_TYPE = Jkbms, I think they are @20cm apart.  That is the log is just  repeating itself. I have used the nightly one too, and is exactly the same error in the logs. I know that this bms use  the protocol or framework jk02_32s (I hope I did remembered correctly). Using a esp32 and the code for esphome I can connect to the bms and receive the info.

mr-manuel, 2023-09-25:
Can you try an older build like https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.0.20230525beta?

itgenmar, 2023-09-26:
Exactly the same:
@40000000651288651134cedc INFO:SerialBattery:
@4000000065128865113b5aa4 INFO:SerialBattery:Starting dbus-serialbattery
@400000006512886511497fbc INFO:SerialBattery:dbus-serialbattery v1.0.0 (20230508)
@400000006512886527c6ea2c INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
@400000006512886527cdcfcc INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
@4000000065128866009ff72c ERROR:SerialBattery:No Bluetooth adapters found.
@400000006512886600a69294 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@40000000651288660ec6cc7c Missing device address argument
@40000000651288671182911c INFO:SerialBattery:
@4000000065128867118c9b6c INFO:SerialBattery:Starting dbus-serialbattery
@4000000065128867119c3f54 INFO:SerialBattery:dbus-serialbattery v1.0.0 (20230508)
@400000006512886727ca3204 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
@400000006512886727d551ac INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
@4000000065128868002b5b9c ERROR:SerialBattery:No Bluetooth adapters found.
@400000006512886800351bb4 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@40000000651288680e45500c Missing device address argument

**I have attached the device info from ESPhome:**

[07:44:59][D][button:010]: 'jk-bms retrieve device info' Pressed.
[07:44:59][D][jk_bms_ble:1247]: Write register: AA.55.90.EB.97.00.00.00.00.00.00.00.00.00.00.00.00.00.00.11 (20)
[07:44:59][I][jk_bms_ble:1158]: Device info frame (300 bytes) received
[07:44:59][I][jk_bms_ble:1210]:   Vendor ID: JK_BD4A24S4P
[07:44:59][I][jk_bms_ble:1211]:   Hardware version: 11.XW
[07:44:59][I][jk_bms_ble:1212]:   Software version: 11.25
[07:44:59][I][jk_bms_ble:1213]:   Uptime: 438900 s
[07:44:59][I][jk_bms_ble:1214]:   Power on count: 13
[07:44:59][I][jk_bms_ble:1215]:   Device name: JK_BD4A24S04
[07:44:59][I][jk_bms_ble:1216]:   Device passcode: 1234
[07:44:59][I][jk_bms_ble:1217]:   Manufacturing date: 230424
[07:44:59][I][jk_bms_ble:1218]:   Serial number: 2082321445
[07:44:59][I][jk_bms_ble:1219]:   Passcode: 0000
[07:44:59][I][jk_bms_ble:1220]:   User data: Input Userdata
[07:44:59][I][jk_bms_ble:1221]:   Setup passcode: 123456
[07:44:59][W][component:204]: Component esp32_ble took a long time for an operation (0.07 s).
[07:44:59][W][component:205]: Components should block for at most 20-30ms.
[07:45:03][I][jk_bms_ble:467]: Cell info frame (version 3, 300 bytes) received

**Devices settings:**

[07:47:35][D][button:010]: 'jk-bms retrieve settings' Pressed.
[07:47:35][D][jk_bms_ble:1247]: Write register: AA.55.90.EB.96.00.00.00.00.00.00.00.00.00.00.00.00.00.00.10 (20)
[07:47:36][I][jk_bms_ble:891]: Settings frame (300 bytes) received
[07:47:36][D][jk_bms_ble:917]:   Unknown6: 0.600000
[07:47:36][I][jk_bms_ble:919]:   Cell UVP: 2.820000 V
[07:47:36][D][number:012]: 'jk-bms cell voltage undervoltage protection': Sending state 2.820000
[07:47:36][I][jk_bms_ble:923]:   Cell UVPR: 2.850000 V
[07:47:36][D][number:012]: 'jk-bms cell voltage undervoltage recovery': Sending state 2.850000
[07:47:36][I][jk_bms_ble:927]:   Cell OVP: 4.200000 V
[07:47:36][D][number:012]: 'jk-bms cell voltage overvoltage protection': Sending state 4.200000
[07:47:36][I][jk_bms_ble:931]:   Cell OVPR: 4.180000 V
[07:47:36][D][number:012]: 'jk-bms cell voltage overvoltage recovery': Sending state 4.180000
[07:47:36][I][jk_bms_ble:935]:   Balance trigger voltage: 0.005000 V
[07:47:36][D][number:012]: 'jk-bms balance trigger voltage': Sending state 0.005000
[07:47:36][I][jk_bms_ble:943]:   Power off voltage: 2.800000 V
[07:47:36][D][number:012]: 'jk-bms power off voltage': Sending state 2.800000
[07:47:36][I][jk_bms_ble:947]:   Max. charge current: 25.000002 A
[07:47:36][D][number:012]: 'jk-bms max charge current': Sending state 25.000002
[07:47:36][I][jk_bms_ble:951]:   Charge OCP delay: 30.000000 s
[07:47:36][I][jk_bms_ble:953]:   Charge OCP recovery delay: 60.000000 s
[07:47:36][I][jk_bms_ble:955]:   Max. discharge current: 40.000000 A
[07:47:36][D][number:012]: 'jk-bms max discharge current': Sending state 40.000000
[07:47:36][I][jk_bms_ble:959]:   Discharge OCP recovery delay: 300.000000 s
[07:47:36][I][jk_bms_ble:961]:   Discharge OCP recovery delay: 60.000000 s
[07:47:36][I][jk_bms_ble:963]:   SCP recovery time: 60.000000 s
[07:47:36][I][jk_bms_ble:965]:   Max. balance current: 0.400000 A
[07:47:36][D][number:012]: 'jk-bms max balance current': Sending state 0.400000
[07:47:36][I][jk_bms_ble:969]:   Charge OTP: 70.000000 °C
[07:47:36][I][jk_bms_ble:971]:   Charge OTP recovery: 60.000000 °C
[... 488 lines cut ...]
[07:42:36][D][sensor:094]: 'jk-bms current': Sending state -1.32700 A with 2 decimals of accuracy
[07:42:36][D][sensor:094]: 'jk-bms power': Sending state -35.18010 W with 2 decimals of accuracy
[07:42:36][D][sensor:094]: 'jk-bms charging power': Sending state 0.00000 W with 2 decimals of accuracy
[07:42:37][D][sensor:094]: 'jk-bms discharging power': Sending state 35.18010 W with 2 decimals of accuracy
[07:42:37][D][sensor:094]: 'jk-bms temperature sensor 1': Sending state 16.00000 °C with 1 decimals of accuracy
[07:42:37][D][sensor:094]: 'jk-bms temperature sensor 2': Sending state 15.80000 °C with 1 decimals of accuracy
[07:42:37][D][sensor:094]: 'jk-bms errors bitmask': Sending state 0.00000  with 0 decimals of accuracy
[07:42:37][D][text_sensor:064]: 'jk-bms errors': Sending state ''
[07:42:37][D][sensor:094]: 'jk-bms balancing current': Sending state 0.00000 A with 2 decimals of accuracy
[... cut ...]
[07:42:37][D][text_sensor:064]: 'jk-bms total runtime formatted': Sending state '5d 1h'
[07:42:37][D][jk_bms_ble:663]: Unknown168: 00.00.00.00.00.00.00.00.00.00.00.00.00.00 (14)
[07:42:37][D][jk_bms_ble:677]: Unknown189: 0x00 0x64
[07:42:37][D][jk_bms_ble:681]: Unknown192: 0x11
[07:42:37][D][jk_bms_ble:683]: Unknown193: 0x00 0xE7 (0x00 0x8D)
[07:42:37][D][jk_bms_ble:685]: Unknown195: 0xA3 0x3E (0x21 0x40)
[07:42:37][W][component:204]: Component esp32_ble took a long time for an operation (0.51 s).
[07:42:37][W][component:205]: Components should block for at most 20-30ms.

mr-manuel, 2023-09-26:
Please provide the logs with timestamp, as we wrote everywhere. See also https://louisvdw.github.io/dbus-serialbattery/troubleshoot/#driver-log-files

Anyway, now the error is another.

```bash
@4000000065128866009ff72c ERROR:SerialBattery:No Bluetooth adapters found.
```

itgenmar, 2023-09-26:
I had a second USB Bluetooth connected, i had removed it, reinstalled, and still the same included the log. 

2023-09-26 09:21:00.146421500 *** CCGX booted (0) ***
2023-09-26 09:22:45.740005500 Missing device address argument
2023-09-26 09:22:47.432327500 INFO:SerialBattery:
2023-09-26 09:22:47.432752500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-26 09:22:47.433665500 INFO:SerialBattery:dbus-serialbattery v1.0.20230525beta
2023-09-26 09:22:47.870858500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:22:47.871406500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:02.831165500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-09-26 09:23:02.831174500 connect and scrape on address: 20:22:08:25:17:41
2023-09-26 09:23:02.831177500 btloop
2023-09-26 09:23:02.831179500 reconnect
2023-09-26 09:23:02.831180500 Exiting bt-loop
2023-09-26 09:23:03.117505500 Missing device address argument
2023-09-26 09:23:04.764602500 INFO:SerialBattery:
2023-09-26 09:23:04.765033500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-26 09:23:04.765953500 INFO:SerialBattery:dbus-serialbattery v1.0.20230525beta
2023-09-26 09:23:05.154074500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:05.154536500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:20.776226500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-09-26 09:23:20.776382500 connect and scrape on address: 20:22:08:25:17:41
2023-09-26 09:23:20.776386500 btloop
2023-09-26 09:23:20.776387500 reconnect
2023-09-26 09:23:20.776389500 Exiting bt-loop
2023-09-26 09:23:21.039026500 Missing device address argument
2023-09-26 09:23:22.115334500 INFO:SerialBattery:
2023-09-26 09:23:22.115969500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-26 09:23:22.116990500 INFO:SerialBattery:dbus-serialbattery v1.0.20230525beta
2023-09-26 09:23:22.493285500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:22.493944500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:37.782339500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-09-26 09:23:37.782491500 connect and scrape on address: 20:22:08:25:17:41
2023-09-26 09:23:37.782495500 btloop
2023-09-26 09:23:37.782497500 reconnect
2023-09-26 09:23:37.782498500 Exiting bt-loop
2023-09-26 09:23:38.051658500 Missing device address argument
2023-09-26 09:23:39.202712500 INFO:SerialBattery:
2023-09-26 09:23:39.203141500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-26 09:23:39.204127500 INFO:SerialBattery:dbus-serialbattery v1.0.20230525beta
2023-09-26 09:23:39.605594500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:39.606058500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:54.778567500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2023-09-26 09:23:54.778798500 connect and scrape on address: 20:22:08:25:17:41
2023-09-26 09:23:54.778802500 btloop
2023-09-26 09:23:54.778803500 reconnect
2023-09-26 09:23:54.778805500 Exiting bt-loop
2023-09-26 09:23:55.044528500 Missing device address argument
2023-09-26 09:23:56.151729500 INFO:SerialBattery:
2023-09-26 09:23:56.152160500 INFO:SerialBattery:Starting dbus-serialbattery
2023-09-26 09:23:56.153071500 INFO:SerialBattery:dbus-serialbattery v1.0.20230525beta
2023-09-26 09:23:56.551272500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2023-09-26 09:23:56.551932500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41

SVstats :
 svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 7 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 10 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 10 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 11 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 12 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 12 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 12 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 13 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 13 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 14 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 14 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 14 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 15 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 15 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 16 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3052) 16 seconds
root@VFIEVOX3:~# svstat /service/dbus-blebattery.*
/service/dbus-blebattery.0: up (pid 3145) 0 seconds

mr-manuel, 2023-09-27:
What is the output of

```bash
bluetoothctl scan on
```

Wait for some seconds, then exit with `CTRL + C`.

mr-manuel, 2023-09-27:
Please tell me also the exact version of the driver you are using and Venus OS. Latest is not a useful answer.

If you are not running driver version `v1.0.20230921beta` then please install the latest `nightly` of the `dev` branch.

Srekrek, 2023-11-02:
Hi,

First of all: thanks for your fine work on this, it's much appreciated !

I have also same sort of a problem:

**Jk BMS :
JK_B1A8S20P
Serial number
Hardware v11.XW
SOFTWARE V11.261

Venus OS device type
Raspberry Pi 4

Venus OS version 3.12

Connection type
Bluetooth

Bluetoothctl info
~~~
Device C8:47:8C:EE:33:E2 (public)
        Name: NKON-125Ah
        Alias: NKON-125Ah
        Paired: no
        Trusted: yes
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffe0-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (f000ffc0-0451-4000-b000-000000000000)
        Modalias: usb:v045Ep0040d0300
~~~

Config file
~~~
[DEFAULT]
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:EE:33:E2
~~~

/data/log/serial-starter/current
~~~
@400000006543bd362daefc04 *** CCGX booted (0) ***
@400000006543bd381516b074 *** starting serial-starter ***
@400000006543bd39220678dc serstart starting
@400000006543bd3923cf2aec INFO: loading config file /etc/venus/serial-starter.conf
@400000006543bd3d1acc7364 INFO: Create daemontools service vedirect-interface.ttyUSB0
@400000006543bd3f1ba0ceac INFO: Create daemontools service mk2-dbus.ttyUSB1
@400000006543bd432894159c INFO: Start service vedirect-interface.ttyUSB0 once
@400000006543bd4601e2db54 INFO: Start service mk2-dbus.ttyUSB1
~~~

/data/log/dbus-blebattery.*/current
~~~
@400000006543c8e70c417574 ERROR:SerialBattery:Unexpected err=ValueError('dbus_fast.signature.Variant size changed, may indicate binary incompatibility. Expected 24 from C header, got 20 from PyObject'),>
@400000006543c8e70c4810dc ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@400000006543c8e70c4b452c
@400000006543c8e7190e15dc Missing device address argument
@400000006543c8e81fc2d23c INFO:SerialBattery:Starting dbus-serialbattery
@400000006543c8e81fd14d44 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
@400000006543c8e836b33644 INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EE:33:E2
@400000006543c8e836badf34 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EE:33:E2
@400000006543c8e9012962dc ERROR:SerialBattery:Unexpected err=ValueError('dbus_fast.signature.Variant size changed, may indicate binary incompatibility. Expected 24 from C header, got 20 from PyObject'),>
@400000006543c8e9012fe6d4 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@400000006543c8e901346b14
@400000006543c8e90e20c494 Missing device address argument
~~~

Any ideas on how to solve this problem ?

mr-manuel, 2023-11-02:
By looking in my crystal ball I assume that you are using the driver version `v1.0.20230531` :-D

A needed Python module named `dbus_fast` was not installed. This is a known issue on weaker devices and was fixed some weeks ago. Just install the latest `nightly` from the `dev` branch and it should work.

Srekrek, 2023-11-02:
It doesn't seem to change things !?

~~~
@400000006543dfee193aafd4 INFO:Bluetooth details
@400000006543dfee1aab27f4 No default controller available
@400000006543dff31cedff8c
@400000006543dff42385316c INFO:SerialBattery:
@400000006543dff4238bd88c INFO:SerialBattery:Starting dbus-serialbattery
@400000006543dff42399a3cc INFO:SerialBattery:dbus-serialbattery v1.0.20231102dev
@400000006543dff43a42c1bc INFO:SerialBattery:Init of Jkbms_Ble at C8:47:8C:EE:33:E2
@400000006543dff43a497c64 INFO:SerialBattery:Test of Jkbms_Ble at C8:47:8C:EE:33:E2
@400000006543dff50785413c Exception in thread Thread-1:
@400000006543dff5078558ac Traceback (most recent call last):
@400000006543dff507856464   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
@400000006543dff507857404     self.run()
@400000006543dff507857bd4   File "/usr/lib/python3.8/threading.py", line 870, in run
@400000006543dff507858b74     self._target(*self._args, **self._kwargs)
@400000006543dff50785972c   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 430, in connect_and_scrape
@400000006543dff507870274     asyncio.run(self.asy_connect_and_scrape())
@400000006543dff507870e2c   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
@400000006543dff507871dcc     return loop.run_until_complete(main)
@400000006543dff507872984   File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
@400000006543dff507873d0c     return future.result()
@400000006543dff50787a29c   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 440, in asy_connect_and_scrape
@400000006543dff50787ba0c     client = BleakClient(self.address)
@400000006543dff50787c1dc   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 518, in __init__
@400000006543dff50787d564     get_platform_client_backend_type() if backend is None else backend
@400000006543dff50787ecd4   File "/usr/lib/python3.8/site-packages/bleak/backends/client.py", line 260, in get_platform_client_backend_type
@400000006543dff50789022c     from bleak.backends.bluezdbus.client import BleakClientBlueZDBus
@400000006543dff5078911cc   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 23, in <module>
@400000006543dff507892554     from dbus_fast.aio import MessageBus
@400000006543dff50789310c   File "/usr/lib/python3.8/site-packages/dbus_fast/__init__.py", line 1, in <module>
@400000006543dff50789a254     from . import introspection, message_bus, proxy_object, service
@400000006543dff50789b1f4   File "/usr/lib/python3.8/site-packages/dbus_fast/introspection.py", line 5, in <module>
@400000006543dff50789c57c     from .errors import InvalidIntrospectionError
@400000006543dff50789d51c   File "/usr/lib/python3.8/site-packages/dbus_fast/errors.py", line 57, in <module>
@400000006543dff50789e8a4     from .message import Message
@400000006543dff5078a427c   File "src/dbus_fast/message.py", line 1, in init dbus_fast.message
@400000006543dff5078a5604 ValueError: dbus_fast.signature.Variant size changed, may indicate binary incompatibility. Expected 24 from C header, got 20 from PyObject
@400000006543dffe1d760bac ERROR:SerialBattery:No BMS found at C8:47:8C:EE:33:E2
@400000006543dffe1d7d202c ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
@400000006543dffe2aff128c
~~~

mr-manuel, 2023-11-02:
Try to run this manually:

https://github.com/Louisvdw/dbus-serialbattery/blob/a6d45e6776077ec7ed4989584108b5a6737dda0a/etc/dbus-serialbattery/reinstall-local.sh#L184-L190

And then as last

```bash
pip3 install dbus-fast
```

Srekrek, 2023-11-02:
Doesn't change much:

~~~
root@raspberrypi2:/data/etc/dbus-serialbattery# export SKIP_CYTHON=false
root@raspberrypi2:/data/etc/dbus-serialbattery# opkg update
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/all/Packages.gz.
Updated source 'all'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/cortexa7hf-neon-vfpv4/Packages.gz.
Updated source 'cortexa7hf-neon-vfpv4'.
Downloading https://updates.victronenergy.com/feeds/venus/release/packages/dunfell/raspberrypi2/Packages.gz.
Updated source 'raspberrypi2'.
root@raspberrypi2:/data/etc/dbus-serialbattery#  opkg install python3-misc python3-pip
No packages installed or removed.
root@raspberrypi2:/data/etc/dbus-serialbattery# echo
root@raspberrypi2:/data/etc/dbus-serialbattery# pip3 install bleak
Requirement already satisfied: bleak in /usr/lib/python3.8/site-packages (0.21.1)
Requirement already satisfied: async-timeout<5,>=3.0.0; python_version < "3.11" in /usr/lib/python3.8/site-packages (from bleak) (4.0.3)
Requirement already satisfied: typing-extensions>=4.7.0; python_version < "3.12" in /usr/lib/python3.8/site-packages (from bleak) (4.8.0)
Requirement already satisfied: dbus-fast<3,>=1.83.0; platform_system == "Linux" in /usr/lib/python3.8/site-packages (from bleak) (2.12.0)
root@raspberrypi2:/data/etc/dbus-serialbattery# pip3 install dbus-fast
Requirement already satisfied: dbus-fast in /usr/lib/python3.8/site-packages (2.12.0)
~~~

mr-manuel, 2023-11-02:
You could try another `dbus-fast` version.

```bash
pip3 install dbus-fast==v1.93.0
```

Srekrek, 2023-11-03:
Thanks for all the support, it does work now
[image]
I don't exactly know what I did (I even once needed to hard reset my rpi), but it now works with dbus-serialbattery v1.0.20230531.
~~~
export SKIP_CYTHON=false
bash /data/etc/dbus-serialbattery/reinstall-local.sh
~~~
the first line was needed because otherwise ik kept hanging during making a wheel for dbus-fast.

mr-manuel, 2023-11-03:
@itgenmar @M0ebiu5 did you solve your problem? Have you changed the pin of the JKBMS?

itgenmar, 2023-11-03:
Hi, I have installed the latest nightly v1.0.20231102dev, i have the default pins : 1234 and 123456 : 
 tried with  dbus-fast==v1.93.0, with export SKIP_CYTHON=false

> 2023-11-03 08:56:42.184893500 Device 20:22:08:25:17:41 (public)
> 2023-11-03 08:56:42.184917500   Alias: JK_BD4A24S04
> 2023-11-03 08:56:42.184919500   Paired: no
> 2023-11-03 08:56:42.184921500   Trusted: yes
> 2023-11-03 08:56:42.184923500   Blocked: no
> 2023-11-03 08:56:42.185006500   Connected: no
> 2023-11-03 08:56:42.185008500   LegacyPairing: no
> 2023-11-03 08:56:42.185010500   RSSI: -48
> 2023-11-03 08:56:42.185949500
> 2023-11-03 08:56:43.326740500 INFO:SerialBattery:
> 2023-11-03 08:56:43.327189500 INFO:SerialBattery:Starting dbus-serialbattery
> 2023-11-03 08:56:43.328121500 INFO:SerialBattery:dbus-serialbattery v1.0.20231102dev
> 2023-11-03 08:56:43.721565500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
> 2023-11-03 08:56:43.722009500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
> 2023-11-03 08:56:46.222166500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!
> 2023-11-03 08:56:48.856816500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
> 2023-11-03 08:56:53.238361500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17:41
> 2023-11-03 08:56:53.238883500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
> 2023-11-03 08:56:53.540250500
> 2023-11-03 08:56:53.540398500 INFO:Bluetooth details
> 2023-11-03 08:56:53.545915500 [CHG] Device 20:22:08:25:17:41 RSSI: -48
> 2023-11-03 08:56:53.545922500 [CHG] Device 20:22:08:25:17:41 RSSI: -52
> 2023-11-03 08:56:53.607303500 Attempting to disconnect from 20:22:08:25:17:41

mr-manuel, 2023-11-03:
Can you enable `LOGGING = DEBUG` in the `config.ini`?

itgenmar, 2023-11-03:
Done

> 2023-11-03 09:13:52.811312500
> 2023-11-03 09:13:52.811441500 INFO:Bluetooth details
> 2023-11-03 09:13:52.833457500 [CHG] Device 20:22:08:25:17:41 RSSI: -48
> 2023-11-03 09:13:52.833464500 [CHG] Device 20:22:08:25:17:41 RSSI: -48
> 2023-11-03 09:13:52.885804500 Attempting to disconnect from 20:22:08:25:17:41
> 2023-11-03 09:13:52.885812500 [CHG] Device D8:E3:5E:A5:4C:BF TxPower is nil
> 2023-11-03 09:13:52.885816500 [CHG] Device D8:E3:5E:A5:4C:BF RSSI is nil
> 2023-11-03 09:13:52.885819500 [CHG] Device 6B:52:83:FD:5D:A4 TxPower is nil
> 2023-11-03 09:13:52.885823500 [CHG] Device 6B:52:83:FD:5D:A4 RSSI is nil
> 2023-11-03 09:13:52.885827500 [CHG] Device 4F:CF:59:55:62:6E TxPower is nil
> 2023-11-03 09:13:52.885831500 [CHG] Device 4F:CF:59:55:62:6E RSSI is nil
> 2023-11-03 09:13:52.885991500 [CHG] Device A5:C2:37:01:AD:77 RSSI is nil
> 2023-11-03 09:13:52.885996500 [CHG] Device 6F:F2:99:5B:BE:32 TxPower is nil
> 2023-11-03 09:13:52.885999500 [CHG] Device 6F:F2:99:5B:BE:32 RSSI is nil
> 2023-11-03 09:13:52.886003500 [CHG] Device 20:22:08:25:17:41 RSSI is nil
> 2023-11-03 09:13:52.886007500 [CHG] Controller B8:27:EB:2A:55:5A Discovering: no
> 2023-11-03 09:13:52.886034500 Successful disconnected
> 2023-11-03 09:13:57.953844500 Device 20:22:08:25:17:41 (public)
> 2023-11-03 09:13:57.953851500   Alias: JK_BD4A24S04
> 2023-11-03 09:13:57.953853500   Paired: no
> 2023-11-03 09:13:57.953854500   Trusted: yes
> 2023-11-03 09:13:57.953856500   Blocked: no
> 2023-11-03 09:13:57.953858500   Connected: no
> 2023-11-03 09:13:57.953860500   LegacyPairing: no
> 2023-11-03 09:13:57.953862500   RSSI: -48
> 2023-11-03 09:13:57.954957500
> 2023-11-03 09:13:59.073141500 INFO:SerialBattery:
> 2023-11-03 09:13:59.073811500 INFO:SerialBattery:Starting dbus-serialbattery
> 2023-11-03 09:13:59.074983500 INFO:SerialBattery:dbus-serialbattery v1.0.20231102dev
> 2023-11-03 09:13:59.475934500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
> 2023-11-03 09:13:59.476630500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
> 2023-11-03 09:13:59.480352500 DEBUG:SerialBattery:scraping thread started -> main thread id: 1995616272 scraping thread: 1979053152
> 2023-11-03 09:13:59.482483500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): Connect and scrape on address: 20:22:08:25:17:41
> 2023-11-03 09:13:59.855391500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): btloop
> 2023-11-03 09:13:59.856167500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): reconnect
> 2023-11-03 09:14:01.228275500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!
> 2023-11-03 09:14:03.854887500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
> 2023-11-03 09:14:09.004290500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17:41
> 2023-11-03 09:14:09.004298500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
> 2023-11-03 09:14:09.286619500

rver2024, 2023-11-05:
I have bluetooth connection issue on my Jkbms (JK-B2A8S20P) and the Cerbo GX running this dbus serial battery driver too. The jkbms connects to Cerbo GX fine after a reboot, but will be disconnected after a while, in which case Cerbo GX doesn’t seem to know about the disconnection and still shows the BMS on the remote console (of course without displaying the correct up-to-date battery information), I know it because the bluetooth indicator on jkbms flashes when it is not connected to something. Any advice on how to go about troubleshooting this issue? Thanks!

mr-manuel, 2023-11-05:
@rver2024 this is a known issue and not yet fixed. See https://github.com/Louisvdw/dbus-serialbattery/issues/720

mr-manuel, 2023-11-05:
@itgenmar can you post a screenshot of the JKBMS displaying the hardware version?

itgenmar, 2023-11-08:
Attached the Image 

[image]

mr-manuel, 2023-11-25:
Sorry but the link is unavailable. Please insert the screenshot directly into GitHub.

mr-manuel, 2023-11-25:
Also try the latest nightly from the dev branch. There were some relevant changes.

itgenmar, 2023-11-25:
[image]

itgenmar, 2023-11-25:
Still no luck, here is the log:

root@raspberrypi2:~# tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64n                                                   local
2023-11-25 09:00:14.945526500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:2                                                   5:17:41
2023-11-25 09:00:16.929148500 INFO:SerialBattery:--> asy_connect_and_scrape(): e                                                   rror while connecting to bt: Characteristic with UUID 00002a24-0000-1000-8000-00                                                   805f9b34fb could not be found!
2023-11-25 09:00:19.371688500 INFO:SerialBattery:--> asy_connect_and_scrape(): E                                                   xit
2023-11-25 09:00:24.461210500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17                                                   :41
2023-11-25 09:00:24.461673500 ERROR:SerialBattery:ERROR >>> No battery connectio                                                   n at Jkbms_Ble
2023-11-25 09:00:24.751878500
2023-11-25 09:00:24.751992500 INFO:Bluetooth details
2023-11-25 09:00:24.753956500 [CHG] Device 20:22:08:25:17:41 RSSI: -62
2023-11-25 09:00:24.753964500 [CHG] Device 20:22:08:25:17:41 RSSI: -62
2023-11-25 09:00:24.753967500 [CHG] Device 20:22:08:25:17:41 RSSI: -41
2023-11-25 09:00:24.803023500 Attempting to disconnect from 20:22:08:25:17:41
2023-11-25 09:00:24.803029500 Successful disconnected
2023-11-25 09:00:29.855923500 Device 20:22:08:25:17:41 (public)
2023-11-25 09:00:29.855930500   Alias: JK_BD4A24S04
2023-11-25 09:00:29.855932500   Paired: no
2023-11-25 09:00:29.855934500   Trusted: no
2023-11-25 09:00:29.855936500   Blocked: no
2023-11-25 09:00:29.855937500   Connected: no
2023-11-25 09:00:29.855939500   LegacyPairing: no
2023-11-25 09:00:29.855941500   RSSI: -57
2023-11-25 09:00:29.856893500
2023-11-25 09:00:31.016673500 INFO:SerialBattery:
2023-11-25 09:00:31.017238500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-25 09:00:31.018425500 INFO:SerialBattery:dbus-serialbattery v1.0.2023111                                                   7dev
2023-11-25 09:00:31.496497500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:2                                                   5:17:41
2023-11-25 09:00:31.496963500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:2                                                   5:17:41
2023-11-25 09:00:34.203320500 INFO:SerialBattery:--> asy_connect_and_scrape(): e                                                   rror while connecting to bt: Characteristic with UUID 00002a24-0000-1000-8000-00                                                   805f9b34fb could not be found!
2023-11-25 09:00:36.387988500 INFO:SerialBattery:--> asy_connect_and_scrape(): E                                                   xit
2023-11-25 09:00:41.013618500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17                                                   :41
2023-11-25 09:00:41.014089500 ERROR:SerialBattery:ERROR >>> No battery connectio                                                   n at Jkbms_Ble
2023-11-25 09:00:41.292225500
2023-11-25 09:00:41.292334500 INFO:Bluetooth details
2023-11-25 09:00:41.294299500 [CHG] Device 20:22:08:25:17:41 RSSI: -57
2023-11-25 09:00:41.294308500 [CHG] Device 20:22:08:25:17:41 RSSI: -56
2023-11-25 09:00:41.338501500 Attempting to disconnect from 20:22:08:25:17:41
2023-11-25 09:00:41.338508500 Successful disconnected
2023-11-25 09:00:46.390658500 Device 20:22:08:25:17:41 (public)
2023-11-25 09:00:46.390664500   Alias: JK_BD4A24S04
2023-11-25 09:00:46.390667500   Paired: no
2023-11-25 09:00:46.390668500   Trusted: no
2023-11-25 09:00:46.390670500   Blocked: no
2023-11-25 09:00:46.390672500   Connected: no
2023-11-25 09:00:46.390674500   LegacyPairing: no
2023-11-25 09:00:46.390676500   RSSI: -63
2023-11-25 09:00:46.392663500
2023-11-25 09:00:47.466390500 INFO:SerialBattery:
2023-11-25 09:00:47.467218500 INFO:SerialBattery:Starting dbus-serialbattery
2023-11-25 09:00:47.468530500 INFO:SerialBattery:dbus-serialbattery v1.0.2023111                                                   7dev
2023-11-25 09:00:47.858728500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:2                                                   5:17:41
2023-11-25 09:00:47.859387500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:2                                                   5:17:41
2023-11-25 09:00:49.264848500 INFO:SerialBattery:--> asy_connect_and_scrape(): e                                                   rror while connecting to bt: Characteristic with UUID 00002a24-0000-1000-8000-00                                                   805f9b34fb could not be found!
2023-11-25 09:00:51.403430500 INFO:SerialBattery:--> asy_connect_and_scrape(): E                                                   xit
2023-11-25 09:00:57.374379500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17                                                   :41
2023-11-25 09:00:57.374882500 ERROR:SerialBattery:ERROR >>> No battery connectio                                                   n at Jkbms_Ble
2023-11-25 09:00:57.655496500
2023-11-25 09:00:57.655604500 INFO:Bluetooth details
2023-11-25 09:00:57.666579500 [CHG] Device 20:22:08:25:17:41 RSSI: -63
2023-11-25 09:00:57.666585500 [CHG] Device 20:22:08:25:17:41 RSSI: -57
2023-11-25 09:00:57.714824500 Attempting to disconnect from 20:22:08:25:17:41
2023-11-25 09:00:57.714831500 Successful disconnected
[... cut ...]
2023-11-25 09:05:53.116444500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2023-11-25 09:05:53.499294500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2023-11-25 09:05:53.499794500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2023-11-25 09:05:54.816510500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!
2023-11-25 09:05:57.868027500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit

itgenmar, 2023-11-25:
is this something that can help you :

### [JK_BD4A24S04]# list-attributes
Primary Service (Handle 0x7dbc)
        /org/bluez/hci0/dev_20_22_08_25_17_41/service0001
        0000ffe0-0000-1000-8000-00805f9b34fb
        Unknown
Characteristic (Handle 0x7440)
        /org/bluez/hci0/dev_20_22_08_25_17_41/service0001/char0004
        0000ffe1-0000-1000-8000-00805f9b34fb
        Unknown
Descriptor (Handle 0x0000)
        /org/bluez/hci0/dev_20_22_08_25_17_41/service0001/char0004/desc0006
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
Characteristic (Handle 0xf9b8)
        /org/bluez/hci0/dev_20_22_08_25_17_41/service0001/char0002
        0000ffe1-0000-1000-8000-00805f9b34fb
        Unknown

mr-manuel, 2023-11-25:
What happens if you try to manually pair the BMS? See https://wiki.debian.org/BluetoothUser#Using_bluetoothctl

itgenmar, 2023-11-26:
Hi it doesn't work : 

**root@VFIEVOX3:~# bluetoothctl
Agent registered
[CHG] Controller B8:27:EB:2A:55:5A Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:2A:55:5A Discovering: yes
[CHG] Device 20:22:08:25:17:41 RSSI: -67
[NEW] Device A5:C2:37:01:AD:77 SP17S005P17S40A
[bluetooth]# scan off
Discovery stopped
[CHG] Device A5:C2:37:01:AD:77 RSSI is nil
[CHG] Device 20:22:08:25:17:41 RSSI is nil
[CHG] Controller B8:27:EB:2A:55:5A Discovering: no
[bluetooth]# connect 20:22:08:25:17:41
Attempting to connect to 20:22:08:25:17:41
[CHG] Device 20:22:08:25:17:41 Connected: yes
Connection successful
[CHG] Device 20:22:08:25:17:41 ServicesResolved: no
[CHG] Device 20:22:08:25:17:41 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled**

mr-manuel, 2023-11-26:
If you cannot connect that way, then you have another problem. Did you change the connection PIN of the BMS?

itgenmar, 2023-11-26:
I have 1234, 123456, and 0000

MisterX1000, 2023-12-13:
Hi all,
any success here? I´m on the same point where I try to pair or connect to the jkbms without success. 

Best regards
Alex

[image]

itgenmar, 2023-12-15:
I think your issue is easy 

You need to install bleak 

opkg update 
opkg install python3-misc python3-pip 
pip3 install bleak

Honusnap, 2024-01-09:
> I think your issue is easy
> 
> You need to install bleak
> 
> opkg update opkg install python3-misc python3-pip pip3 install bleak

Itgenmar, did you found a solution to your problem ?

itgenmar, 2024-01-09:
Not yet :(

Honusnap, 2024-01-09:
> Not yet :(
Seems we got the same proplem : https://github.com/Louisvdw/dbus-serialbattery/issues/909

I've opened an issue on the Bleak project : https://github.com/hbldh/bleak/issues/1493
Cause it seems that even the 2 test scripts from Bleak do not work with my configuration.

My Jkbms is pretty old
[image]

I will certainly buy a BMV712 since it's at a pretty good price.

Honusnap, 2024-01-15:
> Not yet :(

I had the same kind of error then you had, turns out that for me the bluetooth communication ,is not "standard" enough to work with the Bleak API (it's not smart enough to me .. but it's an other problem). I've swapped my old JK for a newer one, and it work perfectly.
The Shunt i've ordered will still be usefull as a DC circuit counter, which is in fact kind mandatory for this whole system to work properly.

pau1phi11ips, 2024-01-18:
> Not yet :(

Am I correct in thinking you are connecting to the JK to a ESP HOME device and to Venus OS at the same time?

AFAIK, it can only connect to 1 device. If I power down the Venus and connect via my phone. It won't re-establish the Bluetooth connection when Venus is restarted.

itgenmar, 2024-01-21:
Just tried again with the latest night v.

 tail -F -n 100 /data/log/dbus-blebattery.*/current | tai64nlocal
2024-01-21 22:05:06.991919500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17:41
2024-01-21 22:05:06.991927500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-01-21 22:05:07.299180500 [CHG] Device 20:22:08:25:17:41 RSSI: -48
2024-01-21 22:05:07.299188500 [CHG] Device 20:22:08:25:17:41 RSSI: -48
2024-01-21 22:05:07.314368500
2024-01-21 22:05:07.314476500 INFO:Bluetooth details
2024-01-21 22:05:07.377149500 Attempting to disconnect from 20:22:08:25:17:41
2024-01-21 22:05:07.377158500 Successful disconnected
2024-01-21 22:05:12.444860500 Device 20:22:08:25:17:41 (public)
2024-01-21 22:05:12.444867500   Alias: JK_BD4A24S04
2024-01-21 22:05:12.444869500   Paired: no
2024-01-21 22:05:12.444871500   Trusted: yes
2024-01-21 22:05:12.444872500   Blocked: no
2024-01-21 22:05:12.444874500   Connected: no
2024-01-21 22:05:12.444876500   LegacyPairing: no
2024-01-21 22:05:12.444878500   RSSI: -31
2024-01-21 22:05:12.445913500
2024-01-21 22:05:27.148724500 *** CCGX booted (0) ***
2024-01-21 22:07:36.537228500
2024-01-21 22:07:36.537524500 INFO:Bluetooth details
2024-01-21 22:07:36.592828500 Attempting to disconnect from 20:22:08:25:17:41
2024-01-21 22:07:36.592835500 Successful disconnected
2024-01-21 22:07:41.677291500 Device 20:22:08:25:17:41 (public)
2024-01-21 22:07:41.677297500   Alias: JK_BD4A24S04
2024-01-21 22:07:41.677299500   Paired: no
2024-01-21 22:07:41.677301500   Trusted: yes
2024-01-21 22:07:41.677303500   Blocked: no
2024-01-21 22:07:41.677304500   Connected: no
2024-01-21 22:07:41.677306500   LegacyPairing: no
2024-01-21 22:07:41.690591500
2024-01-21 22:07:43.583273500 INFO:SerialBattery:
2024-01-21 22:07:43.583278500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-21 22:07:43.583281500 INFO:SerialBattery:dbus-serialbattery v1.1.20240112dev
2024-01-21 22:07:44.494770500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2024-01-21 22:07:44.494778500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2024-01-21 22:07:55.629462500 Exception in thread Thread-1:
2024-01-21 22:07:55.629469500 Traceback (most recent call last):
2024-01-21 22:07:55.629472500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 447, in asy_connect_and_scrape
2024-01-21 22:07:55.629477500     await client.connect()
2024-01-21 22:07:55.629479500   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 605, in connect
2024-01-21 22:07:55.629484500     return await self._backend.connect(**kwargs)
2024-01-21 22:07:55.629487500   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 140, in connect
2024-01-21 22:07:55.629820500     raise BleakDeviceNotFoundError(
2024-01-21 22:07:55.629824500 bleak.exc.BleakDeviceNotFoundError: Device with address 20:22:08:25:17:41 was not found.
2024-01-21 22:07:55.629829500
2024-01-21 22:07:55.629830500 During handling of the above exception, another exception occurred:
2024-01-21 22:07:55.629834500
2024-01-21 22:07:55.629835500 Traceback (most recent call last):
2024-01-21 22:07:55.629838500   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
2024-01-21 22:07:55.629881500     self.run()
2024-01-21 22:07:55.629883500   File "/usr/lib/python3.8/threading.py", line 870, in run
2024-01-21 22:07:55.629887500     self._target(*self._args, **self._kwargs)
2024-01-21 22:07:55.629890500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 433, in connect_and_scrape
2024-01-21 22:07:55.629915500     asyncio.run(self.asy_connect_and_scrape())
2024-01-21 22:07:55.629919500   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
2024-01-21 22:07:55.629977500     return loop.run_until_complete(main)
2024-01-21 22:07:55.629980500   File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
2024-01-21 22:07:55.629985500     return future.result()
2024-01-21 22:07:55.629988500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 468, in asy_connect_and_scrape
2024-01-21 22:07:55.629993500     ) = sys.exc_info()
2024-01-21 22:07:55.629995500 NameError: name 'sys' is not defined
2024-01-21 22:07:55.719759500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17:41
2024-01-21 22:07:55.720015500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-01-21 22:07:56.027600500
2024-01-21 22:07:56.027922500 INFO:Bluetooth details
2024-01-21 22:07:56.092554500 Attempting to disconnect from 20:22:08:25:17:41
2024-01-21 22:07:56.092561500 Successful disconnected
[... 160 lines cut ...]
2024-01-21 22:08:56.628101500 Traceback (most recent call last):
2024-01-21 22:08:56.628104500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 449, in asy_connect_and_scrape
2024-01-21 22:08:56.628109500     await client.read_gatt_char(MODEL_NBR_UUID)
2024-01-21 22:08:56.628113500   File "/usr/lib/python3.8/site-packages/bleak/__init__.py", line 711, in read_gatt_char
2024-01-21 22:08:56.628118500     return await self._backend.read_gatt_char(char_specifier, **kwargs)
2024-01-21 22:08:56.628233500   File "/usr/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 728, in read_gatt_char
2024-01-21 22:08:56.628240500     raise BleakError(
2024-01-21 22:08:56.628242500 bleak.exc.BleakError: Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!
[... cut ...]
2024-01-21 22:08:56.628438500   File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
2024-01-21 22:08:56.628442500     return loop.run_until_complete(main)
2024-01-21 22:08:56.628445500   File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
2024-01-21 22:08:56.628450500     return future.result()
2024-01-21 22:08:56.628452500   File "/opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py", line 468, in asy_connect_and_scrape
2024-01-21 22:08:56.628479500     ) = sys.exc_info()
2024-01-21 22:08:56.628482500 NameError: name 'sys' is not defined

itgenmar, 2024-01-21:
options 
[image]

Honusnap, 2024-01-23:
To check if it is connected to an other device ?.. just start the Android app and see if you can connect to the BMS, if it is connected to an other device .. it will not show in the left list.

mr-manuel, 2024-01-28:
@itgenmar please take a look at https://github.com/Louisvdw/dbus-serialbattery/issues/900#issuecomment-1909477616. Maybe this solves also this issue.

itgenmar, 2024-01-28:
Just tried it with the latest Nightly

~~~2024-01-28 15:13:49.302526500 [CHG] Device 20:22:08:25:17:41 RSSI: -68
2024-01-28 15:13:49.302534500 [CHG] Device 20:22:08:25:17:41 RSSI: -68
2024-01-28 15:13:49.346633500 Attempting to disconnect from 20:22:08:25:17:41
2024-01-28 15:13:49.346640500 Successful disconnected
2024-01-28 15:13:54.405788500 Device 20:22:08:25:17:41 (public)
2024-01-28 15:13:54.405794500   Alias: JK_BD4A24S04
2024-01-28 15:13:54.405796500   Paired: no
2024-01-28 15:13:54.405798500   Trusted: yes
2024-01-28 15:13:54.405800500   Blocked: no
2024-01-28 15:13:54.405802500   Connected: no
2024-01-28 15:13:54.405804500   LegacyPairing: no
2024-01-28 15:13:54.405806500   RSSI: -68
2024-01-28 15:13:54.408237500
2024-01-28 15:13:55.541688500 INFO:SerialBattery:
2024-01-28 15:13:55.542097500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-28 15:13:55.543002500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2024-01-28 15:13:55.915031500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2024-01-28 15:13:55.915467500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2024-01-28 15:13:55.919044500 DEBUG:SerialBattery:scraping thread started -> main thread id: 1995534352 scraping thread: 1979987040
2024-01-28 15:13:55.921040500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): Connect and scrape on address: 20:22:08:25:17:41
2024-01-28 15:13:56.256824500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): btloop
2024-01-28 15:13:56.257239500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): reconnect
2024-01-28 15:13:57.870355500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!
2024-01-28 15:14:00.521282500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-01-28 15:14:05.434178500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17:41
2024-01-28 15:14:05.434635500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble

~~~

itgenmar, 2024-01-28:
And sometimes I am getting this:
~~~
2024-01-28 15:22:27.843173500 INFO:SerialBattery:
2024-01-28 15:22:27.843824500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-28 15:22:27.844843500 INFO:SerialBattery:dbus-serialbattery v1.0.20231117dev
2024-01-28 15:22:28.224290500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2024-01-28 15:22:28.224942500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2024-01-28 15:22:28.228458500 DEBUG:SerialBattery:scraping thread started -> main thread id: 1995669520 scraping thread: 1980122208
2024-01-28 15:22:28.230665500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): Connect and scrape on address: 20:22:08:25:17:41
2024-01-28 15:22:28.587299500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): btloop
2024-01-28 15:22:28.588039500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): reconnect
2024-01-28 15:22:29.352171500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: [org.bluez.Error.Failed] Software caused connection abort
2024-01-28 15:22:29.352991500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-01-28 15:22:52.645988500 Missing device address argument
2024-01-28 15:22:54.309052500 INFO:SerialBattery:Starting dbus-serialbattery
2024-01-28 15:22:54.310023500 INFO:SerialBattery:dbus-serialbattery v1.0.20230531
2024-01-28 15:22:54.729476500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2024-01-28 15:22:54.729949500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2024-01-28 15:23:10.384565500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-01-28 15:23:10.384573500
~~~

itgenmar, 2024-01-28:
Bluetoothctl Log

~~~
[CHG] Device 20:22:08:25:17:41 Connected: yes
[NEW] Primary Service (Handle 0x6884)
        /org/bluez/hci1/dev_20_22_08_25_17_41/service0001
        0000ffe0-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Characteristic (Handle 0x685c)
        /org/bluez/hci1/dev_20_22_08_25_17_41/service0001/char0002
        0000ffe1-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Characteristic (Handle 0x685c)
        /org/bluez/hci1/dev_20_22_08_25_17_41/service0001/char0004
        0000ffe1-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Descriptor (Handle 0x37fc)
        /org/bluez/hci1/dev_20_22_08_25_17_41/service0001/char0004/desc0006
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[CHG] Device 20:22:08:25:17:41 UUIDs: 0000ffe0-0000-1000-8000-00805f9b34fb
[CHG] Device 20:22:08:25:17:41 ServicesResolved: yes
[CHG] Device 20:22:08:25:17:41 ServicesResolved: no
~~~

mr-manuel, 2024-02-16:
What is the current state?

Did you make sure, that only the dbus-serialbattery is connection to the JKBMS?

From all the intel you gave it seems you are trying to connect two devices at thesame time with the JKBMS, which do not work by JKBMS design.

itgenmar, 2024-02-16:
HI Manuel, 

i made sure to disconnect all other devices, i have bought another raspberry pi and still doesn't work. 

here are the logs from dbus-serial battery version:  v1.1.20240128dev with Venus OS version 3.14. and attached Bluethootlctl logs

~~~2024-02-16 12:14:35.776099500
2024-02-16 12:14:35.776104500 INFO:Bluetooth details
2024-02-16 12:14:35.802350500 Attempting to disconnect from 20:22:08:25:17:41
2024-02-16 12:14:35.802356500 Successful disconnected
2024-02-16 12:14:40.847744500 Device 20:22:08:25:17:41 (public)
2024-02-16 12:14:40.847750500   Alias: JK_BD4A24S04
2024-02-16 12:14:40.847752500   Paired: no
2024-02-16 12:14:40.847754500   Trusted: yes
2024-02-16 12:14:40.847756500   Blocked: no
2024-02-16 12:14:40.847758500   Connected: no
2024-02-16 12:14:40.847759500   LegacyPairing: no
2024-02-16 12:14:40.847761500   RSSI: -52
2024-02-16 12:14:40.848721500
2024-02-16 12:14:42.689852500 INFO:SerialBattery:
2024-02-16 12:14:42.690264500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-16 12:14:42.691173500 INFO:SerialBattery:dbus-serialbattery v1.1.20240128dev
2024-02-16 12:14:43.173352500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2024-02-16 12:14:43.173819500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2024-02-16 12:14:43.177375500 DEBUG:SerialBattery:scraping thread started -> main thread id: 1995792400 scraping thread: 1968174176
2024-02-16 12:14:43.180485500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): Connect and scrape on address: 20:22:08:25:17:41
2024-02-16 12:14:43.621520500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): btloop
2024-02-16 12:14:43.622157500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): reconnect
2024-02-16 12:14:44.471306500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #455
2024-02-16 12:14:46.660927500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-02-16 12:14:46.664185500 DEBUG:SerialBattery:scraping thread ended: reseting bluetooth and restarting
2024-02-16 12:14:46.664982500 INFO:SerialBattery:Reset of system Bluetooth daemon triggered
2024-02-16 12:14:46.795317500 [CHG] Device 20:22:08:25:17:41 RSSI: -52
2024-02-16 12:14:46.795324500 [CHG] Device 20:22:08:25:17:41 RSSI: -54
2024-02-16 12:14:46.795328500 [CHG] Device 20:22:08:25:17:41 RSSI is nil
2024-02-16 12:14:48.773704500 Starting bluetooth: bluetoothd.
2024-02-16 12:14:48.801741500 INFO:SerialBattery:System Bluetooth daemon should have been restarted
2024-02-16 12:14:50.805918500 DEBUG:SerialBattery:scraping thread started -> main thread id: 1995792400 scraping thread: 1968174176
2024-02-16 12:14:50.807668500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): Connect and scrape on address: 20:22:08:25:17:41
2024-02-16 12:14:50.808463500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): btloop
2024-02-16 12:14:50.808867500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): reconnect
2024-02-16 12:14:51.725389500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #455
2024-02-16 12:14:53.675049500 INFO:SerialBattery:--> asy_connect_and_scrape(): Exit
2024-02-16 12:14:53.705640500 ERROR:SerialBattery:No BMS found at 20:22:08:25:17:41
2024-02-16 12:14:53.706201500 ERROR:SerialBattery:ERROR >>> No battery connection at Jkbms_Ble
2024-02-16 12:14:54.018067500
2024-02-16 12:14:54.018172500 INFO:Bluetooth details
2024-02-16 12:14:54.057696500 Attempting to disconnect from 20:22:08:25:17:41
2024-02-16 12:14:54.057702500 Successful disconnected
2024-02-16 12:14:59.107082500 Device 20:22:08:25:17:41 (public)
2024-02-16 12:14:59.107088500   Alias: JK_BD4A24S04
2024-02-16 12:14:59.107090500   Paired: no
2024-02-16 12:14:59.107092500   Trusted: yes
2024-02-16 12:14:59.107094500   Blocked: no
2024-02-16 12:14:59.107095500   Connected: no
2024-02-16 12:14:59.107097500   LegacyPairing: no
2024-02-16 12:14:59.107099500   RSSI: -51
2024-02-16 12:14:59.109409500
2024-02-16 12:15:00.310530500 INFO:SerialBattery:
2024-02-16 12:15:00.310944500 INFO:SerialBattery:Starting dbus-serialbattery
2024-02-16 12:15:00.311866500 INFO:SerialBattery:dbus-serialbattery v1.1.20240128dev
2024-02-16 12:15:00.705659500 INFO:SerialBattery:Init of Jkbms_Ble at 20:22:08:25:17:41
2024-02-16 12:15:00.706227500 INFO:SerialBattery:Test of Jkbms_Ble at 20:22:08:25:17:41
2024-02-16 12:15:00.710353500 DEBUG:SerialBattery:scraping thread started -> main thread id: 1996455952 scraping thread: 1971319904
2024-02-16 12:15:00.713371500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): Connect and scrape on address: 20:22:08:25:17:41
2024-02-16 12:15:01.030024500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): btloop
2024-02-16 12:15:01.030441500 DEBUG:SerialBattery:--> asy_connect_and_scrape(): reconnect
2024-02-16 12:15:02.444238500 INFO:SerialBattery:--> asy_connect_and_scrape(): error while connecting to bt: BleakError('Characteristic with UUID 00002a24-0000-1000-8000-00805f9b34fb could not be found!') of type <class 'bleak.exc.BleakError'> in /opt/victronenergy/dbus-serialbattery/bms/jkbms_brn.py line #455
[... cut ...]
        0000ffe1-0000-1000-8000-00805f9b34fb
        Unknown
[NEW] Descriptor (Handle 0xd73c)
        /org/bluez/hci0/dev_20_22_08_25_17_41/service0001/char0004/desc0006
        00002902-0000-1000-8000-00805f9b34fb
        Client Characteristic Configuration
[CHG] Device 20:22:08:25:17:41 ServicesResolved: yes
[CHG] Device 20:22:08:25:17:41 ServicesResolved: no
[CHG] Device 20:22:08:25:17:41 Connected: no
[CHG] Controller B8:27:EB:83:A6:83 Discovering: yes
[CHG] Device 20:22:08:25:17:41 RSSI: -52
~~~

mr-manuel, 2024-02-23:
Fixed with `v1.2.20240223dev`.

closed 2024-02-23 by mr-manuel