Louisvdw/dbus-serialbattery #820: MPPT shows to high charge voltage

Zahnpasta83, 2023-09-24:
### Describe the bug

The MPPT´s show higher charge voltage than set in the app, the settings of serial battery or elsewhere. Why is that?

[image]
[image]

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

I have got an 18S system, set voltage should be 61.2V float voltage and 62.1 (it shows 62.5V there) absorption voltage

### Driver version

latest

### Venus OS device type

Raspberry Pi

### Venus OS version

latest

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

18

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 150.0
MAX_BATTERY_DISCHARGE_CURRENT = 150.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.450

; --------- Bluetooth BMS ---------
; Description: List the Bluetooth BMS here that you want to install
; -- Available Bluetooth BMS:
; Jkbms_Ble, LltJbd_Ble
; Example:
;     1 BMS: Jkbms_Ble C8:47:8C:00:00:00
;     3 BMS: Jkbms_Ble C8:47:8C:00:00:00, Jkbms_Ble C8:47:8C:00:00:11, Jkbms_Ble C8:47:8C:00:00:22
BLUETOOTH_BMS =

; --------- BMS disconnect behaviour ---------
[... 242 lines cut ...]
2023-09-17 11:11:39.592009500 INFO:SerialBattery:Testing HLPdataBMS4S
2023-09-17 11:11:44.105624500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-09-17 11:11:44.105963500 INFO:SerialBattery:Testing Jkbms
2023-09-17 11:11:44.386833500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-09-17 11:11:44.396703500 INFO:SerialBattery:Testing Lifepower
2023-09-17 11:11:44.674326500 ERROR:SerialBattery:>>> ERROR: No reply - returning
2023-09-17 11:11:44.679673500 INFO:SerialBattery:Testing LltJbd
2023-09-17 11:11:45.415823500 INFO:SerialBattery:Connection established to LltJbd
2023-09-17 11:11:45.416296500 INFO:SerialBattery:Battery LLT/JBD connected to dbus from /dev/ttyACM0
2023-09-17 11:11:45.416592500 INFO:SerialBattery:========== Settings ==========
2023-09-17 11:11:45.417150500 INFO:SerialBattery:> Connection voltage: 62.51V | Current: 14.04A | SoC: 100.0%
2023-09-17 11:11:45.417590500 INFO:SerialBattery:> Cell count: 18 | Cells populated: 18
2023-09-17 11:11:45.417975500 INFO:SerialBattery:> LINEAR LIMITATION ENABLE: True
2023-09-17 11:11:45.418309500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 150.0A | MAX BATTERY DISCHARGE CURRENT: 150.0A
2023-09-17 11:11:45.418890500 INFO:SerialBattery:> MAX BATTERY CHARGE CURRENT: 200.0A | MAX BATTERY DISCHARGE CURRENT: 200.0A (read from BMS)
2023-09-17 11:11:45.419236500 INFO:SerialBattery:> CVCM:     True
2023-09-17 11:11:45.419806500 INFO:SerialBattery:> MIN CELL VOLTAGE: 2.9V | MAX CELL VOLTAGE: 3.45V
2023-09-17 11:11:45.420132500 INFO:SerialBattery:> CCCM CV:  True  | DCCM CV:  True
2023-09-17 11:11:45.420564500 INFO:SerialBattery:> CCCM T:   True  | DCCM T:   True
2023-09-17 11:11:45.420891500 INFO:SerialBattery:> CCCM SOC: False | DCCM SOC: False
2023-09-17 11:11:45.463635500 INFO:SerialBattery:DeviceInstance = 1
2023-09-17 11:11:45.464961500 INFO:SerialBattery:com.victronenergy.battery.ttyACM0
2023-09-17 11:11:45.864090500 INFO:SerialBattery:publish config values = 1

```

pos-ei-don, 2023-09-24:
Voltage Threshold is 0.4v, so this voltage is ok!

mr-manuel, 2023-09-25:
Just for curiosity.

1. Did you enable STS in Settings -> DVCC?
2. What is the voltage drop between the battery and the MPPT?

Zahnpasta83, 2023-09-25:
Hi,

1. yes I did.
2. what do you mean by voltage drop between the battery and MPPT? Battery voltage is mostly +- 0.15 the set voltage of float or absorption voltage

mr-manuel, 2023-09-25:
1. If you enabled SVS the voltage difference is normal.
2. What is the voltage you measure with a multimeter on the battery poles and what is the voltage on the MPPT?

Zahnpasta83, 2023-09-25:
I have to check as soon as I’m at home. But atm according to the BMS and the VRM portal it’s the same

LarsAndersenDenmark, 2023-10-11:
If your mppt's can charge more than your inverters are able to convert at forward to the grid (grid feedin), then the system will go 0.4voltage above the set value before it will lower the charging of the mppt's. This is quite normal and also 0.4v shouldn't be a problem...but I must admit I also were a bit terrified when my voltage went above my set max voltage level. I had too much mppt poewr (2x 250v/100A) and only 1 MPII 48v/5000 (at L1). I added an extra MPII to my second phase (L2). This made it possible for my inverters to handle the full charge from the mppt's when battery is full. And now I do go the 0.4v above anymore.