Louisvdw/dbus-serialbattery #823: JK-BMS driver crashes if no Manufacturing Date is given

MuellerM1996, 2023-09-26:
### Describe the bug

My JK BMS has an empty Manufacturing Date (see screenshot). Since Driver version v1.0.20230531 this gets read from the BMS but the BMS just return 0xFF which can't be interpret. 
[image]

Error from the log:
2023-09-26 17:27:23.208204500 INFO:SerialBattery:Testing Jkbms
2023-09-26 17:27:23.325547500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'\xff\xff\xff\xff', 0, 1, 'invalid start byte'), type(err)=<class 'UnicodeDecodeError'>

After some experimenting with old commits I found out that the missing Manufacturing Date lead to this error. I comment this out in the jkbms.py line 174 to 178 (on version v1.0.20230531) and everything works as expected.

### How to reproduce

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior

In case the Manufacturing Date is not available, ignore it (leave it on None)

### Driver version

v1.0.20230531

### Venus OS device type

Raspberry Pi

### Venus OS version

3.10

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 200.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.75
; Max voltage can seen as absorption voltage
MAX_CELL_VOLTAGE   = 3.5
FLOAT_CELL_VOLTAGE = 3.4

; --------- BMS disconnect behaviour ---------
; Description: Block charge and discharge when the communication to the BMS is lost. If you are removing the
;              BMS on purpose, then you have to restart the driver/system to reset the block.
; False: Charge and discharge is not blocked on BMS communication loss
; True: Charge and discharge is blocked on BMS communication loss, it's unblocked when connection is established
;       again or the driver/system is restarted
[... 164 lines cut ...]

; Temperature sensor 2 name
TEMP_4_NAME = Temp 4

; --------- Battery monitor specific settings ---------
; If you are using a SmartShunt or something else as a battery monitor, the battery voltage reported
; from the BMS and SmartShunt could differ. This causes, that the driver never goapplies the float voltage,
; since max voltage is never reached.
; Example:
;     cell count: 16
;     MAX_CELL_VOLTAGE = 3.45
;     max voltage calculated = 16 * 3.45 = 55.20
;     CVL is set to 55.20 and the battery is now charged until the SmartShunt measures 55.20 V. The BMS
;     now measures 55.05 V since there is a voltage drop of 0.15 V. Since the dbus-serialbattery measures
;     55.05 V the max voltage is never reached for the driver and max voltage is kept forever.
;     Set VOLTAGE_DROP to 0.15
VOLTAGE_DROP = 0.00
```

### Relevant log output

```shell
2023-09-26 17:27:23.208204500 INFO:SerialBattery:Testing Jkbms
2023-09-26 17:27:23.325547500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'\xff\xff\xff\xff', 0, 1, 'invalid start byte'), type(err)=<class 'UnicodeDecodeError'>
```

mr-manuel, 2023-09-26:
Thanks, I will fix this is the next days

mr-manuel, 2023-09-27:
Could you please test https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.0.20230927beta?

MuellerM1996, 2023-09-27:
root@raspberrypi2:~# tail -F /data/log/dbus-serialbattery.ttyUSB5/current | tai64nlocal                                                                                                                                                      2023-09-27 08:35:12.921423500 INFO:SerialBattery:-- Testing BMS: 1 of 3 rounds
2023-09-27 08:35:12.921846500 INFO:SerialBattery:Testing Jkbms
2023-09-27 08:35:13.042058500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'\xff\xff\xff\xff', 0, 1, 'invalid start byte'), type(err)=<class 'UnicodeDecodeError'>
2023-09-27 08:35:13.543388500 INFO:SerialBattery:-- Testing BMS: 2 of 3 rounds
2023-09-27 08:35:13.543821500 INFO:SerialBattery:Testing Jkbms
2023-09-27 08:35:13.594388500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'\xff\xff\xff\xff', 0, 1, 'invalid start byte'), type(err)=<class 'UnicodeDecodeError'>
2023-09-27 08:35:14.095854500 INFO:SerialBattery:-- Testing BMS: 3 of 3 rounds
2023-09-27 08:35:14.096441500 INFO:SerialBattery:Testing Jkbms
2023-09-27 08:35:14.145442500 ERROR:SerialBattery:Unexpected err=UnicodeDecodeError('utf-8', b'\xff\xff\xff\xff', 0, 1, 'invalid start byte'), type(err)=<class 'UnicodeDecodeError'>
2023-09-27 08:35:14.646753500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB5

Still the same error.

But if I change the except UnicodeEncodeError to a except UnicodeDecodeError it's working.

mr-manuel, 2023-09-27:
Damn. I rebuild it with the correct exception :-)

closed 2023-09-27 by mr-manuel