Louisvdw/dbus-serialbattery #837: Daly BMS - no data shown in MQTT

gkehrer, 2023-10-09:
### Describe the bug

I got a new Daly BMS with FW 11_230627_001T and here the data is not available in MQTT.
In the old Daly BMS with FW 11_220722_100T all works OK.

Both Daly BMS show data in the Victron console but the new one is not showing all data.
The data in SerialBattery -> Device is missing:
[image]

I installed the latest dev nightly build but the issue is still the same.

### How to reproduce

1. Install latest code
2. go to: SerialBattery -> Device

[image]
[image]
[image]

dbus-spy shows only limited values at the new BMS:
[image]

At the old BMS all data is available with dbus-spy:
[image]

### Expected behavior

Show all values in dbus-spy and MQTT for the new Daly BMS.

### Driver version

1.0.20231009dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.10

### BMS type

Daly Smart BMS

### Cell count

8

### Connection type

Serial USB adapter to RS485

### Config file

```ini
; 1 Seconds
; 2 Time string <days>d <hours>h <minutes>m <seconds>s
; 3 Both seconds and time string "<seconds> [<days>d <hours>h <minutes>m <seconds>s]"
TIME_TO_SOC_VALUE_TYPE = 1
; Specify in seconds how often the TimeToSoc should be recalculated
; Minimum are 5 seconds to prevent CPU overload
TIME_TO_SOC_RECALCULATE_EVERY = 60
; Include TimeToSoC points when moving away from the SoC point [Valid values True, False]
; These will be as negative time. Disabling this improves performance slightly
TIME_TO_SOC_INC_FROM = False

; --------- Additional settings ---------
; Specify only one BMS type to load else leave empty to try to load all availabe
[... 40 lines cut ...]
; Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = -1

; -- ESC GreenMeter and Lipro device settings
GREENMETER_ADDRESS  = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS   = 4
LIPRO_CELL_COUNT = 15
```

### Relevant log output

```shell
Array([], signature=dbus.Signature('i'), variant_level=1), 'Voltages/Sum': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Voltages/Diff': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'TimeToGo': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'CurrentAvg': dbus.Array([], signature=dbus.Signature('i'), variant_level=1), 'Info/Config/PATH_CONFIG_DEFAULT': dbus.String('config.default.ini', variant_level=1), 'Info/Config/PATH_CONFIG_USER': dbus.String('config.ini', variant_level=1), 'Info/Config/default_config_file_path': dbus.String('/opt/victronenergy/dbus-serialbattery/config.default.ini', variant_level=1), 'Info/Config/custom_config_file_path': dbus.String('/opt/victronenergy/dbus-serialbattery/config.ini', variant_level=1), 'Info/Config/DRIVER_VERSION': dbus.String('1.0.20231009dev', variant_level=1), 'Info/Config/zero_char': dbus.String('0', variant_level=1), 'Info/Config/degree_sign': dbus.String('°', variant_level=1), 'Info/Config/MAX_BATTERY_CHARGE_CURRENT': dbus.Double(125.0, variant_level=1), 'Info/Config/MAX_BATTERY_DISCHARGE_CURRENT': dbus.Double(250.0, variant_level=1), 'Info/Config/MIN_CELL_VOLTAGE': dbus.Double(2.9, variant_level=1), 'Info/Config/MAX_CELL_VOLTAGE': dbus.Double(3.555, variant_level=1), 'Info/Config/FLOAT_CELL_VOLTAGE': dbus.Double(3.455, variant_level=1), 'Info/Config/SOC_RESET_VOLTAGE': dbus.Double(3.65, variant_level=1), 'Info/Config/SOC_RESET_AFTER_DAYS': dbus.Boolean(False, variant_level=1), 'Info/Config/BLOCK_ON_DISCONNECT': dbus.Boolean(False, variant_level=1), 'Info/Config/LINEAR_LIMITATION_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/LINEAR_RECALCULATION_EVERY': dbus.Int32(60, variant_level=1), 'Info/Config/LINEAR_RECALCULATION_ON_PERC_CHANGE': dbus.Int32(5, variant_level=1), 'Info/Config/CVCM_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL': dbus.Double(0.01, variant_level=1), 'Info/Config/CELL_VOLTAGE_DIFF_TO_RESET_VOLTAGE_LIMIT': dbus.Double(0.08, variant_level=1), 'Info/Config/MAX_VOLTAGE_TIME_SEC': dbus.Int32(900, variant_level=1), 'Info/Config/SOC_LEVEL_TO_RESET_VOLTAGE_LIMIT': dbus.Int32(90, variant_level=1), 'Info/Config/CCCM_CV_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/DCCM_CV_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/CELL_VOLTAGES_WHILE_CHARGING': dbus.Array([dbus.Double(3.55, variant_level=1), dbus.Double(3.5, variant_level=1), dbus.Double(3.45, variant_level=1), dbus.Double(3.3, variant_level=1)], signature=None, variant_level=1), 'Info/Config/MAX_CHARGE_CURRENT_CV': dbus.Array([dbus.Double(6.25, variant_level=1), dbus.Double(12.5, variant_level=1), dbus.Double(62.5, variant_level=1), dbus.Double(125.0, variant_level=1)], signature=None, variant_level=1), 'Info/Config/CELL_VOLTAGES_WHILE_DISCHARGING': dbus.Array([dbus.Double(2.7, variant_level=1), dbus.Double(2.8, variant_level=1), dbus.Double(2.9, variant_level=1), dbus.Double(3.1, variant_level=1)], signature=None, variant_level=1), 'Info/Config/MAX_DISCHARGE_CURRENT_CV': dbus.Array([dbus.Double(0.0, variant_level=1), dbus.Double(25.0, variant_level=1), dbus.Double(125.0, variant_level=1), dbus.Double(250.0, variant_level=1)], signature=None, variant_level=1), 'Info/Config/CCCM_T_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/DCCM_T_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/TEMPERATURE_LIMITS_WHILE_CHARGING': dbus.Array([dbus.Double(0.0, variant_level=1), dbus.Double(2.0, variant_level=1), dbus.Double(5.0, variant_level=1), dbus.Double(10.0, variant_level=1), dbus.Double(15.0, variant_level=1), dbus.Double(20.0, variant_level=1), dbus.Double(35.0, variant_level=1), dbus.Double(40.0, variant_level=1), dbus.Double(55.0, variant_level=1)], signature=None, variant_level=1), 'Info/Config/MAX_CHARGE_CURRENT_T': dbus.Array([dbus.Double(0.0, variant_level=1), dbus.Double(12.5, variant_level=1), dbus.Double(25.0, variant_level=1), dbus.Double(50.0, variant_level=1), dbus.Double(100.0, variant_level=1), dbus.Double(125.0, variant_level=1), dbus.Double(125.0, variant_level=1), dbus.Double(50.0, variant_level=1), dbus.Double(0.0, variant_level=1)], signature=None, variant_level=1), 'Info/Config/TEMPERATURE_LIMITS_WHILE_DISCHARGING': dbus.Array([dbus.Double(-20.0, variant_level=1), dbus.Double(0.0, variant_level=1), dbus.Double(5.0, variant_level=1), dbus.Double(10.0, variant_level=1), dbus.Double(15.0, variant_level=1), dbus.Double(45.0, variant_level=1), dbus.Double(55.0, variant_level=1)], signature=None, variant_level=1), 'Info/Config/MAX_DISCHARGE_CURRENT_T': dbus.Array([dbus.Double(0.0, variant_level=1), dbus.Double(50.0, variant_level=1), dbus.Double(75.0, variant_level=1), dbus.Double(100.0, variant_level=1), dbus.Double(250.0, variant_level=1), dbus.Double(250.0, variant_level=1), dbus.Double(0.0, variant_level=1)], signature=None, variant_level=1), 'Info/Config/CCCM_SOC_ENABLE': dbus.Boolean(False, variant_level=1), 'Info/Config/DCCM_SOC_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/CC_SOC_LIMIT1': dbus.Double(98.0, variant_level=1), 'Info/Config/CC_SOC_LIMIT2': dbus.Double(95.0, variant_level=1), 'Info/Config/CC_SOC_LIMIT3': dbus.Double(91.0, variant_level=1), 'Info/Config/CC_CURRENT_LIMIT1': dbus.Double(12.5, variant_level=1), 'Info/Config/CC_CURRENT_LIMIT2': dbus.Double(37.5, variant_level=1), 'Info/Config/CC_CURRENT_LIMIT3': dbus.Double(62.5, variant_level=1), 'Info/Config/DC_SOC_LIMIT1': dbus.Double(10.0, variant_level=1), 'Info/Config/DC_SOC_LIMIT2': dbus.Double(20.0, variant_level=1), 'Info/Config/DC_SOC_LIMIT3': dbus.Double(30.0, variant_level=1), 'Info/Config/DC_CURRENT_LIMIT1': dbus.Double(25.0, variant_level=1), 'Info/Config/DC_CURRENT_LIMIT2': dbus.Double(75.0, variant_level=1), 'Info/Config/DC_CURRENT_LIMIT3': dbus.Double(125.0, variant_level=1), 'Info/Config/TIME_TO_GO_ENABLE': dbus.Boolean(True, variant_level=1), 'Info/Config/TIME_TO_SOC_POINTS': dbus.Array([], signature=dbus.Signature('u'), variant_level=1), 'Info/Config/TIME_TO_SOC_VALUE_TYPE': dbus.Int32(1, variant_level=1), 'Info/Config/TIME_TO_SOC_RECALCULATE_EVERY': dbus.Int32(60, variant_level=1), 'Info/Config/TIME_TO_SOC_INC_FROM': dbus.Boolean(False, variant_level=1), 'Info/Config/BMS_TYPE': dbus.Array([dbus.String('Daly', variant_level=1)], signature=None, variant_level=1), 'Info/Config/EXCLUDED_DEVICES': dbus.Array([], signature=dbus.Signature('u'), variant_level=1), 'Info/Config/CUSTOM_BATTERY_NAMES': dbus.Array([], signature=dbus.Signature('u'), variant_level=1), 'Info/Config/AUTO_RESET_SOC': dbus.Boolean(True, variant_level=1), 'In [... cut ...] ariant_level=1), 'Info/Config/LIPRO_CELL_COUNT': dbus.Int32(15, variant_level=1), 'Info/Config/HELTEC_MODBUS_ADDR': dbus.Array([dbus.Int32(1, variant_level=1)], signature=None, variant_level=1), 'Info/Config/VOLTAGE_DROP': dbus.Double(0.0, variant_level=1), 'Settings/HasSettings': dbus.Int32(1, variant_level=1), 'Settings/ResetSoc': dbus.Double(87.8, variant_level=1)}, signature=dbus.Signature('sv'), variant_level=1),) to message with signature v: <class 'ValueError'>: embedded null byte

^C
```

mr-manuel, 2023-10-10:
Could you install the latest nightly of the dev branch again? `1.0.20231009dev` had a bug.

gkehrer, 2023-10-10:
There is no change by installing **1.0.20231010dev** - still not all data visible in dbus-spy and no data in mqtt.

[image]

gkehrer, 2023-10-10:
Now all data is available in dbus-spy and mqtt:

[image]

Thanks for your help !

closed 2023-10-10 by mr-manuel