Louisvdw/dbus-serialbattery #840: SOC Reset persistent?

mdkeil, 2023-10-17:
### Describe the problem

Today I reached the `SOC_RESET_VOLTAGE` for the first time. After 30min of Absorption, Float Voltage was triggered corectly. 
In the evening I have to restart the driver and the CVL is 56.8V again. It should be 55.2V. 

### Driver version

v1.0.20231010beta

### Venus OS device type

Raspberry Pi

### Venus OS version

v3.11

### BMS type

Seplos

### Cell count

16

### Connection type

Serial USB adapter to RS485

### Config file

```ini
MAX_BATTERY_CHARGE_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0

MIN_CELL_VOLTAGE   = 2.900
MAX_CELL_VOLTAGE   = 3.450
FLOAT_CELL_VOLTAGE = 3.375

SOC_RESET_VOLTAGE  = 3.550
SOC_RESET_AFTER_DAYS = 15

;VOLTAGE_DROP = 0.10

LINEAR_LIMITATION_ENABLE = True
CVCM_ENABLE = True

CELL_VOLTAGE_DIFF_KEEP_MAX_VOLTAGE_UNTIL = 0.010

MAX_VOLTAGE_TIME_SEC = 1800

CELL_VOLTAGES_WHILE_CHARGING   = 3.60, 3.50, 3.45, 3.30
MAX_CHARGE_CURRENT_CV_FRACTION =    0, 0.10,  0.5,    1

BMS_TYPE = Seplos

; Exclute this serial devices from the driver startup
; Example: /dev/ttyUSB2, /dev/ttyUSB4
EXCLUDED_DEVICES = /dev/ttyUSB0
```

### Relevant log output

```shell
NTR
```

mr-manuel, 2023-10-17:
https://github.com/Louisvdw/dbus-serialbattery/discussions/827

mdkeil, 2023-10-18:
ok, but at the moment the last reset time will not survive the restart?

mr-manuel, 2023-11-26:
This was introduced with version `v1.0.20231126dev`. Could you please test and report back? I have not enough sun at the moment to test.

mdkeil, 2024-02-08:
"last-soc-reset timestamp" is saved persistently in the victron dbus setting. (`Settings/Devices/serialbattery_SeplosBMS16cells_280_0Ah/SocResetLastReached`) However, the driver does not seem to use the value after a "reinstall" and startet with "Bulk & SoC Reset" again.

mr-manuel, 2024-02-11:
Problem is fixed in the latest nightly and will be in the next beta and stable.

closed 2024-02-11 by mr-manuel