Louisvdw/dbus-serialbattery #841: allowed charge current not increased if more than one BMS is connected/defined

Baxter117, 2023-10-17:
### Describe the bug

I am using 3 Batteries in parallel ( each 14,4 kWh, 200A JKbms, cells can provide 280A current without problems) , connected to my Cerbo GX over bluetooth.
(bluetooth connection also having a lot of problems, like not connecting, restart of carbo, stop working without detection ... but other topics).

in every jkbms hardware settings a charge and discharge current of 200A is defined, as it is the hardware limit of the bms .
In the config.ini the Mac addresses of the 3 BMS are configured (for ble connection)
Also an allowed (dis)charge current of 600A is defined ( 3 batteries in parallel are able to charge with 600A all together) .

But when I want to charge or discharge, it is limited to 200A from the driver.
when I go into one of the connected BMS and Look under "Parameters" the CCL is 200A and the limitation came from "BMS Settings". That is theoretically right, because in the BMS hardware settings I defined this 200A.

### How to reproduce

connect more than one bms.
try to (dis)charge with whole provided battery current.

### Expected behavior

But I would expect that the driver not allowes only the lowest defined current of one of the connected BMS , but instead that the driver allowes the addition of the currents of all connected bms.
In my case 3 BMS are connected ( in parallel of course) and everyone is allowed to (dis)charge with 200A, so the driver should allow 600A . otherwise every BMS is limited to 66A current of its 200A .
[image]

I hope you can add a solution, that I can use the whole current of my batteries.
I think the perfect way would be a detection how many bms are connected (or configured in the config.ini) and allow the addition of all detected hardware limits of the bms ( if all hardware limits are the same of course, otherwise it would case problems ).

another possible solution would be to ignore the hardware setting of the bms and allow the configured current from the config.ini, maybe an extra condition like " Ignore_Hardware_Current_Limitation = true" would be safer so that only people who know what they are doing get that much current allowed :D

thank you a lot for the great work :)

### Driver version

v1.0.20231010dev

### Venus OS device type

Cerbo GX

### Venus OS version

v3.10 large

### BMS type

JKBMS / Heltec BMS

### Cell count

16

### Connection type

Bluetooth

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 600
MAX_BATTERY_DISCHARGE_CURRENT = 600

; --------- Cell Voltages ---------
MIN_CELL_VOLTAGE   = 2.55
; Max voltage ( absorption ) voltage
MAX_CELL_VOLTAGE   = 3.60
FLOAT_CELL_VOLTAGE = 3.375

; --------- Bluetooth BMS ---------
BLUETOOTH_BMS = Jkbms_Ble C8:47:8C:EC:B2:E4, Jkbms_Ble C8:47:8C:E9:30:BD, Jkbms_Ble C8:47:8C:E9:30:7D

; --------- BMS disconnect behaviour ---------
BLOCK_ON_DISCONNECT = True

... I think the rest is not important for this topic
```

### Relevant log output

```shell
--
```

### Any other information that may be helpful

i am using the nightly build dev branch

mr-manuel, 2023-10-18:
There are a lot of questions like this. Therefore a FAQ exists in our documentation where we put a lot of time and effort in it.

[How to aggregate multiple batteries?](https://louisvdw.github.io/dbus-serialbattery/faq/#how-to-aggregate-multiple-batteries)

Without battery aggregator it won't work.

closed 2023-10-18 by mr-manuel

Baxter117, 2023-10-19:
> There are a lot of questions like this. Therefore a FAQ exists in our documentation where we put a lot of time and effort in it.
> 
> [How to aggregate multiple batteries?](https://louisvdw.github.io/dbus-serialbattery/faq/#how-to-aggregate-multiple-batteries)
> 
> Without battery aggregator it won't work.

Thanks for the information, I read mostly all of your descriptions and Issue tickets, but did not connected the word aggregate with the solution, I was searching for :D