Louisvdw/dbus-serialbattery #847: Temperature sensors reversed Overkill BMS

lfhoward, 2023-11-02:
### Describe the bug

Last night the temperature dropped low enough for my battery heater to kick on this morning. My battery heater is programmed to turn on at 39 degrees F and turn off at 45 degrees F between 0600 and 1400. The outdoor temperature last night went down below 30 degrees F, with the battery itself coming down to 37 degrees F before the battery heater came on at 6 AM. 

I have a BMV Victron battery monitor with a wired temperature sensor attached to one of the battery posts. This is the sensor used to turn the battery heater on and off via the BMV relay. The Overkill BMS has 2 of its own temperature sensors, one located in the circuit board (MOSFET temperature) and another on a wire that is embedded between my battery cells (Battery Temperature). 

These two BMS temperature sensors seem to be revered in the Remote Console and VRM. See the following screen captures.

BMV battery temperature showing 45 degrees F after heating the battery this morning
https://i.postimg.cc/ZqVwgDPM/temp-Imaget-G68-Ez.jpg

Overkill BMS sensors reporting 35 degrees F for battery and 46 degrees F for MOSFET temperatures. These are backwards.
https://i.postimg.cc/SxLTLKnq/temp-Imager-AFl-Pj.jpg

The air temperature environment around the battery was 26 degrees F (RuuviTag).
https://i.postimg.cc/kgPYtSXf/temp-Image-Zvh-XTh.jpg

Remote console view showing temperatures.
https://i.postimg.cc/MGWs98BB/temp-Imaged9-Csz8.jpg

VRM view showing temperature history.
https://i.postimg.cc/SsRgp7DN/temp-Image1-V0bp-O.jpg

Thanks for your help!

### How to reproduce

Change the battery temperature quickly and compare to a second battery temperature sensor.

### Expected behavior

The BMS battery temperature reading should mirror the BMV battery temperature sensor. The MOSFET sensor should read closer to the external air temperature sensor.

### Driver version

1.0.20230902dev

### Venus OS device type

Raspberry Pi

### Venus OS version

v 3.12

### BMS type

Smart BMS (LLT, JBD, Overkill Solar)

### Cell count

4

### Connection type

Serial USB adapter to TTL

### Config file

```ini
see screen shots above
```

### Relevant log output

```shell
see screen shots above
```

mr-manuel, 2023-11-02:
Hi, please don't post links to external images, but insert the images directly on GitHub. Else in some weeks/months the images are not available anymore.

Unfortunately I do not understand what exactly the problem is or what you expect. The driver can only read the data the BMS is providing.

lfhoward, 2023-11-02:
I thought about what you said, “The driver can only read the data the BMS is providing,” and figured the names for the 2 BMS temperature sensors had to be able to be set somewhere. I came up with two options. 

One is inside the xiaoxiang BMS app, where “Temp. Sensor 1” and “Temp. Sensor 2” can be renamed. This should fix it at the BMS level. The temperature sensors can be renamed on the front page, by clicking the edit button next to each sensor name. 

The second option is inside the config.default.ini file associated with dbus-serialbattery. (See https://github.com/Louisvdw/dbus-serialbattery/blob/master/etc/dbus-serialbattery/config.default.ini) In that config file there is a way to name the battery temperature sensors as well as choose which sensor(s) to report as the battery temperature. Battery temperature settings begin at line 208 and go through line 227.

I will try both this weekend and report back what works. Meanwhile, thanks for the info about where the battery names come from. That was what got me pointed in the right direction.

mr-manuel, 2023-11-26:
Any news on this?

closed 2023-12-04 by mr-manuel

lfhoward, 2023-12-15:
Just an update - sorry it took me so long to test this out.

1. Changing the names of the temperature sensors inside the JBD bms does not change their names on the VRM side. 

2. Choosing a specific temperature sensor to report in config.ini - did not work as expected. The first battery sensor on the BMS is always reported as "Battery" even though it is actually the temp sensor inside the BMS itself, and the second temperature sensor is always reported as MOSFET even though is actually the remote battery sensor inside the cells. Naming them in the settings has no effect on the GUI. Setting Temp 2 as the primary makes the BMS temp on the GUI disappear.