Louisvdw/dbus-serialbattery #850: Driver Locks itself out

Foxconn11, 2023-11-02:
### Describe the bug

So, i have two Daly BMS connected via USB-TTL to a Cebro (v3.11) and Running the Newest version of dbus-serialbattery to this time.

Problem i found is: My System is discharged right now (BatteryLive cuts at 45%, ESS#1 ESS#2) and no Charge or Discharge happening. BMS going to Sleep after roughly 2 hours. When it Happens, i get 

> 2023-11-01 04:09:01.055014500 INFO:SerialBattery:Testing Daly
> 2023-11-01 04:09:01.195127500 ERROR:SerialBattery:>>> ERROR: No reply - returning
> 2023-11-01 04:09:01.696791500 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0

when this happens its Blocking Charging and Discharging from the whole System, MPPT, Multiplus etc. 
So it cannot start anymore until there is some Current on the BMS. It Locks itself out.

In config.ini i set 

BLOCK_ON_DISCONNECT = False

wich should prevent this or am i wrong?

### How to reproduce

Have A empty Battery
ESS blocks Discharge
BMS goes to Sleep
dbus-serialbattery can't read any Data from BMS
Cebro makes constant Alarm
dbus-serialbattery blocks charge and discharge
System Locked out

### Expected behavior

no Blockage because i set BLOCK_ON_DISCONNECT to False

### Driver version

1.0.20230531

### Venus OS device type

Cerbo GX

### Venus OS version

v3.11

### BMS type

Daly Smart BMS

### Cell count

4

### Connection type

Serial USB adapter to TTL

### Config file

```ini
[DEFAULT]

; --------- Battery Current limits ---------
MAX_BATTERY_CHARGE_CURRENT    = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 200.0

; --------- Cell Voltages ---------
; Description: Cell min/max voltages which are used to calculate the min/max battery voltage
; Example: 16 cells * 3.45V/cell = 55.2V max charge voltage. 16 cells * 2.90V = 46.4V min discharge voltage
MIN_CELL_VOLTAGE   = 2.900
[... 302 lines cut ...]
2023-11-02 15:40:33.113705500 INFO: Start service dbus-cgwacs.ttyUSB1 once
2023-11-02 15:40:44.688612500 INFO: Start service gps-dbus.ttyUSB2 once
2023-11-02 15:40:54.037263500 INFO: Start service vedirect-interface.ttyUSB2 once
2023-11-02 15:40:59.342184500 INFO: Start service gps-dbus.ttyUSB1 once
2023-11-02 15:41:00.573912500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2023-11-02 15:41:05.887997500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2023-11-02 15:41:09.164008500 INFO: Start service vedirect-interface.ttyUSB1 once
2023-11-02 15:41:14.457211500 INFO: Start service dbus-serialbattery.ttyUSB1 once
2023-11-02 15:41:19.799648500 INFO: Start service dbus-cgwacs.ttyUSB1 once
2023-11-02 15:41:31.302098500 INFO: Start service gps-dbus.ttyUSB2 once
2023-11-02 15:41:40.683701500 INFO: Start service vedirect-interface.ttyUSB2 once
2023-11-02 15:41:45.954184500 INFO: Start service gps-dbus.ttyUSB1 once
2023-11-02 15:41:47.223655500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2023-11-02 15:41:52.491059500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2023-11-02 15:41:55.750414500 INFO: Start service vedirect-interface.ttyUSB1 once
2023-11-02 15:42:01.034988500 INFO: Start service dbus-serialbattery.ttyUSB1 once
2023-11-02 15:42:06.348616500 INFO: Start service dbus-cgwacs.ttyUSB1 once
2023-11-02 15:42:17.762081500 INFO: Start service gps-dbus.ttyUSB2 once
2023-11-02 15:42:27.109699500 INFO: Start service vedirect-interface.ttyUSB2 once
2023-11-02 15:42:32.401768500 INFO: Start service gps-dbus.ttyUSB1 once
2023-11-02 15:42:33.609673500 INFO: Start service dbus-serialbattery.ttyUSB2 once
2023-11-02 15:42:38.923130500 INFO: Start service dbus-cgwacs.ttyUSB2 once
2023-11-02 15:42:42.267878500 INFO: Start service vedirect-interface.ttyUSB1 once
2023-11-02 15:42:47.571962500 INFO: Start service dbus-serialbattery.ttyUSB1 once
```

mr-manuel, 2023-11-02:
Did you try the latest nightly from the dev branch?

mr-manuel, 2023-11-02:
Search a bit in the issues and discussions. For Daly you have to set in the Daly BMS settings a specific value very high.

Foxconn11, 2023-11-05:
> Search a bit in the issues and discussions. For Daly you have to set in the Daly BMS settings a specific value very high.

I tried this. I looked through the Issues and found something.
[https://github.com/Louisvdw/dbus-serialbattery/issues/731](url)

I tried to set it to this number, but it won't write the value on either of the Daly BMS (I don't even know what BMS version this is. I can't even find pictures online)

I can write 60000 in the sleep value, and it writes; however, if I read the reported parameter, it only shows 13920, and that's around 4 Hours. I tried it with two different versions of the PC Master Software, but it's the same story.

I bought the batteries with BMS from a guy in Austria who used them. I even considered buying a new BMS like the JK BMS B2A8S20P and replacing the Dalys. As far as I have seen (I already have 2 for Testing purposes, and one is in my smaller Portable battery Bank), they only sleep once the Battery is empty and hits the shutdown voltage.

mr-manuel, 2023-11-05:
Maybe you can upgrade the firmware.

Foxconn11, 2023-11-06:
> Maybe you can upgrade the firmware.

I considered this around a year ago because i got Bluetooth Modules for the Daly BMS but couldn't get it to work. Tried so much and even asked Daly support wich was hard to find. 

Yesterday i ordered the B2A8S20P from JK and i will replace the Daly with them. I hope everything works then.

closed 2023-11-06 by mr-manuel